Freetz-NG/freetz-ng #519: [libcap] cannot compile

harryboo, 2022-04-17:
Since commit 67e5af3d12ba887ead4af0a16df9470abc677a95 (v2.24 -> v2.63) it is not possible to compile for 7490 and old os 6.93.

make libcap.so
make[3]: Verzeichnis „/home/jens/freetz-ng/source/target-mips_gcc-5.5.0_uClibc-0.9.33.2-nptl_kernel-3.10/libcap-2.63/libcap“ wird betreten
/home/jens/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-0.9.33.2-nptl_kernel-3.10/mips-linux-uclibc/bin/mips-linux-uclibc-gcc -march=34kc -mtune=34kc -msoft-float -Ofast -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -fPIC -Dlinux -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -I/home/jens/freetz-ng/source/target-mips_gcc-5.5.0_uClibc-0.9.33.2-nptl_kernel-3.10/libcap-2.63/libcap/../libcap/include/uapi -I/home/jens/freetz-ng/source/target-mips_gcc-5.5.0_uClibc-0.9.33.2-nptl_kernel-3.10/libcap-2.63/libcap/../libcap/include   empty.c   -o empty
objcopy --dump-section .interp=loader.txt empty /dev/null
objcopy: Format von Eingabedatei ‚empty‛ kann nicht bestimmt werden
make[3]: *** [Makefile:112: loader.txt] Fehler 1

fda77, 2022-04-17:
Da deine Ausgabe deutsch ist antworte ich auch so.
Wie wäre es beide Versionen in Freetz zu haben, und je nach Auswahl die entsprechende zu nehmen. Das wird bei anderen Dingen auch so gemacht. 
Dazu müsste man aber noch wissen bei welche Bedingung die alte libcap genommen werden muss, zb: kernel-, gcc-,uclibc-version?

harryboo, 2022-04-17:
Ist glaube ich die beste Loesung...
ich schau mal welche Version noch funktioniert (wann das mit dem EMPTY im Makefile umgestellt wurde)..

harryboo, 2022-04-17:
Also die letzte Version die bei mir funktioniert ist 2.49

Welche Kombi sollte man testen?

fda77, 2022-04-17:
Ich habe mit "LANG=C make" gebaut um die englische Version zu sehen
```
objcopy: Unable to recognise the format of the input file `empty'
```
Dazu findet Google mehr

fda77, 2022-04-18:
Die neue Version 2.6x kann nie funktioniert haben, ich hätte den PR testen sollen...
Version 2.49 funktioniert mit gcc 4 + 7490 6.9x, aber nicht mit einer der ältesten Fritzboxen 3020sl.
Mit dieser und 7170 + 7270v1 funktioniert die 2.24 aber auch nicht -.-
Die älteste die ich finden konnte ist die 7270v2 5.2x - mit der auch 2.49 funktioniert.
Also doch die alte aktualisieren. Auf älteren Geräten hat es bislang auch wohl niemand vermisst

closed 2022-04-18 by fda77

harryboo, 2022-04-18:
Danke fuer den Fix.
Jedoch habe ich mich gefragt warum der Fehler nicht schon vorher aufgefallen ist. Die Lösung war, das die libcap in der config nicht aktiviert ist und trotzdem mit compiliert wird. Somit benutzt tor die lib, welche jedoch nicht mit in das image gepackt wird und somit tor nicht startet.
[image]

harryboo, 2022-04-18:
Ich denke ich habe das Problem gefunden:
Mit dem commit ebe9573e2e173c2288e9d8050f10491a9694c489 gab es einen bump von bind (ich nutze daraus dig). Da ich die alte config immer nur lade wurden die neuen abhaengigkeiten nicht selectiert (aber komischerweise beim compilieren beruecksichtigt). Ich werde jetzt mal dig neu auswaehlen und berichten...

harryboo, 2022-04-18:
Ich habe jetzt dig mal deaktiviert und neu selektiert und es wird die abhaengigkeit libcap nicht mit selektiert aber kompiliert.
Da die libcap nun verfuegbar ist wird sie von tor verwendet, aber nicht mit in das image eingebunden.
Kannst Du das bitte bei Dir mal pruefen, ggf sollte die libcap bei tor als abhaengigkeit mit hinzugefuegt werden.

fda77, 2022-04-18:
Das ist etwas durcheinander. Zuerst allgemein:
 - ein Package kann in seiner Config.in eine lib mit "select" auswählen, die kommt dann auch ins Image
 - ein Package kann in seiner .mk mit .._DEPENDS_ON eine libs auswählen die VOR dem Package gebaut wird

Beide Optionen können unabhängig verwendet werden, manches wird nur zum bauen (zb statisch) gebraucht.
Eine lib kann weiter Abhängigkeiten haben haben

Bind baut immer alle binaries (könnte man evtl ändern), also werden auch alle libs vorher gebaut. Aber nur named braucht libcap auf der box und wählt sie aus.

Wenn man nur tor auswählt und baut sieht man (not-founds da es nicht auf der box ausgeführt wird)
```
toolchain/build/mips_gcc-8.3.0_uClibc-1.0.40-nptl_kernel-4.9/mips-linux-uclibc/bin/mips-linux-uclibc-ldd  packages/target-mips_gcc-8.3.0_uClibc-1.0.40-nptl_kernel-4.9/tor-0.4.6.10/root/usr/sbin/tor
        libz.so.1 => /lib/libz.so.1 (0x00000000)
        libevent-2.1.so.7 => not found (0x00000000)
        libssl.so.1.1 => not found (0x00000000)
        libcrypto.so.1.1 => not found (0x00000000)
        libc.so.0 => not found (0x00000000)
        libc.so.6 => /lib/libc.so.6 (0x00000000)
        /lib/ld-linux.so.2 => /lib/ld-linux.so.2 (0x00000000)
```

Nach einem `make libcap-precompiled` und `make tor-recompile`:
```
toolchain/build/mips_gcc-8.3.0_uClibc-1.0.40-nptl_kernel-4.9/mips-linux-uclibc/bin/mips-linux-uclibc-ldd  packages/target-mips_gcc-8.3.0_uClibc-1.0.40-nptl_kernel-4.9/tor-0.4.6.10/root/usr/sbin/tor
        libz.so.1 => /lib/libz.so.1 (0x00000000)
        libevent-2.1.so.7 => not found (0x00000000)
        libssl.so.1.1 => not found (0x00000000)
        libcrypto.so.1.1 => not found (0x00000000)
>>      libcap.so.2 => not found (0x00000000)
        libc.so.0 => not found (0x00000000)
        libc.so.6 => /lib/libc.so.6 (0x00000000)
        /lib/ld-linux.so.2 => /lib/ld-linux.so.2 (0x00000000)
```

Damit:
 - tor so konfigurieren/patchen dass es die libcap nie verwendet
 - oder tor die libcap in .mk und .in  hinzufügen

Da tor wohl auch ohne läuft (fehlt dadurch irgendwas?) würde ich das 1. nehmen.

Btw, nach einem repo-update reicht ein "olddefconfig" ("make help") um die .config zu aktualisieren. Es wird allerdings nichts automatisch deaktiviert

fda77, 2022-04-18:
Vergessen: https://github.com/Freetz-NG/freetz-ng/commit/693e4272781e28abc1d32f1e9a626210043d6c0c

harryboo, 2022-04-18:
Jo, funzt.
Ist die beste Loesung.

Danke.....