Freetz-NG/freetz-ng #523: Bei Aktivierung von usbutils -> Fehler, da Zugriff auf libatomic.la im falschen Verzeichnis erfolgt

AkteX, 2022-04-22:
Aktuelles Verzeichnis ist 7520-neu-1, aber libatomic.la wird beim make von usbutils-007 im falschen Verzeichnis freetz-ng anstatt im aktuellen Verzeichnis erfolgt

Make log:
libtool: compile:  /home/freetz/7520-neu-1/toolchain/build/arm_gcc-5.5.0_uClibc-1.0.40-nptl_kernel-4.4/arm-linux-uclibcgnueabi/bin/arm-linux-uclibcgnueabi-gcc -DHAVE_CONFIG_H -I. -I.. -I/home/freetz/7520-neu-1/source/target-arm_gcc-5.5.0_uClibc-1.0.40-nptl_kernel-4.4/usbutils-007/usbhid-dump -I/home/freetz/7520-neu-1/source/target-arm_gcc-5.5.0_uClibc-1.0.40-nptl_kernel-4.4/usbutils-007/usbhid-dump/include -DNDEBUG -Os -Wall -mcpu=cortex-a9 -mfpu=neon -mfloat-abi=softfp -marm -Ofast -pipe -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Wl,-I/usr/lib/freetz/ld-uClibc.so.1 -I/home/freetz/7520-neu-1/toolchain/build/arm_gcc-5.5.0_uClibc-1.0.40-nptl_kernel-4.4/arm-linux-uclibcgnueabi/usr/include/libusb-1.0 -MT dev_list.lo -MD -MP -MF .deps/dev_list.Tpo -c dev_list.c -o dev_list.o >/dev/null 2>&1
mv -f .deps/dev_list.Tpo .deps/dev_list.Plo
/bin/bash ../libtool --tag=CC   --mode=link /home/freetz/7520-neu-1/toolchain/build/arm_gcc-5.5.0_uClibc-1.0.40-nptl_kernel-4.4/arm-linux-uclibcgnueabi/bin/arm-linux-uclibcgnueabi-gcc  -Os -Wall -mcpu=cortex-a9 -mfpu=neon -mfloat-abi=softfp -marm -Ofast -pipe -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Wl,-I/usr/lib/freetz/ld-uClibc.so.1 -I/home/freetz/7520-neu-1/toolchain/build/arm_gcc-5.5.0_uClibc-1.0.40-nptl_kernel-4.4/arm-linux-uclibcgnueabi/usr/include/libusb-1.0   -o libuhd.la  dev.lo dev_list.lo iface.lo iface_list.lo libusb.lo  -L/home/freetz/7520-neu-1/toolchain/build/arm_gcc-5.5.0_uClibc-1.0.40-nptl_kernel-4.4/arm-linux-uclibcgnueabi/usr/lib -lusb-1.0
/usr/bin/grep: /home/freetz/**freetz-ng**/toolchain/build/arm_gcc-5.5.0_uClibc-1.0.40-nptl_kernel-4.4/arm-linux-uclibcgnueabi/arm-linux-uclibcgnueabi/lib/libatomic.la: No such file or directory
/usr/bin/sed: can't read /home/freetz/**freetz-ng**/toolchain/build/arm_gcc-5.5.0_uClibc-1.0.40-nptl_kernel-4.4/arm-linux-uclibcgnueabi/arm-linux-uclibcgnueabi/lib/libatomic.la: No such file or directory
libtool: link: `/home/freetz/**freetz-ng**/toolchain/build/arm_gcc-5.5.0_uClibc-1.0.40-nptl_kernel-4.4/arm-linux-uclibcgnueabi/arm-linux-uclibcgnueabi/lib/libatomic.la' is not a valid libtool archive
make[5]: *** [Makefile:286: libuhd.la] Error 1
make[5]: Leaving directory '/home/freetz/7520-neu-1/source/target-arm_gcc-5.5.0_uClibc-1.0.40-nptl_kernel-4.4/usbutils-007/usbhid-dump/lib'
make[4]: *** [Makefile:351: all-recursive] Error 1
make[4]: Leaving directory '/home/freetz/7520-neu-1/source/target-arm_gcc-5.5.0_uClibc-1.0.40-nptl_kernel-4.4/usbutils-007/usbhid-dump'
make[3]: *** [Makefile:282: all] Error 2
make[3]: Leaving directory '/home/freetz/7520-neu-1/source/target-arm_gcc-5.5.0_uClibc-1.0.40-nptl_kernel-4.4/usbutils-007/usbhid-dump'
make[2]: *** [Makefile:726: all-recursive] Error 1
make[2]: Leaving directory '/home/freetz/7520-neu-1/source/target-arm_gcc-5.5.0_uClibc-1.0.40-nptl_kernel-4.4/usbutils-007'
make[1]: *** [Makefile:343: all] Error 2
make[1]: Leaving directory '/home/freetz/7520-neu-1/source/target-arm_gcc-5.5.0_uClibc-1.0.40-nptl_kernel-4.4/usbutils-007'

AkteX, 2022-04-22:
Danke für die schnelle Rückmeldung.

Ich hatte precompiled Toolchain eingestellt und habe mich schon gewundert, warum in den statischen Bibliotheken immer der Pfad freetz-ng enthalten ist.

Ich versuche es jetzt mal mit der selbst gebauten Toolchain.

fda77, 2022-04-25:
-> b2ce78391

closed 2022-04-25 by fda77