Freetz-NG/freetz-ng #524: wireguard.ko invalid module format on 4060 ng22040

robse-git, 2022-04-23:
Hello,
I compiled ng22040 for the 4060.
All fine except that I have this error now on my router when trying wireguard:
`insmod: can't insert 'wireguard.ko': invalid module format`

I also tried in ssh:
`insmod /lib/modules/4.4.60/kernel/drivers/net/wireguard/wireguard.ko`
but it gives the same error.

Some info:
```
root@fritz:/var/mod/root# uname -a
[config.gz](https://github.com/Freetz-NG/freetz-ng/files/8547025/config.gz)

Linux fritz.box 4.4.60 #1 SMP PREEMPT 2021-10-29 armv7l GNU/Linux
```

I fetched the sources with:
`git clone https://github.com/Freetz-NG/freetz-ng ~/freetz-ng --single-branch --branch ng22040`

And compiled everything inside Freetz-Linux-2.0.0-Ubuntu (the downloadable image)

I attach the `.config` used (I did `make menuconfig; make`)

How can I solve this ?

Thank you,
Roberto
[config.txt](https://github.com/Freetz-NG/freetz-ng/files/8547031/config.txt)

fda77, 2022-04-23:
What shows `modinfo /lib/modules/4.4.60/kernel/drivers/net/wireguard/wireguard.ko` ?
Maybe wrong sources, its up to you, see: https://freetz-ng.github.io/freetz-ng/SUPPORT.html#quellcode

robse-git, 2022-04-23:
I don't have modinfo on the router, can you share the binary somewhere so I test ?

fda77, 2022-04-23:
I havo no Freetz on my router

robse-git, 2022-04-23:
> I havo no Freetz on my router

Interesting, what do you run on your router ?

robse-git, 2022-04-23:
I think I'll return this device to Amazon as I need a working wireguard

fda77, 2022-04-23:
Oky

closed 2022-04-23 by fda77