Freetz-NG/freetz-ng #528: radvd patch for mgmt and other flag: what am I missing?

jpsollie, 2022-05-01:
(view attachment)
the patch works, but the webGUI is incomplete:
after checking, the HTML source keeps saying:
```<input id="c3" type="checkbox" name="setother" value="on">
```
Whereas I'd expect a value 'checked' (which is also in the config).
What am I doing wrong?
[mgmt.patch.gz](https://github.com/Freetz-NG/freetz-ng/files/8599093/mgmt.patch.gz)

fda77, 2022-05-01:
Just add .txt to text files to attach here. Maybe you messed "id"s?
Btw, there is a newer and simpler ways for freetz cgis with "cgi_" functions, example: https://github.com/Freetz-NG/freetz-ng/blob/master/make/unbound/files/root/usr/lib/cgi-bin/unbound.cgi

fda77, 2022-05-01:
And im nor sure if "off" is valid, all other use yes/no

fda77, 2022-05-01:
And IDs?

closed 2022-05-20 by fda77

jpsollie, 2022-06-07:
I couldn't rewrite the GUI yet, but here's a temp patch for those interested in this topic:
- Allow to set managed and other config
- provide DNS server list
- set DNS domains to search
[radvd_patch.patch.txt](https://github.com/Freetz-NG/freetz-ng/files/8854024/radvd_patch.patch.txt)

fda77, 2022-06-07:
You also could keek the old syntax. But it seems you deleted "setipv6" from webif but it still exists in the config.
Does the new version now compile with kernel 2.6.13.1 even without 001-fix-NETLINK_NO_ENOBUFS-is-undefined.patch?

fda77, 2022-06-07:
Very old devices (eg kernel 2.6.13.1) only have ipv4 by avm, with these this options should be usefull as freetz adds ipv6.
Maybe this option could be hidden when the device is "newer" like yours
Translation: Just add de+en es it was, but put in german the same as in english, i could translate it later.
Bump: Can you also add this 4 links in the .mk like in https://github.com/Freetz-NG/freetz-ng/commit/80d33169eff88fc9db87a96a86b407cd51c3a248#diff-90b1f4c45a035e6d60e61f0838d4a866cf0fc16d4d1d733e03aebc1c02cc7e5e=

jpsollie, 2022-06-08:
radvd doesn't have a documentation site, so I put the generic website up there
the enobufs patch is already mainlined in 2.19
not yet sure about the libdaemon though ... it may work without it
[radvd_patch.patch.txt](https://github.com/Freetz-NG/freetz-ng/files/8858417/radvd_patch.patch.txt)

fda77, 2022-06-08:
I've changed variable assignment to have always a defined state and changed german strings

fda77, 2022-06-08:
Please test if the last commits works as expected

jpsollie, 2022-06-09:
something which may be interesting for this package (but not necessary):
I saw in the syslogs on my syslogserver the following about radvd:
ˋˋˋ
Jun  9 07:02:31 router/router radvd[8255]: our AdvCurHopLimit on lan doesn't agree with fe80::3ea6:2fff:fe0a:7c7c
Jun  9 07:02:31 router/router radvd[8255]: our AdvValidLifetime on lan for 2a02:578:8538:b00:: doesn't agree with fe80::3ea6:2fff:fe0a:7c7c
Jun  9 07:02:31 router/router radvd[8255]: our AdvPreferredLifetime on lan for 2a02:578:8538:b00:: doesn't agree with fe80::3ea6:2fff:fe0a:7c7c
ˋˋˋ
these messages flood every few seconds.
the link local IPv6 spoken of is the one of the lan interface.
Should we adapt the interface as well?

fda77, 2022-06-09:
I dont know if this is needed for radvd meanwhile. And i dont know how AVM handles IPv6 with newer Fritzos version.
If the loglines are just annoying, they could be patched away...