Freetz-NG/freetz-ng #531: Fehler "unpacking kernel failed"

pfichtner, 2022-05-08:
Mein wöchtentlicher Build bricht beim Bau für eine 3270 sowie eine 7390 ab (während der Build für 7570 bzw. Speedport W920V) funktioniert. 

Trat auf beim Build von df9bb23a93506f89fe138e4ea9dc58ca8c3b8e03, 44d8f160d2d0c1e0d2f23d0cb62f26af2fc0c042  war noch ok: Unterschied zwischen diesen zwei Commits: https://github.com/Freetz/freetz/compare/44d8f16...df9bb23

```
building ... make[1]: Entering directory '/tmp/workspace.KuKaKbeZ7Z/source/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/busybox-1.34.1'
make[1]: Leaving directory '/tmp/workspace.KuKaKbeZ7Z/source/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/busybox-1.34.1'
touch -c source/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/busybox-1.34.1/busybox.links
cat source/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/busybox-1.34.1/busybox.links | sed -r -e 's,/(blkid|wget)$,/\1-busybox,g' > packages/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/busybox/busybox.links
done.
ERROR: unpacking kernel failed
make: *** [make/linux/kernel.mk:158: source/kernel/ref-vr9-3390_06.51/linux-3.10/arch/mips/kernel/avm_kernel_config_area.fritz.box_wlan_3370-06.55.bin] Error 1
```

```
building ... make[1]: Entering directory '/tmp/workspace.pAd6RvaPW5/source/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/busybox-1.34.1'
make[1]: Leaving directory '/tmp/workspace.pAd6RvaPW5/source/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/busybox-1.34.1'
touch -c source/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/busybox-1.34.1/busybox.links
cat source/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/busybox-1.34.1/busybox.links | sed -r -e 's,/(ip|blkid|wget)$,/\1-busybox,g' > packages/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/busybox/busybox.links
done.
ERROR: unpacking kernel failed
make: *** [make/linux/kernel.mk:158: source/kernel/ref-vr9-7490_06.83/linux-3.10/arch/mips/kernel/avm_kernel_config_area.7490.113.06.83.bin] Error 1

```

fda77, 2022-05-08:
Ich tippe auf https://github.com/Freetz-NG/freetz-ng/commit/86698144c5c5c91bc3f1e2d64f8c979a148b30da, schau mal wie bei dir diese Symbole gesetzt sind

pfichtner, 2022-05-08:
Weder vor noch nach `make olddefconfig` ist `FREETZ_KERNEL_VERSION_3_10_MIN` oder `FREETZ_AVM_PROP_KERNEL_CONFIG_AREA_SIZE_0`'  gesetzt (und auch kein auskommentiertert Eintrag mit `is not set`)

fda77, 2022-05-08:
Komisch, an deinen Logs sieht man dass es 3.10 ist (auch wenn es von busybox ist, was hiermit vermutlich nichts zu tun hat).
Schau mal welche kernel Sachen sonst ausgewählt sind, und welche akc größe erkannt wurde

pfichtner, 2022-05-08:
```
$ grep KERNEL_VERSION .config
FREETZ_KERNEL_VERSION_2_6_28=y
FREETZ_KERNEL_VERSION="2.6.28.10"
FREETZ_KERNEL_VERSION_MAJOR="2.6.28"
FREETZ_KERNEL_VERSION_MODULES_SUBDIR="2.6.28.10"
FREETZ_KERNEL_VERSION_2_6_13_MIN=y
FREETZ_KERNEL_VERSION_2_6_19_MIN=y
FREETZ_KERNEL_VERSION_2_6_28_MIN=y
FREETZ_KERNEL_VERSION_2_6_28_MAX=y
FREETZ_KERNEL_VERSION_2_6_32_MAX=y
FREETZ_KERNEL_VERSION_2_6_39_MAX=y
FREETZ_KERNEL_VERSION_2_MAX=y
FREETZ_KERNEL_VERSION_3_10_MAX=y
FREETZ_KERNEL_VERSION_3_12_MAX=y
FREETZ_KERNEL_VERSION_3_MAX=y
FREETZ_KERNEL_VERSION_4_1_MAX=y
FREETZ_KERNEL_VERSION_4_4_MAX=y
FREETZ_KERNEL_VERSION_4_9_MAX=y
FREETZ_KERNEL_VERSION_2=y
```
Bezüglich akc habe ich Angaben wie `make[1]: Entering directory '/tmp/workspace.pAd6RvaPW5/source/host-tools/yf-akcarea-0'` gefunden (alles immer akcarea-0), das sah im funktionierenden Build auch so aus.

fda77, 2022-05-08:
2.6 ist nicht >=3.1
ausserdem passt die .config nicht zu den logs oben

pfichtner, 2022-05-08:
Du hast Recht, der Fehler passiert nachdem die 3270 (bzw, 7390) gebaut wurde. Auf der config für die 3270 baue ich noch für eine 3370 und aus der config für die 7390 noch eine 7490 (das habe ich aber bereits vorher auch gemacht, da ich aber auch dort Änderungen vorgenommen habe, will ich nicht ausschliessen, dass es auch daran liegen könnte). 
Es wird also die 3270 gebaut, dann ein make dirclean aufgerufen, dann die config gepatcht, dann make olddefconfig und make durchlaufen. 
Wenn Du keinen Ansatz hast, warum das ganze jetzt nicht mehr gehen sollte, wird der Fehler bei mir liegen, ich versuche mal den alten commit der ging mit meiner Änderung des builds erneut zu bauen,

pfichtner, 2022-05-09:
Ist reproduzierbar, Typ 3370 (hat also nix mit dem patchen der config zu tun, ich hatte nur fälschlicherweise in der .config vor dem patchen, also 3270 geschaut), so sieht es also "wirklich" in der .config aus: 
```
$ grep 'FREETZ_KERNEL_VERSION_3_10_MIN\|FREETZ_AVM_PROP_KERNEL_CONFIG_AREA' .config
FREETZ_AVM_PROP_KERNEL_CONFIG_AREA_SIZE_64=y
FREETZ_KERNEL_VERSION_3_10_MIN=y
```

Das führt dann zu o.g. Fehler: 
```
building ... make[1]: Entering directory '/workspace/source/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/busybox-1.34.1'
make[1]: Leaving directory '/workspace/source/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/busybox-1.34.1'
touch -c source/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/busybox-1.34.1/busybox.links
cat source/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/busybox-1.34.1/busybox.links | sed -r -e 's,/(blkid|wget)$,/\1-busybox,g' > packages/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/busybox/busybox.links
done.
ERROR: unpacking kernel failed
make: *** [make/linux/kernel.mk:158: source/kernel/ref-vr9-3390_06.51/linux-3.10/arch/mips/kernel/avm_kernel_config_area.fritz.box_wlan_3370-06.55.bin] Error 1
```

pfichtner, 2022-05-09:
Habe mal den angesprochenen commit reverted: 
```
git revert 86698144c5c5c91bc3f1e2d64f8c979a148b30da
make dirclean
make olddefconfig
make
```

Fehler bleibt bestehen: 
```
building ... make[1]: Entering directory '/workspace/source/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/busybox-1.34.1'
make[1]: Leaving directory '/workspace/source/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/busybox-1.34.1'
touch -c source/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/busybox-1.34.1/busybox.links
cat source/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/busybox-1.34.1/busybox.links | sed -r -e 's,/(blkid|wget)$,/\1-busybox,g' > packages/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/busybox/busybox.links
done.
ERROR: unpacking kernel failed
make: *** [make/linux/kernel.mk:159: source/kernel/ref-vr9-3390_06.51/linux-3.10/arch/mips/kernel/avm_kernel_config_area.fritz.box_wlan_3370-06.55.bin] Error 1
```

pfichtner, 2022-05-10:
Das sagt bisect:

b8e45c63c80548bc5c7dc00ff092cd4804994186 is the first bad commit
commit b8e45c63c80548bc5c7dc00ff092cd4804994186
Author: fda77 <fda77@users.noreply.github.com>
Date:   Fri Apr 29 20:48:03 2022 +0200

    tools: build some conditional

 Makefile               |  2 +-
 tools/make/Makefile.in | 33 +++++++++++++++++----------------
 2 files changed, 18 insertions(+), 17 deletions(-)

pfichtner, 2022-05-10:
Bisect manuell verifiziert: 5c5138a91a1a326439abe8cb7768d4882428c6d4 baut, b8e45c63c80548bc5c7dc00ff092cd4804994186 nicht

fda77, 2022-05-10:
Ich kann nicht nachvollziehen was du getan hast. Bei mir gehts

```

$ make distclean
for i in gcc    ; do \
        if [ -f /home/freetz/freetz-ng/toolchain/build/mips_gcc-4.8.5/mips-unknown-linux-gnu/bin-ccache/mips-unknown-linux-gnu-$i ] ; then \
                rm -f /home/freetz/freetz-ng/toolchain/build/mips_gcc-4.8.5/mips-unknown-linux-gnu/bin/mips-unknown-linux-gnu-$i; \
                mv /home/freetz/freetz-ng/toolchain/build/mips_gcc-4.8.5/mips-unknown-linux-gnu/bin-ccache/mips-unknown-linux-gnu-$i /home/freetz/freetz-ng/toolchain/build/mips_gcc-4.8.5/mips-unknown-linux-gnu/bin/; \
        fi; \
done
rm -f -r /home/freetz/freetz-ng/toolchain/build/mips_gcc-4.8.5/mips-unknown-linux-gnu/bin-ccache
rm -f /home/freetz/freetz-ng/toolchain/build/mips_gcc-4.8.5/mips-unknown-linux-gnu/bin/ccache
[ -d "/home/freetz/freetz-ng/source/toolchain-mips_gcc-4.8.5/ccache-4.6" ] && make -j9 -C /home/freetz/freetz-ng/source/toolchain-mips_gcc-4.8.5/ccache-4.6 clean
make: [toolchain/make/kernel/ccache/ccache.mk:92: ccache-kernel-clean] Fehler 1 (ignoriert)
rm -f -r /home/freetz/freetz-ng/source/toolchain-mips_gcc-4.8.5/ccache-4.6
rm -f -r source/toolchain/kernel
rm -f toolchain/kernel
rm -f /home/freetz/freetz-ng/toolchain/.kernel
for i in gcc g++; do \
        if [ -f /home/freetz/freetz-ng/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/mips-linux-uclibc/usr/bin-ccache/mips-linux-uclibc-$i ] ; then \
                rm -f /home/freetz/freetz-ng/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/mips-linux-uclibc/usr/bin/mips-linux-uclibc-$i; \
                mv /home/freetz/freetz-ng/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/mips-linux-uclibc/usr/bin-ccache/mips-linux-uclibc-$i /home/freetz/freetz-ng/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/mips-linux-uclibc/usr/bin/; \
        fi; \
done
rm -f -r /home/freetz/freetz-ng/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/mips-linux-uclibc/usr/bin-ccache
rm -f /home/freetz/freetz-ng/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/mips-linux-uclibc/usr/bin/ccache
[ -d "/home/freetz/freetz-ng/source/toolchain-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/ccache-4.6" ] && make -j9 -C /home/freetz/freetz-ng/source/toolchain-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/ccache-4.6 clean
make: [toolchain/make/target/ccache/ccache.mk:92: ccache-clean] Fehler 1 (ignoriert)
rm -f -r /home/freetz/freetz-ng/source/toolchain-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/ccache-4.6
rm -f -r \
        /home/freetz/freetz-ng/source/toolchain-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/libtool-1.5.26 \
        /home/freetz/freetz-ng/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/mips-linux-uclibc/usr/bin/libtool \
        /home/freetz/freetz-ng/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/mips-linux-uclibc/usr/bin/libtoolize \
        /home/freetz/freetz-ng/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/mips-linux-uclibc/usr/share/aclocal/libtool.m4 \
        /home/freetz/freetz-ng/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/mips-linux-uclibc/usr/share/aclocal/ltdl.m4 \
        /home/freetz/freetz-ng/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/mips-linux-uclibc/usr/share/libtool/
rm -f -r source/toolchain/target
rm -f toolchain/target
rm -f /home/freetz/freetz-ng/toolchain/.target
rm -f -r /home/freetz/freetz-ng/source/host-tools/avm-rle-0
rm -f tools/avm-rle-*
rm -f -r /home/freetz/freetz-ng/source/host-tools/busybox-1.34.1
find tools \( -lname busybox -o -name busybox \) -delete
rm -f -r /home/freetz/freetz-ng/source/host-tools/dtc-1.6.1
rm -f -r /home/freetz/freetz-ng/source/host-tools/e2fsprogs-1.42.13
rm -f tools/e2fsck tools/debugfs tools/tune2fs
rm -f -r /home/freetz/freetz-ng/source/host-tools/fakeroot-1.28
rm -f -r /home/freetz/freetz-ng/tools/build/bin/fakeroot /home/freetz/freetz-ng/tools/build/bin/faked /home/freetz/freetz-ng/tools/build/lib*/libfakeroot*
rm -f -r /home/freetz/freetz-ng/source/host-tools/find-squashfs-0
rm -f tools/find-squashfs
rm -f -r /home/freetz/freetz-ng/source/host-tools/fitimg-0.7.2
rm -f tools/fitimg
rm -f -r /home/freetz/freetz-ng/source/host-tools/genext2fs-4c1cc9468980448ca3e86db1cbe9600a4a084f5e
rm -f tools/genext2fs
rm -f -r /home/freetz/freetz-ng/source/host-tools/gmp-6.1.2
rm -f /home/freetz/freetz-ng/tools/build/lib/libgmp* /home/freetz/freetz-ng/tools/build/include/gmp*.h
rm -f -r /home/freetz/freetz-ng/source/host-tools/kconfig-v5.17
rm -f -r tools/kconfig/
rm -f -r /home/freetz/freetz-ng/source/host-tools/lzma1-465
rm -f tools/lzma
rm -f -r /home/freetz/freetz-ng/source/host-tools/xz-5.2.5
rm -f tools/xz
rm -f -r /home/freetz/freetz-ng/source/host-tools/lzma2eva-0
rm -f tools/lzma2eva tools/eva2lzma tools/bzimage2eva tools/eva2bzimage
rm -f -r /home/freetz/freetz-ng/source/host-tools/mklibs-0.1.34
rm -f -r /home/freetz/freetz-ng/tools/build/bin/mklibs /home/freetz/freetz-ng/tools/build/bin/mklibs-readelf
rm -f -r /home/freetz/freetz-ng/source/host-tools/mpc-1.1.0
rm -f /home/freetz/freetz-ng/tools/build/lib/libmpc* /home/freetz/freetz-ng/tools/build/include/*mpc*.h
rm -f -r /home/freetz/freetz-ng/source/host-tools/mpfr-3.1.6
[... 349 lines cut ...]
fixing ...
 Linking  "source/kernel/ref-iks-7390_06.80/linux-2.6.28/include/linux/new_capi_debug.h" to "../../drivers/isdn/capi_oslib/linux_new_capi_debug.h"
 Creating "source/kernel/ref-iks-7390_06.80/linux-2.6.28/drivers/block/virtual_fat/Makefile"
 Creating "source/kernel/ref-iks-7390_06.80/linux-2.6.28/drivers/char/avm_new/avm_dist_event/Makefile"
[... cut ...]
 Creating "source/kernel/ref-iks-7390_06.80/linux-2.6.28/firmware/dummy/Makefile"
 Creating "source/kernel/ref-iks-7390_06.80/linux-2.6.28/fusiv_src/kernel/drivers/bmedriver/Makefile"
 Creating "source/kernel/ref-iks-7390_06.80/linux-2.6.28/fusiv_src/kernel/drivers/policyVlan/Makefile"
 Creating "source/kernel/ref-iks-7390_06.80/linux-2.6.28/fusiv_src/kernel/drivers/voice/ipvoice/Makefile"
ln -s linux-2.6.28 source/kernel/ref-iks-7390_06.80/linux
touch source/kernel/ref-iks-7390_06.80/.unpacked
```

pfichtner, 2022-05-10:
Ich habe ganz normal mit meiner .config für eine 3370 gebaut (ist eigentlich eine 3270, make menuconfig, dann Typ auf 3370 geändert, ich weiß nicht, ob das relevant ist oder nicht). 

Mit dieser .config: 
`make dirclean && make olddefconfig && make`
5c5138a91a1a326439abe8cb7768d4882428c6d4 --> ok
b8e45c63c80548bc5c7dc00ff092cd4804994186 --> nicht ok

pfichtner, 2022-05-10:
Eine frisch angelegte 3370 .config baut auch bei mir ohne Fehler

pfichtner, 2022-05-10:
So, nachdem ich nach dem ändern von 3270 auf 3370 `distclean` statt `dirclean` aufrufe, funktioniert ist. 
Bis 5c5138a91a1a326439abe8cb7768d4882428c6d4 war ein `dirclean` ausreichend. Sollte das "zufälligerweise" funktioniert haben den Issue bitte schliesen. Sollte eigentlich auch ein `dirclean` ausreichend sein bitte sagen, was an Infos noch benötigt wird bzw. was ich tun kann.

pfichtner, 2022-05-11:
Ich konnte das Problem eingrenzen: 

In meiner .config (für die 3270_v3) findet sich **kein** Eintrag `FREETZ_HOSTTOOLS_DOWNLOAD=y`. Die 3270 baut damit trotzdem. 

Wenn ich nun den Typ auf eine 3370 ändere, funktioniert der build **nicht** (mit dem beschriebenen Fehler). 
Füge ich nun `FREETZ_HOSTTOOLS_DOWNLOAD=y` hinzu, funktioniert der build. 
Wie beschrieben, hat dies vor b8e45c63c80548bc5c7dc00ff092cd4804994186 noch funktioniert (ob gewollt oder nicht, also zufällig, kann ich nicht beurteilen). 

Noch eine Stolperfalle: Wird `make olddefconfig` aufgerufen und das USER_LEVEL ist zu niedrig, wird der Eintrag zu `FREETZ_HOSTTOOLS_DOWNLOAD` wieder aus der .config entfernt!

fda77, 2022-05-11:
Ich weiss es nicht genau.
Jedenfalls ist es so dass die precompiled-Dinge nur verfügbar sind ab "aptguru", dh wenn du kleineren Level nimmst werden die abgewählt. Jetzt könnte es sein dass da ein Fehler in den tools-Abhängigkeiten ist.
Allerdings wird bei dir ja für eine Box mit Kernel 2.6 aus unerfindlichen Gründen ein Verzeichnis mit kernel "3.1" im Namen verwendet. Das kann eigentlich nicht sein, und könnte aus das Fehlen der tools die erst ab kernel 3 nötig sind erklären.
Vielleicht ist da bei deinem "patchen" der .config was falsch. Vergleich doch mal die gepatchte und die ganz neu erstellt.
Man kann durchaus 1 Checkout für mehrere Geräte verwendet. Ich nutze allerdings pro Gerät 1 .config

freetz-ng-mod, 2022-05-19:
ERROR: unpacking kernel failed
make: *** [make/linux/kernel.mk:158: source/kernel/ref-vr9-7490_07.27/linux-3.10/arch/mips/kernel/avm_kernel_config_area.fritz.box_7490-07.29.bin] Fehler 1

Der Fehler kommt auch bei der 7490, sobald man nur 1 im Kernel modules anmacht
freetz neu ausgecheckt nur die box 7490 ausgesucht und auf  Level of user competence (Expert)  --->   gestellt und dann einfach aus Spaß FREETZ_MODULE_ftdi_sio ausgesucht

fda77, 2022-05-19:
Ich hab jetzt nochmals 12 Minuten darauf ver...wendet um zu orakeln was ihr anders macht als ich. Ich weiss es nicht, bei mir gehts:

```
installing modules
  ftdi_sio.ko
  generating modules.dep
  kernel modules installed: 69 entries in modules.dep and 69 .ko-files found.
  removing empty kernel module directories
    kernel/drivers/usb/musb
processing mod base options
  setting colored style as default
  patching security level (0)
  integrating .config is disabled
done.

STEP 3: PACK
  checking for left over version-control-system files
  integrate freetz info file into image
packing var.tar
creating inner-filesystem image (SquashFS4-xz)
  SquashFS block size: 64 kB (65536 bytes)
copying kernel image
  kernel image size: 2.6 MB, max 4.0 MB, free 1.4 MB (1504000 bytes)
creating outer-filesystem image (SquashFS4-xz)
copying filesystem image
  filesystem image size: 30.6 MB, max 48.0 MB, free 17.4 MB (18268160 bytes)
adding checksum to kernel.image
adding checksum to filesystem.image
packing images/7490_07.29.all.image
  packed image file size: 33.6 MB (35194880 bytes)
source firmware: 7490_de-es-fr-it-pl 113.07.29 rev92201 {ALL} [PSQ19P2NL4] (05.11.2021 11:15:11)
  source image file size: 33.1 MB (34672640 bytes)
done.

FINISHED

real    11m55,402s
user    51m18,211s
sys     7m17,690s
```

closed 2022-05-19 by fda77

freetz-ng-mod, 2022-05-20:
Wie @pfichtner es schon sagt, man muss 1-mal FREETZ_HOSTTOOLS_DOWNLOAD benutzen. Dann kann man es wieder aus machen und man kann in Kernel modules Sachen aussuchen ohne FREETZ_HOSTTOOLS_DOWNLOAD einmaliges geht kein Image Bau mit Kernel modules.

fda77, 2022-05-20:
Das ist aber ein bisschen was anderes wie oben das
> Der Fehler kommt auch bei der 7490, sobald man nur 1 im Kernel modules anmacht freetz neu ausgecheckt nur die box 7490 ausgesucht und auf Level of user competence (Expert) ---> gestellt und dann einfach aus Spaß FREETZ_MODULE_ftdi_sio ausgesucht.
Einigt euch auf einen Weg wie jeder das Problem nachstellen kann.
Hab ihr vielleicht noch alte Rest im Verzeichnis? Oder in der .config was altes? 

Es gibt jetzt jedenfalls keine "Fehler 1" Ausgabe mehr, ich hoffe dadurch bekommt ihr es hin eine aussagekräftige Fehlermeldung zu posten: e8af226ca

freetz-ng-mod, 2022-05-20:
Ich sagte ja, freetz ist ganz neu per clone geladen wurden bei meinen test. Eine ganz saubere config von freetz erstellen lassen mit 1 Kernel Module. Daher kann woll nichts altes drin sein.

Aber ich teste das nachher nochmal

fda77, 2022-05-20:
Bei dir muss aber etwas anders sein als bei mir bzw du hast was anders eingestellt. Ich hab ja oben das Ende von "make" angehängt und das lief durch, mit kernelmodul

freetz-ng-mod, 2022-05-20:
Ich habe nur die Änderungen gemacht wie oben schon gesagt. Mehr nicht, aber wie gesagt ich teste es nach der Arbeit noch mal

fda77, 2022-05-20:
Vielleicht hast du eine custom-config.in die irgend was automatisch auswählt?

freetz-ng-mod, 2022-05-20:
Vielleicht benutze ich sowas überhaupt nicht. Und ich sagte es ist ein saubers freetz, mit so was drin wer es nicht mehr sauber

freetz-ng-mod, 2022-05-20:
umask 0022
git clone https://github.com/Freetz-NG/freetz-ng
Klone nach 'freetz-ng' …
remote: Enumerating objects: 193786, done.
remote: Counting objects: 100% (1563/1563), done.
remote: Compressing objects: 100% (749/749), done.
remote: Total 193786 (delta 879), reused 1445 (delta 780), pack-reused 192223
Empfange Objekte: 100% (193786/193786), 69.35 MiB | 4.52 MiB/s, fertig.
Löse Unterschiede auf: 100% (133346/133346), fertig.
cd freetz-ng
umask 0022
make menuconfig
Da habe ich 3 sachen geändert
FREETZ_SHOW_EXPERT=y
FREETZ_TYPE_7490=y
FREETZ_MODULE_ftdi_sio=y
save gemacht
make -j20
Trying libraries: rt m crypt
 Library rt is not needed, excluding it
 Library m is not needed, excluding it
 Library crypt is not needed, excluding it
Final link with: <none>
make[1]: Verzeichnis „/home/xxxx/Ubuntu/freetz-ng/source/target-mips_gcc-5.5.0_uClibc-1.0.40-nptl_kernel-3.10/busybox-1.34.1“ wird verlassen
mkdir -p packages/target-mips_gcc-5.5.0_uClibc-1.0.40-nptl_kernel-3.10/busybox/; cp source/target-mips_gcc-5.5.0_uClibc-1.0.40-nptl_kernel-3.10/busybox-1.34.1/busybox packages/target-mips_gcc-5.5.0_uClibc-1.0.40-nptl_kernel-3.10/busybox/busybox; [ "y" == "y" ] && tools/patchelf --set-interpreter /usr/lib/freetz/ld-uClibc.so.1 packages/target-mips_gcc-5.5.0_uClibc-1.0.40-nptl_kernel-3.10/busybox/busybox || true; /home/xxxx/Ubuntu/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.40-nptl_kernel-3.10/mips-linux-uclibc/bin/mips-linux-uclibc-strip --remove-section={.comment,.note,.pdr} packages/target-mips_gcc-5.5.0_uClibc-1.0.40-nptl_kernel-3.10/busybox/busybox;
cmd() { PATH="/home/xxxx/Ubuntu/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.40-nptl_kernel-3.10/mips-linux-uclibc/bin:/home/xxxx/Ubuntu/freetz-ng/toolchain/build/mips_gcc-5.5.0/mips-unknown-linux-gnu/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin" LD_RUN_PATH="/usr/lib/freetz" FREETZ_LIBRARY_DIR="/usr/lib/freetz" make -j21  "$@"  || { kill $$; printf "\n\\033[33m%s\\033[m\n" "ERROR: Build failed.";  exit 1; } }; 				[ -n "" ] && step="/" || step=""; [ -n "" ] && step="$step/"; case "" in BIN)	echo -n "package/"         >source/.echo_item_tmp ;; LIB)	echo -n "library/"         >source/.echo_item_tmp ;; TOOL)	echo -n "tool/yf-bootmanager-host"           >source/.echo_item_tmp ;; KTC)	echo -n "toolchain/kernel$step" >source/.echo_item_tmp ;; TTC)	echo -n "toolchain/target$step" >source/.echo_item_tmp ;; KRN)	echo -n "kernel"                 >source/.echo_item_tmp ;; esac; if ! diff -q source/.echo_item_tmp source/.echo_item_new >/dev/null 2>&1 || [ ! -e source/.echo_item_1st ]; then 	if [ -e source/.echo_item_end -a -e source/.echo_item_new -a -e source/.echo_item_1st ]; then echo -e "\e[48;5;26mdone\e[49m."; rm -f source/.echo_item_end; fi; [ -s source/.echo_item_tmp ] && cat source/.echo_item_tmp > source/.echo_item_new 2>/dev/null; [ -s source/.echo_item_new ] || cat source/.echo_item_old > source/.echo_item_new 2>/dev/null; if [ -s "source/.echo_item_new" ]; then echo -ne "\e[48;5;90m---> "; cat source/.echo_item_new 2>/dev/null | tee source/.echo_item_old; echo -ne "\e[49m ... "; touch source/.echo_item_end; touch source/.echo_item_1st; fi; fi; echo -ne "\e[48;5;56mbuilding\e[49m ... "; cmd -C source/target-mips_gcc-5.5.0_uClibc-1.0.40-nptl_kernel-3.10/busybox-1.34.1 \
	CC="/home/xxxx/Ubuntu/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.40-nptl_kernel-3.10/mips-linux-uclibc/bin/mips-linux-uclibc-gcc" CROSS_COMPILE="/home/xxxx/Ubuntu/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.40-nptl_kernel-3.10/mips-linux-uclibc/bin/mips-linux-uclibc-" EXTRA_CFLAGS="-march=34kc -mtune=34kc -msoft-float -Ofast -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Wl,-I/usr/lib/freetz/ld-uClibc.so.1" ARCH="mips" \
	busybox.links
building ... make[1]: warning: -j21 forced in submake: resetting jobserver mode.
make[1]: Verzeichnis „/home/xxxx/Ubuntu/freetz-ng/source/target-mips_gcc-5.5.0_uClibc-1.0.40-nptl_kernel-3.10/busybox-1.34.1“ wird betreten
make[1]: Verzeichnis „/home/xxxx/Ubuntu/freetz-ng/source/target-mips_gcc-5.5.0_uClibc-1.0.40-nptl_kernel-3.10/busybox-1.34.1“ wird verlassen
touch -c source/target-mips_gcc-5.5.0_uClibc-1.0.40-nptl_kernel-3.10/busybox-1.34.1/busybox.links
cat source/target-mips_gcc-5.5.0_uClibc-1.0.40-nptl_kernel-3.10/busybox-1.34.1/busybox.links | sed -r -e 's,/(reboot|ip|blkid|wget)$,/\1-busybox,g' > packages/target-mips_gcc-5.5.0_uClibc-1.0.40-nptl_kernel-3.10/busybox/busybox.links
done.
ERROR: unpacking kernel failed
make: *** [make/linux/kernel.mk:158: source/kernel/ref-vr9-7490_07.27/linux-3.10/arch/mips/kernel/avm_kernel_config_area.fritz.box_7490-07.29.bin] Fehler 1

fda77, 2022-05-20:
Mist, da ist ja auch noch eine "Fehler 1" Ausgabe.
Ich glaub ich habs gefunden, sfk-host fehlt

freetz-ng-mod, 2022-05-20:
nun gehrt es ;-)

pfichtner, 2022-05-21:
Ich habe zumindest jetzt mal einen anderen Fehler, Fehler beim Bau von `libgcrypt`. Ich denke, das ist unabhängig von dem hier geschilderten Problem, tritt jetzt vermutlich auf, weil es davor gar nicht bis dorthin kam, vermute ich. 

```
libtool: link: mips-linux-ranlib .libs/libgcrypt.a
libtool: link: rm -fr .libs/libgcrypt.lax
libtool: link: ( cd ".libs" && rm -f "libgcrypt.la" && ln -s "../libgcrypt.la" "libgcrypt.la" )
/bin/bash ../libtool  --tag=CC   --mode=link /workspace/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/mips-linux-uclibc/bin/mips-linux-uclibc-gcc  -march=34kc -mtune=34kc -msoft-float -Os -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 --std=gnu99 -fvisibility=hidden -fno-delete-null-pointer-checks -Wall   -o mpicalc mpicalc-mpicalc.o libgcrypt.la -lgpg-error 
libtool: link: /workspace/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/mips-linux-uclibc/bin/mips-linux-uclibc-gcc -march=34kc -mtune=34kc -msoft-float -Os -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 --std=gnu99 -fvisibility=hidden -fno-delete-null-pointer-checks -Wall -o .libs/mpicalc mpicalc-mpicalc.o  ./.libs/libgcrypt.so /workspace/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/mips-linux-uclibc/usr/lib/libgpg-error.so
./.libs/libgcrypt.so: undefined reference to `_gcry_digest_spec_blake2b_512'
./.libs/libgcrypt.so: undefined reference to `blake2b_vl_hash'
collect2: error: ld returned 1 exit status
make[3]: *** [Makefile:754: mpicalc] Error 1
make[3]: Leaving directory '/workspace/source/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/libgcrypt-1.10.1/src'
make[2]: *** [Makefile:502: all-recursive] Error 1
make[2]: Leaving directory '/workspace/source/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/libgcrypt-1.10.1'
make[1]: *** [Makefile:434: all] Error 2
make[1]: Leaving directory '/workspace/source/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/libgcrypt-1.10.1'
make: *** [make/libs/libgcrypt/libgcrypt.mk:39: source/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/libgcrypt-1.10.1/src/.libs/libgcrypt.so.20.4.1] Terminated
make: *** Deleting file 'source/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/libgcrypt-1.10.1/src/.libs/libgcrypt.so.20.4.1'
```

pfichtner, 2022-05-22:
Baut nach revert von 331488b6e9475c4d19f9dcc5929f8d8c17cbe921, ich mache dazu einen Issue auf

fda77, 2022-05-22:
Ja, das hat hiermit nichts zu tun