Freetz-NG/freetz-ng #536: 7530ax : FIQ watchdog handling (and reboot)

Greg57070, 2022-05-20:
Hello guy,

I've tested Freetz-NG on a 7530AX, but it crash after boot.
I just got enough time to connect through SSH and disable watchdog.
(But even with watchedog disabled, I get some "FIQ watchdog handling" errors which also reboot the box after some minutes)

Please found below some logs :

root@fritz:/var/mod/root# cat /dev/debug 
[   22.609891][F1][tdmlink]ltdm_handledma:error(irq=0): RxDMAHandler own != 1 = 6 (mask=0x3f)!
[   37.642290][0]AVM_WATCHDOG_release(hdl=6 avmnexusd): success
[   39.709854][0]Line 0: xDSL link down
[   45.812530][2]eth3 (Int switch port: 3) (Logical Port: 3) (phyId: b) Link Up at 1000 mbps full duplex
[   46.815062][2]eth3 (Int switch port: 3) (Logical Port: 3) (phyId: b) Link DOWN.
[   47.349729][1]start slab_allocator-trace now  (use cat /proc/slab_allocators)
[   50.444776][2]AVM_WATCHDOG_release(hdl=6 avmnexusd): success
[   50.824513][2]eth3 (Int switch port: 3) (Logical Port: 3) (phyId: b) Link Up at 1000 mbps full duplex
[   53.631809][0]internet: dp_ipmasq_invalidate_masqentries: start: 0 mappings, 0 entries.
[   53.631848][0]internet: dp_ipmasq_invalidate_masqentries: end: 0 mappings, 0 entries.
[   53.636467][0]iptv: dp_ipmasq_invalidate_masqentries: start: 0 mappings, 0 entries.
[   53.636505][0]iptv: dp_ipmasq_invalidate_masqentries: end: 0 mappings, 0 entries.
[   53.636567][0]iptv: dp_ipmasq_invalidate_masqentries: start: 0 mappings, 0 entries.
[   53.636602][0]iptv: dp_ipmasq_invalidate_masqentries: end: 0 mappings, 0 entries.
[   53.647348][0]internet: ipmasq_flush: start: 0 mappings, 0 entries.
[   53.647396][0]internet: ipmasq_flush: end: 0 mappings, 0 entries.
[   53.647459][0]iptv: ipmasq_flush: start: 0 mappings, 0 entries.
[   53.647494][0]iptv: ipmasq_flush: end: 0 mappings, 0 entries.
[   55.210763][0]ubi0 error: ubi_open_volume: cannot open device 0, volume 1, error -16
[   56.382931][0][DECTDRV_ERR] API_FP_MAC_START_STOP_REQ 255 -> 0
[   56.384041][0][DECTDRV_ERR] wHandsetBitMask 0
[   56.409427][0][DECTDRV_ERR] fpCap ext data queueSize 20, avmTpui 16/390, etsiTpui 406/40
[   56.732716][0]AVM_WATCHDOG: System Init Ueberwachung abgeschlossen (243 s noch verfuegbar)
[   57.992819][2][DECTDRV_ERR] API_FP_MAC_START_STOP_REQ 0 -> 0
[   57.994078][2][DECTDRV_ERR] API_FP_MAC_START_STOP_REQ 0 -> 0
[   58.013435][2][DECTDRV_ERR] fpCap ext data queueSize 20, avmTpui 16/390, etsiTpui 406/40
[   58.021419][2][DECTDRV_ERR] fpCap ext data queueSize 20, avmTpui 16/390, etsiTpui 406/40
[   81.993868][0]AVM_WATCHDOG_disable()
[   81.994094][0][ar7wdt_hw_deinit]: Switching off watchdog ...
[  114.226143][F2]FIQ watchdog handling, slave CPU#2 caught!
[  114.226149][F0]FIQ watchdog handling, master: CPU#0, retry counter=10
[  114.226152][F1]FIQ watchdog handling, slave CPU#1 caught!
[  146.258203][F2]FIQ watchdog handling, slave CPU#2 caught!
[  146.258208][F1]FIQ watchdog handling, master: CPU#1, retry counter=9
[  146.258211][F0]FIQ watchdog handling, slave CPU#0 caught!
[  178.290264][F2]FIQ watchdog handling, slave CPU#2 caught!
[  178.290268][F1]FIQ watchdog handling, slave CPU#1 caught!
[  178.290272][F0]FIQ watchdog handling, master: CPU#0, retry counter=8
[  210.322324][F2]FIQ watchdog handling, master: CPU#2, retry counter=7
[  210.322328][F1]FIQ watchdog handling, slave CPU#1 caught!
[  210.322332][F0]FIQ watchdog handling, slave CPU#0 caught!
[  242.354384][F2]FIQ watchdog handling, master: CPU#2, retry counter=6
[  242.354388][F1]FIQ watchdog handling, slave CPU#1 caught!
[  242.354391][F0]FIQ watchdog handling, slave CPU#0 caught!
[  274.386443][F0]FIQ watchdog handling, slave CPU#0 caught!
[  274.386448][F2]FIQ watchdog handling, master: CPU#2, retry counter=5
[  274.386451][F1]FIQ watchdog handling, slave CPU#1 caught!
[  306.418503][F2]FIQ watchdog handling, master: CPU#2, retry counter=4
[  306.418508][F0]FIQ watchdog handling, slave CPU#0 caught!
[  306.418511][F1]FIQ watchdog handling, slave CPU#1 caught!
[  338.450564][F1]FIQ watchdog handling, slave CPU#1 caught!
[  338.450569][F2]FIQ watchdog handling, master: CPU#2, retry counter=3
[  338.450572][F0]FIQ watchdog handling, slave CPU#0 caught!
[  370.482622][F2]FIQ watchdog handling, master: CPU#2, retry counter=2
[  370.482627][F1]FIQ watchdog handling, slave CPU#1 caught!
[  370.482630][F0]FIQ watchdog handling, slave CPU#0 caught!
[  402.514684][F0]FIQ watchdog handling, master: CPU#0, retry counter=1
[  402.514689][F1]FIQ watchdog handling, slave CPU#1 caught!
[  402.514692][F2]FIQ watchdog handling, slave CPU#2 caught!

32 seconds after this last message displayed, a 7530AX restart.
[... 920 lines cut ...]
[   71.794236]   hdl=16, wland, disabled.
[   71.794243]   hdl=17, dsld, disabled.
[   71.794251]   hdl=18, deviceinfod, disabled.
[   71.794258]   hdl=19, kpid, disabled.
[   71.794266]   hdl=20, pbd, disabled.
[   71.794273]   hdl=21, aha, disabled.
[   71.794280]   hdl=22, meshd, disabled.
[   71.794288]   hdl=23, telefon, disabled.
[   71.794295]   hdl=24, voipd, disabled.
[   71.794302]   hdl=25, upnpd, disabled.
[   71.794314] [ar7wdt_hw_deinit]: Switching off watchdog ...
[  104.026432] FIQ watchdog handling, master: CPU#1, retry counter=10
[  104.026437] FIQ watchdog handling, slave CPU#2 caught!
[  104.026440] FIQ watchdog handling, slave CPU#0 caught!
[  136.058491] FIQ watchdog handling, slave CPU#0 caught!
[  136.058496] FIQ watchdog handling, master: CPU#1, retry counter=9
[  136.058499] FIQ watchdog handling, slave CPU#2 caught!
[  168.090549] FIQ watchdog handling, slave CPU#0 caught!
[  168.090554] FIQ watchdog handling, master: CPU#1, retry counter=8
[  168.090557] FIQ watchdog handling, slave CPU#2 caught!
[  200.122608] FIQ watchdog handling, master: CPU#0, retry counter=7
[  200.122612] FIQ watchdog handling, slave CPU#1 caught!
[  200.122616] FIQ watchdog handling, slave CPU#2 caught!

Best regards,

fda77, 2022-05-20:
Here is a discussion about this device https://github.com/Freetz-NG/freetz-ng/discussions/483 and other devices and other things..
 - `cat /proc/avm/wdt` shows what the watchdog is watching and some timeouts
 - workaround1: in menuconfig with user level "developer" you could select a disable-watchdog patch
 - workaround2: if you unlock (see discussion) and select older busybox version it should run
 - maybe "untrustedd" is related here, but i have lost track of the discussion...
 - if i remember correctly only a change in newer busybox causes the problem

I noticed in your logs:
```
[   24.427032] cfg80211: Calling CRDA to update world regulatory domain
[   24.538652] Loading HND Module
[   24.613657] igs_module_init:867:	 IGS 29 create network socket successful
[   27.580094] cfg80211: Calling CRDA to update world regulatory domain
[   29.542284] [0]system-load  56% loadavg 0.50 0.11 0.4 - task runtime:19% max:insmod 14%  pgstat: sum=80686 free=58943 slab=11115 alloc=10316/s fault=6476/s (sleep 4)
[   30.732068] cfg80211: Calling CRDA to update world regulatory domain
[   31.284231] wl: `' invalid for parameter `instance_base'
[   33.884061] cfg80211: Calling CRDA to update world regulatory domain
[   36.606472] wl: `' invalid for parameter `instance_base'
```

I'm not sure this is okay. Have you a serial console to see output with avm image to compare?

From /etc/init.d/bcm-wlan-drivers.sh
```
UNITLIST="0 1 2"
get_unit() {
    for unit in $UNITLIST; do
        wlx=`cat /proc/net/dev |grep wl$unit`
        if [ ${#wlx[@]} == 0 ]; then
            break
        fi
    done

     echo "$unit"
```
... and later where "get_unit" is used

Can you post `cat /proc/net/dev ` ? Seems there is no wl1 wl2 or wl3. And `lsmod`

freetz-ng-mod, 2022-05-25:
cat /proc/net/dev
```
Inter-|   Receive                                                |  Transmit
 face |bytes    packets errs drop fifo frame compressed multicast|bytes    packets errs drop fifo colls carrier compressed
guest_ct1:   15312      87    0    0    0     0          0         0        0       0    0    0    0     0       0          0
 bcmsw: 11288763168 12136208    0   37    0     0          0    145630 11023550256 12056692    0   23    0     0       0          0
  dsl0:       0       0    0    0    0     0          0         0        0       0    0    0    0     0       0          0
ppptty:       0       0    0    0    0     0          0         0        0       0    0    0    0     0       0          0
ip6gre0:       0       0    0    0    0     0          0         0        0       0    0    0    0     0       0          0
  sit0:       0       0    0    0    0     0          0         0        0       0    0    0    0     0       0          0
  ing0:       0       0    0    0    0     0          0         0        0       0    0    0    0     0       0          0
archer: 98451593  204470    0    0    0     0          0         0 100599083  237055    0    0    0     0       0          0
   lan: 43574561  308403    0    1    0     0          0         0 94276239  226867    0    0    0     0       0          0
    lo: 38811512  268583    0    0    0     0          0         0 38811512  268583    0    0    0     0       0          0
 wl1.1:       0       0    0    0    0  3229          0         0    16356      94    0    0    0     0       0          0
 wl0.1:       0       0    0    0    0 160479          0         0    16356      94    0    0    0     0       0          0
  adsl:       0       0    0    0    0     0          0         0        0       0    0    0    0     0       0          0
   wl1: 3780873   20474    0    0    0  3229          0         0 27689540   88632   77    0    0     0       0          0
   wl0:   49796     314    0    0    0 160479          0         0 18190965   67499   80    0    0     0       0          0
  eth0: 7980049141 6842531    0    1    0     0          0     69727 3309538242 5386154    0   23    0     0       0          0
  eth1:   45999     392    0    0    0     0          0        83   402125     551    0    0    0     0       0          0
  eth2: 3205705841 5051029    0   23    0     0          0      1208 7668277883 6452817    0    0    0     0       0          0
ip6tnl0:       0       0    0    0    0     0          0         0        0       0    0    0    0     0       0          0
 guest:   14408      96    0    0    0     0          0         0        0       0    0    0    0     0       0          0
  eth3:       0       0    0    0    0     0          0         0        0       0    0    0    0     0       0          0
  ifb0:       0       0    0    0    0     0          0         0        0       0    0    0    0     0       0          0
  ifb1:       0       0    0    0    0     0          0         0        0       0    0    0    0     0       0          0
lsmod
```
Module                  Size  Used by    Tainted: P  
cifs                  221184  2 
ecb                    16384  0 
arc4                   16384  0 
cmac                   16384  0 
md4                    16384  0 
krtp                  229376  0 
bcm_usb                16384  0 
ohci_platform          16384  0 
ohci_hcd               32768  1 ohci_platform
ehci_platform          16384  0 
ehci_hcd               45056  1 ehci_platform
userman_mod            77824  4 
nf_log_ipv4            16384  0 
nf_log_common          16384  1 nf_log_ipv4
xt_LOG                 16384  0 
xt_limit               16384  0 
xt_state               16384  0 
nf_conntrack_ipv4      20480  0 
nf_defrag_ipv4         16384  1 nf_conntrack_ipv4
xt_conntrack           16384  0 
nf_conntrack           69632  3 xt_state,nf_conntrack_ipv4,xt_conntrack
xt_SKIPLOG             16384  0 
xt_TCPMSS              16384  0 
nfnetlink              16384  0 
ip6table_mangle        16384  0 
ip6_tables             24576  1 ip6table_mangle
iptable_mangle         16384  0 
iptable_filter         16384  0 
ip_tables              20480  2 iptable_mangle,iptable_filter
xt_blog                16384  0 
xt_multiport           16384  0 
xt_mark                16384  0 
xt_u32                 16384  0 
xt_flowlabel           16384  0 
xt_mac_extend          16384  0 
xt_mac                 16384  0 
xt_DSCP                16384  0 
xt_dscp                16384  0 
adsldd                524288  0 
wl                   7028736  0 
[... 14 lines cut ...]
usb_common             16384  1 usbcore
bcmmcast               77824  3 igs,emf
bcm_enet              143360  0 
bcmxtmcfg             102400  1 adsldd
bcmxtmrtdrv            36864  1 bcmxtmcfg
archer                204800  1 wl
cmdlist                69632  1 archer
isdn_fbox_fon5       1286144  6 krtp
pktflow               278528  2 wl,archer
pcmlink               385024  4 avm_dect,capi_codec,isdn_fbox_fon5
bcm_license            45056  2 archer,pktflow
bcm_ingqos            225280  1 archer
chipinfo               16384  0 
bcmvlan               106496  0 
rtc_avm                16384  1 pcmlink
bcmlibs                28672  4 wl,bcm_enet,archer,pktflow
Piglet_noemif          40960  0 
bcm_pcie_hcd           45056  0 
wlcsm                  16384  2 hnd,bcm_pcie_hcd
ext2                   69632  0 
vfat                   20480  0 
fat                    61440  1 vfat
led_module            204800  6 

```

fda77, 2022-05-25:
Sieht gut aus. Aber welche Busybox??

freetz-ng-mod, 2022-05-26:
1.27

fda77, 2022-05-26:
Und mit der anderen?

freetz-ng-mod, 2022-05-26:
Kommt heute abend

fda77, 2022-05-26:
Und auch noch Ausgabe von `dmesg`

freetz-ng-mod, 2022-05-26:
[1.27.txt](https://github.com/Freetz-NG/freetz-ng/files/8781547/1.27.txt)
cat /proc/net/dev
```
Inter-|   Receive                                                |  Transmit
 face |bytes    packets errs drop fifo frame compressed multicast|bytes    packets errs drop fifo colls carrier compressed
guest_ct1:       0       0    0    0    0     0          0         0        0       0    0    0    0     0       0          0
 bcmsw: 1095101    2881    0    0    0     0          0       156  1096557    2896    0    0    0     0       0          0
  dsl0:       0       0    0    0    0     0          0         0        0       0    0    0    0     0       0          0
ppptty:       0       0    0    0    0     0          0         0        0       0    0    0    0     0       0          0
ip6gre0:       0       0    0    0    0     0          0         0        0       0    0    0    0     0       0          0
  sit0:       0       0    0    0    0     0          0         0        0       0    0    0    0     0       0          0
  ing0:       0       0    0    0    0     0          0         0        0       0    0    0    0     0       0          0
archer:  126548    1369    0    0    0     0          0         0   126548    1369    0    0    0     0       0          0
   lan:  921854    1319    0    1    0     0          0         0  1006230    2125    0    0    0     0       0          0
    lo:    5467      98    0    0    0     0          0         0     5467      98    0    0    0     0       0          0
  adsl:       0       0    0    0    0     0          0         0        0       0    0    0    0     0       0          0
  eth0:  951525    1383    0    0    0     0          0        68   137019    1410    0    0    0     0       0          0
  eth1:       0       0    0    0    0     0          0         0        0       0    0    0    0     0       0          0
  eth2:   14513     136    0    0    0     0          0         8    15256     159    0    0    0     0       0          0
ip6tnl0:       0       0    0    0    0     0          0         0        0       0    0    0    0     0       0          0
 guest:       0       0    0    0    0     0          0         0        0       0    0    0    0     0       0          0
  eth3:       0       0    0    0    0     0          0         0        0       0    0    0    0     0       0          0
  ifb0:       0       0    0    0    0     0          0         0        0       0    0    0    0     0       0          0
  ifb1:       0       0    0    0    0     0          0         0        0       0    0    0    0     0       0          0
```
lsmod
```
Module                  Size  Used by    Tainted: P  
krtp                  229376  0 
bcm_usb                16384  0 
ohci_platform          16384  0 
ohci_hcd               32768  1 ohci_platform
ehci_platform          16384  0 
ehci_hcd               45056  1 ehci_platform
userman_mod            77824  4 
nf_log_ipv4            16384  0 
nf_log_common          16384  1 nf_log_ipv4
xt_LOG                 16384  0 
xt_limit               16384  0 
xt_state               16384  0 
nf_conntrack_ipv4      20480  0 
nf_defrag_ipv4         16384  1 nf_conntrack_ipv4
xt_conntrack           16384  0 
nf_conntrack           69632  3 xt_state,nf_conntrack_ipv4,xt_conntrack
xt_SKIPLOG             16384  0 
xt_TCPMSS              16384  0 
nfnetlink              16384  0 
ip6table_mangle        16384  0 
ip6_tables             24576  1 ip6table_mangle
iptable_mangle         16384  0 
iptable_filter         16384  0 
ip_tables              20480  2 iptable_mangle,iptable_filter
xt_blog                16384  0 
xt_multiport           16384  0 
xt_mark                16384  0 
xt_u32                 16384  0 
xt_flowlabel           16384  0 
xt_mac_extend          16384  0 
xt_mac                 16384  0 
xt_DSCP                16384  0 
xt_dscp                16384  0 
adsldd                524288  0 
igs                    28672  0 
emf                    24576  1 igs
hnd                   286720  2 igs,emf
cfg80211              196608  0 
kdsldmod             1548288 21 userman_mod
dect_io                28672  0 
avm_dect              294912  1 dect_io
capi_codec            372736  0 
cchardev               24576  1 kdsldmod
cprocfsmod             20480  2 kdsldmod,cchardev
otp                    16384  0 
bcm_thermal            16384  0 
pwrmngtd               16384  0 
usbcore               143360  4 ohci_platform,ohci_hcd,ehci_platform,ehci_hcd
usb_common             16384  1 usbcore
bcmmcast               77824  3 igs,emf
bcm_enet              143360  0 
bcmxtmcfg             102400  1 adsldd
bcmxtmrtdrv            36864  1 bcmxtmcfg
archer                204800  0 
cmdlist                69632  1 archer
isdn_fbox_fon5       1286144  5 krtp
pktflow               278528  1 archer
bcm_license            45056  2 archer,pktflow
bcm_ingqos            225280  1 archer
pcmlink               385024  4 avm_dect,capi_codec,isdn_fbox_fon5
chipinfo               16384  0 
bcmvlan               106496  0 
bcmlibs                28672  3 bcm_enet,archer,pktflow
rtc_avm                16384  1 pcmlink
Piglet_noemif          40960  0 
bcm_pcie_hcd           45056  0 
wlcsm                  16384  2 hnd,bcm_pcie_hcd
ext2                   69632  0 
vfat                   20480  0 
fat                    61440  1 vfat
led_module            204800  6 
```
[1.34.txt](https://github.com/Freetz-NG/freetz-ng/files/8781577/1.34.txt)

freetz-ng-mod, 2022-05-26:
Kannst nicht mal ein Test Patch dafür schreiben, dass ich es testen kann

fda77, 2022-05-26:
image bauen
build/original/filesystem/etc/init.d/bcm-wlan-drivers.sh editieren
nochmal "make"

zum aufräumen/zurücksetzen: "rm -rf build"

freetz-ng-mod, 2022-05-26:
ok warte ich teste es

fda77, 2022-05-26:
Also du kannst auch auf der box die Datei selbst aufrufen, zb mit
"sh -x /etc/init.d/bcm-wlan-drivers.sh"
Ich weiss jetzt aber nicht ob die Parameter erwartet

freetz-ng-mod, 2022-05-26:
Da habe ich ja wieder 1.27 am Laufen. Denn ich weiß nicht, ob ich das in 2 Min. schaffe, wenn ich auf 1.34 wieder gehe. Daher baue ich das grade mal ganz neu mit 1.34

fda77, 2022-05-26:
Ich verwechsele die 2 fit-boxen mit den Problemen immer ... reicht es bei dieser nicht den Watchdog auszuschalten damit sie nicht rebootet?

freetz-ng-mod, 2022-05-26:
Auch mit dem Dog rebootet sie. Komischer weiße hilft das aber  `echo disable > /dev/watchdog` frag nicht, warum das geht und das andere nicht.
7530AX BB 1.27
5530 geht zurzeit nur mit FREETZ_REMOVE_UNTRUSTEDD (da kann ich aber leider nicht weiter Testen da verkauft)

fda77, 2022-05-26:
Freetz macht das auch mit aktiviertem Patch: https://github.com/Freetz-NG/freetz-ng/blob/master/make/mod/files/root/etc/init.d/rc.mod#L82=
Was steht dem im log (freetz-webif)? Vielleicht wird das zu früh ausgeführt?

freetz-ng-mod, 2022-05-26:
Ich gebe den Befehl ca. 30 SEC nach dem Neustart ein und da bleib sie on. Habe es 3-mal getestet. So das neue Image ist nun auf der Box warte auf den neu start

freetz-ng-mod, 2022-05-26:
Reboot ca. nach 2 min.

fda77, 2022-05-26:
Schau mit "lsmod" ob das "wl" geladen ist.

fda77, 2022-05-26:
Und die ausgabe von 
sh -x /etc/init.d/bcm-wlan-drivers.sh
bzw der Teil wo die ein Fehler auffällt

freetz-ng-mod, 2022-05-26:
```
Module                  Size  Used by    Tainted: P  
krtp                  229376  0 
userman_mod            77824  4 
bcm_usb                16384  0 
ohci_platform          16384  0 
ohci_hcd               32768  1 ohci_platform
ehci_platform          16384  0 
ehci_hcd               45056  1 ehci_platform
nf_log_ipv4            16384  0 
nf_log_common          16384  1 nf_log_ipv4
xt_LOG                 16384  0 
xt_limit               16384  0 
xt_state               16384  0 
nf_conntrack_ipv4      20480  0 
nf_defrag_ipv4         16384  1 nf_conntrack_ipv4
xt_conntrack           16384  0 
nf_conntrack           69632  3 xt_state,nf_conntrack_ipv4,xt_conntrack
xt_SKIPLOG             16384  0 
xt_TCPMSS              16384  0 
nfnetlink              16384  0 
ip6table_mangle        16384  0 
ip6_tables             24576  1 ip6table_mangle
iptable_mangle         16384  0 
iptable_filter         16384  0 
ip_tables              20480  2 iptable_mangle,iptable_filter
xt_blog                16384  0 
xt_multiport           16384  0 
xt_mark                16384  0 
xt_u32                 16384  0 
xt_flowlabel           16384  0 
xt_mac_extend          16384  0 
xt_mac                 16384  0 
xt_DSCP                16384  0 
xt_dscp                16384  0 
adsldd                524288  0 
igs                    28672  0 
emf                    24576  1 igs
hnd                   286720  2 igs,emf
cfg80211              196608  0 
kdsldmod             1548288 19 userman_mod
dect_io                28672  0 
avm_dect              294912  1 dect_io
capi_codec            372736  0 
cchardev               24576  1 kdsldmod
cprocfsmod             20480  2 kdsldmod,cchardev
otp                    16384  0 
bcm_thermal            16384  0 
pwrmngtd               16384  0 
usbcore               143360  4 ohci_platform,ohci_hcd,ehci_platform,ehci_hcd
usb_common             16384  1 usbcore
bcmmcast               77824  3 igs,emf
bcm_enet              143360  0 
bcmxtmcfg             102400  1 adsldd
bcmxtmrtdrv            36864  1 bcmxtmcfg
archer                204800  0 
cmdlist                69632  1 archer
isdn_fbox_fon5       1286144  6 krtp
pktflow               278528  1 archer
pcmlink               385024  4 avm_dect,capi_codec,isdn_fbox_fon5
bcm_license            45056  2 archer,pktflow
bcm_ingqos            225280  1 archer
chipinfo               16384  0 
bcmvlan               106496  0 
bcmlibs                28672  3 bcm_enet,archer,pktflow
rtc_avm                16384  1 pcmlink
Piglet_noemif          40960  0 
bcm_pcie_hcd           45056  0 
wlcsm                  16384  2 hnd,bcm_pcie_hcd
ext2                   69632  0 
vfat                   20480  0 
fat                    61440  1 vfat
led_module            204800  6 
```
Nun hat die Box rebootet.

fda77, 2022-05-26:
Wie es da steht, "start" ist wohl am besten:
`sh -x  /etc/init.d/bcm-wlan-drivers.sh start`

freetz-ng-mod, 2022-05-26:
Mit dog rebootet die Box ca. nach 8 Min.
Muss warten bis sie wieder on ist

freetz-ng-mod, 2022-05-26:
sh -x /etc/init.d/bcm-wlan-drivers.sh start
```
+ trap  3
+ KERNELVER=4.1.52
+ HNDROUTER=y
+ CPEROUTER=y
+ WLAN_BTEST=
+ PROD_FW_PATH=
+ CONFIG_WLDPDCTL=
+ BRCM_CHIP=63178
+ WLAN_REMOVE_INTERNAL_DEBUG=
+ '[' '!' -z ]
+ dhd_console_ms=250
+ '[' '!' -z ]
+ MFG_FW_PATH=/etc/wlan/dhd/mfg
+ WLDPDCTL=0
+ UNITLIST='0 1 2'
+ '[' 1 -lt 1 ]
+ '[' '!' -z y ]
+ '[' -z y ]
+ WLAN_BTEST_FILEPATH=/proc/brcm/blxparms/wlan_btest
+ '[' -e /proc/brcm/blxparms/wlan_btest ]
+ nvram kget wlan_btest
+ btest_mode=
+ '['  '==' 1 ]
+ '['  '==' y ]
+ '[' 63178 '==' 47189 ]
+ '[' '!' -d /lib/modules/4.1.52/extra ]
+ '[' '!' -z  ]
+ '[' '!' -z y ]
+ '[' -f /lib/modules/4.1.52/kernel/net/wireless/cfg80211.ko ]
+ all_wlan_modules='firmware_class cfg80211 hnd emf igs dhd wl'
+ nvram show
+ grep kernel_mods
+ grep -c ^
+ is_nvmodules_list=0
+ '[' 0 -eq 0 ]
+ echo 'no modules list in nvram, using defaults...'
no modules list in nvram, using defaults...
+ modules_list='firmware_class cfg80211 hnd emf igs dhd wl'
+ unload_modules_list='firmware_class '
+ unload_modules_list='cfg80211 firmware_class '
+ unload_modules_list='hnd cfg80211 firmware_class '
+ unload_modules_list='emf hnd cfg80211 firmware_class '
+ unload_modules_list='igs emf hnd cfg80211 firmware_class '
+ unload_modules_list='dhd igs emf hnd cfg80211 firmware_class '
+ unload_modules_list='wl dhd igs emf hnd cfg80211 firmware_class '
+ load_modules
+ '[' '!' -z y ]
+ dhd_module_params='iface_name=wl dhd_console_ms=250'
+ '[' 63178 '==' 47189 ]
+ wl_module_params='intf_name=wl%d'
+ wl_mfgtest_module=wl_mfgtest
+ hnd_mfgtest_module=hnd_mfgtest
+ '[' -f /proc/nvram/wl_nand_manufacturer ]
+ cat /proc/nvram/wl_nand_manufacturer
+ is_mfg=0
+ wldpdctl_init
+ '[' '!' -z y ]
+ '[' '!' -z ]
+ '[' -z 0 ]
+ '[' 0 '=' 1 ]
+ echo 'loading WLAN kernel modules ... firmware_class cfg80211 hnd emf igs dhd wl'
loading WLAN kernel modules ... firmware_class cfg80211 hnd emf igs dhd wl
+ '[' -f /lib/modules/4.1.52/kernel/drivers/base/firmware_loader/firmware_class.ko ]
+ '[' firmware_class '==' dhd ]
+ '[' firmware_class '==' wl ]
+ '[' firmware_class '==' wl_mfgtest ]
+ '[' -e /lib/modules/4.1.52/extra/firmware_class.ko ]
+ insmod /lib/modules/4.1.52/kernel/net/wireless/cfg80211.ko
insmod: can't insert '/lib/modules/4.1.52/kernel/net/wireless/cfg80211.ko': File exists
+ '[' cfg80211 '==' dhd ]
+ '[' cfg80211 '==' wl ]
+ '[' cfg80211 '==' wl_mfgtest ]
+ '[' -e /lib/modules/4.1.52/extra/cfg80211.ko ]
+ module_params=
+ '[' hnd '==' dhd ]
+ '[' hnd '==' wl ]
+ '[' hnd '==' wl_mfgtest ]
+ '[' -e /lib/modules/4.1.52/extra/hnd.ko ]
+ '[' hnd '==' dhd ]
+ nvram kget wl_path
+ nic_nvram=
+ '[' hnd '==' wl ]
+ insmod /lib/modules/4.1.52/extra/hnd.ko
insmod: can't insert '/lib/modules/4.1.52/extra/hnd.ko': File exists
+ module_params=
+ '[' emf '==' dhd ]
+ '[' emf '==' wl ]
+ '[' emf '==' wl_mfgtest ]
+ '[' -e /lib/modules/4.1.52/extra/emf.ko ]
+ '[' emf '==' dhd ]
+ nvram kget wl_path
+ nic_nvram=
+ '[' emf '==' wl ]
+ insmod /lib/modules/4.1.52/extra/emf.ko
insmod: can't insert '/lib/modules/4.1.52/extra/emf.ko': File exists
+ module_params=
+ '[' igs '==' dhd ]
+ '[' igs '==' wl ]
+ '[' igs '==' wl_mfgtest ]
+ '[' -e /lib/modules/4.1.52/extra/igs.ko ]
+ '[' igs '==' dhd ]
+ nvram kget wl_path
+ nic_nvram=
+ '[' igs '==' wl ]
+ insmod /lib/modules/4.1.52/extra/igs.ko
insmod: can't insert '/lib/modules/4.1.52/extra/igs.ko': File exists
+ module_params='iface_name=wl dhd_console_ms=250'
+ '[' dhd '==' dhd ]
+ get_unit
+ cat /proc/net/dev
+ grep wl0
+ wlx=
/etc/init.d/bcm-wlan-drivers.sh: line 416: syntax error: bad substitution
+ idx=
+ instance_base='instance_base='
+ module_params='iface_name=wl dhd_console_ms=250 instance_base='
+ '[' -e /lib/modules/4.1.52/extra/dhd.ko ]
+ module_params='intf_name=wl%d'
+ '[' wl '==' dhd ]
+ '[' wl '==' wl ]
+ get_unit
+ cat /proc/net/dev
+ grep wl0
+ wlx=
/etc/init.d/bcm-wlan-drivers.sh: line 416: syntax error: bad substitution
+ idx=
+ instance_base='instance_base='
+ module_params='intf_name=wl%d instance_base='
+ '[' -e /lib/modules/4.1.52/extra/wl.ko ]
+ '[' wl '==' dhd ]
+ nvram kget wl_path
+ nic_nvram=
+ '[' wl '==' wl ]
+ '[' -f  ]
+ insmod /lib/modules/4.1.52/extra/wl.ko 'intf_name=wl%d' 'instance_base='
insmod: can't insert '/lib/modules/4.1.52/extra/wl.ko': Invalid argument
+ wlaffinity auto
Configuring wlan related threads...
no thread pin applied
Wlan related processes are pinned by default
+ exit 0
```

fda77, 2022-05-26:
Das sieht doch schon mal gut aus:
```
/etc/init.d/bcm-wlan-drivers.sh: line 416: syntax error: bad substitution
```

Und später
```
insmod /lib/modules/4.1.52/extra/wl.ko 'intf_name=wl%d' 'instance_base='
insmod: can't insert '/lib/modules/4.1.52/extra/wl.ko': Invalid argument
```

da muss ich mal suchen

freetz-ng-mod, 2022-05-26:
Soll ich noch mal wieder aufs original umstellen also auf `module_params=$module_params" "$instance_base` und noch mal testen

fda77, 2022-05-26:
Ich kann mir nicht vorstellen dass "module_params" so gewollt ist wie es im original ist.
Ich hab die .31 und .39-inhaus verglichen
```
     for unit in $UNITLIST; do
         wlx=`cat /proc/net/dev |grep wl$unit`
-        if [ ${#wlx[@]} == 0 ]; then
+        if [ -z "$wlx" ]; then
             break
         fi
     done
```

Das könnte der Fehler sein. Entweder wieder in build/original/.... editieren oder wenn du schnell bist auf der Box

```
sed 's/if [ ${#wlx[@]} == 0 ]; then/if [ -z "$wlx" ]; then/g' /etc/init.d/bcm-wlan-drivers.sh > /tmp/bcm-wlan-drivers.sh ; chmod +x /tmp/bcm-wlan-drivers.sh ; sh -x /tmp/bcm-wlan-drivers.sh start
```
ich hoffe der befehl stimmt so

freetz-ng-mod, 2022-05-26:
sed 's/if [ ${#wlx[@]} == 0 ]; then/if [ -z "$wlx" ]; then/g' /etc/init.d/bcm-wlan-drivers.sh > /tmp/bcm-wlan-drivers.sh ; chmod +x /tmp/bcm-wlan-drivers.sh ; sh -x /tmp/bcm-wlan-drivers.sh start
```
+ trap  3
+ KERNELVER=4.1.52
+ HNDROUTER=y
+ CPEROUTER=y
+ WLAN_BTEST=
+ PROD_FW_PATH=
+ CONFIG_WLDPDCTL=
+ BRCM_CHIP=63178
+ WLAN_REMOVE_INTERNAL_DEBUG=
+ '[' '!' -z ]
+ dhd_console_ms=250
+ '[' '!' -z ]
+ MFG_FW_PATH=/etc/wlan/dhd/mfg
+ WLDPDCTL=0
+ UNITLIST='0 1 2'
+ '[' 1 -lt 1 ]
+ '[' '!' -z y ]
+ '[' -z y ]
+ WLAN_BTEST_FILEPATH=/proc/brcm/blxparms/wlan_btest
+ '[' -e /proc/brcm/blxparms/wlan_btest ]
+ nvram kget wlan_btest
+ btest_mode=
+ '['  '==' 1 ]
+ '['  '==' y ]
+ '[' 63178 '==' 47189 ]
+ '[' '!' -d /lib/modules/4.1.52/extra ]
+ '[' '!' -z  ]
+ '[' '!' -z y ]
+ '[' -f /lib/modules/4.1.52/kernel/net/wireless/cfg80211.ko ]
+ all_wlan_modules='firmware_class cfg80211 hnd emf igs dhd wl'
+ nvram show
+ grep kernel_mods
+ grep -c ^
+ is_nvmodules_list=0
+ '[' 0 -eq 0 ]
+ echo 'no modules list in nvram, using defaults...'
no modules list in nvram, using defaults...
+ modules_list='firmware_class cfg80211 hnd emf igs dhd wl'
+ unload_modules_list='firmware_class '
+ unload_modules_list='cfg80211 firmware_class '
+ unload_modules_list='hnd cfg80211 firmware_class '
+ unload_modules_list='emf hnd cfg80211 firmware_class '
+ unload_modules_list='igs emf hnd cfg80211 firmware_class '
+ unload_modules_list='dhd igs emf hnd cfg80211 firmware_class '
+ unload_modules_list='wl dhd igs emf hnd cfg80211 firmware_class '
+ load_modules
+ '[' '!' -z y ]
+ dhd_module_params='iface_name=wl dhd_console_ms=250'
+ '[' 63178 '==' 47189 ]
+ wl_module_params='intf_name=wl%d'
+ wl_mfgtest_module=wl_mfgtest
+ hnd_mfgtest_module=hnd_mfgtest
+ '[' -f /proc/nvram/wl_nand_manufacturer ]
+ cat /proc/nvram/wl_nand_manufacturer
+ is_mfg=0
+ wldpdctl_init
+ '[' '!' -z y ]
+ '[' '!' -z ]
+ '[' -z 0 ]
+ '[' 0 '=' 1 ]
+ echo 'loading WLAN kernel modules ... firmware_class cfg80211 hnd emf igs dhd wl'
loading WLAN kernel modules ... firmware_class cfg80211 hnd emf igs dhd wl
+ '[' -f /lib/modules/4.1.52/kernel/drivers/base/firmware_loader/firmware_class.ko ]
+ '[' firmware_class '==' dhd ]
+ '[' firmware_class '==' wl ]
+ '[' firmware_class '==' wl_mfgtest ]
+ '[' -e /lib/modules/4.1.52/extra/firmware_class.ko ]
+ insmod /lib/modules/4.1.52/kernel/net/wireless/cfg80211.ko
insmod: can't insert '/lib/modules/4.1.52/kernel/net/wireless/cfg80211.ko': File exists
+ '[' cfg80211 '==' dhd ]
+ '[' cfg80211 '==' wl ]
+ '[' cfg80211 '==' wl_mfgtest ]
+ '[' -e /lib/modules/4.1.52/extra/cfg80211.ko ]
+ module_params=
+ '[' hnd '==' dhd ]
+ '[' hnd '==' wl ]
+ '[' hnd '==' wl_mfgtest ]
+ '[' -e /lib/modules/4.1.52/extra/hnd.ko ]
+ '[' hnd '==' dhd ]
+ nvram kget wl_path
+ nic_nvram=
+ '[' hnd '==' wl ]
+ insmod /lib/modules/4.1.52/extra/hnd.ko
insmod: can't insert '/lib/modules/4.1.52/extra/hnd.ko': File exists
+ module_params=
+ '[' emf '==' dhd ]
+ '[' emf '==' wl ]
+ '[' emf '==' wl_mfgtest ]
+ '[' -e /lib/modules/4.1.52/extra/emf.ko ]
+ '[' emf '==' dhd ]
+ nvram kget wl_path
+ nic_nvram=
+ '[' emf '==' wl ]
+ insmod /lib/modules/4.1.52/extra/emf.ko
insmod: can't insert '/lib/modules/4.1.52/extra/emf.ko': File exists
+ module_params=
+ '[' igs '==' dhd ]
+ '[' igs '==' wl ]
+ '[' igs '==' wl_mfgtest ]
+ '[' -e /lib/modules/4.1.52/extra/igs.ko ]
+ '[' igs '==' dhd ]
+ nvram kget wl_path
+ nic_nvram=
+ '[' igs '==' wl ]
+ insmod /lib/modules/4.1.52/extra/igs.ko
insmod: can't insert '/lib/modules/4.1.52/extra/igs.ko': File exists
+ module_params='iface_name=wl dhd_console_ms=250'
+ '[' dhd '==' dhd ]
+ get_unit
+ cat /proc/net/dev
+ grep wl0
+ wlx=
/tmp/bcm-wlan-drivers.sh: line 416: syntax error: bad substitution
+ idx=
+ instance_base='instance_base='
+ module_params='iface_name=wl dhd_console_ms=250 instance_base='
+ '[' -e /lib/modules/4.1.52/extra/dhd.ko ]
+ module_params='intf_name=wl%d'
+ '[' wl '==' dhd ]
+ '[' wl '==' wl ]
+ get_unit
+ cat /proc/net/dev
+ grep wl0
+ wlx=
/tmp/bcm-wlan-drivers.sh: line 416: syntax error: bad substitution
+ idx=
+ instance_base='instance_base='
+ module_params='intf_name=wl%d instance_base='
+ '[' -e /lib/modules/4.1.52/extra/wl.ko ]
+ '[' wl '==' dhd ]
+ nvram kget wl_path
+ nic_nvram=
+ '[' wl '==' wl ]
+ '[' -f  ]
+ insmod /lib/modules/4.1.52/extra/wl.ko 'intf_name=wl%d' 'instance_base='
insmod: can't insert '/lib/modules/4.1.52/extra/wl.ko': Invalid argument
+ wlaffinity auto
Configuring wlan related threads...
no thread pin applied
Wlan related processes are pinned by default
+ exit 0

```

fda77, 2022-05-26:
Sorry, das sed war nicht okay und hat nichts geändert... Noch ein versuch:
```
sed 's/if . $.#wlx.@.. == 0 .; then/if [ -z "$wlx" ]; then/g'   /etc/init.d/bcm-wlan-drivers.sh > /tmp/bcm-wlan-drivers.sh ; chmod +x /tmp/bcm-wlan-drivers.sh ; sh -x /tmp/bcm-wlan-drivers.sh start

```

freetz-ng-mod, 2022-05-26:
sed 's/if . $.#wlx.@.. == 0 .; then/if [ -z "$wlx" ]; then/g'   /etc/init.d/bcm-wlan-drivers.sh > /tmp/bcm-wlan-drivers.sh ; chmod +x /tmp/bcm-wlan-drivers.sh ; sh -x /tmp/bcm-wlan-drivers.sh start
```
+ trap  3
+ KERNELVER=4.1.52
+ HNDROUTER=y
+ CPEROUTER=y
+ WLAN_BTEST=
+ PROD_FW_PATH=
+ CONFIG_WLDPDCTL=
+ BRCM_CHIP=63178
+ WLAN_REMOVE_INTERNAL_DEBUG=
+ '[' '!' -z ]
+ dhd_console_ms=250
+ '[' '!' -z ]
+ MFG_FW_PATH=/etc/wlan/dhd/mfg
+ WLDPDCTL=0
+ UNITLIST='0 1 2'
+ '[' 1 -lt 1 ]
+ '[' '!' -z y ]
+ '[' -z y ]
+ WLAN_BTEST_FILEPATH=/proc/brcm/blxparms/wlan_btest
+ '[' -e /proc/brcm/blxparms/wlan_btest ]
+ nvram kget wlan_btest
+ btest_mode=
+ '['  '==' 1 ]
+ '['  '==' y ]
+ '[' 63178 '==' 47189 ]
+ '[' '!' -d /lib/modules/4.1.52/extra ]
+ '[' '!' -z  ]
+ '[' '!' -z y ]
+ '[' -f /lib/modules/4.1.52/kernel/net/wireless/cfg80211.ko ]
+ all_wlan_modules='firmware_class cfg80211 hnd emf igs dhd wl'
+ nvram show
+ grep -c ^
+ grep kernel_mods
+ is_nvmodules_list=0
+ '[' 0 -eq 0 ]
+ echo 'no modules list in nvram, using defaults...'
no modules list in nvram, using defaults...
+ modules_list='firmware_class cfg80211 hnd emf igs dhd wl'
+ unload_modules_list='firmware_class '
+ unload_modules_list='cfg80211 firmware_class '
+ unload_modules_list='hnd cfg80211 firmware_class '
+ unload_modules_list='emf hnd cfg80211 firmware_class '
+ unload_modules_list='igs emf hnd cfg80211 firmware_class '
+ unload_modules_list='dhd igs emf hnd cfg80211 firmware_class '
+ unload_modules_list='wl dhd igs emf hnd cfg80211 firmware_class '
+ load_modules
+ '[' '!' -z y ]
+ dhd_module_params='iface_name=wl dhd_console_ms=250'
+ '[' 63178 '==' 47189 ]
+ wl_module_params='intf_name=wl%d'
+ wl_mfgtest_module=wl_mfgtest
+ hnd_mfgtest_module=hnd_mfgtest
+ '[' -f /proc/nvram/wl_nand_manufacturer ]
+ cat /proc/nvram/wl_nand_manufacturer
+ is_mfg=0
+ wldpdctl_init
+ '[' '!' -z y ]
+ '[' '!' -z ]
+ '[' -z 0 ]
+ '[' 0 '=' 1 ]
+ echo 'loading WLAN kernel modules ... firmware_class cfg80211 hnd emf igs dhd wl'
loading WLAN kernel modules ... firmware_class cfg80211 hnd emf igs dhd wl
+ '[' -f /lib/modules/4.1.52/kernel/drivers/base/firmware_loader/firmware_class.ko ]
+ '[' firmware_class '==' dhd ]
+ '[' firmware_class '==' wl ]
+ '[' firmware_class '==' wl_mfgtest ]
+ '[' -e /lib/modules/4.1.52/extra/firmware_class.ko ]
+ insmod /lib/modules/4.1.52/kernel/net/wireless/cfg80211.ko
insmod: can't insert '/lib/modules/4.1.52/kernel/net/wireless/cfg80211.ko': File exists
+ '[' cfg80211 '==' dhd ]
+ '[' cfg80211 '==' wl ]
+ '[' cfg80211 '==' wl_mfgtest ]
+ '[' -e /lib/modules/4.1.52/extra/cfg80211.ko ]
+ module_params=
+ '[' hnd '==' dhd ]
+ '[' hnd '==' wl ]
+ '[' hnd '==' wl_mfgtest ]
+ '[' -e /lib/modules/4.1.52/extra/hnd.ko ]
+ '[' hnd '==' dhd ]
+ nvram kget wl_path
+ nic_nvram=
+ '[' hnd '==' wl ]
+ insmod /lib/modules/4.1.52/extra/hnd.ko
insmod: can't insert '/lib/modules/4.1.52/extra/hnd.ko': File exists
+ module_params=
+ '[' emf '==' dhd ]
+ '[' emf '==' wl ]
+ '[' emf '==' wl_mfgtest ]
+ '[' -e /lib/modules/4.1.52/extra/emf.ko ]
+ '[' emf '==' dhd ]
+ nvram kget wl_path
+ nic_nvram=
+ '[' emf '==' wl ]
+ insmod /lib/modules/4.1.52/extra/emf.ko
insmod: can't insert '/lib/modules/4.1.52/extra/emf.ko': File exists
+ module_params=
+ '[' igs '==' dhd ]
+ '[' igs '==' wl ]
+ '[' igs '==' wl_mfgtest ]
+ '[' -e /lib/modules/4.1.52/extra/igs.ko ]
+ '[' igs '==' dhd ]
+ nvram kget wl_path
+ nic_nvram=
+ '[' igs '==' wl ]
+ insmod /lib/modules/4.1.52/extra/igs.ko
insmod: can't insert '/lib/modules/4.1.52/extra/igs.ko': File exists
+ module_params='iface_name=wl dhd_console_ms=250'
+ '[' dhd '==' dhd ]
+ get_unit
+ cat /proc/net/dev
+ grep wl0
+ wlx=
+ '[' -z  ]
+ break
+ echo 0
+ idx=0
+ instance_base='instance_base=0'
+ module_params='iface_name=wl dhd_console_ms=250 instance_base=0'
+ '[' -e /lib/modules/4.1.52/extra/dhd.ko ]
+ module_params='intf_name=wl%d'
+ '[' wl '==' dhd ]
+ '[' wl '==' wl ]
+ get_unit
+ cat /proc/net/dev
+ grep wl0
+ wlx=
+ '[' -z  ]
+ break
+ echo 0
+ idx=0
+ instance_base='instance_base=0'
+ module_params='intf_name=wl%d instance_base=0'
+ '[' -e /lib/modules/4.1.52/extra/wl.ko ]
+ '[' wl '==' dhd ]
+ nvram kget wl_path
+ nic_nvram=
+ '[' wl '==' wl ]
+ '[' -f  ]
+ insmod /lib/modules/4.1.52/extra/wl.ko 'intf_name=wl%d' 'instance_base=0'
+ wlaffinity auto
Configuring wlan related threads...
no thread pin applied
Wlan related processes are pinned by default
set bcmsw_recycle 1415 to aff:2
set bcm_archer_wlan 1388 to aff:2
set wl0-kthrd 13738 to aff:2
set wl1-kthrd 13748 to aff:2
set wl0-kthrd and pid:13738 to prio:5
Failed to set wl0-kthrd to realtime priority 5
set wl1-kthrd and pid:13748 to prio:5
Failed to set wl1-kthrd to realtime priority 5
+ exit 0

```

fda77, 2022-05-26:
Und jetzt geht wlan?

freetz-ng-mod, 2022-05-26:
Nein nun ist die WLAN LED eine Disco

fda77, 2022-05-26:
Eigentlich sieht das gut aus. Diese "priority 5" hat avm auch in der inhaus gefixt, sollten aber egal sein:
```
-            # For 47189 ,it only has single core CPU. Use kthread for dhd_dpc task.
-            if [ "$BRCM_CHIP" == "47189" ]; then
-                dhd_module_params="${dhd_module_params} dhd_dpc_prio=5"
-            fi
```

Vielleicht klappt das nur wenn das ganze beim starten zum richtigen Zeitpunkt ausgeführt wird

build/original/filesystem/etc/init.d/bcm-wlan-drivers.sh :
```
     for unit in $UNITLIST; do
         wlx=`cat /proc/net/dev |grep wl$unit`
-        if [ ${#wlx[@]} == 0 ]; then
+        if [ -z "$wlx" ]; then
             break
         fi
     done
```
dann "make"

freetz-ng-mod, 2022-05-26:
Wlan geht. Nun mal schauen ob reboot kommt nach ca 8 min

freetz-ng-mod, 2022-05-26:
reboot

freetz-ng-mod, 2022-05-27:
Habe ich bis jetzt noch nie gemacht. Daher muss ich mich da erst reinlesen

fda77, 2022-05-27:
Im webif trägst du die ip deines computers oder eines anderes Gerätes bei syslog ein.
Und auf diesem anderen startet du vorher einen Syslog server

freetz-ng-mod, 2022-05-27:
1 Master
2 clint
3 clint
..
11 Rechner lan 
12 Rechner wlan
...

fda77, 2022-05-27:
Ist egal, ein linux mit syslog oder windows mit irgend einer software

freetz-ng-mod, 2022-05-27:
Werde wie gesagt mich nachher einlesen. 

Und es einrichten.

fda77, 2022-05-27:
Das kann syslog https://tinyurl.com/2p8ewn3x

fda77, 2022-05-28:
Wenn man in config/.img/7539--07_4X.in aus GCC_9 ein GCC_8 machst kann man ein inhaus bauen. Auch müssten jetzt alle anderen inhaus baubar sein. Für Kernelmodule: avm nerven

fda77, 2022-05-29:
@freetz-ng-mod Den letzten Post kannst du ignorieren, bis aus "avm nerven". Wie verhält sich die Inhaus, insbesondere Busybox?

Himan2001, 2022-05-29:
Bitte die 5530 neben der 7530ax auch noch im Auge behalten. Viele Sachen treffen leider auch hier zu. 

Bei der Box konnte ich vor etwa 3 Wochen zumindest mit dem deaktivierten trustedd flashen und einwandfrei dry-runs machen.
Segfaults und spontane Reboots habe ich danach nicht mehr beobachten können. Du hast die Box zwar ebenfalls auf "broken" gesetzt aber mit dem git-stand von vor etwa 1 Monat und developer Mode/trustedd deaktivieren rennt die bei mir zumindest auch MIT WLAN "stabil".

freetz-ng-mod, 2022-05-29:
```
  ld_uClibc .........................   12.86 Kb 
  libgcc_s ..........................   14.67 Kb 
  libuClibc .........................  284.19 Kb 
  checking OpenSSL version
    ... used by AVM ...... 1.1.1o	3 May 2022
  checking libc version
    ... used by AVM ...... glibc 2.30
    ... used by Freetz ... uClibc 1.0.41
installing busybox
  replacing busybox
  installing symlinks
    renaming applet build/modified/filesystem/usr/bin/bunzip2
    renaming applet build/modified/filesystem/usr/bin/bzcat
    renaming applet build/modified/filesystem/usr/bin/bzip2
    renaming applet build/modified/filesystem/bin/dmesg
    renaming applet build/modified/filesystem/usr/bin/renice
  checking BusyBox version
    ... used by AVM ...... 1.29.3
    ... used by Freetz ... 1.34.1 (2022-05-29 16:15:38 CEST)
installing packages
  syslogd-cgi-0.2.3 .................    3.21 Kb 
  inetd-0.2 .........................    3.17 Kb 
  authorized-keys-0.1 ...............    2.64 Kb 
  dropbear-2022.82 ..................  124.33 Kb 
  haserl-0.9.36 .....................    9.84 Kb 
  juis_check-1.2 ....................    5.14 Kb 
  mod-1.0 ...........................   88.64 Kb 
  modcgi-0.3 ........................    3.28 Kb 
  checking blkid tools
    ... binary by avm
processing kernel
  checking kernel version
    ... used by AVM ...... 4.19.183
    ... used by Freetz ... 4.19.183
  kernel modules installed: 120 entries in modules.dep and 120 .ko-files found.
  removing empty kernel module directories
processing mod base options
  setting colored style as default
  patching security level (0)
  integrated Box-Info
  integrated FREETZ-Info
  integrated .config (stripped)
  tagging avm webif (tag by freetz-ng)
    build/modified/filesystem/usr/www/all/assets/logos/logo_fritzDiamond.svg
    build/modified/filesystem/usr/www/all/guest/assets/logos/logo_fritzDiamond.svg
    build/modified/filesystem/usr/www.myfritz/all/assets/logos/logo_fritzDiamond.svg
    build/modified/filesystem/usr/www.nas/all/assets/logos/logo_fritzDiamond.svg
stripping leftover unstripped binaries
    build/modified/filesystem/bin/nvram
    build/modified/filesystem/sbin/send_kpi
    build/modified/filesystem/sbin/loadcal
    build/modified/filesystem/sbin/avmwlanlifetest
    build/modified/filesystem/usr/sbin/wpa_supplicant
    build/modified/filesystem/usr/sbin/iw
    build/modified/filesystem/usr/sbin/wpa_cli
    build/modified/filesystem/usr/sbin/hostapd
    build/modified/filesystem/usr/sbin/hostapd_cli
    build/modified/filesystem/usr/lib/libwpa_client.so
    build/modified/filesystem/usr/lib/freetz/libuClibc-1.0.41.so
    build/modified/filesystem/usr/lib/freetz/ld-uClibc-1.0.41.so
    build/modified/filesystem/lib/libnvram.so
done.

STEP 3: PACK/SIGN
  checking for left over version-control-system files
  integrate freetz info file into image
packing var.tar
  checking signature key files
    adding public signature key file
creating filesystem image (SquashFS4-xz)
  SquashFS block size: 64 kB (65536 bytes)
copying kernel image
  kernel image size: 3.1 MB
copying filesystem image
  filesystem image size: 35.6 MB
packing images/7530AX_07.39.all-rev96959_inhaus_freetz-ng-19559M-5f15cec9e_20220529-161606.image
  packed image file size: 39.0 MB (40939520 bytes)
signing packed .image file
  signed image file size: 39.0 MB (40939520 bytes)
source firmware: 7530-AX_de-es-fr-it-nl-pl 256.07.39 rev96959 {ALL} [MOVE21] (25.05.2022 19:23:02) 
  source image file size: 38.7 MB (40550400 bytes)
done.

FINISHED
```
Aber noch nicht auf Box getestet. Alles neu ausgelesen gehabt

freetz-ng-mod, 2022-05-29:
@Himan2001 
Nutze einfach die FW 7.21, da lauft alles. Und da geht, das push auch ohne FREETZ_REMOVE_UNTRUSTEDD. Alle FW da drüber brauchen FREETZ_REMOVE_UNTRUSTEDD sonst geht, das push nicht. Aber das ganze steht da schon
https://github.com/Freetz-NG/freetz-ng/discussions/495

Himan2001, 2022-05-29:
... wie gesagt bei mir rennts mit dem Entfernen des trustedd. Brauchen wir den daemon ueberhaupt im Produktivbetrieb ?
Wie das ganze dann in der Praxis rennt werde ich erst sehen wenn das FTTH geschaltet ist. Kommt auch noch eine 5590 in Kürze, da gibts ja offenbar die Freetz-Probleme nicht (mehr).

fda77, 2022-05-29:
Wundert euch nicht wenn ich nachher wieder alle Geräte durcheinander werfe!
@freetz-ng-mod
Danke fürs testen! Liegt wohl wirklich an syntaxfehler von avm in den wlan scripten die mit der neuen busybox nicht mehr funktionieren
 - Für gcc hat avm immer noch ein paar Patches, scheint auch ohne zu laufen
 - Ohne avm-sourcen geht wohl auch.
 - Hast du die dl-toolchain dfür verwendet (ab aptguru)? 

Wie ist das mit der busybox und der FOS version 7.22? Muss dafür auch die alte 1.2x verwendet werden?

freetz-ng-mod, 2022-05-29:
Nein war nur auf Experte.
Das habe ich noch nicht getestet.

fda77, 2022-05-29:
Compilieren geht, nur obs auch läuft
> Wie ist das mit der busybox und der FOS version 7.22? Muss dafür auch die alte 1.2x verwendet werden?

freetz-ng-mod, 2022-05-29:
```
set -e; : "  COMPAT  include/generated/lisi/linux/simple_mempool.h"; mkdir -p include/generated/lisi/linux/; awk '$1 == "linux/simple_mempool.h" { print "#ifdef __KERNEL__"; print "#include <" $2 ">"; if (match($1, "^uapi/")) { sub(/^uapi\//, "", $1); sub(/^uapi\//, "", $2); print "#else"; print "#include <" $2 ">"; }; print "#endif"; }' < drivers/char/avm_new/avm/conf/compat.headers > include/generated/lisi/linux/simple_mempool.h;
/bin/sh: 1: ./drivers/char/avm_new/generate-event-types: Permission denied
make[3]: *** [drivers/char/avm_new/avm/make/genhdr.make:8: include/generated/uapi/avm/event/gen-types.h] Fehler 126
make[3]: *** Auf noch nicht beendete Prozesse wird gewartet …
make[2]: *** [avm/make/generated/genhdr.make:17: __gen_headers/drivers/char/avm_new] Fehler 2
make[2]: *** Auf noch nicht beendete Prozesse wird gewartet …
set -e; : "  COMPAT  include/generated/lisi/linux/tffs.h"; mkdir -p include/generated/lisi/linux/; awk '$1 == "linux/tffs.h" { print "#ifdef __KERNEL__"; print "#include <" $2 ">"; if (match($1, "^uapi/")) { sub(/^uapi\//, "", $1); sub(/^uapi\//, "", $2); print "#else"; print "#include <" $2 ">"; }; print "#endif"; }' < drivers/char/tffs/avm/conf/compat.headers > include/generated/lisi/linux/tffs.h;
set -e; : "  COMPAT  include/generated/uapi/linux/tffs.h"; mkdir -p include/generated/uapi/linux/; awk '$1 == "uapi/linux/tffs.h" { print "#ifdef __KERNEL__"; print "#include <" $2 ">"; if (match($1, "^uapi/")) { sub(/^uapi\//, "", $1); sub(/^uapi\//, "", $2); print "#else"; print "#include <" $2 ">"; }; print "#endif"; }' < drivers/char/tffs/avm/conf/compat.headers > include/generated/uapi/linux/tffs.h;
make[2]: Verzeichnis „/home/rocco/Ubuntu/tools/freetz-ng/source/kernel/ref-brcma9-7539_07.22/linux-4.1“ wird verlassen
make[1]: *** [GNUmakefile:105: prepare] Fehler 2
make[1]: Verzeichnis „/home/rocco/Ubuntu/tools/freetz-ng/source/kernel/ref-brcma9-7539_07.22/linux-4.1“ wird verlassen
make: *** [make/linux/kernel.mk:166: source/kernel/ref-brcma9-7539_07.22/.prepared] Beendet
rocco@xmg-ultra:~/Ubuntu/tools/freetz-ng$ 
```

freetz-ng-mod, 2022-05-29:
Wenn man ein freetz-ng Verzeichnis hat, wo schon mal der Kernel benutzt worden ist, geht auch git pull und das Bauen des Images. Alles, was neu eingelesen worden ist, über git clone https://github.com/Freetz-NG/freetz-ng geht nichts mehr

fda77, 2022-05-29:
Es gibt aktuell ein Problem beim kernel-bauen, dazu gibts ein issue

Ich bin nun etwas verwirrt...
7530ax hat 3 FOS in freetz: 7.22 7.31 7.39-inhaus
Diese funktionieren nun alle mit busybox 1.3x?
57e04157936695299a1f8e445f72c9acda7ba48f hat also doch gereicht?

Oder gibt es sonst noch irgend etwas was man bei diesen geräten spezielles aus/abwählen muss?

freetz-ng-mod, 2022-05-29:
Bei der 7530AX war es immer nur BB 1.27 wo es ging. 
Das Image, was nun auf der Box ist, halt ein anderes Verzeichnis von gestern wo ich nur git pull gemacht habe. Da ich gestern alles neu angelegt habe (freetz-ng-19559M-5f15cec9e)

fda77, 2022-05-29:
Okay, dann funktioniert dieses Gerät ja nun ohne jegliche Problem :D

freetz-ng-mod, 2022-05-29:
So gesehen ja. Aber ich würde es gerne noch testen, wenn das andere Prob gelöst ist.

closed 2022-05-29 by fda77

fda77, 2022-05-29:
Huch, wenns nicht geht meld dich ..