Freetz-NG/freetz-ng #537: 6591 mit Labor Version

freetz-ng-mod, 2022-05-22:
Da ich mich mit den Kabelboxen total nicht auskenne, kann ich leider dafür kein Fix machen. 
Aber es gibt ein Fehler bei diesen hier!

```
---> kernel ... unpacking ... patching ... #
#kernel version specific patches
    applying patch file make/linux/patches/4.9.250.x86/000-6660.07.26-6660.07.29-delta.patch.xz
    The next patch would create the file linux-4.9/arch/x86/configs/atomp7/intelsdk.config,
    which already exists!  Applying it anyway.
    patching file linux-4.9/arch/x86/configs/atomp7/intelsdk.config
    Hunk #1 FAILED at 1.
    1 out of 1 hunk FAILED -- saving rejects to file linux-4.9/arch/x86/configs/atomp7/intelsdk.config.rej
    can't find file to patch at input line 3638
    Perhaps you used the wrong -p or --strip option?
    The text leading up to this was:
    --------------------------
    |Datei 6660.07.26/linux-4.9/avm/conf/linux-4.9.atomp7.intelsdk ist ein normale Datei, während Datei 6660.07.29/sources/kernel/linux-4.9/avm/conf/linux-4.9.atomp7.intelsdk ein symbolische Verknüpfung ist.
    |--- linux-4.9/drivers/char/avm_new/avm_event_node_pumaX.c	2020-11-24 09:22:13.000000000 +0100
    |+++ linux-4.9/drivers/char/avm_new/avm_event_node_pumaX.c	2021-11-03 15:44:13.000000000 +0100
    --------------------------
    No file to patch.  Skipping patch.
    4 out of 4 hunks ignored
    patching file linux-4.9/drivers/mrpc/sgc_server.c
    Hunk #1 FAILED at 631.
    1 out of 1 hunk FAILED -- saving rejects to file linux-4.9/drivers/mrpc/sgc_server.c.rej
    patching file linux-4.9/include/linux/avalanche/generic/avalanche_pdsp_api.h
    Hunk #1 FAILED at 143.
    1 out of 1 hunk FAILED -- saving rejects to file linux-4.9/include/linux/avalanche/generic/avalanche_pdsp_api.h.rej
    patching file linux-4.9/include/linux/avalanche/generic/avalanche_pp_api.h
    Hunk #1 FAILED at 1.
    Hunk #2 FAILED at 122.
    Hunk #3 FAILED at 503.
    Hunk #4 succeeded at 530 (offset -2 lines).
    Hunk #5 FAILED at 1180.
    Hunk #6 FAILED at 1192.
    Hunk #7 FAILED at 1271.
    Hunk #8 FAILED at 1287.
    Hunk #9 FAILED at 1606.
    Hunk #10 FAILED at 1891.
    Hunk #11 succeeded at 2057 (offset 29 lines).
    Hunk #12 FAILED at 2081.
    Hunk #13 FAILED at 2093.
    Hunk #14 FAILED at 2107.
    Hunk #15 FAILED at 2125.
    Hunk #16 FAILED at 2140.
    Hunk #17 FAILED at 2192.
    Hunk #18 FAILED at 2248.
    Hunk #19 FAILED at 2272.
    Hunk #20 FAILED at 2293.
    Hunk #21 FAILED at 2312.
    19 out of 21 hunks FAILED -- saving rejects to file linux-4.9/include/linux/avalanche/generic/avalanche_pp_api.h.rej
    patching file linux-4.9/include/linux/ti_hil.h
    Hunk #1 FAILED at 1.
    Hunk #2 FAILED at 187.
    Hunk #3 succeeded at 318 with fuzz 2 (offset 12 lines).
    2 out of 3 hunks FAILED -- saving rejects to file linux-4.9/include/linux/ti_hil.h.rej
    patching file linux-4.9/include/net/directconnect_dp_api.h
    Hunk #1 FAILED at 1.
    Hunk #2 FAILED at 116.
    Hunk #3 FAILED at 311.
    Hunk #4 FAILED at 376.
    Hunk #5 FAILED at 553.
    Hunk #6 FAILED at 567.
    Hunk #7 FAILED at 596.
    Hunk #8 FAILED at 604.
    Hunk #9 FAILED at 621.
    Hunk #10 FAILED at 649.
    Hunk #11 FAILED at 695.
    Hunk #12 FAILED at 732.
    Hunk #13 FAILED at 787.
    Hunk #14 FAILED at 804.
    Hunk #15 FAILED at 822.
    Hunk #16 FAILED at 858.
    Hunk #17 FAILED at 869.
    Hunk #18 FAILED at 887.
    Hunk #19 FAILED at 898.
    Hunk #20 FAILED at 916.
    Hunk #21 FAILED at 926.
    Hunk #22 FAILED at 945.
    Hunk #23 FAILED at 986.
    Hunk #24 FAILED at 1006.
    Hunk #25 FAILED at 1017.
    Hunk #26 FAILED at 1033.
    Hunk #27 FAILED at 1060.
    Hunk #28 FAILED at 1080.
    Hunk #29 FAILED at 1094.
    Hunk #30 FAILED at 1146.
    Hunk #31 FAILED at 1220.
    Hunk #32 FAILED at 1247.
    Hunk #33 FAILED at 1259.
    Hunk #34 FAILED at 1287.
    Hunk #35 FAILED at 1306.
    Hunk #36 FAILED at 1335.
    Hunk #37 FAILED at 1392.
    Hunk #38 FAILED at 1401.
    Hunk #39 FAILED at 1445.
    Hunk #40 FAILED at 1481.
    Hunk #41 FAILED at 1525.
    Hunk #42 FAILED at 1543.
    Hunk #43 FAILED at 1561.
    Hunk #44 FAILED at 1585.
    Hunk #45 FAILED at 1613.
    Hunk #46 FAILED at 1638.
    Hunk #47 FAILED at 1667.
    Hunk #48 FAILED at 1691.
    Hunk #49 FAILED at 1717.
    Hunk #50 FAILED at 1734.
    Hunk #51 FAILED at 1758.
    Hunk #52 FAILED at 1803.
    Hunk #53 FAILED at 1816.
    Hunk #54 FAILED at 1834.
    Hunk #55 FAILED at 1848.
    Hunk #56 FAILED at 1861.
    Hunk #57 FAILED at 1874.
    Hunk #58 FAILED at 1885.
    Hunk #59 FAILED at 1900.
    Hunk #60 FAILED at 1915.
    Hunk #61 FAILED at 1928.
    Hunk #62 FAILED at 1939.
    Hunk #63 FAILED at 1950.
    Hunk #64 FAILED at 1961.
    Hunk #65 FAILED at 1971.
    65 out of 65 hunks FAILED -- saving rejects to file linux-4.9/include/net/directconnect_dp_api.h.rej
    patching file linux-4.9/include/net/netfilter/nf_conntrack_tuple.h
    Hunk #1 FAILED at 119.
    1 out of 1 hunk FAILED -- saving rejects to file linux-4.9/include/net/netfilter/nf_conntrack_tuple.h.rej
    patching file linux-4.9/.kernelvariables
    Hunk #1 FAILED at 1.
    1 out of 1 hunk FAILED -- saving rejects to file linux-4.9/.kernelvariables.rej
    patching file linux-4.9/net/avm_pa/avm_pa.c
    Hunk #1 FAILED at 5597.
    Hunk #2 FAILED at 7778.
    2 out of 2 hunks FAILED -- saving rejects to file linux-4.9/net/avm_pa/avm_pa.c.rej
    patching file linux-4.9/net/netfilter/nf_conntrack_core.c
    Hunk #1 FAILED at 13.
    Hunk #2 FAILED at 37.
    Hunk #3 FAILED at 1152.
    3 out of 3 hunks FAILED -- saving rejects to file linux-4.9/net/netfilter/nf_conntrack_core.c.rej
    patching file linux-4.9/net/netfilter/nf_conntrack_proto_tcp.c
    Hunk #1 FAILED at 8.
    Hunk #2 FAILED at 21.
    Hunk #3 FAILED at 860.
    Hunk #4 FAILED at 1074.
    Hunk #5 FAILED at 1085.
    5 out of 5 hunks FAILED -- saving rejects to file linux-4.9/net/netfilter/nf_conntrack_proto_tcp.c.rej
    patching file linux-4.9/net/netfilter/nf_conntrack_standalone.c
    Hunk #1 FAILED at 6.
    Hunk #2 succeeded at 41 with fuzz 2 (offset 8 lines).
    Hunk #3 FAILED at 279.
    2 out of 3 hunks FAILED -- saving rejects to file linux-4.9/net/netfilter/nf_conntrack_standalone.c.rej
ERROR: modpatch: Error in patch-file make/linux/patches/4.9.250.x86/000-6660.07.26-6660.07.29-delta.patch.xz
```

fda77, 2022-05-22:
Wie hast du denn diese Fehler hinbekommen? Hast du was geändert?
Oder ist das so eingecheckt? In dem Fall muss make/linux/patches/4.9.250.x86/000-6660.07.26-6660.07.29-delta.patch.xz wohl in ein Unterverzeichnis make/linux/patches/4.9.250.x86/???????/

freetz-ng-mod, 2022-05-22:
6591 die laber Version ausgesucht
6591 7.29 geht ohne probs

fda77, 2022-05-22:
Das kam wohl irgendwie durcheinander

closed 2022-05-22 by fda77