Freetz-NG/freetz-ng #538: Update von BINUTILS_2_37 zu BINUTILS_2_38

freetz-ng-mod, 2022-05-22:
Habe es mit der 3272 getestet 
[BINUTILS_2_38.zip](https://github.com/Freetz-NG/freetz-ng/files/8749012/BINUTILS_2_38.zip)

fda77, 2022-05-22:
Das wird nur für die neuesten FOS genutzt (mit uclib 1.0.41).
```
config FREETZ_TARGET_BINUTILS_2_38_DEFAULT
	bool
	depends on                                         FREETZ_SEPARATE_AVM_UCLIBC
```
Da durch diesen Bump wieder (wie schon gestern...) 10 dl-toolchains neu gebaut werden müssten lass ich es vorerst offen. Die uclibcng war ein (bisschen) Securityfix

[patch.txt](https://github.com/Freetz-NG/freetz-ng/files/8750351/patch.txt)

freetz-ng-mod, 2022-05-25:
Ich habe mal dein Patch für angepasst wegen der Hash Umstellung
[patch.txt](https://github.com/Freetz-NG/freetz-ng/files/8772676/patch.txt)

fda77, 2022-05-25:
Warte mal ab. Mit der neuen Inhaus die gcc 9.3.0 hat kommen auch noch anderen Dinge wie libtool und binutils mit. Und danach auch dl-Toolchains. Die Versionen davon sollten gleich er von AVM genutzen sein damit das mit den Kernelmodulen funtioniert.
Ich hab aber noch nicht geschaut ob man anhand des Images erkennen kann welche Versionen genutzt wurden, ka pb das überhaupt möglich ist. Sourcen wären besser

fda77, 2022-05-26:
3ad7f7859d2ee831b55a144bdd078e253ade3d5e

closed 2022-05-26 by fda77