Freetz-NG/freetz-ng #540: Undefined references nach libgcrypt bump

pfichtner, 2022-05-22:
Mit 331488b6e9475c4d19f9dcc5929f8d8c17cbe921 baut libgcrypt bei mir auf für die 3270 und 3370 nicht mehr (7570 bzw. W920V, 7390, 7490 und 7590 bauen hingegen noch). 

```
libtool: link: mips-linux-ranlib .libs/libgcrypt.a
libtool: link: rm -fr .libs/libgcrypt.lax
libtool: link: ( cd ".libs" && rm -f "libgcrypt.la" && ln -s "../libgcrypt.la" "libgcrypt.la" )
/bin/bash ../libtool  --tag=CC   --mode=link /workspace/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/mips-linux-uclibc/bin/mips-linux-uclibc-gcc  -march=34kc -mtune=34kc -msoft-float -Os -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 --std=gnu99 -fvisibility=hidden -fno-delete-null-pointer-checks -Wall   -o mpicalc mpicalc-mpicalc.o libgcrypt.la -lgpg-error 
libtool: link: /workspace/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/mips-linux-uclibc/bin/mips-linux-uclibc-gcc -march=34kc -mtune=34kc -msoft-float -Os -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 --std=gnu99 -fvisibility=hidden -fno-delete-null-pointer-checks -Wall -o .libs/mpicalc mpicalc-mpicalc.o  ./.libs/libgcrypt.so /workspace/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/mips-linux-uclibc/usr/lib/libgpg-error.so
./.libs/libgcrypt.so: undefined reference to `_gcry_digest_spec_blake2b_512'
./.libs/libgcrypt.so: undefined reference to `blake2b_vl_hash'
collect2: error: ld returned 1 exit status
make[3]: *** [Makefile:754: mpicalc] Error 1
make[3]: Leaving directory '/workspace/source/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/libgcrypt-1.10.1/src'
make[2]: *** [Makefile:502: all-recursive] Error 1
make[2]: Leaving directory '/workspace/source/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/libgcrypt-1.10.1'
make[1]: *** [Makefile:434: all] Error 2
make[1]: Leaving directory '/workspace/source/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/libgcrypt-1.10.1'
make: *** [make/libs/libgcrypt/libgcrypt.mk:39: source/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/libgcrypt-1.10.1/src/.libs/libgcrypt.so.20.4.1] Terminated
make: *** Deleting file 'source/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/libgcrypt-1.10.1/src/.libs/libgcrypt.so.20.4.1'
```

fda77, 2022-05-22:
Da ist wieder irgend was an gcc/uclibc zu alt um das zu bauen.
Am besten würde man diese alten Boxen einfach rauswerfen...

fda77, 2022-05-22:
https://git.gnupg.org/cgi-bin/gitweb.cgi?p=libgcrypt.git;a=blob;f=NEWS

pfichtner, 2022-05-23:
Scheint mir eher ein "wontix" statt ein "fixed" zu sein. 
Wäre es denn keine Möglichkeit, in solchen Fällen, neben der neuen, weierhin die alte lib version anzubieten und auswählbar zu haben (klar, ohne "Suppor"t auf eigenes Risiko und mit der Gefahr von Folgeproblemen)? 
Es wäre schade, wenn die alten Boxen rausfliegen würden, aber natürlich auch verständlch und nachvollziehbar.

closed 2022-05-23 by fda77

pfichtner, 2022-05-23:
Weil ich jetzt erst gesehen habe, dass Dein commit mein Problem löst :rofl: