Freetz-NG/freetz-ng #544: OpenSSL 3.0.3 auf der 3272 geht nicht

freetz-ng-mod, 2022-05-27:
```
make[2]: *** [Makefile:17695: apps/openssl] Error 1
make[2]: Leaving directory '/home/rocco/Ubuntu/tools/i-matik/freetz-ng/openssl_3_0_x/32xx/source/target-mips_gcc-5.5.0_uClibc-0.9.33.2-nptl/openssl-3.0.3'
make[1]: *** [Makefile:1529: build_sw] Fehler 2
make[1]: Verzeichnis „/home/rocco/Ubuntu/tools/i-matik/freetz-ng/openssl_3_0_x/32xx/source/target-mips_gcc-5.5.0_uClibc-0.9.33.2-nptl/openssl-3.0.3“ wird verlassen
make: *** [make/openssl/openssl.mk:93: source/target-mips_gcc-5.5.0_uClibc-0.9.33.2-nptl/openssl-3.0.3/libssl.so.3] Beendet
make: *** Datei „source/target-mips_gcc-5.5.0_uClibc-0.9.33.2-nptl/openssl-3.0.3/libssl.so.3“ wird gelöscht
cp: -r wurde nicht angegeben, daher wird das Verzeichnis '/home/rocco/Ubuntu/tools/i-matik/freetz-ng/openssl_3_0_x/32xx/images/' ausgelassen
```

Aber OpenSSL 3.0.2 schon

fda77, 2022-05-27:
-.- "Error 1" heisst dass die Befehle DAVOR keine "0" sonder "1" zurückgegeben hatten

https://github.com/Freetz-NG/freetz-ng/blob/master/.github/ISSUE_TEMPLATE/1-bug_report.md

freetz-ng-mod, 2022-05-27:
Oh 7272 auch.

closed 2022-06-02 by fda77

freetz-ng-mod, 2022-06-11:
OpenSSL 3.0.3 geht auf volgenen boxen nicht
3272
7272
7330
7330SL
7390
der fehler wird so angezeigt
```
	-o apps/openssl \
	apps/lib/openssl-bin-cmp_mock_srv.o \
	apps/openssl-bin-asn1parse.o apps/openssl-bin-ca.o \
	apps/openssl-bin-ciphers.o apps/openssl-bin-cmp.o \
	apps/openssl-bin-cms.o apps/openssl-bin-crl.o \
	apps/openssl-bin-crl2pkcs7.o apps/openssl-bin-dgst.o \
	apps/openssl-bin-dhparam.o apps/openssl-bin-dsa.o \
	apps/openssl-bin-dsaparam.o apps/openssl-bin-enc.o \
	apps/openssl-bin-engine.o apps/openssl-bin-errstr.o \
	apps/openssl-bin-fipsinstall.o apps/openssl-bin-gendsa.o \
	apps/openssl-bin-genpkey.o apps/openssl-bin-genrsa.o \
	apps/openssl-bin-info.o apps/openssl-bin-kdf.o \
	apps/openssl-bin-list.o apps/openssl-bin-mac.o \
	apps/openssl-bin-nseq.o apps/openssl-bin-ocsp.o \
	apps/openssl-bin-openssl.o apps/openssl-bin-passwd.o \
	apps/openssl-bin-pkcs12.o apps/openssl-bin-pkcs7.o \
	apps/openssl-bin-pkcs8.o apps/openssl-bin-pkey.o \
	apps/openssl-bin-pkeyparam.o apps/openssl-bin-pkeyutl.o \
	apps/openssl-bin-prime.o apps/openssl-bin-progs.o \
	apps/openssl-bin-rand.o apps/openssl-bin-rehash.o \
	apps/openssl-bin-req.o apps/openssl-bin-rsa.o \
	apps/openssl-bin-rsautl.o apps/openssl-bin-s_client.o \
	apps/openssl-bin-s_server.o apps/openssl-bin-s_time.o \
	apps/openssl-bin-sess_id.o apps/openssl-bin-smime.o \
	apps/openssl-bin-speed.o apps/openssl-bin-spkac.o \
	apps/openssl-bin-storeutl.o apps/openssl-bin-ts.o \
	apps/openssl-bin-verify.o apps/openssl-bin-version.o \
	apps/openssl-bin-x509.o \
	apps/libapps.a -lssl -lcrypto -ldl -pthread -ldl -pthread -latomic
./libcrypto.so: warning: gethostbyname is obsolescent, use getnameinfo() instead.
./libcrypto.so: undefined reference to `freelocale'
./libcrypto.so: undefined reference to `strcasecmp_l'
./libcrypto.so: undefined reference to `strncasecmp_l'
./libcrypto.so: undefined reference to `newlocale'
collect2: error: ld returned 1 exit status
make[3]: *** [Makefile:17694: apps/openssl] Error 1
make[2]: *** [Makefile:1528: build_sw] Fehler 2
make[1]: *** [make/openssl/openssl.mk:93: source/target-mips_gcc-5.5.0_uClibc-0.9.33.2-nptl/openssl-3.0.3/libssl.so.3] Beendet
make[1]: *** Datei „source/target-mips_gcc-5.5.0_uClibc-0.9.33.2-nptl/openssl-3.0.3/libssl.so.3“ wird gelöscht
make: *** [Makefile:46: envira] Beendet
```
Nach dem downgrade auf OpenSSL 3.0.2 lauft es aber durch.

fda77, 2022-06-11:
Geht doch, auch wenn es dauert...
Wenn man danach sucht sieht man dass es ein bug in openssl ist: https://github.com/openssl/openssl/issues/18244
Da ist auch ein patch verlinkt: https://github.com/t8m/openssl/commit/6dc6e949a41f2d20f6d292d73f79ced54b579404
Den könntest du mal ins patches Verzeichnis von openssl3 kopieren und schauen ob das damit läuft

freetz-ng-mod, 2022-06-11:
```
crypto/ctype.c:10:27: fatal error: internal/e_os.h: No such file or directory
compilation terminated.
make[3]: *** [Makefile:13241: crypto/libcrypto-lib-ctype.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [Makefile:1530: build_sw] Fehler 2
make[1]: *** [make/openssl/openssl.mk:93: source/target-mips_gcc-5.5.0_uClibc-0.9.33.2-nptl/openssl-3.0.3/libssl.so.3] Beendet
make: *** [Makefile:46: envira] Beendet
```

fda77, 2022-06-11:
Hm, wenn man https://github.com/t8m/openssl/commit/6dc6e949a41f2d20f6d292d73f79ced54b579404.patch ins patches-Verzeichnis packt kann der nicht fehlerfrei angewendet werden.
 - Nun müsste man den entweder so anpassen dass es geht
 - Oder die nötigen Vorgänger-Patches auch noch (in der richtigen Reihenfolge) dort ablegen

Am wenigsten Aufwand wäre es auf 3.0.4 zu warten...

fda77, 2022-06-22:
@freetz-ng-mod: 6f6413c3f426f1f9411aaea1e4cf2537339fb847 ?

freetz-ng-mod, 2022-06-23:
Es geht wieder. Habe es heute nach durchlaufen lassen