Freetz-NG/freetz-ng #546: precompiled toolchain fuer 6890 rennt nicht mehr auf debian9 und debian10

Himan2001, 2022-05-27:
Mit dem aktuellen trunk lassen sich die precompiled toolschains für die 6890 nicht mehr auf der bisherigen debian9 nutzen, hier gibt es sofort die Meldung das die glibc zu alt wäre. Bei der Debian10 gibts gleich nach dem Start compiler errors und Abbrüche.
Die Dependencies passen aber.

fda77, 2022-05-27:
Ich hab die Toolchains nun mit Ubuntu 18 statt 14 (8 Jahre alt) gebaut.
Mit 14 gab es einfach zu viele Probleme, Sicherheitsupdates gab es dür schon seit Jahren nicht mehr.
Um damit Freetz zu compilieren mussen manuell einige Pakete nachinstalliert werden. Manches wurde wegen diesem alten System nicht geupdatet, wie patchelf-host.
Auch ccache konnte in die dl-toolchains aufgenommen werden, dafür hätten noch einige wetere Pakete manuell aktualisiert werden.
Kurz: Dieses uralte System ist mit zu mühsam!

Möglichkeiten: 
 - Eine aktuelles Linux verwenden
 - die Toolchains selbst bauen
 - ein Docker image unter Debian verwenden

closed 2022-05-27 by fda77

Himan2001, 2022-05-27:
Neue Meldung:

Gerade die precompiled toolchain für 6890 auf einer up-to date ubuntu 18.04 angetestet....
Bricht ebenfalls mit einer illegal instruction kurz nach dem Start ab. Das Hostsystem selber hat gcc-7.5 drauf aktuell.

Evtl. hier: https://freetz-ng.github.io/freetz-ng/PREREQUISITES

...für ubuntu 18.04 noch was vergessen bei den Prerequisites ?

Himan2001, 2022-05-27:
ah jo... das selbst erstellen der toolchain für die 6890 klappt bei debian10 und ubuntu 18.04 nach wie vor ;-)

Himan2001, 2022-05-27:
Hmmmm. sehe gerade für die 18.04 ist auch eine gcc 8.x verfügbar, die hatte ich nicht aktiv und mit "alternatives" auf gcc7 umgestellt vor ner Weile. Ich werde das später nochmal mit der gcc8 und precompiled chain wiederholen.