Freetz-NG/freetz-ng #547: ubuntu18 (und debian10) immer noch compile errors (6890)

Himan2001, 2022-05-27:
Ich habe jetzt mal verschiedene Varianten durch, von der bisherigen debian9, debian10 und aktuell auch die besagte ubuntu 18.04
mit den entsprechenden dependencies.

Weder die precompiled toolchain noch self build rennen durch. Bei der ubuntu18.04 ist die gcc-7 nach wie vor default und gcc-8 muss manuell nachinstalliert werden. nachdem das ueber "alternatives" als default gesetzt ist kommt es beim erstellen der toolchain zu abi-errors, die sich auf gcc-7 beziehen. Mit gcc-7 (default) lässt sich jedoch die toolchain auch nicht kompilieren....

Ich habe jetzt auf meiner bisherigen debian9 einmal den trunk ausgecheckt vom 30. April - hier wird noch die alte
precompiled toolchain verwendet. Ebenso klappt das selber erstellen hier noch.

Ausserdem ist hier das "ccache" abwählbar und nicht geforced. Ergebnis: Das Image für die 6890 wurde, wie auch für die 5530 und 7590 fehlerfrei erstellt.

Der Stand ist jetzt der, das auf keiner vorhandenen VM die auf debian9/debian10/ubuntu18.04 basiert mehr ein trunk erstellt werden kann. 

- Evtl. sollte das force für ccache testweise mal wieder abschaltbar sein ?
- Die Dependancys für ältere Systeme sind offenbar jetzt outdated ?
- Sofern jetzt ubuntu 20+ "Pflicht" ist, sollten die Prerequisites komplett angepasst werden mit dem Hinweis das keine alten
Versionen mehr zum compilen geeignet sind ?

Einfach mal selber frisch ausschecken auf einer neuinstall, die prerequisites verwenden für ubuntu18.04 und ggfs. gcc-8 noch manuell drüber. Weder precompiled toolchain, noch self compiled rennen fehlerfrei durch....

Im weiteren Verlauf wenn es in den Bereich der Kernelheader geht bricht das compiling dann mit einem compiler fatal error ab.

Himan2001, 2022-05-28:
Ich hab nochmal ein wenig mit debian9/10/ubuntu18 gespielt...

bis hierhin rennt noch alles: commit: f5e7df97c5 (precompiled toolchain + host-tools verwendet)
mit dem geforcten ccache und den neueren host-tools fangen dann die compile & glibc probleme an.

Ich bin auch noch drauf gekommen das die automatische cpu-threaderkennung mit "0" bei mir etwas unpassend war:
Ich habe hier eine VM mit 8GB RAM und 8 cores im Einsatz - 8 Threads waren dann zuviel. Ich hab das manuell ohne ccache auf 4 Threads mal gesetzt - dann gabs die compiler fatal errors zumindest nicht mehr.

Himan2001, 2022-05-28:
So. Ich habe jetzt mal eben frisch eine Ubuntu 20.04 minimal aufgesetzt und die entsprechenden prerequisites installiert:

Wenn die pre compiled download chain verwendet wird, derselbe Fatal Error auch hier:
```

WRAPPER_INCLUDEDIR=-I./include ./bin/g++-uc -MT src/sstream.o -MD -MP -MF src/.sstream.o.dep       -Wno-trigraphs -pedantic -Wno-long-long -march=34kc -mtune=34kc -msoft-float -Ofast -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Wl,-I/usr/lib/freetz/ld-uClibc.so.1  -Os -fPIC -c src/sstream.cpp -o src/sstream.o
In file included from src/limits.cpp:20:
./include/limits:24:2: warning: #warning is a GCC extension
 #warning limits header is nowhere complete or accurate
  ^~~~~~~
./include/limits:24:2: warning: #warning limits header is nowhere complete or accurate [-Wcpp]
./include/limits:621:2: internal compiler error: Illegal instruction
  static numeric_type min () { return __DBL_MIN__; }
  ^~~~~~
Please submit a full bug report,
with preprocessed source if appropriate.

```
--snip--

```
make[1]: Leaving directory '/home/freetz/freetz-ng-master/source/target-mips_gcc-8.3.0_uClibc-1.0.41-nptl_kernel-4.9/uclibcxx-2b58f77ee27c60842054fd0e5dd67f5d00e89eb8'
make: *** [make/libs/uclibcxx/uclibcxx.mk:36: source/target-mips_gcc-8.3.0_uClibc-1.0.41-nptl_kernel-4.9/uclibcxx-2b58f77ee27c60842054fd0e5dd67f5d00e89eb8/src/libuClibc++.so.0.2.5] Terminated

```

Das betrifft jetzt gleich am Anfang das compiling der (neuen) uClibcxx

Für Cloud/VM-Installationen ist ubuntu 20.04 immer noch Basis-Standard wg. der Virtualisierungsunterstützung.
Als default compiler ist hier gcc-9.4.0 jetzt aktiv.

Also ubuntu 22.xx-Zwang als unterste Basis mit der precompiled toolchain ist ja nun auch keine Lösung so wirklich...

fda77, 2022-05-28:
Irgendwas machst du falsch, bei mir rennts mit Ubuntu 18: https://github.com/Freetz-NG/freetz-ng/runs/6637734458?check_suite_focus=true

Himan2001, 2022-05-28:
Und jetzt das ganze mit selber bilden der toolchain:

es laufen zuvor unzählige male Warnungen noch durch:

```
cc1plus: warning: -Wabi won't warn about anything [-Wabi]
cc1plus: note: -Wabi warns about differences from the most up-to-date ABI, which is also used by default
cc1plus: note: use e.g. -Wabi=11 to warn about changes from GCC 7

```

Der nächste Abbruch erfolgt jetzt hier:

```
mips-linux-uclibc-ar: src/abi/libsupc/./src/abi/libsupc/array_type_info.o: No such file or directory
make[1]: *** [src/Makefile.in:45: src/libuClibc++.a] Error 1
make[1]: *** Waiting for unfinished jobs....
mips-linux-uclibc-g++: error: src/abi/libsupc/./src/abi/libsupc/array_type_info.o: No such file or directory
mips-linux-uclibc-g++: error: src/abi/libsupc/./src/abi/libsupc/atexit_arm.o: No such file or directory
mips-linux-uclibc-g++: error: src/abi/libsupc/./src/abi/libsupc/atexit_thread.o: No such file or directory
mips-linux-uclibc-g++: error: src/abi/libsupc/./src/abi/libsupc/bad_alloc.o: No such file or directory
mips-linux-uclibc-g++: error: src/abi/libsupc/./src/abi/libsupc/bad_array_length.o: No such file or directory
mips-linux-uclibc-g++: error: src/abi/libsupc/./src/abi/libsupc/bad_array_new.o: No such file or directory
mips-linux-uclibc-g++: error: src/abi/libsupc/./src/abi/libsupc/bad_cast.o: No such file or directory
mips-linux-uclibc-g++: error: src/abi/libsupc/./src/abi/libsupc/bad_typeid.o: No such file or directory
mips-linux-uclibc-g++: error: src/abi/libsupc/./src/abi/libsupc/class_type_info.o: No such file or directory
mips-linux-uclibc-g++: error: src/abi/libsupc/./src/abi/libsupc/dyncast.o: No such file or directory
mips-linux-uclibc-g++: error: src/abi/libsupc/./src/abi/libsupc/eh_arm.o: No such file or directory
mips-linux-uclibc-g++: error: src/abi/libsupc/./src/abi/libsupc/eh_aux_runtime.o: No such file or directory
mips-linux-uclibc-g++: error: src/abi/libsupc/./src/abi/libsupc/eh_call.o: No such file or directory
mips-linux-uclibc-g++: error: src/abi/libsupc/./src/abi/libsupc/eh_catch.o: No such file or directory
mips-linux-uclibc-g++: error: src/abi/libsupc/./src/abi/libsupc/eh_exception.o: No such file or directory
mips-linux-uclibc-g++: error: src/abi/libsupc/./src/abi/libsupc/eh_personality.o: No such file or directory
mips-linux-uclibc-g++: error: src/abi/libsupc/./src/abi/libsupc/eh_ptr.o: No such file or directory
mips-linux-uclibc-g++: error: src/abi/libsupc/./src/abi/libsupc/eh_term_handler.o: No such file or directory
mips-linux-uclibc-g++: error: src/abi/libsupc/./src/abi/libsupc/eh_terminate.o: No such file or directory
mips-linux-uclibc-g++: error: src/abi/libsupc/./src/abi/libsupc/eh_tm.o: No such file or directory
mips-linux-uclibc-g++: error: src/abi/libsupc/./src/abi/libsupc/eh_throw.o: No such file or directory
mips-linux-uclibc-g++: error: src/abi/libsupc/./src/abi/libsupc/eh_type.o: No such file or directory
mips-linux-uclibc-g++: error: src/abi/libsupc/./src/abi/libsupc/eh_unex_handler.o: No such file or directory
mips-linux-uclibc-g++: error: src/abi/libsupc/./src/abi/libsupc/enum_type_info.o: No such file or directory
mips-linux-uclibc-g++: error: src/abi/libsupc/./src/abi/libsupc/function_type_info.o: No such file or directory
mips-linux-uclibc-g++: error: src/abi/libsupc/./src/abi/libsupc/fundamental_type_info.o: No such file or directory
mips-linux-uclibc-g++: error: src/abi/libsupc/./src/abi/libsupc/guard.o: No such file or directory
mips-linux-uclibc-g++: error: src/abi/libsupc/./src/abi/libsupc/guard_error.o: No such file or directory
mips-linux-uclibc-g++: error: src/abi/libsupc/./src/abi/libsupc/hash_bytes.o: No such file or directory
mips-linux-uclibc-g++: error: src/abi/libsupc/./src/abi/libsupc/nested_exception.o: No such file or directory
mips-linux-uclibc-g++: error: src/abi/libsupc/./src/abi/libsupc/pbase_type_info.o: No such file or directory
mips-linux-uclibc-g++: error: src/abi/libsupc/./src/abi/libsupc/pmem_type_info.o: No such file or directory
mips-linux-uclibc-g++: error: src/abi/libsupc/./src/abi/libsupc/pointer_type_info.o: No such file or directory
mips-linux-uclibc-g++: error: src/abi/libsupc/./src/abi/libsupc/si_class_type_info.o: No such file or directory
mips-linux-uclibc-g++: error: src/abi/libsupc/./src/abi/libsupc/tinfo.o: No such file or directory
mips-linux-uclibc-g++: error: src/abi/libsupc/./src/abi/libsupc/tinfo2.o: No such file or directory
mips-linux-uclibc-g++: error: src/abi/libsupc/./src/abi/libsupc/vec.o: No such file or directory
mips-linux-uclibc-g++: error: src/abi/libsupc/./src/abi/libsupc/vmi_class_type_info.o: No such file or directory
mips-linux-uclibc-g++: error: src/abi/libsupc/./src/abi/libsupc/vterminate.o: No such file or directory
mips-linux-uclibc-g++: error: src/abi/libsupc/./src/abi/libsupc/cp-demangle.o: No such file or directory
make[1]: *** [src/Makefile.in:49: src/libuClibc++.so] Error 1
make[1]: Leaving directory '/home/freetz/freetz-ng-master/source/target-mips_gcc-8.3.0_uClibc-1.0.41-nptl_kernel-4.9/uclibcxx-2b58f77ee27c60842054fd0e5dd67f5d00e89eb8'
make: *** [make/libs/uclibcxx/uclibcxx.mk:36: source/target-mips_gcc-8.3.0_uClibc-1.0.41-nptl_kernel-4.9/uclibcxx-2b58f77ee27c60842054fd0e5dd67f5d00e89eb8/src/libuClibc++.so.0.2.5] Terminated

```
Dieselben Probleme hatte ich bereits schon bei debian10 und ubuntu18 ebenso.
Aktuell sind weder prebuild noch selfbuild toolchain für die 6890 möglich.

fda77, 2022-05-28:
Wann ging das noch?
Warnungen interessieren normal nicht

Himan2001, 2022-05-28:
So mal ein Fazit:

Ich habe den commit für das forcen von CCACHE wieder entfernt und ccache im Menu jetzt wieder abgewählt!

Im Ergebnis lässt sich die toolchain jetzt auf debian10/ubuntu18 und ubuntu 20 wieder normal erstellen.
Die Host-tools habe ich übrigens noch als precompiled gesetzt.

Der Imagebau hat einwandfrei hingehaun jetzt auf ubuntu 20.x - ebenso auch auf ubuntu 18.x

Das Problem liegt also an einem fehlerhaften CCACHE, bzw. seiner Konfiguration oder Enviroment.
Die Problematik ist auf allen VMs gleichermassen nachvollziehbar. mit aktivem ccache lässt sich nichts
mehr anständig erstellen.

Bitte mache das ABWÄHLEN von ccache im APT-Guru Menu wieder optional. solange bis die probleme final
gelöst sind.

Himan2001, 2022-05-28:
^^siehe oben. solange wie man das ccache noch ABWÄHLEN konnte ging alles bis zur debian9 runter immer ohne Probleme.

Ich habe jetzt allerdings nur die precompiled & sources fuer die 6890 im gange, wie das mit den anderen precompiled Sachen dann aussieht muss ich erst noch rausfinden.

Es wäre aber zumindest gut, wenn du die precompiled Sachen zumindest auf ubuntu18 noch stützen kannst, bezogen auf den dortigen gcc-7.5 und die "normale Umgebung". Ebenso die precompiled host-tools. Andersfalls können wir den Support für alle älteren Systeme gleich komplett einstampfen ;-)

fda77, 2022-05-28:
> Ich habe den commit für das forcen von CCACHE wieder entfernt und ccache im Menu jetzt wieder abgewählt!

 > Im Ergebnis lässt sich die toolchain jetzt auf debian10/ubuntu18 und ubuntu 20 wieder normal erstellen.

So ein Blödsinn. Wenn du du Toolchain angblich selbst erstellst kannst du ccache ganz normal abwählen. In den precompiled ists halt drin, da bringt abwählen nix

> Das Problem liegt also an einem fehlerhaften CCACHE

Warum gehts hier: https://github.com/Freetz-NG/freetz-ng/runs/6637734458?check_suite_focus=true ? Und lokal bei mir?

> Bitte mache das ABWÄHLEN von ccache im APT-Guru Menu wieder optional. solange bis die probleme final
gelöst sind.

Macht keinen Sinn

> Die Problematik ist auf allen VMs gleichermassen nachvollziehbar. 

Es könnte auch daran liegen was du auf jeder VM gleichermassen gemach hast...
Das weisst nur du
Und die Fehlermeldungen siehst auch nur du

Himan2001, 2022-05-28:
Kannst Du evtl. mach NICHT dockern sondern einfach eine normale ubuntu 18.04 VM verwenden, die Daten aus der PREREQUISITES.md verwenden und mal die 6890 + precompiled toolchain + precompiled hostools auswählen.
- Und dann self compile für toolchain+hosttools auswählen, ob das ebenso durchrennt ?

Dann solltest Du das normal nachvollziehen können - ausser du hast weitere dependencies die in der Auflisting fehlen - was bei einer minimalen, frischen ubuntu 18.04 neuinstall ja nicht der Fall ist ;-)

bei mir macht zumindest der CCACHE Kram all diese Probleme auf allen VMs....

fda77, 2022-05-28:
Ich hab das Ubuntu 18 prerequisites erst gestern angelegt, weil mir aufgefallen ist dass zu vorher noch "libzstd" fehlt. 
https://github.com/Freetz-NG/freetz-ng/commit/d4753cdec07e456b7091bb73c497f59b7f2499ff
Wenn noch was nicht passt ergänz es!

Ich benutz den kram schon seit langem, rennt gut

Himan2001, 2022-05-28:
Bitte, was ist jetzt falsch ?

Eine neue VM erstellen... minimal auf ubuntu 18.04, in meinem fall jetzt auch auf ubuntu 20.x. Die Infos aus der PREREQUISTED.md verwenden vom aktuellen trunk. APT-Guru mode auswählen und precompiled toolchain verwenden. Dann kommem genau die oben beschriebenen Ergebnisse mit den Abbrüchen und Fehlern. Mehr als komplett frisch eine VM anlegen und entsprechend vorgehen kann ich nicht.... Dasselve wenn ich versuche die toolchain selber zu bauen.

Sobald ccache verwendet wird gibts diese Probleme. Entweder fehlen für die Verwendung von ccache noch weitere Dependencys, oder es wird eine bestimmte Version benötigt. Ich weiss es nicht.

Dein Compilat lief unter einem Docker. Es sollte doch evtl. auch in einer regulären Umgebung laufen ?

fda77, 2022-05-28:
Das Compilat weiss gar nicht unter einem Docker läuft.
Ich hab das mal gestartet: https://github.com/Freetz-NG/freetz-ng/actions

Himan2001, 2022-05-28:
bzgl. libzstd hat sich insofern erledigt, da ich ja ganz frische VMs eben erstellt habe, frisch ausgescheckt habe und somit auch die aktuellen prerequisites von gestern natürlich mit drin habe ;-)

Kannst Du das bitte einmal OHNE docker in einer 18.x/20.x Umgebung auch einmal laufen lassen ?

fda77, 2022-05-28:
Ich benutze Fedora. Ubuntu in Docker

Himan2001, 2022-05-28:
Und ich eine native shell auf einem echten ubuntu 18 und 22, da hat man als angelegter shell user mit Sicherheit ein etwas anderes enviroment als in einer .yml config ;-)

in meinem Fall ist der workarround derzeit somit CCACHE abzuschalten (den commit reversen) bis das irgendwann wieder geht :-( - daher auch meine bitte: bitte mach das CCACHE wieder optional abwählbar, das man sich das manuelle reverse
des entsprechenden commits jedes mal sparen kann ;-) -  das würde sicher auch anderen "nativen" shell usern helfen, die keine container nutzen oder ein entsprechendes setup hier bauen wollen.

fda77, 2022-05-28:
>in meinem Fall ist der workarround derzeit somit CCACHE abzuschalten (den commit reversen) bis das irgendwann wieder geht :-( 
Das macht keinen Sinn!! Es ist EGAL ob an oder aus bei precompiled. In der tar ist ccache mit drin und es wird einfch alles entpackt. Deine Option ist wurst! ccache ist immer da.
Und wenn du nicht precompiled verwendest kannst du die eh schalten wie du willst, ohne was zu ändern

Du kannst also damit einfach deinen Fehler(tm) nicht behben!

Also finde zuerst heraus WAS der Fehler ist. DANN die ursache

Himan2001, 2022-05-28:
.... das ich eben mit einer normalen shell nativ arbeite und keinerlei Container oder "remote Services" nutze.
Ich habe einen xen-server mit einer Unmengen an Zielsystemen nativ am laufen nach Bedarf.

Erkläre mir doch bitte wenn ich diesen Patch Reverse:

https://github.com/Freetz-NG/freetz-ng/commit/40a19d68e4b9086412b0e41fff7092025362fbb7

das dann alles einwandfrei wieder rennt - wie schon all die Jahre zuvor auch. Ich habe niemals vorher irgendwelche
Conatiner oder ähnliches benötigt.

Ich bin leider kein Entwickler der auf Conatainern fitt ist - dafür habe ich ja native Installationen für jeden Anwendungsfall.
Ganz einfach: passendes Zielsystem in der Minimalversion, dann die dokumentierten Dependencies drauf, den trunk auschecken und make manuconfig / make. So hab ich das schon immer gemacht - und bei mir war die Option ccache
schon immer deaktiviert gewesen - zumindest macht genau der oben beschriebene commit den Unterschied aus.

Der Fehlerhinweis hier z.B. :
```

mips-linux-uclibc-ar: src/abi/libsupc/./src/abi/libsupc/array_type_info.o: No such file or directory
make[1]: *** [src/Makefile.in:45: src/libuClibc++.a] Error 1
make[1]: *** Waiting for unfinished jobs....
mips-linux-uclibc-g++: error: src/abi/libsupc/./src/abi/libsupc/array_type_info.o: No such file or directory
mips-linux-uclibc-g++: error: src/abi/libsupc/./src/abi/libsupc/atexit_arm.o: No such file or directory
mips-linux-uclibc-g++: error: src/abi/libsupc/./src/abi/libsupc/atexit_thread.o: No such file or directory
mips-linux-uclibc-g++: error: src/abi/libsupc/./src/abi/libsupc/bad_alloc.o: No such file or directory
mips-linux-uclibc-g++: error: src/abi/libsupc/./src/abi/libsupc/bad_array_length.o: No such file or directory
mips-linux-uclibc-g++: error: src/abi/libsupc/./src/abi/libsupc/bad_array_new.o: No such file or directory
mips-linux-uclibc-g++: error: src/abi/libsupc/./src/abi/libsupc/bad_cast.o: No such file or directory
mips-linux-uclibc-g++: error: src/abi/libsupc/./src/abi/libsupc/bad_typeid.o: No such file or directory

```

usw. zeigt mir zumindest das bei gesetzer ccache Option da wohl mit dem cache-system irgendwo etwas fehlt beim Enviroment oder bei einem path ?

Evtl. gibt es auch ein Problem mit der zeitgleichen Verarbeitung bei mehreren Threads. ohne den ccache wird ja auch viel langsamer gebaut alles.

Ich hab zumindest den Fehler für einen regulären Shell-compile mal weitestgehend eingegrenzt, das ich zumindest wieder normal arbeiten kann - egal ob auf debian9/10/ubuntu18 oder ubuntu20 jetzt ;-) - Wenn ich mehr Speed brauche gebe ich der VM einfach mehr CPU-Cores und RAM und gut iss.

fda77, 2022-05-28:
> Erkläre mir doch bitte wenn ich diesen Patch Reverse: das dann alles einwandfrei wieder rennt

Hab ich bereits mehrmals und nun keine Lust mehr. Scroll einfach nach oben

Dieser winzige Ausschnitt der Log bringt überhaupt nichts. Ausserdem kann ich den Fehler bei mir nicht nachvollziehen

--> https://github.com/Freetz-NG/freetz-ng/blob/master/.github/ISSUE_TEMPLATE/1-bug_report.md

Himan2001, 2022-05-28:
...und nachdem ich auf der nativen ubuntu 20.04 wieder auf die precompiled toolchain gegangen bin, ccache manuell abgeschaltet hab, bekomme ich erneut das gleiche Ergebnis beim Versuch hiermit zu arbeiten:

```

WRAPPER_INCLUDEDIR=-I./include ./bin/g++-uc -MT src/sstream.o -MD -MP -MF src/.sstream.o.dep       -Wno-trigraphs -pedantic -Wno-long-long -march=34kc -mtune=34kc -msoft-float -Ofast -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Wl,-I/usr/lib/freetz/ld-uClibc.so.1  -Os -fPIC -c src/sstream.cpp -o src/sstream.o
In file included from src/limits.cpp:20:
./include/limits:24:2: warning: #warning is a GCC extension
 #warning limits header is nowhere complete or accurate
  ^~~~~~~
./include/limits:24:2: warning: #warning limits header is nowhere complete or accurate [-Wcpp]
./include/limits:621:2: internal compiler error: Illegal instruction
  static numeric_type min () { return __DBL_MIN__; }
  ^~~~~~
Please submit a full bug report,
with preprocessed source if appropriate.
```

somit dreht sich das im Kreis und ich finde keinen weiteren Ansatz - zumindest nicht wenn ich Container und Remote compilen nicht nutze. Ich kann nur sagen das die alte precompiled toolchain problemlos auf jedem system bisher immer rannte.
Die aktualisierte Version kommt zumindest auf ubuntu 20.04 auch nicht zurecht.

Mag sein das es auf einer Fedora nativ die Probleme nicht gibt. Ich geb auf an dieser Stelle.

fda77, 2022-05-28:
Ich auch. Ich weiss bis jetzt noch nicht beim compilieren von was der Fehler auftrat

closed 2022-05-28 by fda77

pfichtner, 2022-05-30:
```
> vagrant init ubuntu/bionic64
A `Vagrantfile` has been placed in this directory. You are now
ready to `vagrant up` your first virtual environment! Please read
the comments in the Vagrantfile as well as documentation on
`vagrantup.com` for more information on using Vagrant.
> vagrant up
Bringing machine 'default' up with 'virtualbox' provider...
==> default: Importing base box 'ubuntu/bionic64'...
==> default: Matching MAC address for NAT networking...
==> default: Checking if box 'ubuntu/bionic64' version '20220523.0.0' is up to date...
==> default: Setting the name of the VM: TMP-freet-issue-552_default_1653947822172_46025
==> default: Clearing any previously set network interfaces...
==> default: Preparing network interfaces based on configuration...
    default: Adapter 1: nat
==> default: Forwarding ports...
    default: 22 (guest) => 2222 (host) (adapter 1)
==> default: Running 'pre-boot' VM customizations...
==> default: Booting VM...
==> default: Waiting for machine to boot. This may take a few minutes...
    default: SSH address: 127.0.0.1:2222
    default: SSH username: vagrant
    default: SSH auth method: private key
    default: Warning: Remote connection disconnect. Retrying...
    default: Warning: Connection reset. Retrying...
    default:
    default: Vagrant insecure key detected. Vagrant will automatically replace
    default: this with a newly generated keypair for better security.
    default:
    default: Inserting generated public key within guest...
    default: Removing insecure key from the guest if it's present...
    default: Key inserted! Disconnecting and reconnecting using new SSH key...
==> default: Machine booted and ready!
==> default: Checking for guest additions in VM...
    default: The guest additions on this VM do not match the installed version of
    default: VirtualBox! In most cases this is fine, but in rare cases it can
    default: prevent things such as shared folders from working properly. If you see
    default: shared folder errors, please make sure the guest additions within the
    default: virtual machine match the version of VirtualBox you have installed on
    default: your host and reload your VM.
    default:
    default: Guest Additions Version: 5.2.42
    default: VirtualBox Version: 6.1
==> default: Configuring proxy for Apt...
==> default: Configuring proxy environment variables...
==> default: Configuring proxy for Git...
==> default: Mounting shared folders...
    default: /vagrant => /home/nonroot/TMP-freet-issue-552
> vagrant ssh
Welcome to Ubuntu 18.04.6 LTS (GNU/Linux 4.15.0-180-generic x86_64)

 * Documentation:  https://help.ubuntu.com
 * Management:     https://landscape.canonical.com
 * Support:        https://ubuntu.com/advantage

  System information as of Mon May 30 21:57:48 UTC 2022

  System load:  0.26              Processes:             101
  Usage of /:   2.9% of 38.71GB   Users logged in:       0
  Memory usage: 13%               IP address for enp0s3: 10.0.2.15
  Swap usage:   0%

0 updates can be applied immediately.

New release '20.04.4 LTS' available.
Run 'do-release-upgrade' to upgrade to it.

vagrant@ubuntu-bionic:~$

```

```
sudo apt-get -y update
sudo apt-get -y upgrade
sudo apt-get -y dist-upgrade
sudo apt-get -y install pv cpio rsync kmod execstack imagemagick inkscape graphicsmagick subversion git bc unar wget sudo gcc g++ binutils autoconf automake autopoint libtool-bin make bzip2 libncurses5-dev libreadline-dev zlib1g-dev flex bison patch texinfo tofrodos gettext pkg-config ecj fastjar          perl libstring-crc32-perl ruby ruby1.9 gawk python libusb-dev unzip intltool libacl1-dev libcap-dev libc6-dev-i386 lib32ncurses5-dev gcc-multilib bsdmainutils lib32stdc++6 libglib2.0-dev ccache cmake lib32z1-dev libsqlite3-dev sqlite3 libzstd-dev netcat curl openssl build-essential
```

```
vagrant@ubuntu-bionic:~/freetz-ng$ cat /etc/lsb-release 
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=18.04
DISTRIB_CODENAME=bionic
DISTRIB_DESCRIPTION="Ubuntu 18.04.6 LTS"
vagrant@ubuntu-bionic:~/freetz-ng$ uname -r
4.15.0-180-generic

```

```
umask 0022
git clone https://github.com/Freetz-NG/freetz-ng.git
cd freetz-ng/
wget -O.config https://github.com/Freetz-NG/freetz-ng/files/8800302/config.txt 
make olddefconfig
make
```

```
[...]
STEP 3: PACK/SIGN
  checking for left over version-control-system files
  integrate freetz info file into image
packing var.tar
  checking signature key files
    creating private signature key file
    creating public signature key file
    adding public signature key file
creating filesystem image (SquashFS4-xz)
  SquashFS block size: 64 kB (65536 bytes)
copying kernel image
  kernel image size: 6.0 MB, max 8.0 MB, free 2.0 MB (2090240 bytes)
copying filesystem image
  filesystem image size: 39.8 MB, max 44.0 MB, free 4.2 MB (4390912 bytes)
adding checksum to kernel.image
adding checksum to filesystem.image
packing images/6890_07.29.all_freetz-ng-19584M-1ed47e6fe_20220530-223932.image
  packed image file size: 46.0 MB (48240640 bytes)
signing packed .image file
  signed image file size: 46.0 MB (48240640 bytes)
source firmware: 6890-LTE_en-de-es-fr-it-nl-pl 162.07.29 rev93213 {ALL} [PSQ19P2NL4] (07.12.2021 17:32:22) 
  source image file size: 43.7 MB (45864960 bytes)
done.

FINISHED
```