Freetz-NG/freetz-ng #549: Fehler beim Bau von kernel drivers/crypto

pfichtner, 2022-05-29:
Fehler tritt sowohl beim Bau für eine 7390 (kernel/ref-iks-7390_06.80/linux-2.6.28) als auch eine 3270 (kernel/ref-vr9-3390_06.51/linux-3.10) auf. 

```
mips-unknown-linux-gnu-ld  -m elf32btsmip   -r -o fs/yaffs2/yaffs.o fs/yaffs2/yaffs_ecc.o fs/yaffs2/yaffs_vfs.o fs/yaffs2/yaffs_guts.o fs/yaffs2/yaffs_checkptrw.o fs/yaffs2/yaffs_packedtags1.o fs/yaffs2/yaffs_packedtags2.o fs/yaffs2/yaffs_nand.o fs/yaffs2/yaffs_tagscompat.o fs/yaffs2/yaffs_tagsvalidity.o fs/yaffs2/yaffs_mtdif.o fs/yaffs2/yaffs_mtdif1.o fs/yaffs2/yaffs_mtdif2.o fs/yaffs2/yaffs_nameval.o fs/yaffs2/yaffs_attribs.o fs/yaffs2/yaffs_allocator.o fs/yaffs2/yaffs_yaffs1.o fs/yaffs2/yaffs_yaffs2.o fs/yaffs2/yaffs_bitmap.o fs/yaffs2/yaffs_verify.o 
   mips-unknown-linux-gnu-ld  -m elf32btsmip   -r -o fs/yaffs2/built-in.o fs/yaffs2/yaffs.o 
   mips-unknown-linux-gnu-ld  -m elf32btsmip   -r -o fs/built-in.o fs/open.o fs/read_write.o fs/file_table.o fs/super.o fs/char_dev.o fs/stat.o fs/exec.o fs/pipe.o fs/namei.o fs/fcntl.o fs/ioctl.o fs/readdir.o fs/select.o fs/fifo.o fs/dcache.o fs/inode.o fs/attr.o fs/bad_inode.o fs/file.o fs/filesystems.o fs/namespace.o fs/seq_file.o fs/xattr.o fs/libfs.o fs/fs-writeback.o fs/pnode.o fs/drop_caches.o fs/splice.o fs/sync.o fs/utimes.o fs/stack.o fs/buffer.o fs/bio.o fs/block_dev.o fs/direct-io.o fs/mpage.o fs/ioprio.o fs/inotify.o fs/inotify_user.o fs/eventpoll.o fs/anon_inodes.o fs/signalfd.o fs/timerfd.o fs/eventfd.o fs/aio.o fs/locks.o fs/nfsctl.o fs/binfmt_script.o fs/binfmt_elf.o fs/mbcache.o fs/nfs_common/built-in.o fs/proc/built-in.o fs/partitions/built-in.o fs/sysfs/built-in.o fs/devpts/built-in.o fs/dcookies.o fs/ext3/built-in.o fs/jbd/built-in.o fs/ext2/built-in.o fs/squashfs/built-in.o fs/ramfs/built-in.o fs/nfs/built-in.o fs/lockd/built-in.o fs/nls/built-in.o fs/fuse/built-in.o fs/yaffs2/built-in.o 
make[1]: Leaving directory '/workspace/source/kernel/ref-iks-7390_06.80/linux-2.6.28'
make: *** [make/linux/kernel.mk:211: source/kernel/ref-iks-7390_06.80/linux-2.6.28/vmlinux.eva_pad] Terminated
```

rufe ich make dann erneut auf erhalte ich
```
make -f scripts/Makefile.build obj=fs/yaffs2
make -f scripts/Makefile.build obj=drivers/char/ubik2
make -f scripts/Makefile.build obj=drivers/clocksource
make -f scripts/Makefile.build obj=drivers/crypto
/workspace/source/kernel/ref-iks-7390_06.80/linux-2.6.28/drivers/crypto/Makefile:7: /workspace/source/kernel/ref-iks-7390_06.80/linux-2.6.28//drivers/crypto/Makefile.ifx: No such file or directory
make[3]: *** No rule to make target '/workspace/source/kernel/ref-iks-7390_06.80/linux-2.6.28//drivers/crypto/Makefile.ifx'.  Stop.
make[2]: *** [scripts/Makefile.build:361: drivers/crypto] Error 2
make[1]: *** [Makefile:942: drivers] Error 2
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/workspace/source/kernel/ref-iks-7390_06.80/linux-2.6.28'
make: *** [make/linux/kernel.mk:211: source/kernel/ref-iks-7390_06.80/linux-2.6.28/vmlinux.eva_pad] Terminated
```

Falls .config oder weitere Angaben benötigt werden bitte um Info.

fda77, 2022-05-29:
Ich bin mir nicht sicher an welcher der 2 Neuerungen es liegt: https://freetz-ng.github.io/freetz-ng/NEWS
Es betrifft wohl kernel von älteren Geräten

freetz-ng-mod, 2022-05-29:
nach ein git pull
```
make/linux/kernel.mk:39: Warnung: Die Befehle für das Ziel „dl/“ werden überschrieben
make/linux/kernel.mk:35: Warnung: Alte Befehle für das Ziel „dl/“ werden ignoriert
make/sundtek-rtspd/sundtek-rtspd.mk:14: Warnung: Die Befehle für das Ziel „dl/“ werden überschrieben
make/linux/kernel.mk:39: Warnung: Alte Befehle für das Ziel „dl/“ werden ignoriert
make/linux/kernel.mk:39: Warnung: Die Befehle für das Ziel „dl/“ werden überschrieben
make/linux/kernel.mk:35: Warnung: Alte Befehle für das Ziel „dl/“ werden ignoriert
make/sundtek-rtspd/sundtek-rtspd.mk:14: Warnung: Die Befehle für das Ziel „dl/“ werden überschrieben
make/linux/kernel.mk:39: Warnung: Alte Befehle für das Ziel „dl/“ werden ignoriert
configuration written to .config
```

fda77, 2022-05-29:
Als workaround `Download options  --->  Use old AVM konglomerat` aktivieren

@freetz-ng-mod
Deine .config ist zu alt -> `make olddefconfg`

pfichtner, 2022-05-29:
Hier die Bestätigung, git bisect sagt: 

```
4d29c10d54154be99b61cf7dc95acf751d3b4303 is the first bad commit
commit 4d29c10d54154be99b61cf7dc95acf751d3b4303
Author: fda77 <fda77@users.noreply.github.com>
Date:   Sat May 28 17:50:39 2022 +0200

    kernel: switch to vanilla

 config/mod/download.in | 16 ++++++++++++----
 make/linux/kernel.mk   | 41 ++++++++++++++++++++++++++++-------------
 2 files changed, 40 insertions(+), 17 deletions(-)
bisect run success

```

er13, 2022-05-29:
Das ist alles womöglich die Folge von https://github.com/Freetz-NG/freetz-ng/commit/4d29c10d54154be99b61cf7dc95acf751d3b4303#r74849870

Fehlt tools/xz werden all diese Link/Exec-"Fixes" nicht angewandt, der Build-Prozess läuft aber munter weiter, bis er eben irgendwo tief im Kernel-Bau fehlschlägt, womöglich weil die Link/Exec-"Fixes" nicht angewandt wurden oder gar unvollständig sind.

fda77, 2022-05-29:
Dein Fehler ist klar eine Folge des fehlenden xz. Ich nutze aber immer dl-hosttools, xz ist also immer vorhanden.
Dennoch bekomme ich für die aktuelle 7490

```
make -f scripts/Makefile.build obj=drivers/misc/lis3lv02d
make -f scripts/Makefile.build obj=drivers/misc/ti-st
make -f scripts/Makefile.build obj=drivers/mmc
make -f scripts/Makefile.build obj=drivers/mtd
/home/freetz/7490-freetz/source/kernel/ref-vr9-7490_07.27/linux-3.10/drivers/mtd/Makefile:38: /home/freetz/7490-freetz/source/kernel/ref-vr9-7490_07.27/linux-3.10/drivers/mtd/Makefile.ifx: Datei oder Verzeichnis nicht gefunden
make[4]: *** Keine Regel, um „/home/freetz/7490-freetz/source/kernel/ref-vr9-7490_07.27/linux-3.10/drivers/mtd/Makefile.ifx“ zu erstellen.  Schluss.
make[3]: *** [scripts/Makefile.build:455: drivers/mtd] Fehler 2
make[2]: *** [Makefile:806: drivers] Fehler 2
make[2]: Verzeichnis „/home/freetz/7490-freetz/source/kernel/ref-vr9-7490_07.27/linux-3.10“ wird verlassen
make[1]: *** [GNUmakefile:109: vmlinux.eva_pad] Fehler 2
make[1]: Verzeichnis „/home/freetz/7490-freetz/source/kernel/ref-vr9-7490_07.27/linux-3.10“ wird verlassen
make: *** [make/linux/kernel.mk:230: source/kernel/ref-vr9-7490_07.27/linux-3.10/arch/mips/boot/vmlinux.eva_pad] Beendet
```

Weshalb weiss ich noch nicht. Ich hab die Tabelle am Anfang der 71 Patches so erstellt:
```
#$(for x in $(find avm/linux* -type d -empty        ); do echo -e "    Dirs# ${x#avm/linux*/}                   "; done)
$(for x in $(find avm/linux* -type l               ); do echo -e "    Link# ${x#avm/linux*/} -#- $(readlink $x)"; done)
$(for x in $(find avm/linux* -type f -a -executable); do echo -e "    Exec# ${x#avm/linux*/}"                   ; done)
```
Leere Verzeichnisse von AVM sind in den Patches noch nicht enthalten, das sah ich erst später

er13, 2022-05-29:
`drivers/mtd/Makefile.ifx` ist in dem AVM-Baum leer. `diff` ignoriert leere Dateien, daher ist diese in dem Patch nicht enthalten.

fda77, 2022-05-29:
Nachdem ich den Fehler bekam waren in  `source/kernel/ref-vr9-7490_07.27/linux-3.10/drivers/mtd`  39 Dateien

Man kann einen Linux kernel  `~/.freetz-dl/linux-*.tar.xz` entpacken und in diesem Verzeichnis dann
```
vanilla2avm() {
  $HOME/freetz-ng/tools/freetz_patch ./ $1
  find ./ -type l -exec rm -f {} ';'
  xz -d $1 -c | grep -E '^    (Dirs|Link|Exec)# .*' | while read a b c d; do
    [ "$a" == "Dirs#" ] && mkdir -p ./$b
    [ "$a" == "Link#" ] && mkdir -p ./${b%/*}
    [ "$a" == "Link#" ] && ln -sf $d ./$b
    [ "$a" == "Exec#" ] && chmod +x ./$b
  done
}
#vanilla2avm ~/.freetz-dl/kernel_*-avm.patch.xz
```

ausführen. Sieht soweit gleich aus. Abgesehen von den Timestamps

closed 2022-05-30 by fda77

fda77, 2022-05-30:
Sollte jetzt mit allen Kerneln funktionieren.
Die 72 Patches haben zusammen 280MB, ein kompletter kernel von avm je 50-100MB, etwa 7gb zusammen
Das Script zum erstellen ist hier, falls es weitere Probleme gibt: 36d90ce94d6a3f065c9194fcf681e426ae6d37ed
Die AVM-Sourcen von 7582 07.15 sind nicht wie alle anderen gepackt, dessen Patch muss noch angepasst werden

freetz-ng-mod, 2022-05-30:
Also könnte ich damit dann zb ein Patch für denn 1200AX machen. Frage nur zur sichherit nach

fda77, 2022-05-30:
Ich verstehe nicht ganz. Fehlen in Freetz vom 1200ax noch sourcen?

freetz-ng-mod, 2022-05-30:
Man kann zb nicht cifsmount nutzen da gibt es bis heute morgen immer noch Fehler. Daher hatte ich ja zur sicherheit damals die source abgefragt bei avm, und ja gepostet

fda77, 2022-05-30:
Die 1200ax nutzt aktuell in freetz sourcen der 6000. Da die 1200ax ein FIT ist sollte das nicht unbedingt passen.
http://osp.avm.de/fritzwlan/fritzrepeater-1200-ax/source-files-FRITZ.Repeater_1200_AX-07.30.tar.gz einthält eine "linux-4.4.176.tar.xz" (gleiche md5 wie von kernel.org), die Box läuft aber mit kernel 4.4.60. 4.4.176 gibts eigentlich auf keinem avm-Gerät.
Bleibt nur das gleiche wie immer -> AVM nerven