Freetz-NG/freetz-ng #554: 6890v2 wireshark / update fuer push_firmware

Himan2001, 2022-05-31:
So, bin mal eben dazu gekommen:

[image]

Hier mal der benötigte Wireshark-Teil aus dem original AVM Recovery 7.29 für die 6890v2:

....mtd1... 0x8c000000 als offset und natürlich das entsprechende memsize entsprechend der Grösse des zu flashenden Binarys.
RAM/Flash exakt 0x1000000 Unterschied.

Wichtig ist halt, das es bei einer nagelneuen Box keine "fs_linux_start" variable gibt, es wird immer von mtd1 erstmalig ausgegangen!
(heisst das auslesen mit push_firmware, wo es derzeit noch 2 Restarts braucht, meldet halt im ersten Durchgang den obligaten Fehler
und der Default sollte dann wohl "Bank 0" sein - also mtd1). Im Grunde könnte push_firmware direkt nach dem Auslesen des
Enviroment schon loslegen - ein 2. Neustart ist hier wie gesagt unnötig, sobald die ENV Variablen bekannt sind, einfach
die ftp commands absetzen und los.

Am Ende des Flashvorgangs tut sich dann auch erst mal eine Weile nix mehr. mit eva_to_memory gabs zumindest auch keine
Rückmeldung. Da habe ich einfach die Box LEDs beobachtet und irgendwann wars dann durch.... Etwa 3-4 Minuten später war die Box dann wieder pingbar.

fda77, 2022-05-31:
> ....mtd1...

lesen!!! Zum Glück hast du einen Screenshot angehängt

> bei einer nagelneuen Box keine "fs_linux_start" variable gibt,

Es ist eigentlich dass nach einem "recover" diese variable nicht existiert und so "0" ist

> meldet halt im ersten Durchgang den obligaten Fehler

Hö?

>  "Bank 0" sein - also mtd1

Wos? Partitions "set" 0 oder 1 sind je mehrere partitionen...

> ein 2. Neustart ist hier wie gesagt unnötig, sobald die ENV Variablen bekannt sind, einfach
die ftp commands absetzen und los.

Nö, eva ist dann müde und muss restarten

> am Ende des Flashvorgangs tut sich dann auch erst mal eine Weile nix mehr

Quatsch, das ins ram geladene system startet und flasht sich danach selbst. sieht du dann mit serieller console. Steht da nicht was von "wartenn" beim flashen?

Es wäre halt ganz gut wenn jemand Wissen sammeln würde UND irgend wo aufschreiben würde, zb unter https://github.com/Freetz-NG/freetz-ng/tree/master/docs/wiki . Dazu hat aber niemand lust

fda77, 2022-05-31:
Der screenshot reicht für die infos. Optiomal wäre das Log der Verbindung auf 20/tcp. Rechtsklick auf eine der zeilen, dann "follow stream" oder so

0x944e500  148mb fürs laufende linux
0x8944e500 - 0x944e500 = ram beginnt bei 0x80000000
0x80000000 bis 0x8c000000 = C000000 = 192 mb

192-148= dein image ist 44mb (das ist auch im dump drin)

Dh der Wert den das recovery nutzt ist 192MB, obwohl die box 512MB ram hat -> nur ein dump zeigt den wirklichen wert der im recovery hinterlegt ist.
Man müsste mal die ganzen .exe auslesen. Wofür die restlichen 320mb reserviert sind ist nicht bekannt

closed 2022-05-31 by fda77

fda77, 2022-06-02:
Wie lange hat es denn nun gedauert alle Toolchains zu bauen?