Freetz-NG/freetz-ng #555: FB6890v2 Labor: "FREETZ_MODIFY_DSL_SPECTRUM" als optionaler Patch geht hier aktuell nicht mehr

Himan2001, 2022-05-31:
Beim aktuellen 6890v2-Labor funktioniert dieser Patch nicht mehr und muss deaktiviert bleiben erstmal.

Bei der Final 7.29 passt das allerdings noch. Statt zu deaktivieren würde ein rebase reichen, da es einen REJECT
hier nur gibt - das Template hat sich hier etwas geändert

Betrifft dieses File: "fritzbox-6890-labor-95589.zip"

-> # FREETZ_MODIFY_DSL_SPECTRUM is not set

Die anderen Patches passen mit etlichen Hunks immer noch problemlos.

freetz-ng-mod, 2022-05-31:
Dann benutze doch mal die aktuelle Labor 96061. Denn der Patch paßt.

Himan2001, 2022-05-31:
WTF? ich habe den auf Labor "95589" vor 2 Tagen angewendet, da gabs noch den Error.

Allerdings musste das Script 6 Server befragen bis ueberhaupt ein Server das Labor als .zip mal liegen hatte, und das war wohl ein recht altes.... ?

Ist gut wissen das da im Tagestakt offenbar für die Kiste Labore kommen?

Habe bei AVM x-mal gemeckert über ipv6 Bugs und Unzulänglichkeiten die seit 2 Jahren dort schon drin bestehen, wie z.b. kein ipv6 mit 3GPP Version 10 - aber Version ist CAT6 - und Version 8 entspricht CAT 4 - somit kein Carrierbunding, dafür aber ipv6. Hatten die nie gefixt gehabt so richtig.

Dann gibts für den DSL/WAN Teil kein "Telekom" bei den Voip-Settings nur Telekom LTE - und das geht ueberhaupt nicht.

Auch das Thema ipv6 im Mischbetrieb dann nicht mehr über LTE habe ich bemängelt ebenso fehlende extra DNS-Server nur für VoIP. Einige Anbieter wie Telekom & Co mögen nicht wirklich alle "freien" DNS-Server als Resolver - besonders wenn man ueber deren Mobilnetz (LTE-Backup, usw rein kommt... da bekomme ich aktuell "Host not found" und so Witze...

werde dann mal das frische Labor jetzt testen. Danke für die Info :-)

fda77, 2022-05-31:
Wenn du es magst das manuell zu machen: https://freetz-ng.github.io/freetz-ng/JUIS
Hat jemand von euch ein avm-powerline Gerät !ohne! FOS? Davon bräuchte ich noch einen juis-tcpdump

freetz-ng-mod, 2022-06-01:
Ich habe kein poweline Geräte

fda77, 2022-06-01:
Schade, die fehlen noch auf der JUIS Seite weil nicht bekannt ist wie die angefragt werden

Himan2001, 2022-06-01:
Ich habe jetzt bei der 6890v2 das Labor 07.39-96061 BETA (die Version wo gerade noch 0.1 MB frei sind) kompiliert und ueber freetz eingespielt und jetzt, trotz Werksreset das Problem:

"Die Internetverbindung wird initialisiert. Bitte warten Sie einen Augenblick und klicken Sie dann auf "Aktualisieren"."

Klartext: es wird kein LTE-Modul mehr gefunden, segfaults sehe ich in der shell auch nicht. mit dem aktuellsten Labor + Freetz gehts einfach nicht mehr....

Frage an @freetz-ng-mod 

Was hast Du für Settings, bei Dir scheint das aktuelle Labor das LTE Modul wohl noch zu finden ?
Stehe gerade etwas an und muss Raten wieso ich jetzt diese Kuriosität habe nach Swiwtching zum Labor.....

Gibts da jetzt irgend einen Patch dem in ICHT aktivieren darf für das aktuelle Labor ?

Strange alles.....

Himan2001, 2022-06-01:
ja das weiss ich eh, lebt eh noch alles. Ich hab jetzt das neue image mit weniger patches mal im AVM webinterface geflasht nicht im freetz, zumindest meint er jetzt es wurde geflsht und er rebootet. mal sehen was gleich passiert :-)

danach mal trustedd deaktivieren, kann bei der labor sicherlich was ausmachen - in der shell gibt es allerdings keinerlei hinweise auf fehlfunktionen, die kernel module sind alle geladen und er erkennt auch eine SIM-Karte. aber es werden keine Mobilfunknetze gefunden, und ich bekomme eben eine Fehlermeldung -1 das kein Internet vorhanden sein, die Kommunikation mit dem LTE Modul tut einfach nicht mehr. Auch eben Werksreset fürs Labor und alles sauber neu confen dasselbe Ergebnis, geht nur DSL kram noch ;-)

Himan2001, 2022-06-01:
Fehler offenbar gefunden!!

Wenn ich das freet-labor uebers AVM webinterface flashe geht LTE wieder, uber das freetz Firmware geht nicht, Ich habe jetzt aber 2 Optionen noch geändert:

- "REMOVE Leftovers" habe ich weggelassen, heisst die sind jetzt drin, das teste ich gleich nochmal ob es nun am Freetz flashing oder am AVM webinterface gelegen hat.....

Himan2001, 2022-06-01:
SO, fehler gefunden:

a.)
Beim letzten Labor der 6890v2 darf REMOVE Leftovers NICHT gesetzt sein. Wenn das gesetzt wird geht LTE nicht mehr!

b.)
Das Fashen im Freetz-Webinterface ist nicht einwandfrei, es MUSS im Moment direkt ubers AVM-Webinterface geflashed werden.

Jetzt rennts.

Speicher ist aber wie gesagt am Ende: 0.2 MB noch frei, Da kannst kein MyFreetz und tr-069 mehr rein tun, oder man muss auf jegliche anderen Pakete die man braucht verzichten. Fühle mich wie bei der 7390 wo auch für nix Platz ist ;-)