Freetz-NG/freetz-ng #559: 5530 7.29 trusted/untrusted flashing, working, not-working, serielle logfiles + config

Himan2001, 2022-06-03:
So, ich hatte mal etwas Luft und habe 4 serielle Logfiles erstellt zur 5530:

[flashing_freetz_with_untrusted.txt](https://github.com/Freetz-NG/freetz-ng/files/8828132/flashing_freetz_with_untrusted.txt)
[flashing_freetz_without_untrusted.txt](https://github.com/Freetz-NG/freetz-ng/files/8828133/flashing_freetz_without_untrusted.txt)
[recovery_mit_gbic0729.txt](https://github.com/Freetz-NG/freetz-ng/files/8828134/recovery_mit_gbic0729.txt)
[recovery_ohne_gbic0729.txt](https://github.com/Freetz-NG/freetz-ng/files/8828135/recovery_ohne_gbic0729.txt)

dazu noch die entsprechende test-config:

[.config.txt](https://github.com/Freetz-NG/freetz-ng/files/8828138/default.config.txt)

Folgendes geben die oberen Logfiles mal wieder/Beobachtungen:

Grundsätzlich endet das AVM original 7.29-recovery immer mit einem segfault (!), wobei es hier noch einen Unterschied gibt:

a.) Recovery OHNE ein eingestecktes GBIC durchführen: Nach dem direkten Flashen und ersten Starten tritt am Ende immer ein segfault auf und die Box restartet im Anschluss. Danach ist sie erreichbar über webinterface.

b.) Recovery MIT eingestecktem GBIC durchführen: In diesem Fall kann es bis zu 3 (!) neustarts benötigen, und jeder neue reboot
erzeugt ebenfalls einen segfault. Nach etwa 5-7 Minuten ist dann die Box ueberhaupt erstmalig erreichbar....

So zur Thematik mit "broken" betreffs "untrusted" und flashing:

Dazu sind oben auch 2 serielle Logs angefügt (wie gewünscht):

Mit AKTIVEM untrustedd wird (offensichtlich) auch vollständig geflashed, jedoch wird Freetz dabei NICHT aktiv, also das bekannte Problem.

Im letzten Durchgang habe ich bei der beigefügten config lediglich untrustedd DEAKTIVIERT. in diesem Fall ist Freetz anschliessend vorhanden und die Box ist "stabil" nach wenigen Minuten erreichbar und es können alle möglichen Dinge dann ueber das AVM-Webinterface durchgespielt werden.

Flashing von Firmware über das Freetz-Webinterface habe ich nicht weiter durchgespielt - dort steht Bank0/1 Flashing für die 5530 aktuell auch nicht zur Verfügung und war auch nicht Sinn dieses Tests.

Interessant ist jedoch, das selbst das original Recovery von AVM immer mit einem segfault bei der Version 7.29 schon unterwegs ist und das man bei einem recovery am besten immer das Gbic aussteckt (!) - schon etwas problematich wenn das so schon losgeht....

Thema serielle Schnittstelle: Für Nachmacher und Selberlogger:
unbedingt den TX-pin zum PL2303 oder FTDI Adapter NICHT anschliessen. Die 5530 hat dieselben Macken wie die 7520/7530:
Ein aktiver TX-Pin kommend vom Terminal kann zu sporadischen segfaults oder reboots oder hängern führen - daher habe ich nur mit dem RX-Pin geloggt in Empfangsrichtung zum puTTY!

Remove Leftovers habe ich ich NICHT verwendet, um zu vermeiden das es zu noch mehr Nebeneffekten kommt.

Leider erschliessen sich mir aus den reinen consolen Logfiles kein Grund, wieso ein mit push_firmware hochgeschobenes Image
mit aktivem "untrusted" letztlich "nicht aktiv" wird und es beim Original Image von AVM bleibt - oder ich habe da irgend etwas uebersehen in den vielen Logausgaben ?

Im letzten Durchgang dann wieder meine gewohnte Version mit deaktivem trusted. Diese Version wird wie bisher auch dann uebernommen - bis auf das Bank 0/1 nicht zur Verfügung stehen im Freetz.

Seit mehr als 1 Stunde sind WLAN, VoIP und alles Weitere stabil und dmesg zeigt keinerlei Auffäligkeiten, errors, segfaults
oder ähnliche Unarten.

Ich hoffe mit dem 4 logfiles und der config kommen wir ein wenig weiter die Ursache mit dem trustedd jetzt herauszufinden.

Gerne verschieb das Ganze auch als Diskussion wo dann mehrere Besitzer der 5530 auch Ihre Erfahrungen mit reinposten können.

Zur Aussage: "nimm 7.21 oder 7.26": Da kommt dann das VoIP-Problem bei Telekomanschlüssen mit den neueren DNS-Abfragen wieder zum tragen - ebenso beim WLAN einige bekannte Probleme zurück....?

Leider gibt es offenbar keine Inhouse oder neuere Labor zur 5530 bislang. Jedenfalls spricht es schon Bände wenn selbst das original recovery von AVM grundsätzlich mit segfaults restartet und mit angesteckten Gbic nach einem recovery nochmal 2-3x durchstartet bis die Box dann endlich wieder erreichbar ist....

@fda77 

Bitte sag mir welche logs du gerne als nächstes hättest, nachdem Du jetzt die seriellen Logs, eine entsprechende config mit der ich freetz teste, usw alles vorliegen hast.

Aus meiner Sicht wären da Versuche wie binutils auf gleiche Version wie AVM downzugraden oder wie schon ein Kollege schrieb, die busybox Version und die aktuelle Busybox config meine nächsten Verdachtskandidaten.

Was auch nicht vergessen werden sollte: es wird immer vom dsld & co. ausgegangen....

bei der Fiber Box rennen aber noch andere daemons, die wir evtl. mit den bisherigen Patches? nicht berücksichtigt haben ausreichend:

```
root@fritz:/var/mod/root# ps w
  PID USER       VSZ STAT COMMAND
    1 root      1916 S    init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
[... 269 lines cut ...]

Die DG hat mir ferner auch mitgeteilt, das es wärend der gesamten Ausbauphase eines Ortes in der Anfangszeit auch vielfach noch zu ungeplanten Ausfällen noch in der Anfangszeit kommen kann - erst nach Abschluss eines ganzen Ortes sollte dann auch eine entsprechende Stabilität dauerhaft vorhanden sein. -> Für Leute die aufs Festnetz angewiesen sind dauerhaft kann das schnell zu einem Problem werden wärend der Bauphase nach einem Wechsel der Zugangs-Technologie.

Ich habe zusätzlich mit der DG dieses Thema VoIP und Portierung auch noch weitergehend abklären können, was viele nicht wissen:
Rufnummern vom Altanbieter können über den Kundenservice auch nachträglich rüberportiert werden, es muss NICHT zum Schaltungstermin das Risiko des harten Umschaltens geben! - Will man die Doppelzahlung wg. Restlaufzeit vermeiden kann man aber, sofern man mehrere Nummern beim alten Anbieter hat eine "Trashnummer" zur DG portieren aus seinem "Pool" - dann kommt man doch in den Genuss der Frei-Monate, solange der Altvertrag noch Bindung hat.

Sofern man es sich finanziell übergangsweise leisten kann wäre mein dringender Rat, wenn man auf Internet & Telefonie
beruflich angewiesen ist im HomeOffice den Alt-Vertrag NICHT per Stichtag zu kündigen und dann ins kalte Wasser zu springen,
sondern aufgrund der Gesetzesänderung mit 30 Tagen nach Ablauf der Mindestlaufzeit den DSL-Vertrag noch eine Weile
weiterlaufen zu lassen, somit auch die gewohnte Telefonie erst einmal dort zu belassen mit den "wichtigen" Rufnummern.

Und dann schaut man sich das FTTH mit der einen portierten "Trashnummer" und der Verfügbarkeit erst mal 1-3 Monate in Ruhe an, und kündigt den Altvertrag erst, wenn der neue FTTH-Anschluss wirklich sauber, stabil mit Internet UND Telefonie verlässlich
rennt. Auch dazu habe ich eine Auskunft erhalten: Das Störungsbehebungen durchaus mal 2-3 Tage andauern können, Bei den DSL-Anbietern, bzw. Leitungen die über die Telekom geschaltet werden ist eine Entstörung oft zig fach schneller meistens möglich!!

- Eine Option, wie ich sie habe, "Expressentstörung 8 Stunden" gegen kleinen Aufpreis" bietet die DG im Privatsegment aktuell nicht an!! - Bei Telekom-Kunden kann jeder Privatkunde diese "Business-Option" ueber die Serviceline jederzeit mit einer 12 monatigen Mindestlaufzeit zu allen Privat-Tarifen auch dazu buchen. Aber nur Telefonisch nicht über das Kundenportal ;-)

Wichtige "Support-Info" noch zu der Warnung der DG wenn Kundeneigener Router verwendet wird, wo es dann ja "keinen offiziellen" Support geben soll (...) - Dazu habe ich folgende Auskunft bekommen:

Da bei dieser Konfig die DG mit dem tr069 der Fritzbox aktuell wohl nocht nicht arbeitet hat die DG dann keinerlei Einsicht auf die Kundenseite?! Bei einem Ausfall muss der Kunde dann gewisse Daten wie die ModemID, MAC-Adresse usw. seiner eigenen Fritzbox
dem Support selber übermitteln damit die trotzdem ein Mindestmass an Diagnosen durchführen können. Es gibt dort dhcp anstelle von PPPoE - somit stehen bei OFFLINE dem Support so gut wie keine Infos zur Verfügung - es gibt dann eben kein Glasfasermodem mehr wo sie bis zum Kunden reinschauen können. 1und1 mit seiner OEM-Version nutzt wie immer schon das tr069 und hat bei den Wholesales Sachen dann einen besseren Einblick, zumindest auf Geräte die von denen gebrandet bezogen worden sind.

[... cut ...]
Als Quickfix kann man aber diese beiden Daten im Bastelgerät auch "spoofen" und die Daten des "Vertragsgerätes" verwenden.
Die MAC kann ueber das AVM-Webinterface geändert werden, die ModemID müsste ueber das Enviroment im EVA geändert werden oder über eine Freetz/Shell-Option ;-) - dabei hängt es bei aktuellen Bootloadern natürlich davon ab, ob diese Änderungen auch permanent dann übernommen werden - siehe 1und1 - aktuelle Geräte wo man das branding auch nicht mehr dauerhaft auf "avm" abspeichern kann ;-)

fda77, 2022-06-03:
Ja, das sind genügend logs .. vielleicht sogar zu viele, da ich bei den ganzen Crashes nicht durchblicke
 - Das flashen interessiert nicht. Wenns drauf ist ists gut. Dass bei AVM vieles nicht optimal umgesetzt ist war schon immer so. Zb bei der 7270 durfte man nichts am Syslog ändern, da sonst Dect nicht mehr funktioniert das syslog gegrept hat -.-
 - Bei deinem DG resüme hast du vergessen dass es keine IPv4 mehr gibt, selbst in dem Tarif für über 100 Euro monatlich!
 - Die hat 3 kerne, steht auch in deinen logs so. Vielleicht von der Resterampe bei denen der 4. kaputt war :)
 - Das env kann man bei neuen nicht mehr ändern, es gibt aber im menuconfig optionen zum überlagern (hab ich nie benötigt+benutzt)
> unbedingt den TX-pin zum PL2303 oder FTDI Adapter NICHT anschliessen. 

Das klingt nach Einstrahlung in die TX-Leitung und Phantomeingaben. Mal das Kabel von der Platine hochbiegen oder so. Hatte ich auch schon
 
Für die Logs: Firmware flashen, mal AVM, mal mir freetz, mal ohne den untrustedd (trustedd läuft auf einem core an dem freetz nix macht).
DANACH Strom rausziehen und ein Log erstellen vom Start bis die Box komplett hochgefahren ist. Falls sie mehrere anläufe braucht die alten Versuche wegwerfen

Himan2001, 2022-06-03:
DG: Ja das mit Zwangsnat kann man für "schlappe" 35 euro aufpreis und ggfs. verhandeln umgehen. einfach das gleiche produkt als "Business-Entry" buchen ist man teilweise immer noch unterm Telekompreis aber hat immer noch massiv mehr upstream :-) - ich glaub dann gibts auch static v4. Bis zu 8 IPs kann man bekommen bei diesen Varianten. Iss schon happig für eine einzelne IP, normal bekommt man sowas auch um 5 euro bei Vodafone im Cable (wenn sie wollen am Ende....)
Zusätzlich im Business-Entry ist die Entstörzeit bei Ausfällen erheblich schneller als beim Endkundenprodukt, also auch
am Wochenende direkt nach Meldung - stand ja selbst vor der Entscheidung obs mir den kleinen Aufpreis wert ist :-)

Box:
Hmmm. hatte ich Dir nicht genau diese Logs gemacht ? die haben alle mit einem Kaltstart angefangen. die ersten waren Recovery von AVM, einmal mit eingestecktem gbic, einmal ohne.

Die anderen beiden waren mit trusted und untrusted ebenfalls direkt nach dem kaltstart, das da zwischendurch
auch nen crash drin war und sich das ganze dann nochmal wiederholt ist ja auch ein Teil des Problems, ich habe nicht bewusst extra viel geloggt nur so lange bis ich aufs webinterface dann auch einloggen konnte. Ich habe dann auch im Webinterface geconfed um zu schaun ob weitere Ausgaben noch erscheinen aber mehr ist nicht passiert.

Ohne trustedd rennt die einfach, keine reboots, und WLAN einwandfrei. einzig FTTH dran konnte ich nicht testen, aber gbic werden beide Typen auch einwandfrei erkannt. Auch wechseln im laufenden Betrieb ohne neuen segfault ging im laufenden Betrieb.

Kannst Du bei den gemachten logfiles mal vom Ende an zurückscrollen bis zum bootloader-start, dann hast du genau die logs die du kleiner haben wolltest, das waren die kaltstarts NACH dem Flashen - Am Anfang hat das log den flashvorgang geloggt weil wir ja auch rausfinden wollen wieso sich das ganze ohne trustedd gar nicht erst flashen lies. Da habe ich jetzt durch die console die Bestätigung gesehen das auch das geflashed wird, aber es wird dann nicht in diese Bank umgeschaltet und sofort wieder das Original verwendet - der nächste Boot dauerte daher auch länger.

Das gibt aber das serielle Log nicht aus - hatte ich irgendwie drauf gehofft gehabt. (Nicht das der Bootloader jetzt auch certificate drin hat und schon vor dem boot das image verweigert, lol) - oder stellt das Teil bei vorhandenem trustedd fest das da was nicht original ist und switched die BANK um/reboot ?

Ich sehe selber im consolenlog bei den vielen Zeilen nicht wirklich ein Hinweis....

Himan2001, 2022-06-03:
hrhrhr, na bei den aktuellsten Modellen ist avm ja bald alles zuzutrauen, signiert in sich selber ist der bootloader ja schon. eine 7520 hatte ich ja damals durch nen patch schon gekilled und dann war die box tod :-)

Was ich jetzt gerne mache ist dieselben logfiles nehmen und nochmal kürzen für dich wo der kaltstart alleine drin ist, zum besseren vergleichen dann hast es einfacher :-)

Himan2001, 2022-06-03:
[freetzboot_with_untrusted.txt](https://github.com/Freetz-NG/freetz-ng/files/8833098/freetzboot_with_untrusted.txt)
[freetzboot_without_untrusted.txt](https://github.com/Freetz-NG/freetz-ng/files/8833099/freetzboot_without_untrusted.txt)

Das sind jetzt die beiden COLDBOOT files, einmal mit und einmal ohne trustedd.

Sind im Grunde die letzten Teile aus den längeren Teilen.

Der Anfang aus den anderen Logs beginnt mit dem push_firmware. einmal trusted einmal untrusted und den reboots nach dem flaschen ;-) -  und ganz am ende ist dann bei beiden files der kaltstart den du ja haben wolltest ;-)

fda77, 2022-06-03:
> Der ANFANG aus den anderen logs sollte uns eigentlich einen Hinweis geben wieso das flashen mit aktivem trustedd nicht angenommen wird 

Das ist wieder eine dieser abstrusen Theorien. Was ist "angenommen" ? Kein lan-Verbingung? Wenn was nciht glatt läuft sieht man das meist auf der console und muss vergleichen

fda77, 2022-06-03:
es fehlt noch ein log avm-only booten

Mit untrustedd ist dies mehr
```
 [    7.826283] cadence-qspi 19300000.qspi: missing lantiq,qspi-physize; default to 128 MB
 [    7.832775] cadence-qspi 19300000.qspi: DMA NOT enabled
+[    7.838558] cadence-qspi 19300000.qspi: timeout waiting for qspi dll lock
 [    7.845076] cadence-qspi 19300000.qspi: master is unqueued, this is deprecated``
```
aussderdem wird das "cpunet" früher geladen und hat 2 "link is not ready". 

kommt später:
```
phy-xrx500 16210100.phy-prx300: booting GPHY firmware for PRX300
```

Diese Meldungen fehlen mit untrusted, vielleicht eine 2. cpu die nicht will:
```
[   13.509292] [avm_debug] standard kernel-messages
[   13.519072] [module-mem] give 0xc000 bytes at 71911000 to module 'Piglet_noemif' (0x8b8000 bytes left)
[   13.529542] Could not find Env 'kernel_args' in the device tree 
[   13.534167] [avm-gpio-shim] avm_get_hw_config: ignoring p_param pointer!
[   13.540788] [avm-gpio-shim] avm_get_hw_config: ignoring p_param pointer!
[   13.547472] [piglet]use settings for 257(2 gpios from hw_config)
[   13.547472] 
[   13.554968] [piglet]no TDM-Check because Host-CPU is master
[   13.573254] [piglet]patch_dectfw: patch_adress: a0 max_len: 53 with Version 1 Len: 17
[   13.579685] [piglet]patch_dectfw: RFPI tag found 039478b9e8
[   13.585236] [piglet]get_tag_value: tag 9 not found -> ignore
[   13.590836] [piglet]patch_dectfw: dectmode 4
[   13.595660] [piglet]patch_dectfw: c07c0000, totallen = 51637(c9b5) crc=0xac
[   13.628795] [piglet]slow: dect_loader-bootmode
...
```

Wäre interessant wies es ganz ohne freetz sein sollte

fda77, 2022-06-03:
Zur Erinnerung: untrustedd musste man mit 7590 ab einem gewissen FOS (7.2?) und ähnlichen geräten anfangs auch rauswerfen. Ursache war das eine modules.bin (oder so) fehlte und das Modul nicht geladen werden konnte. Das kann man hier irgendwo in einem langen Thread nachlesen

Himan2001, 2022-06-03:
Sekunde eben ich grab eben ein bootlog der originalfirmware noch aus....

Himan2001, 2022-06-03:
[boot_original_mit gbic0729.txt](https://github.com/Freetz-NG/freetz-ng/files/8833276/boot_original_mit.gbic0729.txt)

Hier ist der original-AVM- coldboot direkt nach dem flashen mit eingestecktem gbic

Himan2001, 2022-06-03:
also das AVM usb treiber in den kernel einer nicht usb-box statisch mit drin hat und die treiber läd ist ja auch sinnfrei sehe ich gerade :-)

Himan2001, 2022-06-03:
> Zur Erinnerung: untrustedd musste man mit 7590 ab einem gewissen FOS (7.2?) und ähnlichen geräten anfangs auch rauswerfen. Ursache war dass eine modules.bin (oder so) fehlte und das Modul nicht geladen werden konnte. Das kann man hier irgendwo in einem langen Thread nachlesen

Frage, da Du da fitter bist... kannst Du mal ein Original image auspacken und mal die Ordner nach ".bins" durchstöbern ?
Ich hab da auch diese eeproms 24c02 gesehen. Es kann durchaus sein, das irgend ein Intel-Chip eine Firmware noch braucht
die wir evtl. vergessen haben ? - Evtl. benötigt der PHY für das GBic sowas ?

fda77, 2022-06-03:
-> `tools/fwdu`

"random: fast init done" kommt bei avm viel früher. Es war schonmal im Verdacht dass Zufall fehlt, bei einem anderen Gerät

"cadence-qspi 19300000.qspi: timeout waiting for qspi dll lock" ist nur mit freetz+untrustedd

Himan2001, 2022-06-03:
...und hilft Dir das jetzt irgendwie weiter ?

mir sagt das gerade gar nichts da ich damals die 7590er sache nicht verfolgt hatte. Ich war ueber nen halbes Jahr raus allem mit freetz rausgewesen und es hat sich viel getan bei vielen Kisten ;-)

fda77, 2022-06-03:
Wir sind auf dem gleichen Stand, ich habs mittlerweile vergessen :D

fda77, 2022-06-04:
Jetzt im richtigen issue:

Damit ich die Dateien nicht verliere, mit ein paar Leerzeilen, zu 99% gleich:
[boot_original_mit.gbic0729.txt](https://github.com/Freetz-NG/freetz-ng/files/8837400/boot_original_mit.gbic0729.txt)
[freetzboot_with_untrusted.txt](https://github.com/Freetz-NG/freetz-ng/files/8837401/freetzboot_with_untrusted.txt)
[freetzboot_without_untrusted.txt](https://github.com/Freetz-NG/freetz-ng/files/8837402/freetzboot_without_untrusted.txt)

Bei "without untrustedd" fehlt für cpu3 das `Outside WAIT got IPI interrupt to launch from cpu 3`, das gibts nur für 1+2

@Himan2001:
Noch ein Klassiker, versuch mal:
Die Busyboy NICHT ersetzen und die von AVM in Freetz behalten. Dafür gibt es keine Option im menuconfig, das musst du selbst basteln!

Himan2001, 2022-06-04:
Ja, so eine idee hatte ich auch, um erst einmal auf den kleinsten Nenner zu kommen. Damit das aber nicht zu viel Durcheinander wird, würde ich gern erst mal die 5590 mit dem gbic-problem lösen - die 5530 hat ja "nur" ein untrustedd-Problem, rennt aber soweit erst mal "dry".

Bei der 5530 hatte ich Dir erstmal wie gewünscht die Bootlogs seriell erstellt gehabt, das haben wir jetzt ja durch und nun will die eigentliche Ursache noch gefunden werden ^^

fda77, 2022-06-04:
Es ist hat nicht so gut dass nicht einfach ne Fehlermeldung da drin ist an der man erkennen kann woran es liegt.
Nur dass gewisse Dinge in anderer Reihenfolge starten

fda77, 2022-06-05:
Vielleicht hilft c4370f578 weiter

freetz-ng-mod, 2022-06-22:
Fritz_Box_HW257 (FRITZ!Box 5530 Fiber)
da gibt es nun eine inhaus Version

fda77, 2022-06-22:
Ich kann bezüglich untrustedd keine Unterschiede zwischen 7.20+ und 7.25+ feststellen. Vielleicht hattest du den remove patch beim testen noch an

fda77, 2022-07-04:
beim booten mit freetz image:
```
Eva_AVM >
Eva_AVM >............................................................................[    0.000000] Linux version 4.9.231 (jenkins@jenkins-ci.avm.de) (gcc version 8.3.0 (Buildroot 2018.11.4) ) #1 SMP 2021-09-17
```

und mit avm image:
```
Eva_AVM >..........................4Kec starting @ 0x88299590
............................................................................[    0.000000] Linux version 4.9.231 (jenkins@jenkins-ci.avm.de) (gcc version 8.3.0 (Buildroot 2018.11.4) ) #1 SMP 2021-09-17
```

dies ist die Adresse des 2. kernel, aus der dti:
```
        prxB_HW0257_kernel {
            load = <0x88000000>;
            entry = <0x88299590>;
```
dieser 5. blob im fit (6+7 sind seine DTs) liegt hinter dem 4. mit dem (geänderten) Dateisystem. Der Hautpkernel ist 1,  mit seinen DT in 2+3

Vielleicht wird die "signatur" die von avm um das fit gebastelt wurde geprüft...?

Jedenfalls ist später das Problem dass das cpunet nicht richtig läuft. Auf dem Hauptsystem gibt es "fd62:c5d3:4303:c662::2" über die untrusted auf (trusted an) "fd62:c5d3:4303:c662::2" von "B" versucht zu verbinden. Und untrustedd darauf den Watchdog auslöst

fda77, 2022-07-04:
Wenn man einem AVM fit-image die 512bit "signatur" am anfang durch 00 überschreibt und dies ohne weiter Änderungen flasht gibt es auch einen Crash:

```
[  177.331685] [2]AVM_WATCHDOG_reboot(hdl=3, untrustedd): reboot (current: swapper/2 oom_score 0)
[  177.338838] pagefaults absolut 8882 since last untrustedd-trigger 362
[  177.345470]   hdl= 2 supervisor    pid  745 triggered before:  10.240 s(avg  16.940 s) state S cpu0 pgfault      0 oom_score 19
[  177.356725]   hdl= 3 untrustedd    pid  879 triggered before: 163.592 s(avg   0.000 s) state S cpu2 pgfault      0 oom_score 203
[  177.368256]  force SIGBUS for untrustedd (pid= 879)
[  177.373126] ar7wdt_hw_reboot: kernel context for untrustedd (pid= 879):
[  177.379716] untrustedd      S    0   879    745 0x08100000
[  177.385184] Stack : 9ca0bd0c 0000000d 1f78a400 00000200 0000000d 709c0000 9ca0bd78 7016b6fc
[  177.393515]         91c2abc8 0000000d 1f78a400 957ec0f0 9ca0be10 00000000 709c0000 9ca0be1c
[  177.401849]         9d0f88e0 ffffffff 00000002 7fba3e24 563522b4 7079a358 9ca0be10 00000000
[  177.410182]         9ca0be10 00000000 709c0000 700a4c44 00000000 716ea7b8 709c0000 9d0f88e0
[  177.418516]         00000002 000409e7 00000000 ffffffea 00000000 000409e7 00000000 00000000
[  177.426849]         ...
[  177.429337] Classified pointer on stack:
[  177.433189] 9ca0bd0c [stack(untrustedd): 9ca08000+0x3d0c]
[  177.438621] 9ca0bd78 [stack(untrustedd): 9ca08000+0x3d78]
[  177.443975] 7016b6fc current_time+0xa0/0xc8
[  177.448159] 91c2abc8 [slab: type:shmem_inode_cache size:424 start:0x91c2ab98+0x30 allocated by:0x700ff018 shmem_alloc_inode+0x1c/0x30]
[  177.460216] 957ec0f0 [slab: type:vm_area_struct size:120 start:0x957ec0f0+0x0 allocated by:0x7011dc38 mmap_region+0x1e8/0x53c]
[  177.471572] 9ca0be10 [stack(untrustedd): 9ca08000+0x3e10]
[  177.476963] 9ca0be1c [stack(untrustedd): 9ca08000+0x3e1c]
[  177.482337] 9d0f88e0 [page: type:alloc]
[  177.486208] 7079a358 schedule+0x78/0x9c
[  177.489976] 9ca0be10 [stack(untrustedd): 9ca08000+0x3e10]
[  177.495366] 9ca0be10 [stack(untrustedd): 9ca08000+0x3e10]
[  177.500747] 700a4c44 futex_wait_queue_me+0x128/0x174
[  177.505696] 716ea7b8 [page: type:alloc]
[  177.509507] 9d0f88e0 [page: type:alloc]
[  177.513397] 700a5cc4 futex_wait+0x108/0x280
[  177.517539] 91c67a40 [slab: type:mm_struct size:480 start:0x91c67a40+0x0 allocated by:0x7002da1c mm_alloc+0x20/0x68]
[  177.527998] 9ca0bdd4 [stack(untrustedd): 9ca08000+0x3dd4]
[  177.533370] 7011deac mmap_region+0x45c/0x53c
[  177.537620] Call Trace:
[  177.540058] [<70799ea0>] 0x70799ea0 __schedule+0x430/0x870
[  177.545533] [<7079a358>] 0x7079a358 schedule+0x78/0x9c
[  177.550649] [<700a4c44>] 0x700a4c44 futex_wait_queue_me+0x128/0x174
[  177.556899] [<700a5cc4>] 0x700a5cc4 futex_wait+0x108/0x280
[  177.562370] [<700a86f0>] 0x700a86f0 do_futex+0xb8/0x190
[  177.567576] [<700a8924>] 0x700a8924 SyS_futex+0x15c/0x1e8
[  177.572971] [<70017880>] 0x70017880 syscall_common+0x34/0x54
[  177.578596] ar7wdt_hw_reboot: user context for untrustedd:
        at: 00000001 v0: 0000108e v1: 00000000
        a0: 7fba3e24 a1: 00000080 a2: 00000002 a3: 00000000
        t0: 773e3000 t1: 00000000 t2: 00000040 t3: 9cf545b4
        t4: 9cf545b0 t5: 00100000 t6: fffffffc t7: 00001072
        s0: 7fba3d04 s1: 7730f000 s2: 7730f000 s3: 7730d800
        s4: 00000001 s5: 56341e40 s6: 7fba3f70 s7: 000000b7
        t8: 00000000 t9: 7732ed70
        gp: 773f72a0 fp: 00000000
        sp:  7fba3d28 (start_stack 7fba40f0)
        epc: 7732edb4
        ra:  7739c998
[  177.622259] Mem-Info:
[  177.624535] active_anon:4307 inactive_anon:29680 isolated_anon:0
[  177.624535]  active_file:0 inactive_file:0 isolated_file:0
[  177.624535]  unevictable:0 dirty:0 writeback:0 unstable:0
[  177.624535]  slab_reclaimable:3326 slab_unreclaimable:0
[  177.624535]  mapped:398 shmem:33909 pagetables:15 bounce:0
[  177.624535]  free:139646 free_pcp:410 free_cma:4277
[  177.656909] Node 0 active_anon:17228kB inactive_anon:118720kB active_file:0kB inactive_file:0kB unevictable:0kB isolated(anon):0kB isolated(file):0kB mapped:1592kB dirty:0kB writeback:0kB shmem:135636kB writeback_tmp:0kB unstable:0kB pages_scanned:0 all_unreclaimable? no
[  177.680874] DMA free:77096kB min:296kB low:368kB high:440kB active_anon:0kB inactive_anon:0kB active_file:0kB inactive_file:0kB unevictable:0kB writepending:0kB present:262144kB managed:260992kB mlocked:0kB slab_reclaimable:4kB slab_unreclaimable:0kB kernel_stack:0kB pagetables:0kB bounce:0kB free_pcp:0kB local_pcp:0kB free_cma:17108kB
[  177.710539] lowmem_reserve[]: 0 587 587
[  177.714380] Normal free:481488kB min:2952kB low:3688kB high:4424kB active_anon:17228kB inactive_anon:118720kB active_file:0kB inactive_file:0kB unevictable:0kB writepending:0kB present:741376kB managed:640600kB mlocked:0kB slab_reclaimable:13300kB slab_unreclaimable:0kB kernel_stack:1344kB pagetables:60kB bounce:0kB free_pcp:1640kB local_pcp:680kB free_cma:0kB
[  177.746217] lowmem_reserve[]: 0 0 0
[  177.749690] DMA: 2*4kB (EC) 4*8kB (UEC) 4*16kB (UEC) 2*32kB (EC) 4*64kB (UEC) 5*128kB (UMEC) 5*256kB (MEC) 4*512kB (UMEC) 3*1024kB (UEC) 4*2048kB (UMEC) 15*4096kB (MC) = 77096kB
[  177.765489] Normal: 48*4kB (UME) 10*8kB (UME) 4*16kB (UME) 6*32kB (UME) 1*64kB (M) 1*128kB (M) 4*256kB (UE) 3*512kB (UME) 1*1024kB (M) 3*2048kB (UME) 115*4096kB (M) = 481488kB
[  177.781115] 33909 total pagecache pages
[... 223 lines cut ...]
[  186.719126] [<703ffff4>] 0x703ffff4 panic_function+0xe8/0xf8
[... cut ...]
[  186.765734] [<7000a050>] 0x7000a050 r4k_wait_irqoff+0x34/0x40
[  186.771471] [<70074a80>] 0x70074a80 cpu_startup_entry+0x130/0x170
[  186.777542] [<70798da0>] 0x70798da0 smp_boot▒
ROM VER: 2.1.0
CFG 0c
B
DdrOk
Building bbt 1024 Blocks
Bad block 0xBA detected @ 0x1740000
Bad block 0x300 detected @ 0x6000000

(AVM) EVA Revision: 1.11334 fc5825de

```

Sieht für mich unfreetzbar aus - solang man die 2. cpu nutzen möchte. Fraglich welche Funktionen die übernimmt,
Oder mach packt eine Alternative wie Openwrt drauf...

closed 2022-07-07 by fda77