Freetz-NG/freetz-ng #560: 5590 is (now) broken - GBIC wird nicht mehr erkannt / Providersettings-Menu Broken

Himan2001, 2022-06-03:
Ich habe nach https://github.com/Freetz-NG/freetz-ng/discussions/534

jetzt auch mal ein image erstellt aus dem aktuellsten trunk.

Leider gibt es nach wie vor kein Recovery für die Box und es war ein riesen Zirkus wieder auf die Stock-Firmware zurückzukommen.

Problem nach freezen: Kontinuierliche reboots, dann ueber verschiedene Versuche Optionen geändert:

trustedd deaktiviert. remove leftovers on/off, im Ergebnis lief da zwar die Box ohne sporadische reboots, jedoch
wird das GBIC nach dem freezen NICHT mehr erkannt.
Zusätzlich ist das Menu mit der Providerauswahl für FTTH auch verburkt und man hat keine ISP-config-Optionen mehr.

-> Anpassung beim menuconfig: Es gibt auch ein branding fuer 1und1 neben avm und avme!

Hier ist die letzte config mit der ich getestet habe:

[.config.txt](https://github.com/Freetz-NG/freetz-ng/files/8829946/default.config.txt)

Flashen des Original-Images ueber push_firmware ist sofort abgebrochen, eva2memory ein in-memory-file lies sich aus dem
Stock-binary wg. anderem format auch nicht erstellen und so habe ich nochmal ein Image erstellt wo die Originalkeys wieder drin sind und konnte dann glücklicherweise über das AVM Webinterface wieder aufs Original zurück.

Auch ein Versuch über das Freetz Webinterface das Original-Image zu flashen schlug fehl.
Eigentlich hätte dort ja das Script mit Status ablaufen sollen (wurde ja disuktiert). Bei mir blieb alles leer und es hat sich nichts geändert. Auch von Bank 0/1 war nirgend wo etwas zu sehen.

Folgendes Enviroment hat meine Box:

```
Found AVM bootloader: AVM EVA Version 1.11822 0x0 0xC6409
Environment read from device:
DMC                   SL1
HardwareFeatures      cpu=389.2.0,emmc=5.0.17.0.2
HWRevision            272
HWSubRevision         2
ProductID             Fritz_Box_HW272
SerialNumber          P15381530xxxxxx
annex                 Ohne,pon
autoload              yes
bootloaderVersion     1.11822
country               049
firstfreeaddress      0x50500000
firmware_info         272.07.29,slot1=07.29-95450,slot0=07.29-95450
firmware_version      avme
flashsize             nor_size=0MB sflash_size=0KB nand_size=0MB emmc_size=1888MB
gpon_serial           AVMG72xxxxxx
language              de
linux_fs_start        0
memsize               0x40000000
mtd0                  0x0,0x0
mtd1                  0x3000000,0x8000000
mtd2                  0x0,0x2000000
mtd3                  0x2000000,0x3000000
mtd4                  0x8000000,0xD000000
mtd5                  0xD000000,0x75BFC000
my_ipaddress          192.168.178.1
prompt                Eva_AVM
ptest
usb_device_id         0x0000
usb_device_name       USB DSL Device
usb_manufacturer_name  AVM

```

Interessant ist auch Folgendes:

Die Box rennt unter avme, nicht avm. Obwohl das eine "freie" Box ist kann man das enviroment hier auch nicht dauerhaft
von avme auf avm aendern, nach einem reboot steht wieder "avme" drin. Das Image selber für freetz habe ich beide Varianten durch,
also image für avm oder avme. Auch da wurde das Gbic nicht mehr erkannt.

Ich habe, als freetz noch drauf war, per ssh mir das dmesg angesehen und die processliste. Nach dem Boot wurde an keiner Stelle mehr das Gbic initialisiert und im Webinterface stand nur die Info "unknown" beim GBIC.

Die Box kam von Notebooksbilliger.de, da das die Einzigsten waren, die ueberhaupt vor 3 Tagen noch sofort liefern konnten.
Auch auf der Verpackung steht nichts von einer intl-Version drauf.

Wie gehts jetzt weiter ?

- binutils downgraden auf AVM-Version ?
- busybox downgraden auf AVM-Version ?

sinnvoll loggen kann ich nur wenn freetz und ssh drauf sind.

Ich hab jetzt wieder die Stock-Firmware drauf, dort Werksreset gemacht und jetzt wird auch das Gbic wieder erkannt.

Ob das vor 3-4 Wochen im trunk noch lief kann ich nicht sagen, da ich die Box zu der Zeit noch nicht vorliegen hatte.

(ich könnte höchstens die debian10 mit dem commitmentstand vor der Änderung der neuen toolchain hier auch noch mal antesten, obwohl ich die chain sowieso selber baue)

fda77, 2022-06-03:
Wenn man eine neues Gerät verwenden IMMER ALLE remove patches nicht auswählen!
Was heisst "GBIC wird nicht mehr erkannt"? Wann wurde es noch erkannt? Mit älterem FOS?
>  Anpassung beim menuconfig: Es gibt auch ein branding fuer 1und1 neben avm und avme!

Was heisst das? Die werden eigentlich automatisch erkannt

Und hör mal auf mit deinen flash-theorien, da bickt doch niemand durch... Wenn ein Image auf der Box ist, ist es so. Egal wie es dahin kam. Wenn es nicht läuft schlaten neuere Bootloader (wohl) auf das andere um

Himan2001, 2022-06-03:
Also ich habe die remove patches alles wieder rausgenommen genommen gehabt - selbstverständlich, die Diskussion auch komplett nochmal nachgelesen und nachgebaut wo der Kollege die screenshots gemacht hatte....,

GBIC nicht erkannt heisst:

Der uebliche Kernel-init und die entsprechenden daemons werden gar nicht mehr geladen. Ich habe etwa 20 verschiedene images gebaut und alle möglichen optionen reihenweise deaktiviert bis zur minimalsten Änderung. Beim GBIC Menu kam dann nur noch die Meldung "unknown" - und das menu für Providersetup zeigte nur noch "andere Provider", wenn man dort dann Telekom auswählte kamen auf einmal 1&1 varianten, natürlich haben werksresets nach dem freetzen auch keine änderungen gebracht. also die üblichen Varianten "trustedd" ON/OFF und dieselben spielchen wie auch bei der 5530 noch durchgspielt. Im Ergebnis habe ich ueber einige Umwege dann die Originalfirmware wieder draufbekommen und schon wurde das gbic wieder erkannt. Und das komplette Internet-Zugangsmenü war wieder da.

Ich hab auch genau die config genommen aus der diskussion mit selben settings zum testen.....

Ich habs deshalb als ticket aufgemacht, weil die diskussion sich auf einen trunk bezogen hatte wo der Kollege offenbar erfolg hatte - hatte er denn auch ein gbic drin was erkannt wurde, das fehlte in der Diskussion ?

Freetz drauf geht ja, nur wird eben kein gbic mehr gefunden - und das muss ja irgend einen Grund haben....

soll ich die Box auch aufschrauben und ne serielle dran pappen ? anders sehen wir beim boot nicht wieso da treiber nicht geladen werden oder lieber auf nen alten trunk zurück ?

Mehr verwundert war ich ueber AVME statt AVM - was sich nicht fix ändern lies, das image selber wurde aber für AVM gebaut.
Natürlich hab ich das dann für AVME auch zum testen nochmal gebaut, aber auch das brachte keine Unterschiede.

natürlich mit FOS 7.29 - da gibts bis heute kein recovery oder eine ältere Version, nix. Deshalb war ich ja heilfroh das ich uebers AVM webinterface wieder das Originalimage überhaupt reinbekam. push_firmware hat das originalimage gleich mal rejected und eva_to_memory mag keine Stock-Images und Freetz flashing ging ja ebenfalls nicht, da brach schon leichte Panik aus (ohne recovery.exe) :-)

Im folder /config/.kos gibt es für die 5590 auch kein config-file, nur für die 5530 oder hat das .kos damit nichts zu tun ?

Habe nochwas gefunden: laut boxmatrix kannst du dem push_firmware bzw. dem freetz noch eine ID hinzufügen: Es gibt die 5590 noch mit der ID 283 (oem/branded)

Hier die Refferenz dazu: https://boxmatrix.info/wiki/FRITZ!Box_5590_Fiber

Im Header des original 07.29-AVM Images stehen uebrigens alle 3 brandings drin - die firmware ist für alle Modelle vorgesehen offenbar.

Wär aber trotzdem interessant zu wissen wieso ich AVME drin habe statt AVM, evtl. habe ich vom dealer eine intl. Box bekommen. ? - Hab ja niemandem zum Vergleichen :-(

Habe gerade nochmal ein image gebaut und dieses mal auch die host-tools selber nochmal gebaut, da hatte ich auch was gelesen das ich neue brauche an irgend einer anderen Stelle.

Ich würde jetzt einfach mal auf die Busybox-Version oder host-tools oder die uclibc++ als Nächstes tippen ?

fda77, 2022-06-03:
> Habe nochwas gefunden: laut boxmatrix kannst du dem push_firmware bzw. dem freetz noch eine ID hinzufügen: Es gibt die 5590 noch mit der ID 283 (oem/branded)

Könntest es sein dass du eine Firmware die nicht für deine Hardware geeignet ist geflash hast??? Und deshalb das Modul nicht lädt? Das ist dem bootloader (relativ) egal beim flashen
Akzeptiert das avm-webif hw272 und 273 firmware? Normal sind das "andere" Geräte.
Besorg doch mal so ein 283 image

 > Ich würde jetzt einfach mal auf die Busybox-Version oder host-tools oder die uclibc++ als Nächstes tippen ?

Nur zu. Oder lass die uclibc++ ganz weg, die wird nur ausgewählt wenn eines DEINER paket diese braucht

Himan2001, 2022-06-03:
das war mein erster gedanke. siehe meinen quote vom enviroment. Ich habe defintiv die HW272 und die ist auch im push_firmware drin. original AVM Boxen dürften alle HW272 haben. denke dann 1und1 und OEM haben dann HW273.
Das kann ich ausschliessen.

Mit nem binary viewer findest du in der 7.29er AVM.firmware (die, die auch freetz beim building runter läd) die info: HW272 und als OEMS avm avme und 1und1, diese Version ist also richtig gewesen.

> 
> Nur zu. Oder lass die uclibc++ ganz weg, die wird nur ausgewählt wenn eines DEINER paket diese braucht

Ich bereite fürs Wochenende gerade absolut minimale Sachen vor. ich habe jetzt erstmalig auch die self-compiled host-tools komplett selber mit ins build genommen und dann schraube ich die Busybox mal ne Stufe runter, auch da hab ich einige Optionen schon rausgenommen das ich so wenig Unterschiede wie möglich habe zum Original.... Ich hasse try und error wenn man keine exakten Logs bekommt und man nicht direkt am Problem arbeiten kann ;-)

KEIN Labor/Inhaus für die 5530 und 5590 machens aber auch nicht einfacher, im Grunde sind beide Firmwares auch ohne Freetz nicht ganz ausgereift für FTTH, alles noch early Bird und schon ausverkauft.... :-)

ABER immerhin: LTE mit nem Stick rennt auch auf der 5590, altbacken ohne IPV6, ohne NCM, so wie die letzten 20 Jahre schon :-)

fda77, 2022-06-03:
Ich versteh nicht was du immer mit docker/precompiled/toolchain/tools/binutils/busybox hast

Himan2001, 2022-06-03:
....weil mich betreffs der host-tools eine diskussion darauf gebracht hat, das ich evtl. mal die "aktuellen" host-tools auch selber bauen sollte, weil da irgendetwas aktueller ist als bei den precompiled.

In wieder einer anderen Diskussion betreffs WLAN Probleme war eine neuere Busybox Version dann ein Problem.

Wenn ich was zu 753x lese kann ich mit Annehmen, sowas dann auch die die 553x er ebenfalls in Betracht ziehen. Wäre mal einfache Logik, da alles auf 7.29 basiert.... Gewisse Übereinstimmungen sind schon da wenn ich an meine 7520er denke von vor ueber nem Jahr ;-)

fda77, 2022-06-03:
> ....weil mich betreffs der host-tools eine diskussion darauf gebracht hat, das ich evtl. mal die "aktuellen" host-tools auch selber bauen sollte, weil da irgendetwas aktueller ist als bei den precompiled. - bzw. genau anders rum, es kann sein das die älteren host-tools benötigt werden die noch aus der 14.04 Zeit sind. (Ausprobieren....)

Es könnte auch sein dass das egal ist und nichts hiermit zu tun hat

fda77, 2022-06-03:
> In wieder einer anderen Diskussion betreffs WLAN Probleme war eine neuere Busybox Version dann ein Problem.

Lies nicht nur die Überschrift, sondern schau dir an woran es lag

> Wenn ich was zu 753x lese kann ich mit Annehmen, sowas dann auch die die 553x er ebenfalls in Betracht ziehen.

Das bringt doch alles nichts

fda77, 2022-06-03:
So, Wochenende. Ciao!

Himan2001, 2022-06-03:
Genau ^^ Denkpause und dann frisch ans Werk und alles mal einwirken lassen :-)

fda77, 2022-06-03:
Damit ich die Dateien nicht verliere, mit ein paar Leerzeilen, zu 99% gleich:
[boot_original_mit.gbic0729.txt](https://github.com/Freetz-NG/freetz-ng/files/8835025/boot_original_mit.gbic0729.txt)
[freetzboot_with_untrusted.txt](https://github.com/Freetz-NG/freetz-ng/files/8835026/freetzboot_with_untrusted.txt)
[freetzboot_without_untrusted.txt](https://github.com/Freetz-NG/freetz-ng/files/8835027/freetzboot_without_untrusted.txt)

Himan2001, 2022-06-03:
...und wie kann sowas entstehen, da wird doch im Grunde erst mal nur der Kernel gebootet und der ist doch original AVM ?

da muss ich doch glatt bei der 6890v2 auch noch mal genauer hinsehen, die rennt ja mit "WLAN on" auch noch nicht ganz rund, ausser ich nehm das Labor. War aber auch eine meiner Fragen, wieso Freetz 3 CPUs anzeigt und ich dann fragte: Wird da ab "0" gezählt das dort 3 CPU angezeigt wird ^^

fda77, 2022-06-04:
Falsches issue

fda77, 2022-06-04:
Ich weiss gar nicht so genau was hier nicht gehen soll. Sowei ich mitbekommen hab
 - flashen ist nicht grün
 - ftth ist verbunker
 
?!?
Das hat bei anderen schon funktioniert

Himan2001, 2022-06-04:
Moin. Naja... das eben das GBIC nicht mehr erkannt wurde nach dem freetzen und das Providerauswahlmenu verborkt war.

Das ich dann im Anschluss etwa 20 Images mit verschiedenen Optionen noch gebaut habe um das Problem einzugrenzen und nicht drauf gekommen bin wieso das gbic nicht mehr erkannt wird. Und letztlich eine Rückflash-Odyssey hatte um wieder auf die Stockfirmware zu kommen (mangels recovery.exe)

Ich gehe ab morgen nochmal frisch an die 5590 - mit einem image was nur minimalste Änderungen drin hat. Ich hab mich ja an dem orientiert was diskutiert wurde und an den Screenshots welche Optionen der Kollege verwendet hatte.

Letztlich habe ich die Frage gestellt, ob der Kollege denn auch aktives FTTH und gbic ebenso drin hatte, wie ich......

...ist ja Eines ein Image zu erstellen womit eine Box startet - und was Anderes ob dann auch alle Komponenten noch erkannt werden ;-)

Sollte ich erneut nach dem Freetzen kein gbic vorfinden, werd ich wohl oder übel auch diese Box aufmachen müssen und seriell loggen um herauszufinden, wieso der WAN-Teil nicht mehr erkannt wird......

fda77, 2022-06-04:
Danke, habs jetzt richtig kapiert. Vielleicht fehlen in dem Menü nur deshalb die Einträge, weil kein Modul erkannt wurde

fda77, 2022-06-04:
"Zurückflashen" geht genau so mit push firmware wie mit jedem anderen auch

Himan2001, 2022-06-04:
Also mit push_firmware und der Original 7.29 gabs sofort ne Fehlermeldung - Abbruch beim Flashen. Wird wohl von der Box nicht akzeptiert worden sein. Ging wie gesagt erst nachdem ich ein image wieder erstellt hatte wo die original Zertifkate wieder drin waren von AVM und dann ueber das Webinterface der Box selber zurück..

Ich habe nebenbei etwas gefunden:

```
freetz@freetzdevel:~/freetz-ng-master/build/modified/filesystem/lib/systemd/system$ cat fiber.service
[Service]
Before=voip.service net.service
ExecStart=/etc/init.d/E40-fiber
ExecStop=/etc/init.d/e40-fiber <----------------  das File gibts nicht
RemainAfterExit=yes
Type=oneshot

[Install]
WantedBy=multi-user.target

```

denke zwar nicht, das es das Problem ist, aber der Stop-befehl kann mal nicht ausgeführt werden, da im /etc/init.d/ ordner das E40 nur als E40... und nicht als e40... vorhanden ist....

Ist nur das da: -rwxrwx--- 1 freetz freetz   318 Mar 25 10:32 E40-fiber

Sehe gerade den Fehler hat AVM im Original ebenso drin ^^

fda77, 2022-06-04:
> Also mit push_firmware und der Original 7.29 gabs sofort ne Fehlermeldung - Abbruch beim Flashen.

Glaube ich nicht
1) ist das wurscht
2) gibts keine entsprechenden Ausgaben

>  Wird wohl von der Box nicht akzeptiert worden sein. Ging wie gesagt erst nachdem ich ein image wieder erstellt hatte wo...

Dreh den Stromstecker doch mal um 180°, vielleicht hast du schlechten Strom erwischt. Bei NOR Chips wäre es egal, aber die NAND sind wählerisch. Ich hoffe AVM steigt bald auf NDIV um

>  ein image wieder erstellt hatte wo die original Zertifkate wieder drin waren von AV

Wie war das mit dem KEINE REMOVE PATCHES AUSWÄHLEN?
Also wenn das bei andere geht, und bei dir nicht rennt könnte es an deiner Auswahl im menuconfig etc liegen

Himan2001, 2022-06-04:
hier ist die config die zum selben ergebnis führt....
(gbic wird nicht mehr erkannt / Providersettings verborkt)

[.config.txt](https://github.com/Freetz-NG/freetz-ng/files/8838998/default.config.txt)

minimaler gehts kaum noch zum testen....

Himan2001, 2022-06-05:
So, da es ja fortlaufend immer nur heisst "worksforme/close", hier jetzt mal ein weiterer Test:

Mal für ganz Blöde, obwohl ich 15 Jahre niemals Probleme mit freetzen hatte und Images bauen bislang für zig Modelle....

selbe .config wie oben.

Dann nach aktuellsten info für Fedora 36 hier nach gegangen:

Fedora Server/netinstall/minimal image vom .iso auf dem Xen-Server erstellt.
config: 4 cores Intel, 16 GB Ram, 4GB swap... als Basis

Root login ueber ssh und dann wie beschrieben:

dnf -y update && sudo systemctl daemon-reload

gefolgt von:

dnf -y groupinstall 'Development Tools' 'Development Libraries'
dnf -y install pv cpio rsync kmod execstack sqlite.i686 sqlite-devel cmake zlib-devel.i686 libstdc++-devel.i686 openssl xz bc unar inkscape ImageMagick subversion ccache gcc gcc-c++ binutils autoconf automake libtool make bzip2 libglade2-devel qt5-qtbase-devel ncurses-devel ncurses-term zlib-devel flex bison patch texinfo gettext pkgconfig ecj perl perl-String-CRC32 wget glib2-devel git util-linux libacl-devel libattr-devel libcap-devel ncurses-devel.i686 glibc-devel.i686 libgcc.i686 libzstd-devel.x86_64 libstdc++-devel.x86_64 netcat curl

zusätzlich die fehlenden dependencies für "ftp" und "ncftp" noch manuell erledigt (für push_firmware fehlt das nämlich....).

user "freetz" erstellt und ist via "wheel" der sudo group auch hinzugefügt.

depmod -a
ldconfig

zur sicherheit am Ende nochmal hinterher. Überflüssiger Weise sogar noch mal einen kompletten reboot.

Login ueber ssh/root; ins homedir vom freetz -> su freetz

und den aktuellen trunk ausgecheckt. die config eine Nachricht höher wieder verwendet. (Minimalste Änderungen, alles auf default, keine remove Patches ausser dtrace)

Precompiled host-tools und crosscompiler, wie auch schon bei ubuntu > 18.x nicht verwendet da die bereits dokumentierten
compile errores auch hier 1:1 auftraten wie auch bei ubuntu 18+20 - somit beides selber erstellt (die .config zeigt das eh....)

Ist durchgelaufen und dasselbe Ergebnis:

Gbic wird nicht mehr gefunden + Provider-Menu verborkt. Selbes Ergebnis also wie bei ubuntu 18.04.06.

Und jetzt erkläre mir doch einfach wieso es seit dem Wechsel ab Trunk: "08f711812f14dbef732576979e7524b613f4f20c"
auf allen Geräten, egal ob mit ubuntu 18, 20 & und dann die die von Dir verwendete Fedora, die ich jetzt auch abgearbeitet habe, es immer wieder auf allen möglichen Geräten zu Bröseln kommt.... ?

So jetzt wirds mit Sicherheit so laufen, das WORKSFORME bleibt, wie üblich abgehackt ist, aka Du hälst mich weiterhin für Blöd,
und ich warte (wie damals auch schon einfach mal 2-3 Monate ab, bis das eigentliche Problem im "Enviroment" dann auch für diese Art des normalen Erstellens auch gefunden wird.

Ein typisches Beispiel, worauf ich auch schon mit Vermutungen öfters hingewiesen hatte, hatten wir ja gerade erst:

https://github.com/Freetz-NG/freetz-ng/commit/79673245fe3b285897caa8776f34de4c2e77481c

aber das hattest Du immer abgetan und mir unterstellt. ich mache wohl "alles falsch"... bei mir sei alles nicht korrekt,
und ich könnte nicht mal ein Image "korrekt" Flashen" (....)

Abgesehen von einigen push_firmware adds für 6890v2, usw, was etwas später noch kam, klappt es nach wie vor mit der debian10 die ich seit Jahren verwende bis zum dem Punkt wo die alte ubuntu 14.04 crosstools oder auch hier noch die alte AVM-Methode mit dem Kernel, usw. verwendet wurde erneut auch hier wieder problemlos, und Überaschung: Eine weitere Box bei der auf einmal das gbic wieder gefunden wurde, oder es die von mir geschilderten probleme auf magische Weise jetzt wieder mal nicht gibt...

aber ich bin ja zu dumm ;-)

Es hat auch bei Freshtomato bis heute seinen Grund wieso das Basis-System hier immer noch debian10 ist - weil das Ursprungssystem selber einfach älter ist und neuere Versionen hier immer wieder zu Problemen geführt hatten, von daher
kenne ich genau solche Phenomene wenn das seit Jahren laufende System jetzt so "hart" geändert wurde, weil (verständlicherweise) die Pflege eines ubuntu 14.04 Sets mit sehr viel Aufwand verbunden ist.

Und ich wette mir Dir. Da wird aus dem Forum mit Sicherheit noch ein weiteres Finding Beizeiten noch kommen, und auf einmal wird es für normale "Shell-User "auf magische Weise dann auch wieder ganz normal laufen alles ;-)

Ich habe mir gestern auch die Mühe gemacht, von Hand die beiden unpacked Sachen original und modified aus dem build Dir nochmals zu vergleichen, ob da wirklich nicht etwas vergessen wurde, fehlende firmware, bin-files, hex-files.... Bis auf den von AVM eingebauten Fehler mit "e41 statt eE40" beim Stop, was wohl zweitrangig ist, ist da auch nichts erkennbar gewesen was im
freetz-build vergessen wurde aus dem Original-Image mit zu übernehmen. Einzig die extra "inserts" für den Freetz-Mode
waren an den bekannten stellen im Startup mehr mit drin.

Dummerweise habe ich nach dem Wechsel von freetz/not-working -> avm-update -> original-Image -> nächster Test,
noch das Problem, das das Freetz selber mit der kompletten config auch nach einem Werksreset in einer Original erhaltenen
freetz-config von vorherigen freetz-Test weiterhin im Flash verblieben ist, mangels recovery.exe bislang war auch keine 100% ge
"Restauration" des TFFS machbar um wirklich nochmal komplett clean weitere Versuche zu machen. Was es evtl. auch nicht einfacher macht: Werksreset über Webinterface nach Rückspielen der Stock-Firmware cleant nicht die Flashbereiche die
[... cut ...]
**Also: WORKSFORME: debian10 bis zum genannten commit, und alle 3 Boxen 5530, 5590 und 6890v2 LAUFEN genauso wie
die 7590 und 7490 ohne die neuen Brösel und alles ist wieder wie es all die Jahre zuvor auch immer gelaufen ist.**

Das push_firmware hingegen nehm ich aus dem aktuellsten trunk, da dort die neuen HW-IDs mit drin sind.

Für mich ist der Fall erledigt, in einigen Monaten wirds sicher auch mit der neuen Methode dann endlich wieder laufen, mein Feedback hierzu in Endlosschleife ist ja leider nie durchgekommen so wirklich.....

Himan2001, 2022-06-05:
Ich habe jetzt nochmal den git-Stand aus der Diskussion verwendet:

https://github.com/Freetz-NG/freetz-ng/discussions/534

auf der debian10 ein image für die 5590 gebaut. Witzigerweise auch hier erneut das Ergebnis das
kein Gbic vorhanden ist und die ISP-Config verborkt ist.

der bbpd daemon ist geladen, allerdings sehe ich im logfile dann sowas:

```
root@fritz:/var/log$ cat broadband_ui.log
[BROADBAND_UI MAIN][ERR]Error while polling for avm_event_id_dsl_get after registration.
[BROADBAND_UI BBCFG_MAIN][INFO]broadband_ui_bbcfg_main__convert_ar7cfg: Converting fiber section succeeded with 0.
[BROADBAND_UI BBCFG_MAIN][INFO]broadband_ui_bbcfg_main__clear_ar7cfg: Clearing fiber section succeeded with 0.
[BROADBAND_UI MAIN][INFO]convert_cfg: Converting to broadbandcfg and clearing ar7cfg succeded.
[BROADBAND_UI BBCFG_MAIN][INFO]broadband_ui_bbcfg_main__update_bbcfg: Fresh bbcfg - no need to convert
[BROADBAND_UI BBCFG_MAIN][INFO]broadband_ui_bbcfg_main__update_bbcfg: Updating bbcfg version from 0 to 3.
[BROADBAND_UI DSL][ERR]get_dslparams failed at 21074 msec.
[BROADBAND_UI DSL][ERR]get_dslparams failed at 30048 msec.
[BROADBAND_UI DSL][ERR]get_dslparams failed at 31603 msec.
[BROADBAND_UI DSL][ERR]get_dslparams failed at 33106 msec.
[BROADBAND_UI DSL][ERR]get_dslparams failed at 75366 msec.
[BROADBAND_UI DSL][ERR]get_dslparams failed at 80668 msec.
[BROADBAND_UI DSL][ERR]get_dslparams failed at 88736 msec.
[BROADBAND_UI DSL][ERR]get_dslparams failed at 90270 msec.
[BROADBAND_UI DSL][ERR]get_dslparams failed at 103438 msec.
[BROADBAND_UI DSL][ERR]get_dslparams failed at 104987 msec.
[BROADBAND_UI DSL][ERR]get_dslparams failed at 106510 msec.
[BROADBAND_UI DSL][ERR]get_dslparams failed at 108027 msec.
[BROADBAND_UI DSL][ERR]get_dslparams failed at 109728 msec.
[BROADBAND_UI DSL][ERR]get_dslparams failed at 311354 msec.
[BROADBAND_UI DSL][ERR]get_dslparams failed at 315674 msec.
[BROADBAND_UI DSL][ERR]get_dslparams failed at 317192 msec.
[BROADBAND_UI DSL][ERR]get_dslparams failed at 318700 msec.
[BROADBAND_UI DSL][ERR]get_dslparams failed at 320210 msec.
[BROADBAND_UI DSL][ERR]get_dslparams failed at 321834 msec.
[BROADBAND_UI DSL][ERR]get_dslparams failed at 380793 msec.

```

zusätzlich habe ich mit "ldd" sicherheitshalber bei einigen binarys nochmal geprüft ob da mit den libs auf der Box selber
auch alles passt.

Die serielle hab ich noch nicht dran. Ich hatte allerdings ständig TFFS/error -4 - weil ja immer noch kein recovery.exe
von AVM da war womit ich alte freetz sachen noch restlos entfernen kann. Habe die config/TFFS aber mit "bin/setfactorydefaluts" dann zumindest mal wieder sauber bekommen.

Damit die Box nicht in Endlosschleife rebootet habe ich den Watchdog und untrusted erst mal temporär dazu gepackt um in der shell der Box ein wenig länger als nur 40 sekunden spielen zu können.

Also wenn ich einige Logausgaben in verschiedenen Files richtig interpretiert habe, wird der gbic kram auch ueber den kdsld mitgemacht - wo ja im freetz der dsl teil für die 5590 draussen ist ?

so sieht jetzt die processliste aus:

```
root@fritz:/var/mod/root# ps w
  PID USER       VSZ STAT COMMAND
    1 root      1656 S    init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    4 root         0 SW   [kworker/0:0]
    5 root         0 SW<  [kworker/0:0H]
    6 root         0 SW   [kworker/u8:0]
    7 root         0 SW   [rcu_preempt]
    8 root         0 SW   [rcu_sched]
    9 root         0 SW   [rcu_bh]
   10 root         0 SW   [migration/0]
   11 root         0 SW   [watchdog/0]
   12 root         0 SW   [watchdog/1]
   13 root         0 SW   [migration/1]
   14 root         0 SW   [ksoftirqd/1]
   15 root         0 SW   [kworker/1:0]
[... 170 lines cut ...]
        libz.so.1 => /usr/lib/libz.so.1 (0x00000000)
        libewnwnet.so.0 => /lib/libewnwnet.so.0 (0x00000000)
        libwebsrv.so.2 => /lib/libwebsrv.so.2 (0x00000000)
        libavmauth.so.1 => /lib/libavmauth.so.1 (0x00000000)
        libavmhmac.so.2 => /lib/libavmhmac.so.2 (0x00000000)
        libewnwlinux.so.2 => /lib/libewnwlinux.so.2 (0x00000000)
        libdputil.so.2 => /lib/libdputil.so.2 (0x00000000)
        libwdt.so.1 => /lib/libwdt.so.1 (0x00000000)
        libminneapolis.so => /usr/lib/libminneapolis.so (0x00000000)
        not a dynamic executable

```

Haben wir da jetzt evtl. etwas zuviel rausgenommen, vonwegen "box hat ja kein dsl mehr" - damit meine ich nicht die Webinterface-Patches aus der Diskussion.... ?

mehr sehe ich erst wenn ich die stock-firmware wieder drauf habe und die serielle dann dran ist um wieder beide bootlogs miteinander vergleichen zu können. die ar7.cfg wird - passend zur verborkten ISP-config - dann auch auf "dsl_other"
gesetzt, sofort nach einem reset - wohl als resultat das kein gbic oder zumindest (k)ein leerer gbic port erkannt wird.....

Zum nachvollziehen:
git-revision nach der Diskussion: e1389b793517fd2b75ce62d4932419278e9fb870

und die test-config dazu:
[.config.txt](https://github.com/Freetz-NG/freetz-ng/files/8840682/default.config.txt)

das ergebnis mit dem aktuellen trunk ist jetzt in diesem fall identisch was das Verhalten angeht...

Himan2001, 2022-06-06:
SO, hier sind mal 2 Logfiles:

[5590_log_stock0729.txt](https://github.com/Freetz-NG/freetz-ng/files/8845038/5590_log_stock0729.txt)
[5590_log_freetz0729.txt](https://github.com/Freetz-NG/freetz-ng/files/8845040/5590_log_freetz0729.txt)

Was sofort auffällt ist, ist das beim freetz diese ganzen upx-logging sachen niemals kommen, die bei der Stock-Firmware kontinuierlich ueber die serielle console laufen:

```
luacgi[2042]: <<< 220 bb_shm_manager ready.
luacgi[2042]: >>> get_overview_info
luacgi[2042]: <<< 225 sending
luacgi[2042]: <<< 220 bb_shm_manager ready.
luacgi[2042]: >>> get_overview_info
luacgi[2042]: <<< 225 sending
luacgi[2049]: <<< 220 bb_shm_manager ready.
luacgi[2049]: >>> get_overview_info
luacgi[2049]: <<< 225 sending
luacgi[2049]: <<< 220 bb_shm_manager ready.
luacgi[2049]: >>> get_overview_info
luacgi[2049]: <<< 225 sending
luacgi[2053]: <<< 220 bb_shm_manager ready.
luacgi[2053]: >>> get_overview_info
luacgi[2053]: <<< 225 sending
luacgi[2053]: <<< 220 bb_shm_manager ready.
luacgi[2053]: >>> get_overview_info
luacgi[2053]: <<< 225 sending
luacgi[2053]: <<< 220 bb_shm_manager ready.
luacgi[2053]: >>> get_fiber_optic

```

Beim Original-Image wird zumindest das erfolgreich geladen:

```
[Supervisor] Info: start upx-hwk-kpanic-recvd.service
[Supervisor] Info: start upx-hwk-time-sendd.service
[Supervisor] Info: start upx-hwk-ucrash-recvd.service
[Supervisor] Info: start upx-hwk-ulog-recvd.service
[Supervisor] Info: capitcp.service exit success
[Supervisor] Info: start upx-hwk-mgmtd.service
[Supervisor] Info: stop upx-hwk-serlogd.service

```

Ich schätze mal, das da: "Eva_AVM >{arch_fit_signature_fail} not implemented"

ist wohl normal nach dem flashen eines freetz images ?

Deutlich ist nach dem freetzen hier der Unterschied:

```
[Supervisor] Info: timeout triggered for upx-hwk-ucrash-recvd.service
[Supervisor] Info: start upx-hwk-ucrash-recvd.service
[Supervisor] Info: upx-hwk-ucrash-recvd.service exit error
[Supervisor] Info: timeout triggered for upx-hwk-kpanic-recvd.service
[Supervisor] Info: start upx-hwk-kpanic-recvd.service
[Supervisor] Info: upx-hwk-kpanic-recvd.service exit error
[Supervisor] Info: timeout triggered for upx-hwk-ulog-recvd.service
[Supervisor] Info: start upx-hwk-ulog-recvd.service
[Supervisor] Info: upx-hwk-ulog-recvd.service exit error
[Supervisor] Info: timeout triggered for upx-hwk-time-sendd.service
[Supervisor] Info: start upx-hwk-time-sendd.service
tcpsvd: applet not found
[Supervisor] Info: upx-hwk-time-sendd.service exit error
[Supervisor] Info: timeout triggered for upx-hwk-ucrash-recvd.service
[Supervisor] Info: start upx-hwk-ucrash-recvd.service
[Supervisor] Info: upx-hwk-ucrash-recvd.service exit error
[Supervisor] Info: timeout triggered for upx-hwk-kpanic-recvd.service
[Supervisor] Info: start upx-hwk-kpanic-recvd.service
[Supervisor] Info: upx-hwk-kpanic-recvd.service exit error
[Supervisor] Info: timeout triggered for upx-hwk-ulog-recvd.service

```

Kommen wir jetzt evtl. etwas weiter ?

Himan2001, 2022-06-06:
Zwischenbereicht:

Mit der neuen Developer-Busybox Option aus dem Trunk: Keinerlei änderung: "Unbekanntes Gbic" (gleiche test-config wie oben

das upx.... luacgi zeuchs, was beim original-image,  auf der seriellen console endlos durchläuft startet bei der gefreetzen version nach wie vor nicht....

Himan2001, 2022-06-06:
Hier zum Thema: Flashen über push_firmware des original image "würde auch problemlos" gehen:

```
* Analyzing 'FRITZ.Box_5590-07.29.image' ...
 * Product: Fritz_Box_HW272 (FRITZ!Box 5590)
 * Using command: ftp
 * Target host: 192.168.178.1
 * Outgoing IP: 192.168.178.99
 * Flash mode: fit-boot
 * Allowed memory size: 1024 MB
 * Designated linux_fs_start: <other>

 !!! WARNING !!! WARNING !!! WARNING !!! WARNING !!! WARNING !!!
 !!!  THERE IS NO WARRANTY AT ALL !!! USE AT YOUR OWN RISK   !!!

 * Are you sure, that you want to flash this file to the device?
   FRITZ.Box_5590-07.29.image
   Proceed? (y/[n]) y

 * You should now reboot your box (192.168.178.1). Waiting for shut down.
   Switch off, if reboot is not detected because it happens too quickly.
   Some newer bootloader versions allow to flash on power-cycle only.

 * MAPSTART=0x40000000
 * FULLSIZE=0x40000000   (1024 MB)
 * MAPLIMIT=0x80000000
 * FILESIZE=0x02ab6d77   (~42 MB)
 * FREESIZE=0x3d549289   (~981 MB)
 * MTDSTART=0x7d549289

 * No reply from box, assuming switch-off or restart. Trying to re-detect box.
   Waiting .......... found!

 * Box is back up again, initiating transfer.

Debugging on (debug=1).
---> TYPE I
---> MEDIA SDRAM
---> SETENV memsize 0x3d549289
---> SETENV kernel_args_tmp "avm_fwupdate mtdram1=0x7d549289,0x80000000 mtdparts_ext=update-image.0:0x02ab6d77@0x0(fit-image)"
---> PASV
---> STOR 0x7d549289 0x80000000
Execution failed.
---> QUIT

done

```

eben nicht - es ist bei der Box nicht egal. Es geht nur der Weg übers AVM-Webinteface aktuell und das original certificate, das ein originalimage auch angenommen wird muss auch im freetz image im Moment weiterhin drin sein....

Himan2001, 2022-06-06:
Eine Frage... laut Boxmatrix:

https://boxmatrix.info/wiki/Property:Falcon

arbeitet als 2. CPU für das Fiberfrontend - das ist identisch mit der 5530. Wurde dieser MIPS Teil korrekt berücksichtigt ?

freetz-ng-mod, 2022-06-07:
Vielleicht solltest du erst mal abwarten bis es überhaupt ein recover für die Box gibt oder ein source.

Himan2001, 2022-06-07:
@freetz-ng-mod 

Wg. recovery hatte ich am am Wochenende zeitgleich noch ne Supportanfrage an AVM gestellt, evtl. schaffen sie es kurzzeitig im downloadbereich den fehlenden 5590 ordner einzubinden oder mir ein recovery.exe direkt zukommen zu lassen ;-)

Aber andere Frage, da ich ja auch mit deinen infos aus dem Thread gearbeitet hatte... Ist bei Dir das gbic nach dem freetzen noch verfügbar ? Falls ja, hättest du mal Deine .config und evtl. infos deiner cross-umgebung zum Vergleich mehr für mich ?
evtl. auch nen git-hash, mit dem Du (noch) Erfolg hattest ?

Die "Reste" vom vorherigen freetz, die nach dem Rückflashen der Stockfirmware weiterhin über geblieben sind, habe ich mithilfe des AVM-shell-script "/bin/restorefactory" oder wie das script hies, dann halbwegs wegbekommen. Zumindest wurde damit der TFFS komplett leer gemacht und im Anschluss stecker ziehen und neu flashen.

in-memory files mit den eva_tools erstellen ist ja auch nicht mehr drin, da hier der "fehlende" kernel im FIT-Image bemängelt wird. Die 5530 jedenfalls rennt bei mir (mit der Einschränkung das ich hier weiterhin ohne den trustedd arbeite), dort wird das gbic allerdings erkannt - und laut Boxmaxtrix verwenden beide Kisten den gleichen MIPS-Teil für den Fiberbereich, also quasi als 2. CPU, wenn ich das so richtig verstanden habe. Das sollte aber normal auch kein Problem sein, da freetz ja an den AVM-Binarys keine Änderungen vornimmt und die aus dem Original-Image 1:1 übernimmt. Evtl. bei der 5590 nur ein Timing-Problem beim Init wg. der extra Freetz-AVM mods, die vorweg aktiv werden ? Bootlogs von beiden Varianten hängen ja etwas höher als Anhang mit dran. Das logfile vom freetz ist auch etwas länger, da es noch den Flashvorhang ueber push_firmware am Anfang mit dabei hat um auch dort zu sehen wie das image initialisiert wird.

wie fda77 schon bemerkt hat: bei der 5530 mit trusted werden nur 3 CPUs erkannt, bei der 5590 passt aber hier alles.
4 CPUs, laut freetz-Info, ist mit deinem Screenshot aus dem Thread auch identisch.

Mir ist jedoch eine Sache extra noch aufgefallen: Direkt nach dem flashen / werksreset wird für fiberboxen die ar7.cfg wg. fiber beim "kdsl" Teil einmalig modifiziert in der config-sektion "DSL". Das verbleibt dann auch. Beim rückflashen auf das Stock-Image wird ja OHNE werksreset dann das gbic auch nicht mehr erkannt - weil die ar7.cfg schon verborkt ist. Sauber wirds halt entweder über den Weg stock-image flashen und sofort wieder werksreset oder vor dem rückflashen von freetz aus mit ssh eben noch das /bin/factoryrestore.sh einmal ausführen und dann stecker ziehen, kaltstart und dann stock image ueber das AVM webinterface wieder rein. push_firmware mit stock-image führt leider auch zum oben gepasteten "error" und abbruch dann.

Himan2001, 2022-06-07:
... so erste Rückmeldung von AVM ist auch gerade gekommen:

"Es steht aktuell dem Support-Team zur 5590 noch nichts zur Verfügung.... wie z.b. eine recovery.exe zumindest..."

Himan2001, 2022-06-07:
Gerade gesehen:

https://boxmatrix.info/wiki/FIT-Image#EVA-Upload

```
The [5590](https://boxmatrix.info/wiki/5590) also has an 80MB size limit for the complete [fit-image](https://boxmatrix.info/wiki/fit-image), covering 3 kernels and 3 filesystems.
But unlike the [7530ax](https://boxmatrix.info/wiki/7530ax), [5530](https://boxmatrix.info/wiki/5530), [6000](https://boxmatrix.info/wiki/6000) and [4060](https://boxmatrix.info/wiki/4060) its main filesystem is not the last binary in FIT.
The remaining size needs to be calculated for the filesystem, not the filesyetem2. TODO!

$ fitimg -l 5590.fit -f -m
41208000-41543a43  3390019 kernel.image
41208000-41b950cc 10014924 * decompressed lzma *
41208000-41cd1000 11309056 * max size, 89% used *
41cd1000-41cd4f88    16264 flatdt_0.image
43000000-44ac1000 28053504 filesystem.image
88000000-881230df  1192159 kernel3.image
88000000-883b10a1  3870881 * decompressed lzma *
88000000-884f2000  5185536 * max size, 75% used *
884f2000-884f245e     1118 flatdt3_0.image
8b000000-8b0f2667   992871 filesystem3.image
70000000-703304dc  3343580 kernel2.image
70000000-70b598e5 11901157 * decompressed lzma *
70000000-70fac000 16433152 * max size, 72% used *
70fac000-70fadbda     7130 flatdt2_0_pon.image
70fac000-70fadbd4     7124 flatdt2_0_aon.image
73000000-7378d000  7917568 filesystem2.image
73000000-75cb4d40 46878016 * max size, 17% used *

```

Einfache Frage... schrauben wir denn nach dem zerlegen des Images den Originl-kernel auch wieder richtig zusammen,
bzw. handhabt das push_firmware das aktuell auch korrekt ?

Diese beide Dinge:

70fac000-70fadbda     7130 flatdt2_0_pon.image
70fac000-70fadbd4     7124 flatdt2_0_aon.image

Sollten wohl unverändert an den richtigen Speicherstellen auch stehen ?

Himan2001, 2022-06-07:
wg. der angesprochenen gpon upx-sachen, die beim freetz-image ja nicht aktiv sind, habe ich dann noch das gefunden:

https://boxmatrix.info/wiki/Property:ar7-lite
https://www.ip-phone-forum.de/threads/software-fitimg-avm-fit-image-manipulator.308941/page-4

Kann es sein, das wir mit freetz diesen teil nicht mit initialisieren und daher auch der gbic nicht mehr verfügbar ist ?
Sas spiegen ja auch die beiden consolen-lofiles irgendwie wieder - womit unter normalen vorrausetzungen ohne deaktivem watchdog zum testen ja auch recht schnell ein reboot kommt....

ar7-lite is a [FIT-Image](https://boxmatrix.info/wiki/FIT-Image) transfer and boot command using simple [FTP](https://boxmatrix.info/w/index.php?title=FTP&action=edit&redlink=1) to a remote [EVA](https://boxmatrix.info/w/index.php?title=EVA&action=edit&redlink=1) bootloader, like a [Recovery.exe](https://boxmatrix.info/wiki/Recovery.exe) does.
It is used in [upx-hwk-boot-prx](https://boxmatrix.info/wiki/upx-hwk-boot-prx) on a [Hawkeye](https://boxmatrix.info/wiki/Hawkeye) to first set the remote [Bootloader-Environment](https://boxmatrix.info/wiki/Bootloader-Environment) and then transfer and boot the
[Falcon](https://boxmatrix.info/wiki/Falcon) firmware [fit-image](https://boxmatrix.info/wiki/fit-image) on a [5590](https://boxmatrix.info/wiki/5590).

@fda77 

Handhaben wir das aktuell richtig ?:

```

ar7-lite is a [FIT-Image](https://boxmatrix.info/wiki/FIT-Image) transfer and boot command using simple [FTP](https://boxmatrix.info/w/index.php?title=FTP&action=edit&redlink=1) to a remote [EVA](https://boxmatrix.info/w/index.php?title=EVA&action=edit&redlink=1) bootloader, like a [Recovery.exe](https://boxmatrix.info/wiki/Recovery.exe) does.
It is used in [upx-hwk-boot-prx](https://boxmatrix.info/wiki/upx-hwk-boot-prx) on a [Hawkeye](https://boxmatrix.info/wiki/Hawkeye) to first set the remote [Bootloader-Environment](https://boxmatrix.info/wiki/Bootloader-Environment) and then transfer and boot the
[Falcon](https://boxmatrix.info/wiki/Falcon) firmware [fit-image](https://boxmatrix.info/wiki/fit-image) on a [5590](https://boxmatrix.info/wiki/5590).

The name of this program likely comes from the fact that the remote side is not the [Falcon](https://boxmatrix.info/wiki/Falcon) PRX CPU itself but an [AR7](https://boxmatrix.info/w/index.php?title=AR7&action=edit&redlink=1) lite
helper CPU, which also received the upload of the bootloader through a serial interface in [upx-hwk-send-urlader-prx](https://boxmatrix.info/wiki/upx-hwk-send-urlader-prx).
The mostly crypted bootloader binary contains the string 4Kc, which is the [MIPS](https://boxmatrix.info/w/index.php?title=MIPS&action=edit&redlink=1) core of an [AR7](https://boxmatrix.info/w/index.php?title=AR7&action=edit&redlink=1). The [Falcon](https://boxmatrix.info/wiki/Falcon) uses a 34Kc
[MIPS](https://boxmatrix.info/w/index.php?title=MIPS&action=edit&redlink=1) core. This is just an assumption for the moment but may well be true.
```

fda77, 2022-06-07:
Frag doch einfach direkt Hippie. Den kannst du aber nur im IRC oder per eMail erreichen, Github ist ihm zu neumodisch

freetz-ng-mod, 2022-06-07:
Seit Wann installiert dir freetz überhaupt was, das must du alles doch selber installieren.

https://freetz-ng.github.io/freetz-ng/PREREQUISITES

Himan2001, 2022-06-07:
geh komm. die nummer mit den prerequisites und eine grosse Auswahl von VMs von debian10,ubuntu 18,20 ist doch sowieso samt aller prerequistes vorhanden und rennen auch problemlos mit den anderen Boxen. Ich habe eigentlich mal zum vergleich weiter oben 2 serielle logs zur verfügung gestellt und drauf verwiesen das nach dem freetzen die ganzen UPX-sachen nicht aktiv sind. Alles was ich selber an unterschieden gefunden hatte, habe ich ausführlich gepostet gehabt....

Die Frage ob DEIN Gbic mit DEINER config nach dem freetzen rennt hast mir wieder nicht beantwortet - und falls ja, wie Du das hinbekommen hast. Ich schrieb bereits ich hab auch die i-matik sache spassenshalber weise auch mal durchgespielt ;-)

freetz-ng-mod, 2022-06-07:
Wenn man nicht dazu kommt es zu testen, kann man auf Fragen auch nicht richtig antworten. 

Und ich bleibe lieber bei Ubuntu 20.04 und nicht bei 18.04.

Himan2001, 2022-06-07:
ich hab eine ubuntu 20.04 laufen - damit werden die images einwandfrei gebaut - auch für meine 5530, 6890v2, 7590.
Ich baue seit mehr als 15 Jahren problemlos freetz-images, und in diesem speziellen Fall kann ich nur die Logs anbieten
und die Infos die sich ergeben weitergeben....

Ohne selber Programmierer zu sein, sieht es für mich so aus, als ob beim boot der 2. CPU-Teil, welcher über die UPX-Boot gemanaged wird NICHT initialiert wird - dafür sind ja im Image auch die anderen Kernel und Module für pon mit enthalten.

Laut dem fitimg log sind die auch im freetz-image die module korrekt enthalten - aber werden nach dem freetzen wohl nicht initialisiert ?

Aber schau einfach selber mal die seriellen logfiles zwischen stock & freetz an, die ich gefertigt habe....

Ich hab @fda77  access inkl. remote-reset & seriellen Port via ser2net angeboten. Mehr kann ich nicht machen um zu unterstützen....

freetz-ng-mod, 2022-06-07:
Hast du auch das installiert was für fit-image gebraucht wir oder optimal wer. Des Weiteren weist du auch das ich das fretzen per push einfach getestet habe ich bin mir nicht mal 100% sicher das die Adresse wirklich richtig passt. Denn wie du weißt gibt ja nicht mal eine recover gibt.

fda77, 2022-06-07:
Wie gesagt, frag hippie wegen fit. Das wird von freetz aufgerufen und das wars.
Vielleicht hat freetz-ng-mod erfolg mit einer älteren version von fitimg? Die aktuellste hat zumindest 1 bekannten bug der aber in freetz behoben ist: https://github.com/Freetz-NG/freetz-ng/commit/64603a7472a758eb12d18181f2508e73d90b3fa7

fda77, 2022-06-07:
Wie Himan2001 das ganze testet hab ich eh nicht verstanden. Laut ihm werden avm-images die mit PF hochgeladen werden vom bootloader wegen falscher signatur zurückgewiesen.
Dem bootloader ist tatsächlich es aber völlig egal was da ins ram hochgelaten wird. Er hat keinen Code zur Prüfung (wie wollte man sonst Freetz installieren???), davon abgesehen wird die Sigantur gar nicht mit hochgeladen

Himan2001, 2022-06-07:
Ok, ja war etwas viel, aber chronologisch von oben nach unten war der thread trotzde:

- nochmal zusammengefasst:

- Freetz bauen
- dann mit push_firmware das eben gebaute freetz image zur Box
- fertig

Nach dem nächsten reboot ist dann eben die 2. CPU für das Fiber-Modul nicht mehr initialisiert.
Wenn man meine beiden seriellen bootlogfiles vergleicht sieht man das auch sofort: Da gehts rein um die UPX-Sachen die mit der MIPS-CPU kommunzieren - es nach dem freetzen aber nicht mehr tun.

Beim originalimage sieht man dann auch die oben weiter dokuemtnierten ausgaben mit dem luacgi wo ständig der Status abgefragt wird.

Auch was das Thema "Original-Image" mit push_firmware zurück zur Box flashen (statt avm webinterface der box verwenden) ist nachvollziehbar. Dieses führt wenige Sekunden nach dem Flashvorgang zu einem Abbruch. Ich werde dazu nochmal das debug log des PF extra einschalten das man das genau sieht, wieso original-image flashen bei der 5590 ueber pf eben NICHT hinhaut.
(Im Gegensatz zu meinen anderen Boxen....)

So ich habe da gleich ein paar Sachen für @fda77 zum nachchecken eben beim direkten vergleich aus dem build dir gefunden....

fda77, 2022-06-07:
Vielleicht hast du irgend welche Remove Patches aktiviert die die Kommunikation zwischen den CPUs verhindern?

fda77, 2022-06-07:
> somit auch keine leftovers entfernt.

Das ist in der config im 1. post noch drin...

> hier habe ich etwas was im original und auch im freetz image vorhanden ist:

Wenn man möglichst wenig auswählt im menuconfig sollte viele gleich sein...

>  timeout triggered for upx-hwk-kpanic-recvd.service

Der startet halt nicht richtig. Schau in die service Datei und führ von hand aus was im start ist. Vielleicht mit irgend einem debugging noch

fda77, 2022-06-07:
Deinstallier PV, dann wird nicht genutzt

freetz-ng-mod, 2022-06-07:
@fda77 
ich habe die schon vor 3 Wochen erfragt. Aber wie du es sagst. Nichts bis jetzt

Himan2001, 2022-06-07:
Zitat:

"Der startet halt nicht richtig. Schau in die service Datei und führ von hand aus was im start ist. Vielleicht mit irgend einem debugging noch"

Da ich, wie geschrieben, auf den innereien von freetz selber nicht seit 15 jahren fitt bin, kannst du mir für die ssh-shell unter laufendem freetz eben die systemd-comands kurz nennen die ich direkt eingeben soll, damit ich da passende antworten bekomme ? - ich frage deshalb weil ich kein "gewohntes" systemctl auf der box finde - das ist bei jedem Linux einfach da sofort bei jedem teil an einen status und nen rückgabewert zu kommen...

Das was ich dir aus "boxmatrix" gepostet hatte mit den infos zur "2. CPU" und dem boot sollte Dir einen hinweis geben wo evtl. nach dem Start das Problem liegen kann. Herausgelesen habe ich zumindest mal eines. Es gibt für die 5530 UND die 5590.
nur die 5530 rennt ja (sofern man den trustedd derzeit deaktiviert) - auch hier gibts diese 2 MIPS-cpu die beim kaltstart initialisiert wird.

Es gibt 3 kernel im Image + entsprechende extra Module (pon kram....). Die sind auch mit einem fitimage-test im freetz nach wie vor erhalten, das sollte also auch nicht das Problem sein. Stehen auch an denselben Startadressen wie im Original.

Sowie ich die Erklärungen aus dem forum und der boxmatrix aber verstanden habe, werden direkt nach dem kaltstart auch der kernel für die MIPS-CPU gebootet, da stand auch was von kernel-commandline aus dem enviroment wenn ich das richtig gelesen habe.

JA: Ich hatte AVM am Wochenende dazu kontaktiert und bekam vor 3 Stunden die Antwort das es bislang weder ein recovery.exe noch irgend etwas zu der box gibt beim "offiziellen" Support. Der fehlende 5590 Downloadbereich ist bereits bemängelt worden von mir.

Aber: Ich habe 3 INHOUSE Versionen für 7.27 gefunden im Forum, die waren auch manuell problemlos noch zum herunterladen von AVM... denke die bringen uns aber auch nicht viel weiter....

fda77, 2022-06-07:
@freetz-ng-mod: Dann hat der Praktikant bei AVM aktuell wohl Urlaub...

Laut https://boxmatrix.info/wiki/Property:tcpsvd haben (nur) 5530+5590 dieses applet (freetz per default nicht).

Aus dem "berühmten" "gpl" Sourcecode von AVM:

```
grep TCPSVD make/busybox/avm/*55*
make/busybox/avm/07.21-5530--busybox.config.prxI:# CONFIG_TCPSVD is not set
make/busybox/avm/07.29-5530--busybox.config.prxI:# CONFIG_TCPSVD is not set
```

fda77, 2022-06-07:
@Himan2001: Du hast doch oben selbst mit "ls" die service Dateien gezeigt. Schau dir mit "cat" an was drin steht

freetz-ng-mod, 2022-06-07:
Das mit dem Original Image per push, ging bei meiner 7530AX auch nicht, aber das habe ich aber immer unter der Sache eingestuft wegen des Fehlers in meinen Speicher Bock. Da es bei dem 6000 ging.

Himan2001, 2022-06-07:
DAS werde ich in die eben gepostete .config eben nochmnal manuell hinzufügen. SONST weiter nichts :-)

Ansonsten habe ich im originalimage endlose ausgaben des "luacgi" wo der fiberstatus in endlosschleife ausgegeben wird - sofern die 2. CPU auch erfolgreich gestartet worde - das wäre dann der Normalzustand. Wieso AVM das alles mitten in die Console haut ist nicht natürlich auch nicht einfacher fürs serielle loggen.... ;-)

Das meine ar7.cfg so verborkt ist - was sich auf das Provider-Menu ja auch auswirkt, hat schlicht und ergreifend damit zu tun, das beim allersten neustart nach einem flashen eine diff zur "normalen" ar7.cfg erstellt wird, wo dann Anpassungen zum PON statt finden. Geht das aber wegen der UPX-Probleme in die Hose verbleibt eine "kaputte" ar7.cfg die auf "dsl" basiert. Das ist dann nur eine Folgeerscheinung, das beim initialboot eben die "PON-CPU" nicht inisiert werden kann. Steht auch in den logfiles
drin, das die Änderungen durchgeführt werden wenn es sich um eine fiber-box handelt ;-) -  das passiert eben nur genau dieses einzige mal und bei einem Werksreset.

Himan2001, 2022-06-07:
> @Himan2001: Du hast doch oben selbst mit "ls" die service Dateien gezeigt. Schau dir mit "cat" an was drin steht

ja klar sehe ich den inhalt:

```
root@freetzdevel:/home/freetz/freetz-ng/build/original/filesystem/lib/systemd/system# cat upx-hwk-kpanic-recvd.service
[Service]
ExecStart=/bin/upx-hwk-tcp-recvd upx-hwk-kpanic-recvd 4344 /proc/avm/panic2.log
Restart=always
Type=notify
Before=upx-hwk-mgmtd.service
After=avmipcd.service

[Install]
WantedBy=multi-user.target

--

root@freetzdevel:/home/freetz/freetz-ng/build/original/filesystem/lib/systemd/system# cat upx-hwk-mgmtd.service
[Service]
ExecStart=/bin/upx-hwk-mgmtd.sh
Restart=always
After=avmipcd.service

[Install]
WantedBy=multi-user.target

--

root@freetzdevel:/home/freetz/freetz-ng/build/original/filesystem/lib/systemd/system# cat upx-hwk-pppdlite.service
[Service]
ExecStart=/bin/upx-hwk-pppdlite
Restart=always
Before=avmipcd.service

[Install]
WantedBy=multi-user.target

--

root@freetzdevel:/home/freetz/freetz-ng/build/original/filesystem/lib/systemd/system# cat upx-hwk-serlogd.service
[Service]
ExecStart=/bin/upx-hwk-serlogd
Restart=always

[Install]
WantedBy=multi-user.target

--

root@freetzdevel:/home/freetz/freetz-ng/build/original/filesystem/lib/systemd/system# cat upx-hwk-time-sendd.service
[Service]
ExecStart=/bin/upx-hwk-time-sendd 4350
Restart=always
After=avmipcd.service

[Install]
WantedBy=multi-user.target

--

root@freetzdevel:/home/freetz/freetz-ng/build/original/filesystem/lib/systemd/system# cat upx-hwk-ucrash-recvd.service
[Service]
ExecStart=/bin/upx-hwk-tcp-recvd upx-hwk-ucrash-recvd 4345 /proc/avm/crash2.log
Restart=always
Type=notify
Before=upx-hwk-mgmtd.service
After=avmipcd.service

[Install]
WantedBy=multi-user.target

--

root@freetzdevel:/home/freetz/freetz-ng/build/original/filesystem/lib/systemd/system# cat upx-hwk-ulog-recvd.service
[Service]
ExecStart=/bin/upx-hwk-tcp-recvd upx-hwk-ulog-recvd 4346 /proc/avm/user2.log
Restart=always
Type=notify
Before=upx-hwk-mgmtd.service
After=avmipcd.service

[Install]
WantedBy=multi-user.target

```

Ich habs für alle die die Box gerade nicht vor der Nase haben und das original-Image ausgepackt liegen haben jetzt mal gepostet was in allen files drin ist.

Evtl. siehst Du da ja auch noch einen weiteren "Service" wo evtl. busybox Sachen noch interessant sein könnten ?

fda77, 2022-06-07:
> Das mit dem Original Image per push, ging bei meiner 7530AX auch nicht

Es wird die "fit" Datei aus der image.-tar hochgeladen. Was man dorthin lädt ist völlig egal. Der EVA bootloader ist absolut minimal, der prüft gar nichts.. Wenn es Käse ist startet die Box nachher eben nicht. Weshalb da so ist sieht man dann aber im seriellen log. Manche neue bootloader dann wieder (angeblich?!) auf das alte linux_fs_start zurük
Wenn das von freetz "nachgebaut" image bootet, müsste das von avm erst recht.
Aber ohne logs und irgend welchen Hinweisen kann man da nichts machen

fda77, 2022-06-07:
> Evtl. siehst Du da ja auch noch einen weiteren "Service" wo evtl. busybox Sachen noch interessant sein könnten ?

Mein grep oben hat 2 services gefunden die tcpsvd  nutzen

fda77, 2022-06-07:
Versuch 
```
PS1=blubb
```

fda77, 2022-06-07:
Antwort: Das steht doch oben schon.

fda77, 2022-06-07:
> wollte nur rückmelden das die "Testoption" für AVM die originale busybox mit den original-symlinks ...

Es ist klar dass es da bei den FREETZ-SACHEN viele Fehler geben wird! Du sollst einfach nur schauen ob ein avm-dameon der vorher nicht wollte nun startet.

Übrigens: Damit wäre dieser "tcpsvc" drin geblieben und es hätte funktionieren sollen !?

Himan2001, 2022-06-07:
...und wenn ich schon dabei bin... das "netca"t ist in der busybox config auch nicht drin - das wird aber wohl nur im rahmen eines debug-logfiles erstellen verwendet wie ich das auf die schnelle gesehen habe....

Interessant ist jedenfalls nach dem erstellen die beiden build-ordner "original" und "modified" verzeichnis für verzeichnis
manuell zu vergleichen. selbst ohne busybox config sieht man ja schon menge bei den Symlinks wo es da Unterschiede gibt ;-)

fda77, 2022-06-07:
Steck mal das Kabel rein

fda77, 2022-06-07:
Und gibt mal ein "tcpsvc --help"

Himan2001, 2022-06-07:
ah es hat sich "ps" was getan. obowhl wieder alles verborkt ist rennt jetzt das tcpsvc der busybox zumindest als process.

ich logge die ssh shell. kommt gleich.....

Himan2001, 2022-06-07:
hier:

root@fritz:/var/log$ tcpsvd --help
BusyBox v1.34.1 (2022-06-07 18:22:49 CEST) multi-call binary.

Usage: tcpsvd [-hEv] [-c N] [-C N[:MSG]] [-b N] [-u USER] [-l NAME] IP PORT PROG

Create TCP socket, bind to IP:PORT and listen for incoming connections.
Run PROG for each connection.

        IP PORT         IP:PORT to listen on
        PROG ARGS       Program to run
        -u USER[:GRP]   Change to user/group after bind
        -c N            Up to N connections simultaneously (default 30)
        -b N            Allow backlog of approximately N TCP SYNs (default 20)
        -C N[:MSG]      Allow only up to N connections from the same IP:
                        new connections from this IP address are closed
                        immediately, MSG is written to the peer before close
        -E              Don't set up environment
        -h              Look up peer's hostname
        -l NAME         Local hostname (else look up local hostname in DNS)
        -v              Verbose

Environment if no -E:
PROTO='TCP'
TCPREMOTEADDR='ip:port' ('[ip]:port' for IPv6)
TCPLOCALADDR='ip:port'
TCPORIGDSTADDR='ip:port' of destination before firewall
        Useful for REDIRECTed-to-local connections:
        iptables -t nat -A PREROUTING -p tcp --dport 80 -j REDIRECT --to 8080
TCPCONCURRENCY=num_of_connects_from_this_ip
If -h:
TCPLOCALHOST='hostname' (-l NAME is used if specified)
TCPREMOTEHOST='hostname'

[75 more replies]