Freetz-NG/freetz-ng #564: Nach git Pull Abhängigkeitsprobleme...

SvenLuebke, 2022-06-09:
Hallo!

Hab gerade ein "git pull" gemacht (899d9bb66d4325670f5c3ff215be750d8d99969a) und wollte mittels "make" durchbauen. Zumindest auf meinem System (Debian 10) scheinen die Dependencies nicht mehr aufgelöst zu werden. Ich bekam die Meldung, dass "libfdt.h" nicht gefunden werden kann. Dann weiter herumprobiert, es baute weiter, aber später kam einer neuer Fehler. Also "make distclean" ausgeführt. und nun bekomme ich wieder den libfdt.h-Fehler (siehe Log). Hatte vor zwei oder drei Tagen ebenfalls ein Update durchgeführt und dort lief noch alles.

Seit ein paar Wochen scheint die makefile bash completion auch nicht mehr zu funktionieren. Wenn ich "make" TAB eingebe, dann erscheint immer nur envira. Das war vorher schöner :-)

[config.txt](https://github.com/Freetz-NG/freetz-ng/files/8869653/config.txt)
[freetz-NG_error.txt](https://github.com/Freetz-NG/freetz-ng/files/8869683/freetz-NG_error.txt)

Viele Grüße
Sven

harryboo, 2022-06-09:
auch hier auf ubuntu 20.04 gleiches problem mit der libfdt.h

fda77, 2022-06-09:
Mit 046530b2a hat sich er Pfad geändert in dem die *fdt* installiert werden und mit bddfc4540 wird dieser benutzt.
Ich vermute dies hilft: `make dtc-host-recompile`

SvenLuebke, 2022-06-09:
Nach einem `make distclean` hätte das dann doch eigentlich funktionieren müssen, oder? https://github.com/Freetz-NG/freetz-ng/blob/9918df807e13fa124d563babdb692348a60985e2/tools/make/yf-akcarea-host/yf-akcarea-host.mk#L7 ist auch vorhanden. Deswegen hatte ich vermutet, dass es ein Dependency-Problem gibt.

`make dtc-host` meldete beim Kopieren der Includefiles, dass das Directory "tools/build/include" nicht gefunden werden konnte. Hätte `make dtc-host-recompile` geholfen?

@fda77 besteht die Möglichkeit, die bash completion wieder zum Laufen zu bringen...oder ist das so gewollt?

fda77, 2022-06-09:
> Nach einem make distclean hätte das dann doch eigentlich funktionieren müssen, oder?

Ja...

fda77, 2022-06-09:
> make dtc-host meldete beim Kopieren der Includefiles, dass das Directory "tools/build/include" nicht gefunden werden konnte.

auch nach dem "dtc-host: create dirs if they dont exist" oben?

SvenLuebke, 2022-06-09:
> > make dtc-host meldete beim Kopieren der Includefiles, dass das Directory "tools/build/include" nicht gefunden werden konnte.
> 
> auch nach dem "dtc-host: create dirs if they dont exist" oben?

Ja, denke schon!

Hat nun eine Zeitlang weiter gebaut nun kann er `python` nicht finden (siehe [freetz-NG_NEWerror.txt](https://github.com/Freetz-NG/freetz-ng/files/8870159/freetz-NG_NEWerror.txt) )

fda77, 2022-06-09:
Das dtc wurde seit 5fe3e29c10621df40becaac52ea5d98ce2690397 gar nicht mehr gebaut da von der übergeordneten Funktion _DEPENDS_ON statt _DEPENDS erwartete

SvenLuebke, 2022-06-09:
Ahh, ok! sorry, nein, ich hatte zwichendurch kein `git pull` gemacht. Mach ich jetzt mal

SvenLuebke, 2022-06-09:
Läuft weiter! 👍

fda77, 2022-06-09:
> Seit ein paar Wochen scheint die makefile bash completion auch nicht mehr zu funktionieren. Wenn ich make TAB eingebe, dann erscheint immer nur envira. Das war vorher schöner :-)

Ich wusste gar nicht dass make das kann. Jedenfalls geht das seit diesem "if" nicht mehr: https://github.com/Freetz-NG/freetz-ng/commit/
Dieses geschachtelte make hat den Zweck dass im PATH die freetz eigenen Tools immer zuerst genutzt werden (in make und fwmod) und dass UMASK richtig gesetzt ist sowie die Dateiberechtigungen.
Wie man die bash-completion  jetzt wieder hinbekommt weiss ich allerdings nicht

SvenLuebke, 2022-06-09:
> Ich wusste gar nicht dass make das kann. Jedenfalls geht das seit diesem "if" nicht mehr: https://github.com/Freetz-NG/freetz-ng/commit/ Dieses geschachtelte make hat den Zweck dass im PATH die [...]

Kannst Du den Link bitte nochmal korrigieren? Wollte mal schauen, wodurch das zustande kommt.
Ja, die Completion ist schon ziemlich cool. Benutzte ich recht häufig.

fda77, 2022-06-09:
https://github.com/Freetz-NG/freetz-ng/blob/master/Makefile#L20

fda77, 2022-06-09:
Bist du sicher dass die bash-completion je in freetz funktioniert hat?
Wenn ich Zeile 20-53 lösche und die allerletzte auch gibt es gar keine completion

fda77, 2022-06-09:
Dependencies funktionieren nach https://github.com/Freetz-NG/freetz-ng/commit/5fe3e29c10621df40becaac52ea5d98ce2690397 wieder

fda77, 2022-06-09:
Also grundsätzlich geht die shell-completion ja, aber nur in dieser kurze envira-Zweig. Wäre aber nett das überall zu haben. `make olddefconfig -pRrq` zeigt jedenfalls an was es alles so gibt

fda77, 2022-06-09:
```
complete -W "$(grep '^[^\.][-a-z\.0-9_\/]*:' ./Makefile | sed 's/:.*//' | sort -u)" make
make <tab>
```

SvenLuebke, 2022-06-09:
So funktioniert es bei mir auch!

Vorher ging die Completion aber sogar noch in die einzelnen `make/*/*.mk` rein. Damit konnte man `make perl<tab>` eintippen und hat dann zusätzlich sämtliche "Subcommands" angezeigt bekommen. Z.B. `make perl-recompile`.

SvenLuebke, 2022-06-09:
Danke fürs Fixen des anderen Problems!

SvenLuebke, 2022-06-09:
https://github.com/Freetz-NG/freetz-ng/commits/master?since=2022-02-02&until=2022-02-04

fda77, 2022-06-09:
> Vorher ging die Completion aber sogar noch in die einzelnen make/*/*.mk rein.

Ich wüsste nicht wie das funktioniert haben könnte, ausser auf deinem Sytem war was ganz tolles in `/etc/bash_completion.d/`.
Damit noch mehr geht müsste man den Befehl erweitern

SvenLuebke, 2022-06-09:
War noch am Suchen! Ich dachte, ich hätte den Unterschied dort gefunden, aber das war ein Fehler. Inzwischen funkt die `.config` irgendwie dazwischen. Wenn man sie weglöscht, dann läuft es wieder mit der unten stehenden CommitID:
```git reset --hard 7799769666236c366d0aa83fe6079dcc91f314fe
kmeleon@er00186v:/media/projects/fritzbox/freetz-ng_aarch64$ git show
commit 7799769666236c366d0aa83fe6079dcc91f314fe (HEAD -> master)
Author: fda77 <fda77@users.noreply.github.com>
Date:   Mon May 30 20:16:54 2022 +0200

    fix README

diff --git a/README.md b/README.md
index c23d7640b..d5a737e9e 100644
--- a/README.md
+++ b/README.md
@@ -14,7 +14,7 @@ More features - less bugs!

 ### Requirements:
  * You need an up to date Linux System with some [prerequisites](docs/PREREQUISITES.md).
- * Or download a ready-to-use VM like Gismotro's [Freetz-Linux](https://freetz.digital-eliteboard.com/?dir=Teamserver/Freetz/Freetz-VM/VirtualBox/). (admin & pass: freetz)
+ * Or download a ready-to-use VM like Gismotro's [Freetz-Linux](https://freetz.digital-eliteboard.com/?dir=Teamserver/Freetz/Freetz-VM/VirtualBox/). (user & pass: `freetz`)
  * Your linux user needs to have set `umask 0022` before checkout and during make.

 ### Clone the main branch:
kmeleon@er00186v:/media/projects/fritzbox/freetz-ng_aarch64$ make
ERROR: The program 'depmod' was not found in path
ERROR: The program 'ldconfig' was not found in path
Makefile:143: *** Some build prerequisites are missing! See '.prerequisites' for why. Please install the missing packages before trying again. See https://freetz-ng.github.io/freetz-ng/PREREQUISITES for installation hints.  Stop.
kmeleon@er00186v:/media/projects/fritzbox/freetz-ng_aarch64$ make
Display all 8287 possibilities? (y or n)
kmeleon@er00186v:/media/projects/fritzbox/freetz-ng_aarch64$ make ^C
kmeleon@er00186v:/media/projects/fritzbox/freetz-ng_aarch64$ make
Display all 8287 possibilities? (y or n)
kmeleon@er00186v:/media/projects/fritzbox/freetz-ng_aarch64$ make perl
perl                   perl-build-prereq      perl-dirclean          perl-distclean         perl-list              perl-precompiled       perl-recompile         perl-uninstall
perl-autofix           perl-clean             perl-dirclean--int     perl-download          perl-package           perl-precompiled--int  perl-source            perl-unpacked
kmeleon@er00186v:/media/projects/fritzbox/freetz-ng_aarch64$ make perl
perl                   perl-build-prereq      perl-dirclean          perl-distclean         perl-list              perl-precompiled       perl-recompile         perl-uninstall
perl-autofix           perl-clean             perl-dirclean--int     perl-download          perl-package           perl-precompiled--int  perl-source            perl-unpacked
kmeleon@er00186v:/media/projects/fritzbox/freetz-ng_aarch64$ make perl
perl                   perl-build-prereq      perl-dirclean          perl-distclean         perl-list              perl-precompiled       perl-recompile         perl-uninstall
perl-autofix           perl-clean             perl-dirclean--int     perl-download          perl-package           perl-precompiled--int  perl-source            perl-unpacked
kmeleon@er00186v:/media/projects/fritzbox/freetz-ng_aarch64$ make perl-
perl-autofix           perl-clean             perl-dirclean--int     perl-download          perl-package           perl-precompiled--int  perl-source            perl-unpacked
perl-build-prereq      perl-dirclean          perl-distclean         perl-list              perl-precompiled       perl-recompile         perl-uninstall
kmeleon@er00186v:/media/projects/fritzbox/freetz-ng_aarch64$ make perl-
````

fda77, 2022-06-09:
Für sowas benutze ich svn, mit git ist das viel zu umständlich. Aktuell ist r19720. getestet hab ich 19500 19000 18500. Bei keiner Version funktioniert das bei mir.
Es scheint doch Vorteile von Debian gegenüber Fedora zu geben :) Mit der aktuellen Version geht allerdings "envira" bei mir -.-

fda77, 2022-06-09:
Vielleicht hilft es das "if" in der Makefile  zu negieren und dessen true und false part zu tauschen...

fda77, 2022-06-09:
Vielleicht einfach so:
```
_makefile_freetz() {
    [ -e docs/wiki/20_Advanced/make_targets.en.md ] || return

    # most(tm) freetz targets, no k+t toolchains
    local x targets="$( (
        for x in make/*/*.mk make/libs/*/*.mk tools/make/*/*.mk; do x="${x##*/}"; x="${x%.mk}"
         echo "$x-precompiled $x-recompile $x-autofix $x-unpacked $x-clean $x-dirclean $x-distclean"; done
        grep -vE "^(#|$|\\$|\`| )" docs/wiki/20_Advanced/make_targets.en.md | cut -f1 -d " "
    ) | tr '\n' ' ')"

    # 1 target is enough, deny further completion
    for x in ${COMP_WORDS[@]}; do [ "${COMP_WORDS[COMP_CWORD]}" != "$x" ] && echo " $targets " | grep -q " $x " && return; done

    COMPREPLY=( $(compgen -W "${targets}" ${COMP_WORDS[COMP_CWORD]}) )
}
complete -o nospace -F _makefile_freetz make

$ make 
Display all 3055 possibilities? (y or n)

$ make perl-
perl-autofix      perl-clean        perl-dirclean     perl-distclean    perl-precompiled  perl-recompile    perl-unpacked
```

Das legt nur alle anderen makefiles still...