Freetz-NG/freetz-ng #570: FIT, 5530, 5590, 7530ax etc

fda77, 2022-06-18:
Weiterführung von https://github.com/Freetz-NG/freetz-ng/commit/bc00d8f8b280fd10603358955381f1917d028c08

fda77, 2022-06-18:
> frontend ist nur ein Symlink, der von fitdump.sh (bei passenden Optionen) erzeugt wird 

Werden diese Unterverzeichnisse beim zusammenpacken auch beachtet? Oder wie ohne nur anhand der .its?

https://github.com/Freetz-NG/freetz-ng/blob/master/tools/fwdu#L141= erstellt als workaround zu den fehlenden Links einfach filesystem auf die größte Datei, kernel auf die 2. größte.
Das entpacken davon ging: https://github.com/Freetz-NG/freetz-ng/actions/runs/2520434098
Und das Ergebnis hier: https://github.com/Freetz-NG/freetz-ng/commit/daead25a21ae6550f152cb651b1259b7e916a042

Die 4060-inhaus wurde vorhin erst aufgenommen: https://github.com/Freetz-NG/freetz-ng/commit/63f4fc32a34ad224651b31cbc8f1ece9716f0562

Die 5590 kann den fiber-core laden, auch wenn ein avm-image keine "sigantur" aussenherum in den extras hat, WENN man ar7-lite durch einen ncftp-wrapper ersetzt. Allerdings funktioniert ein Freetz image in dem nur das haupt-fs ersetzt hat nicht -> evtl fit-problem. Aber vorerst nicht so wichtig da der workaround mit fit-image in der storage funktioniert

Die 5530 funktionier ja schon soweit, bis auf dass untrustedd entfernt oder der watchdog ausgeschaltet werden muss:  https://github.com/Freetz-NG/freetz-ng/discussions/495
-> Die Dateisysteme sind wohl (zumindest vom Hauptsystem) richtig gepackt, rest evlt fit-problem

7530ax-innhaus bootet immerhin, aber auch so lala, die Fehlerbeschreibung ist aber diffus: https://github.com/Freetz-NG/freetz-ng/discussions/483#discussioncomment-2806037
-> freetz erstell die "ramdisk" wohl korrekt, rest fit-problem?

Da das alles evtl an dem perl-fitimg liegt weiss ich nicht genau ob dies zusammenhängt
Ich hab jedenfalls Zzugrriff auf serielle consolen von 5530 + 5590 zum herumtesten, flashen etc

> Was genau willst Du mit dem hier

Kann ich nicht beantworten, der link geht nicht

fda77, 2022-06-19:
5530: Bootet nun auch mit `mkimage`, für dieses muss man aber `SOURCE_DATE_EPOCH` gesetzt haben

PeterPawn, 2022-06-19:
Ich habe NICHT in `fwdu` nachgesehen, was Du da machst - aber ich hatte zuvor festgestellt, daß eine reine "Erkennung" anhand der Größe und/oder Position im FIT-Image (was ich zuerst auch so implementiert hatte) nur sehr unzuverlässig arbeitet (das mit der Größe klappt vielleicht noch für das Dateisystem ... spätestens beim dazugehörenden richtigen(!) Kernel wird das wackelig, weshalb ich auch in den `configurations`-Nodes nach dem richtigen Kernel für das (identifizierte) Dateisystem-Image suche) und daher habe ich das "etwas komplizierter" implementiert - dafür sollte es "länger halten", wenn AVM nicht erneut solche "game changers" einbaut, mit denen dann das Dateisystem nicht mehr zuverlässig identifiziert werden kann. 

Aber dagegen hilft auch KEINE andere Methode ... aber gegen simplere "Ärgernisse" wie Änderungen der Namen von Nodes und/oder deren Reihenfolge kann man sich auf diesem Weg einigermaßen absichern und damit "Zuordnungen", wie sie in `fitimg` im Perl-Code erfolgen, überflüssig machen.

Das Skript, um aus der Verzeichnisstruktur wieder eine `image.its` als Eingabe für `mkimage` zu erstellen, ist eben (wie zuvor schon geschrieben) noch NICHT fertig ... nur ist das halt keine "rocket science", so etwas zu machen, weil das nur das rekursive Abgrasen der Verzeichnisstruktur erfordert. Mehr Aufwand ist es, wenn man das alles auch noch "schön einrücken" wollte - das wäre aber wieder nur für den menschlichen Leser interessant, denn das `mkimage` juckt es nicht, solange die (geschweiften) Klammern für die Nodes nur richtig sitzen.

Wenn es jetzt tatsächlich gebraucht werden sollte (das war ja bisher nicht abzusehen und für mich hat(te) es keine Priorität), kann man das ja mal in Angriff nehmen. Bis dahin kannst Du ja erst mal richtig testen, ob Du überhaupt die richtigen Stellen in den Properties auch richtig anpaßt ... dafür braucht man ja nicht gleich funktionierenden Code, weil man DAS dann wirklich erst mal (meinetwegen auch automatisiert mit `sed`, wenn man's nicht von Hand machen will) an der "fertigen" `its`-Datei ausprobieren kann, damit man auch sicher ist, daß die Änderungen tatsächlich den gewünschten Erfolg bringen. 

DANACH kann man dann immer noch überlegen, wie genau man diese Änderungen in die Datei kriegt ... ich habe meine Idee für den richtigen Weg dabei, bei der (nach meiner Ansicht) dann auch keine Unwägbarkeiten mehr übrig bleiben, daß mit irgendwelchen "search & replace"-Aktionen entweder gar keine Änderungen erfolgen (weil die "Anker" nicht stimmen) oder - auf den Rest der enthaltenen Daten hat man nun mal auch keinen Einfluß und der kann/wird in der nächsten Version von AVM schon wieder ein anderer sein - daß dabei zuviele Änderungen erfolgen, weil das "Suchmuster" zufällig noch an anderer Stelle auftaucht, wo man es aber gar nicht ändern WILL. 

Nach meiner Erfahrung sind solche Ansätze zwar "praktisch", aber eben auch gefährlich - "sicheres Implementieren" verbietet solches "search & replace", solange der "Code", der das ausführen will, KEINE AHNUNG vom tatsächlichen Aufbau der Eingangsdaten hat. 

Schon wenn Du die `load`-Property für ein Dateisystem ändern willst, müßtest Du WISSEN, welche die richtige ist, wenn da mehrere in einem FIT-Image definiert sind. Du kannst ja mal spaßeshalber nach `load` suchen lassen in so einer `image.its` ... ähnliches gilt auch für das Finden der richtigen `kernel-args` in so einer (Text-)Datei, denn wie Du Dir bei der 5590 (07.2<irgendwas>) ansehen kannst, gibt es da auch mehrere Nodes, in denen `kernel-args` vorkommt. 

DAS ist auch der Grund (WEIL es bei der Firmware der 5590 bisher schon einen solchen (zweiten) Eintrag für `prxI_HW273_squashFS_filesystem` gab, der aber eben NICHT der Node für das Dateisystem fürs Frontend ist und dennoch die Zeichenfolge `rootfs_ram` enthält), warum ich die Änderung beim `fitdump.sh` für die neuen Inhouse-Versionen erst einmal TESTEN will, bevor ich sie auf die Allgemeinheit loslasse - denn genau dieser zweite Eintrag war bisher der Grund dafür, daß ich mich bei der (automatischen) Suche nach dem Node mit den Dateisystem für das Frontend NICHT auf die Zeichenfolge `rootfs_ram` beschränkt hatte.

fda77, 2022-06-19:
`fwdu` wird nicht vom make also vom Benutzer genutzt, sondern einfach nur zum manuelle einpacken das Haupt-"Filesystems" (wie es aus immer heissen mag) und zum generieren von config/*img - bei dem aktuell etwas nicht geht: https://github.com/Freetz-NG/freetz-ng/actions/workflows/generate_img.yml

Der aktuelle Stand: [fit.patch.txt](https://github.com/Freetz-NG/freetz-ng/files/8934993/fit.patch.txt)
Damit booten 5530+5590 mit denen ich testen kann. Das trifft aber keine Aussage über die untrusted und watchdog Sachen der 5530

[image]

Sollte man evtl den AVM-"Header" der avm-fit einfach kopieren wie er ist statt zu 00en? Vielleicht ist es mehr als eine Signatur

Mir ist klar dass dieses ersetzen in der .its nicht so optimal ist, vorerst aber okay. Mit dem Patch oben werden durch make `build/(original|modified)/fit-dump/` Verzeichnisse zum vergleichen erzeugt

PeterPawn, 2022-06-19:
Immer noch, ohne in Deine Änderungen zu schauen (ich sehe bisher keine Notwendigkeit dafür):

Meine "Warnung" hinsichtlich rein größenbasierter Erkennung (für Kernel-Images) und die Suche nach dem richtigen(!) `kernel-args`-Eintrag, bezieht sich ausdrücklich auf die 5590 und zwar mit der Version 07.27 (wenn ich mich richtig erinnere). DORT kannst Du nachlesen, daß es (a) mehrere `load`-Einträge gibt, (b) zwei unterschiedliche Kernel, die sich in ihrer Größe nur marginal unterscheiden (3.390.019 vs. 3.343.580 Bytes, was bei der nächsten Version schon wieder ganz anders aussehen kann - und zwar in beide Richtungen -> größere oder kleinere Differenz, wenn's schlecht läuft, sogar "Umkehr" der Relation) und (c) auch mehr als einen Dateisystem-Node mit `kernel-args`, `mtdram` und `rootfs_ram` ... und letzteren brauchst Du auch, wenn Du die richtigen Speicheradressen finden willst, an denen bei AVM die Images geladen werden und die Du (wenn sich Dein Dateisystem unterscheidet) ändern mußt.

Wenn Du das also nicht alles noch irgendwo "definieren" willst (was dann Deine automatische(!) Anpassung von Freetz-NG an neue Versionen, die dank JUIS-Abfrage nicht mehr für jede neue AVM-Version eine Freetz-NG-Änderung braucht, ja irgendwie ad absurdum führen würde - die Werte können (und werden) sich desöfteren ändern, weil schon die Ladeadressen immer von der konkreten Größe der Daten in der jeweiligen Version abhängen dürften), brauchst Du schon einen zuverlässigen Weg, diese Werte für das jeweils vorliegende Image zu ermitteln. 

Für die richtigen(!) `kernel-args` für das Frontend-System könntest Du z.B. (mal unterstellt, mein `fitdump.sh` arbeitet korrekt - wenn nicht, muß man DAS dann halt anpassen/korrigieren) einfach den Inhalt der Datei `frontend/avm,kernel-args` auslesen (im `fit-dump`-Verzeichnis), ebenso natürlich für alle weiteren Properties, die zu dem Node gehören.

Dann hättest Du (einigermaßen sicher) auch den richtigen Eintrag, egal in welcher Reihenfolge die Daten durch AVM in (auch künftigen) FIT-Images definiert wurden ... bei der Verwendung von `sed` sähe ich schon für das Ermitteln der richtigen Ladeadressen (im Fall der o.a. Version wären das 0x43000000 und 0x44b00000) Probleme, wenn man die richtigen Adressen nicht kennt - denn dann kann man auch nicht entscheiden, ob das nun die erste, zweite oder gar dritte Zeile in der `image.its` ist, in der ein `mtdram=rootfs_ram,...` auftaucht. 

Und ohne den richtigen "Ursprungswert" kann man weder selbst rechnen (um die passenden Änderungen zu ermitteln), noch die richtige Stelle für "search & replace" finden ... und so ziehen sich die Probleme dann weiter - auch das richtige `load` findet man halt nur, wenn die richtigen Adressen in einem `kernel-args`-Eintrag bekannt sind (sogar jedes FDT-Image in so einem FIT-Image hat seine eigene `load`-Property).

fda77, 2022-06-19:
Das ist schon klar... deshalb such ich nicht nach load. Sondern wie in den ~20 Zeilen die der Patch oben hinzufügt zu sehen:
```
XXX_LOAD="$(sed -rn "s/.*mtdram=ram-filesystem,([^,]*),.*:$NEW_SIZE@.*/\1/p" "$FITIMAGE_MOD_DIR/image.its")"
```
Die Zeile mit den Kernel args in der die größe der ersetzen Partition steh. Da ist es sehr unwahrscheinlich dass es Kollisionen gibt °-°
Dennoch, wie gesagt, noch nicht optimal. Da ich noch nicht so genau weiss was ob bzw was genau du erweitern möchtest warte ich mal ab

Dieses Sortien nach Größen war bei den 2 Geräten bei denen die Links fehlten, der workaround as ist nun weg: https://github.com/Freetz-NG/freetz-ng/commit/e256a10512f3ed962cd0914a7562b13a977cf31a
Diese Geräte hatten auch nur 1 "filesystem"

fda77, 2022-06-19:
Mist, durch die letzte Änderung wird bei der 5590 das falsche Filesystem verlinkt: https://github.com/Freetz-NG/freetz-ng/commit/feaccc394e995aef1a7388f9d46953886dc2c791

PeterPawn, 2022-06-19:
> Sollte man evtl den AVM-"Header" der avm-fit einfach kopieren wie er ist statt zu 00en? Vielleicht ist es mehr als eine Signatur

Das ist doch reines Rätselraten ... die "Signatur" (die ersten vier Bytes) ist praktisch "glasklar", ebenso die nächsten vier Bytes, die (in allen AVM-Images) mit der Größe des (originalen) FDT für das FIT-Format (so, wie das im `u-boot` spezifiziert ist - DESHALB funktioniert ja auch das `mkimage` von dort für den "inneren" Teil) befüllt sind ... und danach bleiben nur noch die 64 Bytes (512 Bits) übrig, die als "Hash" (von mir) interpretiert wurden - was sich mit dem Auftauchen von `ar7-lite` dann ja irgendwie bestätigte und damit auch fast nicht mehr in Zweifel zu ziehen wäre. Danach kommen dann halt die FDT-Inhalte (also die Ausgabe von `mkimage` und ganz hinten dran hängen dann (bei AVM auch) noch 8 Bytes mit Nullen - nach meiner Interpretation ein Relikt, weil da bei anderen Image-Formaten (`kernel.image` und `filesystem.image`) noch die TI-Prüfsumme käme - da diese Bytes keine (erkennbare) Funktion haben, tippe ich (immer noch) auf irgendeine Stelle, wo eben (immer) 8 Bytes zur Dateigröße hinzugezählt wurden bzw. werden bei AVM.

Was genau davon willst Du ändern? Wenn es ein ED25519-basierter Hash ist (genauer wäre ja "ED25519-basierte Signatur"), dann ist es ein SHA(2)-512-Hash, der mit dem privaten Teil des ED25519-Keys verschlüsselt wurde und mit dem öffentlichen Teil (der irgendwo dann auch in `ar7-lite` enthalten sein sollte, wenn das tatsächlich selbst die Signatur prüft) entschlüsselt und mit dem selbst berechneten Hash über die Daten verglichen werden kann. Da GIBT es dann (nach meiner Ansicht) auch nichts mehr, was daran irgendwie "geheimnisvoll" wäre ... wenn bestätigt ist, daß AVM da auf ED25519 setzt (und die Funktionsnamen in `ar7-lite` dürften als Bestätigung ausreichen), dann bleiben da auch nicht wirklich weitere Spielräume für eine andere Interpretation.

---

> Sondern wie in den ~20 Zeilen

Und wieder ohne in den Patch zu schauen ... DAS, was Du da als Code zitierst, funktioniert ja dann bei den Inhouse-Versionen mit den geänderten `kernel-args` auch nicht mehr.

> warte ich mal ab

Was Du ja dennoch schon mal machen kannst, wäre das Ersetzen der Werte nicht nur per `sed` in der `image.its`, sondern auch gleich in der Verzeichnisstruktur - dann mußt Du nur noch auf das neue Skript warten (das WIRD nur die Werte aus den Dateien benutzen, um eine Text-Datei zu generieren - da ändert sich an der Idee garantiert nichts mehr) und wenn es vorhanden ist und funktioniert, anstelle der `sed`-Änderungen in der `image.its` mit diesem Skript eine neue generieren lassen. 

Wo ich noch am Überlegen bin, ist die Frage, ob man in diesem Schritt auf die "einzelnen Images" (als `image_nnn.bin` im "Hauptverzeichnis") zurückgreift (und Freetz-NG damit auch dort die Datei mit dem eigenen Image ersetzen müßte) oder ob man das aus dem `data`-File unterhalb des Nodes generiert (und Freetz-NG DEN dann ersetzen müßte), wo die Daten ja auch noch (eigentlich zum zweiten Mal) liegen. 

Da ich ja immer auch die direkte Verwendung auf der FRITZ!Box im Hinterkopf habe, ist dieser doppelte Platzbedarf nichts, mit dem ich mich abfinden will. Eine Option, mit der die Daten im "Hauptverzeichnis" (`fit-dump`) nicht mehr erzeugt werden sollen (weder die `image.its`, noch die einzelnen `image_nnn.bin`-Files), ist bereits vorgesehen (`-i` bzw. `--no-its`), aber noch nicht (richtig) implementiert (nur in der Hilfe-Anzeige), weil es bisher eben das Skript für den entgegen gesetzten Weg noch nicht gibt.

Aber die Links im "Hauptverzeichnis" (`kernel.image` und `filesystem.image` bzw. `ramdisk.image`) könnten eben auch problemlos auf die jeweiligen `data`-Files im entsprechenden Node verlinken ... daher wäre es in jedem Fall gut, wenn Du Dich beim Ersetzen einer dieser Dateien durch Freetz-NG dann auch am jeweiligen Symlink orientierst - dann MUSS Freetz-NG gar nicht wissen, wohin der nun genau zeigt.

---

> Mist, durch die letzte Änderung wird bei der 5590 das falsche Filesystem verlinkt

Etwas in der Richtung dachte ich mir schon (siehe oben), daher ja meine (wiederholte) Warnung (für Deinen Code) und mein Verlangen nach entsprechenden Tests (durch mich), BEVOR ich das wirklich freigebe. Hier kann (bzw. muß) man dann doch wieder mit der Größe arbeiten (beim Dateisystem sind die Unterschiede auch viel gravierender, so daß die Unsicherheit, das falsche zu erwischen, kleiner ist) - bisher habe ich das (iirc) nur bei mehreren `ramdisk`-Images in einem FIT-Image gemacht (weil da die Erkennung über `kernel-args` nicht machbar war).

fda77, 2022-06-19:
Achso, es war noch nicht freigegeben :) Ich hab aber mehr

> Wenn es ein ED25519-basierter Hash ist 

Weil es halt nicht klar ist. Vielleicht sind das wieder calibrationsdate oder so... wer weiss

> Was Du ja dennoch schon mal machen kannst, wäre das Ersetzen der Werte nicht nur per sed in der image.its, sondern auch gleich in der Verzeichnisstruktur 

Mach ich, bislang hatte ich ohne "-f" entpackt

> Hier kann (bzw. muß) man dann doch wieder mit der Größe arbeiten

Ich glaub mehr was an dem Statement ist nicht korrekt, aus der its
```
        qcaarmv8_HW272_squashFS_filesystem {
            description = "squashFS for qcaarmv8_HW272";
            #address-cells = <0x00000001>;
            avm,data-align = <0x00001000>;
            data = /incbin/("image_003.bin"); // size: 28033024, offset=0x0033f98c
            type = "filesystem";
            compression = "none";
            load = <0x43000000>;
            avm,kernel-args = "mtdram=ram-filesystem,0x43000000,0x44b00000 mtdparts_ext=ram-filesystem:28033024@0x0(rootfs_ram)";
            hash_0 {
                algo = "crc32";
                value = <0xb7830827>;
            };
        };
...
        prxI_HW273_squashFS_filesystem {
            description = "squashFS for prxI_HW273";
            #address-cells = <0x00000001>;
            avm,data-align = <0x00001000>;
            data = /incbin/("image_010.bin"); // size: 7802880, offset=0x02345924
            type = "filesystem";
            compression = "none";
            load = <0x73000000>;
            avm,kernel-args = "mtdram=rootfs_ram,0x33000000,0x33800000";
            hash_0 {
                algo = "crc32";
                value = <0x5bdc525c>;
            };
        };
```

Bei freetz ist das ganze original ein "cache", dann muss nicht nochmal entpackt werden

PeterPawn, 2022-06-19:
> Ich glaub mehr was an dem Statement ist nicht korrekt, aus der its

Also DAS kann ich (mit einiger Gewißheit) ausschließen ... bisher war es halt so, daß das "Frontend"-System mit dem `mtdparts_ext`-Parameter daher kam, aber das zweite Filesystem mit dem `mtdram=rootfs_ram,...` existierte (für einen anderen Prozessor) ebenfalls (bei der 5590 - daher ja auch mein (bisheriger) Weg der "Vorhersage" des richtigen Dateisystems). 

Wenn sich das Format aus den Inhouse-Versionen für 7530AX und 4060 jetzt (oder in einer der nächsten Versionen) auch für die 5590 durchsetzen sollte bei AVM, hat man am Ende (mind.) ZWEI Einträge mit diesem "Schlüssel", nur halt mit unterschiedlichen Lade-Adressen und DAS wäre dann spätestens auch das Aus für `sed`-basierte Änderungen.

Ich habe IMMER NOCH NICHT in Deinen Patch gesehen ... aber wenn ich den restlichen Texten trauen will, dann verzichtest Du auf das Anpassen des `load`-Entrys für das ersetzte Dateisystem? DAS würde ich (wenn es so ist) für einen Fehler halten ... ich vermute mal, der Bootloader wird sich beim Laden eher am Inhalt der `load`-Property orientieren und nicht am Inhalt von `kernel-args` (was er auch selbst erst passend parsen müßte - das ist die Verwendung von `load` einfacher), den er wohl nur 1:1 an den Kernel durchreichen wird. 

Da wäre es dann (für mich jedenfalls) eher verwunderlich, wenn der Kernel das (an eine andere Stelle geladene) Dateisystem finden würde, wenn er an der Stelle sucht, die in `kernel-args` übermittelt wurde.

fda77, 2022-06-19:
Vielleicht solltest du doch mal reinschauen? Es wird das uppler-limit/size geändert, load bleibt also gleich

Nach deinem if-Statement sollte doch zuerst filesystem_indicator_marker1, dann filesystem_indicator_marker2 gesucht werden wenn 1 nicht da ist - der ist aber da! Siehe mein letzte post

fda77, 2022-06-19:
Ach neee... du suchst image 001 bis 010
bei 003 wird `"filesystem_found=1` gesetzt
und ebenso bei 010, das das zuerst gefunden überschreibt

Die Priorität filesystem_indicator_marker1/filesystem_indicator_marker2 gibt es also nicht, sondern das letzte gewinnt

PeterPawn, 2022-06-19:
> Es wird das uppler-limit/size geändert, load bleibt also gleich

Gut, es könnte ja beides funktionieren. Nur stelle ich mir dann halt auch die Frage, wie Du zu DEINER Annahme kommst - daß ich das noch nicht genauer untersucht habe, habe ich ja (am "Ursprungsort") schon "zugegeben". Welche (AVM-)Versionen hast Du denn miteinander verglichen und wo sind dabei dann unterschiedliche Werte aufgetreten, die in Dir die Überzeugung reifen ließen, daß man problemlos auch die OBERE Adresse ändern könne? Hast Du Dir jemals eine Übersicht erstellt, an welche Speicheradressen bei AVM die verschiedenen Images aus so einem FIT-Image geladen werden und wo da - ggf. noch mit entsprechendem Padding - die unterschiedlichen Bereiche aneinander grenzen?

> sondern das letzte image gewinnt

Ich weiß nicht, wie ich das NOCH DEUTLICHER formulieren soll, als mit einem expliziten Kommentar im Commit-Text, daß das UNGETESTET war (an anderer Stelle steht auch, weshalb das so ist): https://github.com/PeterPawn/YourFritz/commit/399b791a6bd999ca05343274e9102dacc37cdb68

Üblicherweise mache ich eben auch Regressionstests und wie ich schon oben geschrieben habe, GIBT es (für `ramdisk`) auch schon den "das Größere gewinnt"-Check: https://github.com/PeterPawn/YourFritz/blob/399b791a6bd999ca05343274e9102dacc37cdb68/fit_tools/fitdump.sh#L338 ... nur war das bisher bei `filesystem` eben nicht erforderlich, WEIL da die Erkennung durch `mtdparts_ext` schon zum gewünschten Ergebnis führte.

Liest Du manchmal auch das, was ich schreibe? Mal abgesehen davon, daß Du ja - wenn Du die letzten Änderungen jetzt schon testest, OBWOHL sie eben noch nicht von mir getestet und "freigegeben" wurden - dann offenbar den Branch `fit-tools` verwendest? Macht es wirklich Sinn, mir Fehler (und -Meldungen) über zu helfen für Versionen, die ich noch gar nicht selbst getestet habe? Ich habe darauf auch eine Antwort: Es macht KEINEN Sinn.

Wenn Du "ungeduldig" bist und das selbst machen willst ... nur zu. Passende (und korrekte, was die richtige Behandlung ALLER bisher von AVM veröffentlichten Images einschließt) Änderungsvorschläge nehme ich auch gerne an ... aber ich weiß jetzt seit Donnerstag abend, kurz vor 21 Uhr, daß Du diese Implementierung (a) überhaupt verwenden willst (obwohl es sie auch schon seit mehr als drei Monaten gibt) und (b) welche "Bedürfnisse" es in Freetz-NG (vor und nach der Formatänderung bei AVM) gibt, die bisher noch nicht von den Skripten in diesem (Unter-)Projekt abgedeckt werden. Muß ich da tatsächlich jetzt (Sonntag mittag) schon eine "Lösung" (die dann möglichst auch noch durchdacht sein soll und bestenfalls auch noch funktioniert) parat haben?

Es GAB nun mal bisher für mich keine Notwendigkeit, den umgekehrten Weg ebenfalls zu unterstützen (auch wenn ich das für die Zukunft durchaus angedacht hatte) ... und wenn ich das jetzt in Angriff nehmen will, dann auch so, daß es meinen eigenen Ansprüchen an Flexibilität und Allgemeingültigkeit gerecht wird - ansonsten mußt Du das eben (s.o.) selbst machen. 

Und zu einer (für andere auch nützlichen(!)) Lösung gehört es dann auch, daß man die selbst entsprechend dokumentiert (und sei es in einem Commit-Text, auf den man verlinken kann) ... gäbe es den (und die Info, wo er zu finden ist) für die Frage hinsichtlich der "Flexibilität" der `load`-Property (die obere Grenze kann man eben nur dann (uneingeschränkt) verschieben, wenn dahinter keine anderen Daten gespeichert sind und dazu bräuchte es mindestens mal eine "map" für das Ergebnis des Ladens so eines FIT-Images), wären meine Zweifel, welche Richtung nun korrekt sein möge und vor allem, wieviel Platz man in jeder Richtung noch nutzen könnte, ohne sich mit anderen Daten ins Gehege zu kommen, bereits ausgeräumt.

fda77, 2022-06-19:
> I've to get at least one of the new original images first"

klingt halt etwas abenteuerlich nachdem ich vorab schon alles verlinkt hatte. Davon abgesehen funktionieren die neuen ja. Nur 1 altes Gerät passt nicht mehr. Ich bin nicht sicher ob du das so schnell gefunden hättest

> Liest Du manchmal auch das, was ich schreibe? 

Bis hierhin bin ich jedenfalls gekommen. War das ein Test???

> Wenn Du "ungeduldig" bis

Ja

PeterPawn, 2022-06-19:
> klingt halt etwas abenteuerlich nachdem ich vorab schon alles verlinkt hatte.

Ach weißt Du ... schau Dir einfach noch einmal die Zeitstempel meiner drei "Veröffentlichungen" von heute morgen **genau** an (1x Commit, 2x Kommentare in verschiedenen Threads) und überdenke, was ich meinerseits zu Deinem "inhaus"-Link geschrieben habe und zur "Erkennbarkeit" als Quelle für eine URL bei AVM.

>  Ich bin nicht sicher ob du das so schnell gefunden hättest

Und auch hier dachte ich, daß ich schon VOR Deiner Feststellung, daß es nun für die 5590 nicht mehr funktionieren würde, auf das (erwartbare) Problem hingewiesen hatte. Ich WUSSTE das also tatsächlich schon vorher und hätte mich nur gerne - und zwar auch BEVOR ich dann für `filesystem` ebenfalls den bei `ramdisk` bereits (länger) realisierten Größen-Test aktiviere - durch einen Test davon überzeugt, was DIESE Änderung für die bisher verfügbaren Images GENAU bewirkt.

Dein "Fund" steht in diesem Thread mit einer "creation time" von 11:45 Uhr (heute) und meine "Warnung", daß die 5590 Probleme machen WIRD, wurde bereits um 11:17 Uhr geschrieben (https://github.com/Freetz-NG/freetz-ng/issues/570#issuecomment-1159664641) bzw. die Notwendigkeit, das GERADE bei der 5590 erst mal richtig zu testen, stand schon seit 10:15 Uhr am Ende dieses Beitrags: https://github.com/Freetz-NG/freetz-ng/issues/570#issuecomment-1159651418

Vielleicht sind es ja "Bemerkungen" wie diese von Dir, die mich zur Frage, ob Du das auch wirklich liest, verleitet haben? Aber ich will das eigentlich gar nicht weiter "ausdiskutieren", weil die dafür verbrauchte Zeit im Moment vielleicht besser an anderer Stelle eingesetzt werden könnte.

freetz-ng-mod, 2022-06-19:
Nur so mal zur Information, dass man nun überhaupt nichts mehr bauen kann bei der 5590 
```
umask 0022
git clone https://github.com/Freetz-NG/freetz-ng test-ng
Klone nach 'test-ng' …
remote: Enumerating objects: 197655, done.
remote: Counting objects: 100% (1690/1690), done.
remote: Compressing objects: 100% (787/787), done.
remote: Total 197655 (delta 984), reused 1478 (delta 868), pack-reused 195965
Empfange Objekte: 100% (197655/197655), 72.13 MiB | 4.85 MiB/s, fertig.
Löse Unterschiede auf: 100% (136066/136066), fertig.
rocco@xmg-ultra:~/Ubuntu$ cd test-ng
rocco@xmg-ultra:~/Ubuntu/test-ng$ make menuconfig
Freetz-NG 19818-feaccc394 master 2022-06-19
..................................................................................................
---> tools/kconfig-host ... 
preparing ... mkdir -p /home/rocco/Ubuntu/test-ng/source/host-tools/kconfig-v5.18; tar -C /home/rocco/Ubuntu/test-ng/source/host-tools -xf dl/kconfig-v5.18.tar.xz 
set -e; shopt -s nullglob; for i in tools/make/kconfig-host/patches/*.patch*; do case $i in *.patch|*.patch.gz|*.patch.bzip2|*.patch.bz2|*.patch.bz|*.patch.xz|*.patch.lz|*.patch.lzma|*.patch.Z|*.patch.diff) ;; *) continue ;; esac; tools/freetz_patch /home/rocco/Ubuntu/test-ng/source/host-tools/kconfig-v5.18 $i; done; 
    applying patch file tools/make/kconfig-host/patches/300-main_makefile.patch
    patching file Makefile
    ----------------------------------------------------------------------
    applying patch file tools/make/kconfig-host/patches/310-fix_make.patch
    patching file scripts/kconfig/Makefile
    ----------------------------------------------------------------------
    applying patch file tools/make/kconfig-host/patches/320-unchangable_style.patch
    patching file scripts/kconfig/mconf.c
    ----------------------------------------------------------------------
    applying patch file tools/make/kconfig-host/patches/330-comment_style.patch
    patching file scripts/kconfig/mconf.c
    ----------------------------------------------------------------------
    applying patch file tools/make/kconfig-host/patches/340-fail_on_unmet_direct_dependency.patch
    patching file scripts/kconfig/symbol.c
    ----------------------------------------------------------------------
    applying patch file tools/make/kconfig-host/patches/350-max_item_count_for_choice_dialog.patch
    patching file scripts/kconfig/mconf.c
    ----------------------------------------------------------------------
    applying patch file tools/make/kconfig-host/patches/360-save-hidden-prompts-to-file.patch
    patching file scripts/kconfig/menu.c
    ----------------------------------------------------------------------
    applying patch file tools/make/kconfig-host/patches/370-multiline_comments_hack.patch
    patching file scripts/kconfig/mconf.c
    patching file scripts/kconfig/menu.c
    ----------------------------------------------------------------------
    applying patch file tools/make/kconfig-host/patches/380-project_name.patch
    patching file scripts/kconfig/mconf.c
    ----------------------------------------------------------------------
    applying patch file tools/make/kconfig-host/patches/390-menu_help.patch
    patching file scripts/kconfig/mconf.c
    ----------------------------------------------------------------------
    applying patch file tools/make/kconfig-host/patches/391-no_symbol_prefix.patch
    patching file scripts/kconfig/lkc.h
    ----------------------------------------------------------------------
    applying patch file tools/make/kconfig-host/patches/392-use_explicit_flex_call.patch
    patching file scripts/kconfig/Makefile
    ----------------------------------------------------------------------
    applying patch file tools/make/kconfig-host/patches/393-no_POSIX_warning_from_bison.patch
    patching file scripts/kconfig/Makefile
    ----------------------------------------------------------------------
    applying patch file tools/make/kconfig-host/patches/400-do_not_create_autoconf_stuff.patch
    patching file scripts/kconfig/confdata.c
    ----------------------------------------------------------------------
cmd() { make -j --quiet "$@"  || { [ "" = "0" ] && echo && cat .build.log 2>/dev/null; kill $$ 2>/dev/null || kill $$$$ 2>/dev/null; printf "\n\\033[33m%s\\033[m\n" "ERROR: ";  exit 1; } }; 			mkdir -p source; [ -n "" ] && step="/" || step=""; [ -n "" ] && step="$step/"; case "" in BIN)	echo -n "package/"         >source/.echo_item_tmp ;; LIB)	echo -n "library/"         >source/.echo_item_tmp ;; HTL)	echo -n "tools/"           >source/.echo_item_tmp ;; KTC)	echo -n "toolchain/kernel$step" >source/.echo_item_tmp ;; TTC)	echo -n "toolchain/target$step" >source/.echo_item_tmp ;; KRN)	echo -n "kernel"                 >source/.echo_item_tmp ;; esac; if ! diff -q source/.echo_item_tmp source/.echo_item_new >/dev/null 2>&1 || [ ! -e source/.echo_item_1st ]; then 	if [ -e source/.echo_item_end -a -e source/.echo_item_new -a -e source/.echo_item_1st ]; then echo -e "\e[48;5;26mdone\e[49m."; rm -f source/.echo_item_end source/.echo_item_1st; fi; [ -s source/.echo_item_tmp ] && cat source/.echo_item_tmp > source/.echo_item_new 2>/dev/null; [ -s source/.echo_item_new ] || cat source/.echo_item_old > source/.echo_item_new 2>/dev/null; if [ -s "source/.echo_item_new" ]; then echo -ne "\e[48;5;90m---> "; cat source/.echo_item_new 2>/dev/null | tee source/.echo_item_old; echo -ne "\e[49m ... "; [ "" != "0" ] && echo; touch source/.echo_i [... cut ...]
  HOSTLD  scripts/kconfig/mconf
mkdir -p tools/kconfig/; cp /home/rocco/Ubuntu/test-ng/source/host-tools/kconfig-v5.18/scripts/kconfig/mconf tools/kconfig/mconf;
done.

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.

rocco@xmg-ultra:~/Ubuntu/test-ng$ make -j20
Freetz-NG 19818-feaccc394 master 2022-06-19
make/libs/libatomic/libatomic.mk:1: *** Undefined version for PKG_INIT in make/libs/libatomic/libatomic.mk.  Schluss.
make: *** [Makefile:46: envira] Beendet

```

fda77, 2022-06-19:
Mir ist es momentan eh zu warm für irgend was
@freetz-ng-mod: Einfach die vorletzte revision nehmen

Himan2001, 2022-06-19:
Ich glaube wir sollten wieder "Milestones" einführen die man konkret "namentlich" ausschecken kann.
- Dann ist der trunk wie bei OpenWRT dauerhaft work in progress :-)

fda77, 2022-06-19:
Gibt es, nennt sich tag: https://github.com/Freetz-NG/freetz-ng#or-clone-a-single-tag

`fdtget `ist ganz nett. Vielleicht nimmt man fürs setzen der` kernel-args` einfach `fdtput`. Scheinbar brauch man die hashes nicht zu aktualieren, da ds bei `mkimage` automatisch geht
```
for z in *; do echo -e "\n ## $z"; for x in $(fdtget $z /images -l); do y="$(fdtget $z /images/$x type)"; [ "$y" == ramdisk -o "$y" == filesystem ] && echo -e "$y\t$x" ; done ; done

 ## FRITZ.Box_4060-07.30.itb
filesystem      qcaarmv8_HW261_squashFS_filesystem

 ## FRITZ.Box_4060-07.39-97200-Inhaus.itb
filesystem      qcaarmv8_HW261_squashFS_filesystem

 ## FRITZ.Box_5530-07.29.itb
ramdisk         prxI_HW257_ramdisk
ramdisk         prxB_HW0257_ramdisk

 ## FRITZ.Box_5590-07.29.itb
filesystem      qcaarmv8_HW272_squashFS_filesystem
ramdisk         prxB_HW0273_ramdisk
filesystem      prxI_HW273_squashFS_filesystem

 ## FRITZ.Box_7510-07.30.itb
filesystem      maple_HW271_squashFS_filesystem

 ## FRITZ.Box_7530_AX-07.31.itb
filesystem      brcma9_HW256_squashFS_filesystem

 ## FRITZ.Box_7530_AX-07.39-96959-Inhaus.itb
filesystem      brcma9_HW256_squashFS_filesystem

 ## FRITZ.Repeater_1200_AX-07.30.itb
filesystem      maple_HW268_squashFS_filesystem

 ## FRITZ.Repeater_6000-07.29.itb
filesystem      qcaarmv8_HW253_squashFS_filesystem
```

fda77, 2022-06-20:
Letzte commits beachten! Die uboot & dtc tools laufen recht flott, leider weigert sich mkimage ohne openssl zu compilieren und es gibt Problem je ob Verion 1.1 oder 3 installiert ist. Das menuconfig von uboot hat unmengen dependency-Probleme. Ich hätte nicht gedacht dass das in so schlechtem Zustand ist

PeterPawn, 2022-06-20:
fdtget/fdtput

Wenn man damit tatsächlich all das abbilden könnte, was man beim Entpacken, Ändern und Neupacken von AVM-Images im FIT-Format braucht, wäre ich sicherlich auch darauf angesprungen (immerhin gehört das `fdtdump` als Basis für das `fitdump` (als C-Version) auch zu diesen Tools, die ich mir halt vorher genau angesehen habe). 

Aber diese beiden Programme arbeiten eben nur noch mit dem binären Format des FDT (und kennen ihrerseits auch keine Optionen für einen Offset oder zum Scannen) - aber das Entfernen des AVM-Headers mit `fit-remove-avm-header.sh` gibt es am Ende auch nur, WEIL die Verwendung dieser Tools eine denkbare Option wäre, was man auch schon im entsprechenden Kommentar nachlesen kann: https://github.com/PeterPawn/YourFritz/blob/836ff09fc1d51a47f39a12be61ef2a37d8ac5d4f/fit_tools/fit-remove-avm-header.sh#L34 - für etwas anderes wird das (zumindest bei mir) gar nicht genutzt, da das "komplette" Zerlegen des FIT-Images, inkl. des "Erratens" des richtigen(!) Images für das Dateisystem, ja in `fitdump.sh` erfolgt (aber sogar da habe ich eine Option eingebaut, mit der man auch nur noch das "native" FIT-Image ohne den AVM-Header verarbeiten lassen kann).

Für mich ist das (als endgültige Lösung, die auch auf der Box selbst arbeitsfähig sein soll) aber auch wieder keine Option ... ich will/werde nicht zusätzlich noch weitere (binäre) Dateien für irgendein Zielgerät ins Spiel bringen (u.a. auch, WEIL ich mir bewußt war, daß die "vanilla sources" für `u-boot` einiges an Vorbereitungen/Einstellungen brauchen und ich wüßte jetzt aus dem Stand keine Distribution, die NICHT auch ein fertiges Paket für `mkimage` bereithalten würde, wobei die Verfügbarkeit der anderen "FIT tools" durchaus - je nach Distro - schwankt) ... sogar das `mkimage` ist mir am Ende zuviel und wenn's möglich ist, will ich sogar das am Ende durch Shell-Code ersetzt haben (solange die Performance einigermaßen akzeptabel ist), weil das - im gesamten Konstrukt, wenn man mal von "squashfs-tools" absieht - das einzige Binärprogramm wäre und bei genauerer Betrachtung sogar das Zusammensetzen eines neuen Images relativ simpel bleibt, weil man die ganzen Optionen des `mkimage`, die nur bei der Benutzung OHNE (umfassendes) `its`-File eine Rolle spielen, gar nicht benötigt.

Aber viele Wege führen nach Rom - ich hatte(!) eben diese beiden Tools auch mal auf dem Schirm und habe mich aber dagegen entschieden, sie für meine kleine "FIT toolchain" zu verwenden. 

Wobei für die "Analyse" eines kompletten Images dann fast wieder `fdtgrep` interessanter sein dürfte (wenn Du weiter experimentieren willst), damit hat man noch deutlich mehr Optionen als mit `fdtget` und auch eine flexiblere Syntax, wenn es um Alternativen bei den "Treffern" geht.

---

Daß das `mkimage` (bzw. darin verwendeter Code) einiges selbst erledigt und was das genau ist, findet man dann auch in dessen Quelltext (https://github.com/u-boot/u-boot/blob/700cff2328333ad81ef7e0e303edb336224614b4/tools/fit_image.c#L664) ... hinsichtlich des "Zeitstempels" hatte ich das ja auch schon mal irgendwo angetextet, weil sich deshalb keine 1:1-Kopie der Ausgangsdaten ergibt, wenn man aus einer its-Datei ein neues (FDT-/FIT-)Image erzeugt. Daher (also wg. der oben verlinkten "Quelle") habe ich (wenn ich mich richtig erinnere) auch nichts davon geschrieben, daß man die CRC32-Werte SELBST aktualisieren müsse (auch nicht unter dem Commit, aus dem das hier hervorging).

fda77, 2022-06-20:
fdtgrep ist mir ganz entgangen, ich hab mir hier ein paar Sachen zum anzeigen und vergleichen von its hinterlegt: https://github.com/Freetz-NG/freetz-ng/blob/master/tools/fit.sh#L82=

Zu den Checksummen hat niemand was geschrieben! Ausser dass ich fetsgestellt hab dass man die doch nicht manuell anpassen muss! Der zeitstempel ist mit EPOCH kein Problem: https://github.com/Freetz-NG/freetz-ng/blob/master/fwmod#L2136

Der Vorteil von diesen Tool ist vor allem dass sie auf dem Computer schlicht viel schneller laufen. Für das fitdump-Script hatte ich einen pipe durch `pv`  vorgesehen damit man erkennt ob sich noch was tud :)

Im menuconfig von uboot kann man ssl zwar deaktivieren, danach compiliert es aber nicht -.-
Dynamisch gelinkt gibt es Probleme ob openssl 1.1 oder 3 installiert ist, dann läuft die Binary nämlich nicht ^^

Kleiner Helfer zum analysieren aller dtb in einem Verzeichnis:
```
$ ~/freetz-ng/tools/fit.sh s .

 ## ./FRITZ.Box_4060-07.28-92702-Inhaus.dtb
filesystem   qcaarmv8_HW261_squashFS_filesystem
kernel       qcaarmv8_HW261_kernel
kernel-args  mtdram=ram-filesystem,0x43000000,0x44a00000 mtdparts_ext=ram-filesystem:26976256@0x0(rootfs_ram)

 ## ./FRITZ.Box_4060-07.30.dtb
filesystem   qcaarmv8_HW261_squashFS_filesystem
kernel       qcaarmv8_HW261_kernel
kernel-args  mtdram=ram-filesystem,0x43000000,0x44a00000 mtdparts_ext=ram-filesystem:26955776@0x0(rootfs_ram)

 ## ./FRITZ.Box_4060-07.39-97259-Inhaus.dtb
filesystem   qcaarmv8_HW261_squashFS_filesystem
kernel       qcaarmv8_HW261_kernel
kernel-args  mtdram=rootfs_ram,0x43000000,0x44d00000

 ## ./FRITZ.Box_5530-07.21.dtb
filesystem   prxI_HW257_ramdisk
kernel       prxI_HW257_kernel

 ## ./FRITZ.Box_5530-07.29.dtb
filesystem   prxI_HW257_ramdisk
kernel       prxI_HW257_kernel

 ## ./FRITZ.Box_5590-07.27-94541-Inhaus.dtb
filesystem   qcaarmv8_HW272_squashFS_filesystem
kernel       qcaarmv8_HW272_kernel
kernel-args  mtdram=ram-filesystem,0x43000000,0x44b00000 mtdparts_ext=ram-filesystem:28053504@0x0(rootfs_ram)

 ## ./FRITZ.Box_5590-07.29.dtb
filesystem   qcaarmv8_HW272_squashFS_filesystem
kernel       qcaarmv8_HW272_kernel
kernel-args  mtdram=ram-filesystem,0x43000000,0x44b00000 mtdparts_ext=ram-filesystem:28033024@0x0(rootfs_ram)

 ## ./FRITZ.Box_7510-07.30.dtb
filesystem   maple_HW271_squashFS_filesystem
kernel       maple_HW271_kernel
kernel-args  mtdram=ram-filesystem,0x43000000,0x44c00000 mtdparts_ext=ram-filesystem:28626944@0x0(rootfs_ram)

 ## ./FRITZ.Box_7530_AX-07.22.dtb
filesystem   brcma9_HW256_squashFS_filesystem
kernel       brcma9_HW256_kernel
kernel-args  mtdram=ram-filesystem,0x19c00000,0x1bb00000 mtdparts_ext=ram-filesystem:31862784@0x0(rootfs_ram)

 ## ./FRITZ.Box_7530_AX-07.31.dtb
filesystem   brcma9_HW256_squashFS_filesystem
kernel       brcma9_HW256_kernel
kernel-args  mtdram=ram-filesystem,0x19c00000,0x1bb00000 mtdparts_ext=ram-filesystem:32489472@0x0(rootfs_ram)

 ## ./FRITZ.Box_7530_AX-07.39-96959-Inhaus.dtb
filesystem   brcma9_HW256_squashFS_filesystem
kernel       brcma9_HW256_kernel
kernel-args  mtdram=rootfs_ram,0x19c00000,0x1c000000

 ## ./FRITZ.Repeater_1200_AX-07.30.dtb
filesystem   maple_HW268_squashFS_filesystem
kernel       maple_HW268_kernel
kernel-args  mtdram=ram-filesystem,0x43000000,0x44200000 mtdparts_ext=ram-filesystem:17981440@0x0(rootfs_ram)

 ## ./FRITZ.Repeater_6000-07.29.dtb
filesystem   qcaarmv8_HW253_squashFS_filesystem
kernel       qcaarmv8_HW253_kernel
kernel-args  mtdram=ram-filesystem,0x43000000,0x44000000 mtdparts_ext=ram-filesystem:16609280@0x0(rootfs_ram)
```

Ich hab jetzt diese Bedingung zur Erkennung der richtigen Partition genommen (die wird im scan-Aufruf vorher auch angewendet): https://github.com/Freetz-NG/freetz-ng/blob/master/tools/fit.sh#L13= bis auf 5590 brauchte es keine Ausnahme
```
		[ "${PART/TZ_HW/}" != "$PART" ] && continue # TrustZone
		[ "${PART/_HW0/}"  != "$PART" ] && continue # other CPU
		[ "${PART/HW273/}" != "$PART" ] && continue # also 5590
```

PeterPawn, 2022-06-20:
> Der Vorteil von diesen Tool ist vor allem dass sie auf dem Computer schlicht viel schneller laufen. 

Wenn's Dir nur darum geht und Du ohnehin nicht im Text-Format in der `its`-Datei ändern willst, kannst Du ja auch GLEICH die C-Version von `fitdump` nutzen (die war ja die "Vorlage" für das, was ich danach in Shell umgesetzt habe, damit das auch ohne zusätzliche Binär-Dateien arbeiten kann): https://github.com/PeterPawn/dtc

Das Zeug DAZU hast Du ohnehin schon in der Toolchain (weil z.B. die `libfdt` für das `avm_kernel_config`-Zeug daraus kommt) ... nur sucht Dir dieses Programm eben auch nicht automatisch das passende Image für Kernel und Dateisystem in einem AVM-Image ... und daß da durchaus mehrere existieren können, siehst Du spätestens bei der 5530 und der 5590. 

Wie weit da Deine Bedingungen dann für die Auswahl des richtigen Systems funktionieren, hängt für meinen Geschmack viel zu sehr davon ab, daß AVM an den Namen (die auch in einem FIT-Image nur Schall und Rauch sind - die kann man JEDERZEIT auch ändern) nichts weiter dreht ... das ist für mich genauso "sicher", wie das Abzählen der Images, weil der Kernel für das "Hauptsystem" bei Boxen mit Trust-Zone vielleicht (bisher) immer der zweite war.

> Du hast nicht rein zufällig was herumliegen?

Nein ... brauche ich nicht. Für `mkimage` gibt es in meiner Distribution ein passendes Paket und das andere Zeug (`fdt...irgendwas`) hat ja mit `u-boot` gar nichts zu tun, das gehört zum `dtc`-Projekt (device tree compiler). Ich habe das `mkimage` auch nur verwendet/angesprochen, weil es eben ein "Standard-Tool" ist (wo man selbst erst mal nichts machen muß(te), um ein neues Image zu erstellen) und ich damit die korrekte Funktion meiner "Zerlegungen" dahingehend testen konnte, daß damit auch wieder ein korrektes Image erstellt werden kann.

Warum nimmst Du nicht einfach auch das `mkimage` (wenn Du's verwenden willst) aus der jeweiligen Distribution? Warum muß man das immer alles selbst übersetzen wollen? Es ist doch gar nicht erforderlich ... das Programm läuft (bei Freetz(-NG)) nur auf dem Build-Host und anstatt jetzt irgendwelche Konfigurationsorgien zu veranstalten, um aus den `u-boot`-Quellen ein EIGENES Binary zu erstellen, kann man ohne weiteres auch das aus der eigenen Distro nehmen ... deren Maintainer werden sich dann schon mit diesen Fragen/Einstellungen herumgeschlagen haben und ob das Zeug nun OpenSSL 1.1 oder 3.0 will, gehört zu den Abhängigkeiten des jeweiligen Pakets.

---

Ich habe jedenfalls bei meinen bisherigen Versuchen, aus der Verzeichnisstruktur eine `its`-Datei zu erzeugen, sogar festgestellt, daß es vermutlich einfacher (oder zumindest nur "genauso schwer") ist, aus der Struktur gleich wieder ein fertiges Image zu machen, anstatt erst noch den Weg über die `its`-Datei und `mkimage` zu gehen ... zumindest für meine Bestrebungen, daß auch AUF einer FRITZ!Box machen zu können, ist das wohl sogar der bessere Weg. Ob man dann das Erzeugen einer `its`-Datei quasi als "Sahnehäubchen" überhaupt noch braucht, kann und will ich jetzt noch nicht abschließend beurteilen ... auf alle Fälle wird es wohl doch eher ein neues Skript werden, was gleich den passenden FDT ausgibt (die AVM-Verpackung kann man dann immer noch dranpappen) und die `its`-Datei (die man ja mit derselben Rekursion über die Struktur zusammensuchen muß) wird nur "optional" sein.

Himan2001, 2022-06-20:
Bei debian und ubutnu gabs jahrelang z.b. "openssl-compat" in der art. das war die sache das man 0.9.8 immer zeitgleich die libs für 1.x und 0.9.x auch drauf hat. nur bei dee headern muss dann halt wählen. Auch auf Redhat & Co. muesste es ein "compat-openssl" paket noch geben" - zumindest gabs das damals immer auf rpm-basierten distros...

fda77, 2022-06-20:
@Himan2001 Das müsste man dann auch wieder installieren wenn man so eine Binary verwenden möchte. Ich schaue mal ob es sowas für Fedora und Openssl 1.1 gibt..
Man müsste das blöde uboot nur statisch oder ohne ssl bauen können

@PeterPawn 

> (die auch in einem FIT-Image nur Schall und Rauch sind - die kann man JEDERZEIT auch ändern)

Sehe ich auch so. Da man aber nicht weiss was kommt ist das so für den moment für mich okay

> Warum nimmst Du nicht einfach auch das mkimage

Halt das alte Thema, dann kommt wieder jemand mit einer uralten (Debian..) Distribution wo das Ding zu alt ist...
Ob die erfoderlichen libs zum compilieren besser sind ist ne andere Frage - ich weiss nicht so recht was am besten wäre.
Ich überlege auch ob autoconf etc  host-tools werden sollten. Für die uralten Compiler der ersten fritzboxen gibt es immer mehr Fehlermeldungen, vor kruzem hatte ich in dem Zusammenhang auch einen Fehler gehabt (und ignoriert+vergessen was es genau war)

fda77, 2022-06-20:
Wie gesagt, ich weiss nicht genau was das mit fit werden soll. Ich hab mir das erst vor ~2 Tagen angesehen. Ich hab das fitimg als blackbox betrachet - solang es geht ists gut.
Jedenfalls hat hippie angekündig (nicht persönlich sonder pseudo-allgemein auf seiner lahmen webseite) sein perl-scripte nur noch non-public weiterzumachen. Und da in diesem für jedes neuer Gerät (vermutlich auch die 2 neuen inhaus) Änderungen nötig sind und ich keine Lust auf diesen Perl-Unsinn hab musste was anderes her.
So wie es jetzt ist funktioniert es jedenfalls schonmal auf der 5530+5590, bei den anderen 5 fit geht ich auch davon aus

> irgendetwas unbedingt selbst "nachprogrammieren" wolltest

will ich nicht! Nur wenn man andere Script nutzt und dann noch ganz viel drumherum machen muss ist die Frage was mehr aufeand ist. Was mich wirklich an dem fitdump-Script gestört hat ist die Laufzeit. Das einpacken der 5590 07.20 dauert bei mir 20 sekunden. Mit den binaries die fit.sh nutzt sind es noch 3,5. Ein komplettes minimales image nach distclean bauen braucht bei mir insgesammt ca 2 minuten, da fällt das ins gwicht
Nichts gegen das Script, das ist ja um auf der Box genutzt zu werden ursprünglich, für die bootmanager-erkennung

> Mit dem passenden Parameter für das originale fdtdump (das kennt nämlich sogar einen "scan"-Modus), müßtest Du sogar das Extrahieren des "originalen" FIT-Images aus der AVM-Verpackung gar nicht mehr machen lassen.

Interessant, schaue ich mal

 ---

Mit den alten Geräten ist das so ne Sache. Wenn ich binaries oder webif test nehm ich dafür ne 7390 :D 
Diese System mit der 5530+5590 hat zb ein komisches verhalten: Das ubuntu 18 produziert mit den dl-toolchains "illegal instruction" fehler - keine Ahnung was das verursacht

>  und man sollte weit vor dem Mittelalter dann einen Schlußstrich ziehen.

Nur wo?
Eigentlich müsste meine neueste, fast 10 jahre alte 7490 dann auch raus...
Und die 6590 die noch Kernel 2.6 hat
Wenn irgend was mit altem nicht geht frier ich einfach die Version ein und nur noch neuer Geräte bekommen was neues, das muss sich dann jeder selbst überlegen

freetz-ng-mod, 2022-06-20:
@fda77 
Dir ist aber schon klar das man wieder keine Images bauen kann für die 5590.
```
STEP 1: UNPACK
source firmware file: dl/fw/FRITZ.Box_5590-07.29.image
checking signature: valid
unpacking firmware image
dumping fit-image file
/home/rocco/Ubuntu/freetz-ng-test/tools/fdtget: error while loading shared libraries: libfdt.so.1: cannot open shared object file: No such file or directory
/home/rocco/Ubuntu/freetz-ng-test/tools/fdtget: error while loading shared libraries: libfdt.so.1: cannot open shared object file: No such file or directory
/home/rocco/Ubuntu/freetz-ng-test/tools/fit.sh: Zeile 31: [: -le: Einstelliger (unärer) Operator erwartet.
mv: Aufruf von stat für 'build/original/fit-image/image.0' nicht möglich: Datei oder Verzeichnis nicht gefunden
/home/rocco/Ubuntu/freetz-ng-test/tools/fit.sh: Zeile 36: [: -le: Einstelliger (unärer) Operator erwartet.
ln: auf 'build/original/fit-image/filesystem.image' konnte nicht zugegriffen werden: Datei oder Verzeichnis nicht gefunden
ERROR: cannot find kernel.image (build/original/firmware/var/tmp/kernel.image)
make[1]: *** [Makefile:376: firmware-nocompile] Fehler 1
make: *** [Makefile:46: envira] Beendet
```
Daher wer es doch sinnvoller, wenn Du das erst alles mal testest oder ein Testbereich machst, wo andere Leute dann (die Lust haben) Deine Änderungen testen

fda77, 2022-06-20:
Klappt wunderbar. pebkac?
Die Sache mit den Logs posten ist nicht so deine sache? Das wichtigste fehlt wieder, wie so oft.
Ansonsten: https://github.com/Freetz-NG/freetz-ng#or-clone-a-single-tag

freetz-ng-mod, 2022-06-20:
```
shes/rmd128.o src/hashes/rmd160.o src/hashes/rmd256.o src/hashes/rmd320.o src/hashes/sha1.o src/hashes/sha2/sha224.o src/hashes/sha2/sha256.o src/hashes/sha2/sha384.o src/hashes/sha2/sha512.o src/hashes/sha2/sha512_224.o src/hashes/sha2/sha512_256.o src/hashes/sha3.o src/hashes/sha3_test.o src/hashes/tiger.o src/hashes/whirl/whirl.o src/mac/blake2/blake2bmac.o src/mac/blake2/blake2bmac_file.o src/mac/blake2/blake2bmac_memory.o src/mac/blake2/blake2bmac_memory_multi.o src/mac/blake2/blake2bmac_test.o src/mac/blake2/blake2smac.o src/mac/blake2/blake2smac_file.o src/mac/blake2/blake2smac_memory.o src/mac/blake2/blake2smac_memory_multi.o src/mac/blake2/blake2smac_test.o src/mac/f9/f9_done.o src/mac/f9/f9_file.o src/mac/f9/f9_init.o src/mac/f9/f9_memory.o src/mac/f9/f9_memory_multi.o src/mac/f9/f9_process.o src/mac/f9/f9_test.o src/mac/hmac/hmac_done.o src/mac/hmac/hmac_file.o src/mac/hmac/hmac_init.o src/mac/hmac/hmac_memory.o src/mac/hmac/hmac_memory_multi.o src/mac/hmac/hmac_process.o src/mac/hmac/hmac_test.o src/mac/omac/omac_done.o src/mac/omac/omac_file.o src/mac/omac/omac_init.o src/mac/omac/omac_memory.o src/mac/omac/omac_memory_multi.o src/mac/omac/omac_process.o src/mac/omac/omac_test.o src/mac/pelican/pelican.o src/mac/pelican/pelican_memory.o src/mac/pelican/pelican_test.o src/mac/pmac/pmac_done.o src/mac/pmac/pmac_file.o src/mac/pmac/pmac_init.o src/mac/pmac/pmac_memory.o src/mac/pmac/pmac_memory_multi.o src/mac/pmac/pmac_ntz.o src/mac/pmac/pmac_process.o src/mac/pmac/pmac_shift_xor.o src/mac/pmac/pmac_test.o src/mac/poly1305/poly1305.o src/mac/poly1305/poly1305_file.o src/mac/poly1305/poly1305_memory.o src/mac/poly1305/poly1305_memory_multi.o src/mac/poly1305/poly1305_test.o src/mac/xcbc/xcbc_done.o src/mac/xcbc/xcbc_file.o src/mac/xcbc/xcbc_init.o src/mac/xcbc/xcbc_memory.o src/mac/xcbc/xcbc_memory_multi.o src/mac/xcbc/xcbc_process.o src/mac/xcbc/xcbc_test.o src/math/fp/ltc_ecc_fp_mulmod.o src/math/gmp_desc.o src/math/ltm_desc.o src/math/multi.o src/math/radix_to_bin.o src/math/rand_bn.o src/math/rand_prime.o src/math/tfm_desc.o src/misc/adler32.o src/misc/base64/base64_decode.o src/misc/base64/base64_encode.o src/misc/burn_stack.o src/misc/compare_testvector.o src/misc/crc32.o src/misc/crypt/crypt.o src/misc/crypt/crypt_argchk.o src/misc/crypt/crypt_cipher_descriptor.o src/misc/crypt/crypt_cipher_is_valid.o src/misc/crypt/crypt_constants.o src/misc/crypt/crypt_find_cipher.o src/misc/crypt/crypt_find_cipher_any.o src/misc/crypt/crypt_find_cipher_id.o src/misc/crypt/crypt_find_hash.o src/misc/crypt/crypt_find_hash_any.o src/misc/crypt/crypt_find_hash_id.o src/misc/crypt/crypt_find_hash_oid.o src/misc/crypt/crypt_find_prng.o src/misc/crypt/crypt_fsa.o src/misc/crypt/crypt_hash_descriptor.o src/misc/crypt/crypt_hash_is_valid.o src/misc/crypt/crypt_inits.o src/misc/crypt/crypt_ltc_mp_descriptor.o src/misc/crypt/crypt_prng_descriptor.o src/misc/crypt/crypt_prng_is_valid.o src/misc/crypt/crypt_prng_rng_descriptor.o src/misc/crypt/crypt_register_all_ciphers.o src/misc/crypt/crypt_register_all_hashes.o src/misc/crypt/crypt_register_all_prngs.o src/misc/crypt/crypt_register_cipher.o src/misc/crypt/crypt_register_hash.o src/misc/crypt/crypt_register_prng.o src/misc/crypt/crypt_sizes.o src/misc/crypt/crypt_unregister_cipher.o src/misc/crypt/crypt_unregister_hash.o src/misc/crypt/crypt_unregister_prng.o src/misc/error_to_string.o src/misc/hkdf/hkdf.o src/misc/hkdf/hkdf_test.o src/misc/mem_neq.o src/misc/pk_get_oid.o src/misc/pkcs5/pkcs_5_1.o src/misc/pkcs5/pkcs_5_2.o src/misc/pkcs5/pkcs_5_test.o src/misc/zeromem.o src/modes/cbc/cbc_decrypt.o src/modes/cbc/cbc_done.o src/modes/cbc/cbc_encrypt.o src/modes/cbc/cbc_getiv.o src/modes/cbc/cbc_setiv.o src/modes/cbc/cbc_start.o src/modes/cfb/cfb_decrypt.o src/modes/cfb/cfb_done.o src/modes/cfb/cfb_encrypt.o src/modes/cfb/cfb_getiv.o src/modes/cfb/cfb_setiv.o src/modes/cfb/cfb_start.o src/modes/ctr/ctr_decrypt.o src/modes/ctr/ctr_done.o src/modes/ctr/ctr_encrypt.o src/modes/ctr/ctr_getiv.o src/modes/ctr/ctr_setiv.o src/modes/ctr/ctr_start.o src/modes/ctr/ctr_test.o src/modes/ecb/ecb_decrypt.o src/modes/ecb/ecb_done.o src/modes/ecb/ecb_encrypt.o src/modes/ecb/ecb_start.o src/modes/f8/f8_decrypt.o src/modes/f8/f8_done.o src/modes/f8/f8_encrypt.o src/modes/f8/f8_getiv.o src/modes/f8/f8_setiv.o src/modes/f8/f8_start.o src/modes/f8/f8_test_mode.o src/modes/lrw/lrw_decrypt.o src/modes/lrw/lrw_done.o src/modes/lrw/lrw_encrypt.o src/modes/lrw/lrw_getiv.o src/modes/lrw/lrw_process.o src/modes/lrw/lrw_setiv.o src/modes/lrw/lrw_start.o src/modes/lrw/lrw_test.o src/modes/ofb/ofb_decrypt.o src/modes/ofb/ofb_done.o src/modes/ofb/ofb_encrypt.o src/modes/ofb/ofb_getiv.o src/modes/ofb/ofb_setiv.o src/modes/ofb/ofb_start.o src/modes/xts/xts_decrypt.o src/modes/xts/xts_done.o src/modes/xts/xts_encrypt.o src/modes/xts/xts_init.o src/modes/xts/xts_mult_x.o src/modes/xts/xts_test.o src/pk/asn1/der/bit/der_decode_bit_string.o src/pk/asn1/der/bit/der_decode_raw_bit_string.o src/pk/asn1/der/bit/der_encode_bit_string.o src/pk/asn1/der/bit/der_encode_raw_bit_string.o src/pk/asn1/der/bit/der_length_bit_string.o src/pk/asn1/der/boolean/der_decode_boolean.o src/pk/asn1/der/boolean/der_encode_boolean.o src/pk/asn1/der/boolean/der_length_boolean.o src/pk/asn1/der/choice/der_decode_choice.o src/pk/asn1/der/generalizedtime/der_decode_generalizedtime.o src/pk/asn1/der/generalizedtime/der_encode_generalizedtime.o src/pk/asn1/der/generalizedtime/der_length_generalizedtime.o src/pk/asn1/der/ [... cut ...]
mv: Aufruf von stat für 'build/original/fit-image/image.0' nicht möglich: Datei oder Verzeichnis nicht gefunden
/home/rocco/Ubuntu/freetz-ng-test/tools/fit.sh: Zeile 36: [: -le: Einstelliger (unärer) Operator erwartet.
ln: auf 'build/original/fit-image/filesystem.image' konnte nicht zugegriffen werden: Datei oder Verzeichnis nicht gefunden
ERROR: cannot find kernel.image (build/original/firmware/var/tmp/kernel.image)
make[1]: *** [Makefile:376: firmware-nocompile] Fehler 1
make: *** [Makefile:46: envira] Beendet

```

fda77, 2022-06-20:
Aha, kein caching der alten Sachen...
```
/home/rocco/Ubuntu/freetz-ng-test/tools/fdtget: error while loading shared libraries: libfdt.so.1: cannot open shared object file: No such file or directory
```
Das neue Tool findet die seine lib nicht.

Du hast vorhin mit https://github.com/MasterRoCcO/i-matik/commit/e7374ffa1041fadae9994fe923090d9346cf941f uboot rausgeworfen ... hast du es auch deinstalliert? Und irgend was wichtiges mit abgeräumt?

Aber ich denke ich mache es am besten wie PeterPawn meinte: Alles wieder rauswerfen, und nur vom host-system installiertes nutzen, und wer ein inkompatibles Linux nutzt hat Pech

freetz-ng-mod, 2022-06-20:
Ubuntu ist neu installiert worden vor 1H
der 6000 macht den gleichen Fehler
`sudo apt-get -y install pv cpio rsync kmod execstack imagemagick inkscape graphicsmagick subversion git bc unar wget sudo gcc g++ binutils autoconf automake autopoint libtool-bin make bzip2 libncurses5-dev libreadline-dev zlib1g-dev flex bison patch texinfo tofrodos gettext pkg-config ecj fastjar perl libstring-crc32-perl ruby gawk python2 python-is-python3 libusb-dev unzip intltool libacl1-dev libcap-dev libc6-dev-i386 lib32ncurses5-dev gcc-multilib bsdmainutils lib32stdc++6 libglib2.0-dev ccache cmake lib32z1-dev libsqlite3-dev sqlite3 libzstd-dev netcat curl libssl-dev uuid-dev libgnutls28-dev`

uname -a
`Linux xmg-ultra 5.15.0-10033-tuxedo #34tux1 SMP Thu May 19 01:09:52 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux`
lsb_release -a
```
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04 LTS
Release:	22.04
Codename:	jammy
```

fda77, 2022-06-20:
Wie gesagt, ich werfe die neuen host-tools wieder raus und es werden dann die vom host verwendet. Dann muss man eben ein kompatibles Linux nutzen. Dazu brauch ich mich nicht mit uboot und openssl herumzuschlagen und die dl-tools können aktualisiert werden.

@PeterPawn 

```
$ time fdtdump -s fit-image  > x

**** fdtdump is a low-level debugging tool, not meant for general use.
**** If you want to decompile a dtb, you probably want
****     dtc -I dtb -O dts <filename>

real    0m1,712s
user    0m1,618s
sys     0m0,269s

$ ../../../tools/yf/fit_tools/fit-remove-avm-header.sh  fit-image > fok-image
$ time  dtc -I dtb -O dts fok-image  > y

real    0m1,560s
user    0m1,513s
sys     0m0,206s

 ```` 

Der unterschied sind leerzeilen, tabs statts leerzeichen und die herader-kommentar. Also egal welches man nimmt. Sieht auch https://github.com/Freetz-NG/freetz-ng/blob/master/tools/fit.sh#L81 ff

freetz-ng-mod, 2022-06-20:
@fda77 
nach dem ich das hier gemacht habe `sudo apt install device-tree-compiler` kann ich ein image bauen. Das sollte dann aber auch gesagt werden, dass man dieses brauch

fda77, 2022-06-20:
Wie gesagt, ich werfe die neuen host-tools wieder raus und es werden dann die vom host verwendet. Dann muss man eben ein kompatibles Linux nutzen. Dazu brauch ich mich nicht mit uboot und openssl herumzuschlagen und die dl-tools können aktualisiert werden.

freetz-ng-mod, 2022-06-20:
Ich wollte dir, das nur sagen, dass das halt fehlte. Bin auch nur durch Zufall drauf gekommen

fda77, 2022-06-20:
Sekunden, gleich fertig. Compiliert noch

PeterPawn, 2022-06-20:
> Mit den binaries die fit.sh nutzt sind es noch 3,5.

> `time fdtdump -s fit-image  > x`

Falsches Tool. Ich rede von `fitdump` und nicht von `fdtdump`, wenn es um "Geschwindigkeit" geht. DAS habe ich oben auch verlinkt.

```
peh@vidar:/home/GitHub/YourFritz/fit_tools/fit-images/FB5590> sudo rm -r fit-dump/; time ../../../../dtc/fitdump fit-image; ls -l fit-dump/
[sudo] password for root:

real    0m0.141s
user    0m0.000s
sys     0m0.141s
total 43892
-rw-r----- 1 peh users  3390019 Jun 20 21:53 image_001.bin
-rw-r----- 1 peh users    16264 Jun 20 21:53 image_002.bin
-rw-r----- 1 peh users 28053504 Jun 20 21:53 image_003.bin
-rw-r----- 1 peh users  1192159 Jun 20 21:53 image_004.bin
-rw-r----- 1 peh users     1118 Jun 20 21:53 image_005.bin
-rw-r----- 1 peh users   992871 Jun 20 21:53 image_006.bin
-rw-r----- 1 peh users  3343580 Jun 20 21:53 image_007.bin
-rw-r----- 1 peh users     7130 Jun 20 21:53 image_008.bin
-rw-r----- 1 peh users     7124 Jun 20 21:53 image_009.bin
-rw-r----- 1 peh users  7917568 Jun 20 21:53 image_010.bin
-rw-r--r-- 1 peh users     7389 Jun 20 21:53 image.its
peh@vidar:/home/GitHub/YourFritz/fit_tools/fit-images/FB5590>
```

Das ist also AUCH das Zerlegen einer 07.27 für die 5590 ... in < 0,15(!) Sekunden (wobei da noch das Setzen der Flags auf `640` fehlte in den C-Quellen, was ich schnell noch eingebaut und veröffentlich habe).

DAS ist es halt, was ich nicht verstehe. Deine Implementierung soll (a) schnell sein (damit fällt meine Shell-Implementierung dann aus) und stört sich (b) auch nicht daran, wenn dazu weitere Binär-Dateien benötigt werden (was ja mein Grund war, das IN SHELL selbst zu machen). 

Gleichzeitig HAST Du aber schon das Kompilieren von Tools aus `dtc` in Deinem Build-Prozess ... es bräuchte also für das Bauen von `fitdump` in der C-Version nur noch einen einzigen zusätzlichen Patch - meinetwegen könnte man die drei Commits aus meinem `dtc`-Klon eben sogar noch zu einem "squashen" und hätte dann auch nur einen einzigen Patch.

---

Warum Du dann das Zerlegen (zumindest in die `its`-Datei und die einzelnen Image-Files, wobei Du für `fit-image -> image.itb` ja noch mein Shell-Skript nimmst) dennoch selbst machen willst (mit viel `sed`-Zauberei, die immer noch - ich kann mich nur wiederholen - wacklig ist ... was passiert denn bei Dir in dem `sed`, wenn auf einmal die `data`-Properties NICHT mehr jeweils als (zig Megabyte lange) einzelne Zeile ausgegeben werden?), erschließt sich mir nicht. 

Selbst wenn Dir das, was ich meinerseits in `fitdump.c` ggü. `fdtdump.c` geändert habe, nicht reicht, kannst Du da ja noch Teile aus Deinen "analyze" einbauen (oder die sogar IMMER NOCH zusätzlich im Shell-Code verwenden). Schaut man sich das `unpack` in Deinem `fit.sh` an, macht das (wenn ich's richtig lese, wovon ich jetzt erst mal ausgehe, weil ich mich schon länger mit dem Thema befaßt habe) EXAKT das (beim Erzeugen der `image.its` und mit dem `dumpimage`), was `fitdump` (in der C-Version) auch machen würde, bevor es (als letztes) dann `analyze` aufruft.

Wenn Du unbedingt mit `fdtget`/`fdtgrep` dann noch an die "Struktur" willst (die mehrfachen Schleifen machen das im `analyze` auch nicht gerade gut lesbar - mit `fdtgrep type <itb-file>` und anschließendem `sed`-"Zauber" geht das auch in EINEM einzigen Aufruf, was da in der ersten Schleife praktiziert wird in `analyze`), dann bleibt Dir das ja unbenommen. 

Mich selbst stört(e) an den anderen `fdt...`-Programmen aber halt auch die Tatsache, daß das nur mit der binären Form des FDT funktioniert, auch wenn man mit einem `dtc -I dts -O dtb <its-file> > <itb-file>` ja jederzeit aus der Quelle auch die binäre Form machen kann (das funktioniert auch, daß man den `dtc` zum "Übersetzen" in die binäre Form eines FIT-Images verwendet - nur werden dann eben die (irgendwo hier vorher verlinkten) Aktionen, die beim `mkimage` automatisch erfolgen (Timestamp, Hash berechnen, etc.) nicht ausgeführt. Wobei es für mich egal bleibt (bei meinen eigenen Zielen), ob ich nun `dtc` oder `mkimage` als Binärdatei bräuchte - ich will beides nicht haben.

---

Abgesehen davon ... Du weißt schon, daß man mit einem `printf '%03u' "$count"` auch IN EINEM ZUG zu einer dreistelligen Zahl mit Vornullen kommen kann? (https://github.com/Freetz-NG/freetz-ng/blob/3ce8567d35c9cac88ec6a4bb8372a88b402638cf/tools/fit.sh#L54)

Warum genau machst Du da eigentlich AUCH drei Stellen draus? Was hält Dich davon ab (wenn DU das schon so "kompliziert" auswertest, ob der Zähler nun schon zweistellig ist oder nicht), dann einfach GAR KEINE Vornullen bei der Nummerierung der Image-Dateien zu verwenden?

---

fda77, 2022-06-20:
Nachdem heut bereits 2 herumgenölt haben wegen prerequisites, libs usw hab ich die 2 Hosttools wieder rausgeworfen. Ich hatte auch wenig lust dieses dynamic-openssl zu lösen, du fandest die als host-tools ja auch nicht so gut.
So werd ich ftdgrep aber nicht ausprobieren können, bei fedora scheint das in keinem package zu sein

> Gleichzeitig HAST Du aber schon das Kompilieren von Tools aus dtc in Deinem Build-Prozess

Nicht mehr. Schade das hätte Zeit gespart. Das war auch etwas trick mit dem AKC, den dessen dtc-lib ist 32 bit zum kernel passend und ich muss die 2x erstellen in verschiedenen Verzeichnissen

> mit viel sed-Zauberei, die immer noch - ich kann mich nur wiederholen - wacklig ist ... 

hatte nichts besseres. Warum kann das tool das nicht gleich so exportieren :-/

> Warum genau machst Du da eigentlich AUCH drei Stellen draus? 

Weil das besser aussieht... :) Ich weiss dass die meisten davon ichts halte, mit er13 hatte ich da auch so manche diskussion

> Abgesehen davon ... Du weißt schon, daß man mit einem printf

Nein, bei printf bin ich noch nicht durchgestiegen! Ich schaue nochmal

Unabhängig von dem geschriebene ist meine Motivation so ca bei 0

fda77, 2022-06-20:
Ich muss sagen dass mir das fdtgrep am besten von den optionen her gefällt, gleich noch mit version und header dabei. 
Nur sehr seltsam dass das bei fedora nicht mit normalem paketmanager installierbar ist

```
$  dnf whatprovides fdtgrep
Fehler: No matches found. If searching for a file, try specifying the full path or using a wildcard prefix ("*/") at the beginning.

$ dnf whatprovides fdtget
dtc-1.6.1-2.fc35.x86_64 : Device Tree Compiler
Paketquelle      : @System
Übereinstimmung von:
Dateiname     : /usr/bin/fdtget

dtc-1.6.1-2.fc35.x86_64 : Device Tree Compiler
Paketquelle      : fedora
Übereinstimmung von:
Dateiname     : /usr/bin/fdtget

```

7530ax geht auch mit den allerneusten fit änderungen

fprint ist ganz praktisch... ffeeedc6c

fda77, 2022-06-22:
Mit der neuen (und einzigen) 5530 inhaus funktioniert `fdtdump` als auch `dtc -I dtb -O dts ` und ein selbst gebautes `fdtgrep` nicht richtig

```
grep entry build/o*/fit-image/image.its
build/modified/fit-image/image.its:                     entry = "p{o";
build/modified/fit-image/image.its:                     entry = <0x8829b200>;
```

das `fitdump.sh` script funktioniert:
```
grep entry  fit-dump/*
fit-dump/image.its:            entry = <0x707b6f00>;
fit-dump/image.its:            entry = <0x8829b200>;
```

Vielleicht doch besser fitdump selbstbau tool...

PeterPawn, 2022-06-23:
Die Daten sind gleichwertig ... ich habe nur (der "Schönheit" wegen) eine gesonderte Behandlung von "4-Byte-Werten" eingebaut, die man auch als Zeichenkette (also drei druckbare Zeichen mit einer binären Null im Byte danach) interpretieren könnte: https://github.com/PeterPawn/YourFritz/commit/21e6d8692d91cdaf01f755550232e6e74ae06299

fda77, 2022-06-23:
Aber zu mir was sagen von wegen schöne Dateinamen :)
Avm hat in den neuen inhaus noch ein 2. fit im fit

fda77, 2022-06-23:
Ich weiss nicht genau was mit der 7530ax das Problem ist, jedenfalls kamen da am Anfang bei eva 2-3 Zeilen mit sowas wie "problem beim unflattenig" . Muss ich nochmal genau schauen

kamasudo, 2022-06-23:
Wenn der dt nicht ausgerollt werden kann startet der kernel nicht. dann geht - gar nichts!

fda77, 2022-06-27:
Danke, ich denkte das funktioniert soweit

closed 2022-06-27 by fda77