Freetz-NG/freetz-ng #572: Error building fakeroot as host dependency

TheOneWithTheBraid, 2022-06-22:
While trying to build for Fritz!Box 7490, the `make` aborts on a particularly useless error message about the host dependency fakeroot.

I am building on Arch Linux 5.18. Did anyone else encounter this problem? I am pretty sure it's just because some missing dependencies...

The following output is shown:

<details>
<summary> `make` logs </summary>

```
 make
Freetz-NG 19866-6ef58b771 master 2022-06-21
---> tools/fakeroot-host ... 
configuring ... (mkdir -p /home/void/freetz-ng/source/host-tools/fakeroot-1.29/build/biarch; cd /home/void/freetz-ng/source/host-tools/fakeroot-1.29/build/biarch; rm -f config.cache; \
	CC="gcc" \
	CXX="g++" \
	CFLAGS="-Wall -O2 -m32" \
	CXXFLAGS="-Wall -O2 -m32" \
	LDFLAGS="" \
	../../configure \
	--prefix=/home/void/freetz-ng/tools/build \
	--enable-shared \
	--with-ipc=sysv \
	 \
	--disable-nls \
	 \
	 \
);
configure: WARNING: unrecognized options: --disable-nls
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking target system type... x86_64-pc-linux-gnu
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a race-free mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether to enable maintainer-specific portions of Makefiles... no
checking whether make sets $(MAKE)... (cached) yes
checking how to print strings... printf
checking whether make supports the include directive... yes (GNU style)
checking for gcc... gcc
checking whether the C compiler works... no
configure: error: in `/home/void/freetz-ng/source/host-tools/fakeroot-1.29/build/biarch':
configure: error: C compiler cannot create executables
See `config.log' for more details
make[1]: *** [tools/make/fakeroot-host/fakeroot-host.mk:55: /home/void/freetz-ng/source/host-tools/fakeroot-1.29/build/biarch/.configured] Error 77
make: *** [Makefile:46: envira] Terminated
```

</details>

Attachments:

- [config.txt](https://github.com/Freetz-NG/freetz-ng/files/8955924/config.txt)
- [config.log](https://github.com/Freetz-NG/freetz-ng/files/8955931/config.log)

fda77, 2022-06-22:
Searching for
```
configure: error: C compiler cannot create executables
```
shows me something on your system needed to compile is missing - and freetz obviousy does not check it yet. Maybe this helps: https://askubuntu.com/questions/647583/how-to-fix-configure-error-c-compiler-cannot-create-executables

TheOneWithTheBraid, 2022-06-22:
Do you know whether freetz requires a particular libc version? I have `libc++` 13 and `glibc` 2.35 installed.

fda77, 2022-06-22:
I'm not sure, but i think your compiler should be able to use it

TheOneWithTheBraid, 2022-06-22:
Well, and if I manually compile the "normal" fakeroot binary of my OS ([AUR](https://aur.archlinux.org/packages/fakeroot-git)), it properly works. Are there any patches or special configs applied to the fakeroot source?

fda77, 2022-06-22:
Do you properly compile it for 32 bit and 64 bit too?

closed 2022-06-27 by fda77