Freetz-NG/freetz-ng #579: Hardware series (4&5)  ---> bei Expert Ansicht

freetz-ng-mod, 2022-06-30:
@fda77 ist das normal das nun bei der Expert Ansicht Hardware series (4&5) gleich angegeben wird und nicht all. Also freetz ohne .config gestartet

closed 2022-07-02 by fda77

fda77, 2022-07-02:
Ne, so war das nicht gedacht! Müsste jetzt passen.
Sag mal, wie findest du das beginner/advanced/expert? 2 Level sollten doch reichen

freetz-ng-mod, 2022-07-02:
Nun steht das da.
```
Level of user competence (Expert)  --->   
Hardware series (7490/7590)  ---> 
Hardware type (7590)  --->      

```
Immer noch nicht all.

Für die meisten sollte ja Beginner reichen. Und daher sollte die Aufteilung, die in freetz drin ist, schon ganz ok sein.

fda77, 2022-07-02:
Jo, passt so.
Also Beginner findest du gut. Und dann advanced+expert zusammen legen, bzw advanced streichen?
Das käme auf
 > 2 Level sollten doch reichen

freetz-ng-mod, 2022-07-02:
Ich glaube, das hat jeder eine andere Meinung. Der eine will gleich alles, der andere will Sicherheit. Daher ist das ganze nicht so einfach zu beantworten.  Mir reicht zB Expert, da DEVELOPER einfach viele Sachen hat, die ich nicht brauche ok, wenn ich Test halber halt eine 5530 bauen will dann brauche ich es schon

fda77, 2022-07-02:
Also `developer` muss bleiben, da sind sachen dahinter die man nur nutzen sollte wenn man weiss dass man sie braucht. (zb 7530ax labor/inhaus und watchdog..)
Eigentlich wäre es gut `aptguru `auch wegzumachen, nur dann haben viele Probleme mit prerequisites und das Geschrei geht wieder los...
`Beginner` nur mit Minimalem ist auch okay.
Dann blieben noch die 2 von denen ich 1 überflüssig finde...

freetz-ng-mod, 2022-07-02:
Klar sollte `developer` bleiben, ich versuche mich ja grade ein wenig an `aptguru` wegen Bauen meiner eigenen Toolcains und so. Dass er die auch dann benutzt, aber steige noch nicht so durch. Und `Beginner` ist klar ;-). Wie gesagt, ich nutze eigentlich sehr oft `expert`. Das letzte habe ich noch nie benutzt, ganz ehrlich gesagt. Was macht das überhaupt oder wofür ist das da.

Wenn ein Tipp hast, was ich einstellen muss, dass mir mein System eigenen Toolcains baut und sie dann auch beim Bauen nutzt, wer ich für jeden Tipp dankbar

fda77, 2022-07-02:
Gut, du würdest den `advanced` auch nicht vermissen.

Toolchains baue ich einfach mir `tools/dl-toolchains_make` - 8h warten, da kommen dann 500mb für ~25 toolchains raus. Es gitb auch eine github actions die das machen könnten: https://github.com/Freetz-NG/freetz-ng/actions/workflows/dl-toolchains_new.yml
Wenn du deren Version `rLOCAL` nennst gibt es ein unsichtbares Sysmbol das man in die custom.in machen könnte, um die zu benutzen.
Was willst du denn mit selbst gebauten? Die allerneueste Version läuft auch auf CPUs ohne AVX2

freetz-ng-mod, 2022-07-02:
Ich bin dabei Freetz immer besser kennen zulernen und versuche mich halt nach und nach an allen einmal um es besser zu verstehen. Da ich ja öfters mal ein paar Images baue, einfach aus Spaß. Würde das ganze dann ja ein wenig schneller gehen beim Bauen, wenn ich das freetz Verzeichnis wieder lösche.

Ich werde mir das mal genauer anschauen, mit den beiden Sachen 

Ja ich würde denn nicht vermissen da ich denn noch nie genutzt habe ;-)

fda77, 2022-07-02:
Wenn du `aptguru` aktiviert hast und die dl toolchains+tools aktiviert sind werden die von mir erstellten genutzt.
So kann man nach einem distclean ein minimales image ohne packages+kernelmodule in ca 2 min bauen.
Selbst bauen brauchst du aber nur wenn du den binaries von mir nicht traust oder die bei dir nicht laufen sollten (zb kein AVX(1))
Schau dir in dem script aber mal das `miniconfig` an, statt den ganzen sed!

freetz-ng-mod, 2022-07-03:
Also bei mir brauch `tools/dl-toolchains_make` 94 minutes and 15 seconds

```
--- tools/dl-toolchains_make	2022-07-02 23:53:23.780325485 +0200
+++ tools/dl-toolchains_make	2022-07-03 09:24:52.309077193 +0200
@@ -2,6 +2,7 @@
 # packs toolchains to dl/*freetz-VERSION-shared-glibc.tar.lzma in 20-25min each
 #Arg $1: <VERSION>
 #Arg $2: <new|old|all|FREETZ_TYPE_...>
+begin=$(date +"%s")
 
 VERSION="$(tools/freetz_revision short)"
 VERSION="${1:-r${VERSION%-*}}"
@@ -135,5 +136,8 @@
 [ "$new" == "new" ] && dltc "FREETZ_TYPE_6660    FREETZ_TYPE_FIRMWARE_07_2X"  # SEPARATE
 
 
+termin=$(date +"%s")
+difftimelps=$(($termin-$begin))
+echo "$(($difftimelps / 60)) minutes and $(($difftimelps % 60)) seconds elapsed for Script Execution."
 exit 0
 
```

fda77, 2022-07-03:
>  #Arg $2: <new|old|all|FREETZ_TYPE_...>

> Es sind 330,6 MB

Das ist aufgeteilt da eine github acrtion max 6h laufen darf. Zur Zeitmussing einfach mit time aufrufen:
`time sleep 2`

freetz-ng-mod, 2022-07-03:
Also muss ich das abändern 
```
	a|all)		old="old" new="new"       ;;
	o|old)		old="old" new=""          ;;
	n|new|"")	old=""    new="new"       ;;
```
auf all aber so recht verstehe ich noch nicht welches der new ich auf all setzen muss

fda77, 2022-07-03:
Gar nichts musst du ändern.... "arg" ist die kurzform von "argument". Man sagt auch "parameter" dazu. Also das was du hinter den aufruf schreibst:
`./script arg1 arg2 arg3`
-.-

> #Arg $2: <new|old|all|FREETZ_TYPE_...>

sind die möglichen werte. Und nein, '...' schreibt man nicht hin sondern was passenden, falls man es denn überhaupt möchte

freetz-ng-mod, 2022-07-03:
Dann teste ich es mal mit all all all ;-) mal schauen wie lange er brauch

fda77, 2022-07-03:
Nicht ganz das doppelte, 2,5h +

freetz-ng-mod, 2022-07-03:
```
tools/dl-toolchains_make all all all 
tools/dl-hosttools
tools/dl-toolchains_eval
```
Sehe ich das richtig, dass ich das alles brauche. Und dann die passenden Dateien anpassen ;-).
Wie gesagt ich teste nur ein wenig herum, um das Ganze besser zu verstehen und zulernen.

fda77, 2022-07-03:
Welche Datein brauchst du? Die Scripte erstellen Dateien. Was welches Scipt genau macht steh in der 2. Zeile, direkt nach Shebang

freetz-ng-mod, 2022-07-03:
Habe es nun, er baut mir nun die Sachen so `mipsel_gcc-4.6.4_uClibc-0.9.28-freetz-r19984-shared-glibc.tar.lzma` davor war es so `mipsel_gcc-4.6.4_uClibc-0.9.28-freetz-all-shared-glibc.tar.lzma`
Wieder was dazugelernt. Habe mir dazu auch ein Patch geschrieben, der dann gleich `default "r19900"` anpasst, voll automatisch. So dass ich Test halber meine toolchains dann auch mal testen kann. 

Mal schauen, was ich mir als Nächstes dann mal genauer anschauen werde. Gibt es eigentlich ein Script was alle FW Downloadet

fda77, 2022-07-03:
Weiss ich nicht, es gibt aber ein make target das alle fw überprüft. Wie das genau heisst weiss ich nicht, da musst du in den makefules schauen. Mit juis ist das aber eigentlich egal
Schreib noch einen Patch der alle Checksummen anpasst, das fehlt noch ;-)

freetz-ng-mod, 2022-07-03:
Das mit dem Patch für Checksummen habe ich auch schon dran gedacht, aber da ich sie ja lokal habe brauche ich ja kein Check. Aber mal schauen wie ich, das vielleicht doch hinbekomme. Denn wie gesagt, ich lerne ja noch. Vieles kenne ich noch nicht, aber doch schon einiges.

fda77, 2022-07-03:
Mach mal. Mir war es zu stressig das automatisch zu machen und mache es lieber manuell...

freetz-ng-mod, 2022-07-03:
/test.sh
```
./test.sh
Bereits aktuell.
ALT_TOOLS 2022-06-27
ALT_HASH 6dc5409e00a344184e91dbe423d7d3361b3f5539edd4486477f67beec0cccc11
NEU_TOOLS r19984
NEU_HASH ac141a20b915d8d91cfd9f454301097d4fb26ff86a0d082c69503e26fa4841ff
```
```
#!/bin/bash
TOOLS_PATH=$(cd `dirname $0` && pwd)
VERSION_ALT="$(tools/freetz_revision short)"
VERSION_ALT="${1:-r${VERSION_ALT%-*}}"
git pull
VERSION_NEU="$(tools/freetz_revision short)"
VERSION_NEU="${1:-r${VERSION_NEU%-*}}"

rm -f dl/tools-${VERSION_ALT}.tar.xz

tools/dl-hosttools $VERSION_NEU > tools/tools-${VERSION_NEU}.sha256sum 

ALT_TOOLS="$(grep '^$(call TOOLS_INIT,' tools/make/tools-host/tools-host.mk | awk -F',' '{print $2}' | xargs)"
echo "ALT_TOOLS ${ALT_TOOLS:0:10}"

ALT_HASH="$(grep '^$(PKG)_HASH:=' tools/make/tools-host/tools-host.mk | awk -F'=' '{print $2}' | xargs)"
echo "ALT_HASH $ALT_HASH"

NEU_TOOLS="$(grep '^OUTPUT:=' tools/tools-${VERSION_NEU}.sha256sum  | awk -F'=' '{print $2}' | xargs)"
echo "NEU_TOOLS ${NEU_TOOLS:6:6}"

NEU_HASH="$(grep '^SHA256:=' tools/tools-${VERSION_NEU}.sha256sum  | awk -F'=' '{print $2}' | xargs)"
echo "NEU_HASH $NEU_HASH"

rm -f tools/tools-${VERSION_NEU}.sha256sum 2>/dev/null

sed -i "s+${ALT_TOOLS:0:10}+${NEU_TOOLS:6:6}+g" tools/make/tools-host/tools-host.mk 2>/dev/null
sed -i "s+$ALT_HASH+$NEU_HASH+g" tools/make/tools-host/tools-host.mk 2>/dev/null
```

fda77, 2022-07-04:
Ja...neee
Es ging doch um die 40 Dateien der toolchains! Die 1 von den tools gibt es schon: https://github.com/Freetz-NG/freetz-ng/blob/master/.github/workflows/dl-hosttools.yml#L96=

freetz-ng-mod, 2022-07-04:
```
MIPS_KERNEL_5_ALT 2b21f4bb4df1a32955870cb4f09083f9fc5a1182d51b83094f9749c41f84a890
MIPS_KERNEL_5_NEU 24db311a492f2bb5236a8b968dc733648df08d293a223eeaaf35e54a92c20773
sed -i s+2b21f4bb4df1a32955870cb4f09083f9fc5a1182d51b83094f9749c41f84a890+24db311a492f2bb5236a8b968dc733648df08d293a223eeaaf35e54a92c20773+g config/mod/download.in_neu
MIPS_TARGET_KERNEL_4_4_ALT b945bbfa71fe47259e73284746ef1f6e59887d2b2abf559b71d49daba5b374fd
MIPS_TARGET_KERNEL_4_4_NEU 865018423bacbe8e85a9f89e88d2fe1142765abb83211b2c2ebfaa6a8896a7a9
```
Das habe ich schon mal, nun muss ich alles nur noch richtig zuordnen

fda77, 2022-07-04:
> nur

Genau... Die alten müssen nicht immer die gleiche Revision haben, wenn nur uclib eine neue Version bekommt, brauchen nur die "new" gebaut zu werden. Evtl kommt auch mal eine nue Toolchain dazu die dann die aktuelle Revision hat usw

freetz-ng-mod, 2022-07-04:
Du meinst das hier 
```
UCLIBC_ALT=FREETZ_TARGET_UCLIBC_1_0_41
UCLIBC_NEU=FREETZ_TARGET_UCLIBC_1_0_41
^ der Teil soll das dann im Script ändern und in der download.in 
UCLIBC_NEU_TAG=uClibc-1.0.41
^ der Teil ist dann dafür da mips_gcc-8.3.0_uClibc-1.0.41-nptl_kernel-4.9-freetz-r19984-shared-glibc.tar.lzma
```
das meinst du

fda77, 2022-07-04:
Nein, die Version der toolchains. In einer .in die "default" Werte

freetz-ng-mod, 2022-07-04:
Erst mal muss das jetzt gehen, dann kann man weiter schauen.

freetz-ng-mod, 2022-07-04:
```
arm_gcc-5.5.0-bcm_uClibc-1.0.41-nptl_kernel-4.1-freetz-r19984-shared-glibc.tar.lzma FREETZ_TARGET_ARCH_ARM                           && FREETZ_TARGET_GCC_5   && FREETZ_TARGET_UCLIBC_1_0_41 && FREETZ_KERNEL_VERSION_4_1 && FREETZ_SYSTEM_TYPE_BCM63138
arm_gcc-5.5.0_uClibc-1.0.14-nptl_kernel-4.4-freetz-r19984-shared-glibc.tar.lzma FREETZ_TARGET_ARCH_ARM                           && FREETZ_TARGET_GCC_5   && FREETZ_TARGET_UCLIBC_1_0_14 && FREETZ_KERNEL_VERSION_4_4
arm_gcc-5.5.0_uClibc-1.0.41-nptl_kernel-4.1-freetz-r19984-shared-glibc.tar.lzma FREETZ_TARGET_ARCH_ARM                           && FREETZ_TARGET_GCC_5   && FREETZ_TARGET_UCLIBC_1_0_41 && FREETZ_KERNEL_VERSION_4_1
arm_gcc-5.5.0_uClibc-1.0.41-nptl_kernel-4.4-freetz-r19984-shared-glibc.tar.lzma FREETZ_TARGET_ARCH_ARM                           && FREETZ_TARGET_GCC_5   && FREETZ_TARGET_UCLIBC_1_0_41 && FREETZ_KERNEL_VERSION_4_4
arm_gcc-9.3.0_uClibc-1.0.41-nptl_kernel-4.19-freetz-r19984-shared-glibc.tar.lzma FREETZ_TARGET_ARCH_ARM                           && FREETZ_TARGET_GCC_9   && FREETZ_TARGET_UCLIBC_1_0_41 && FREETZ_KERNEL_VERSION_4_19
i686_gcc-4.7.4_uClibc-0.9.33.2-nptl-freetz-r19984-shared-glibc.tar.lzma FREETZ_TARGET_ARCH_X86                           && FREETZ_TARGET_GCC_4_7 && FREETZ_TARGET_UCLIBC_0_9_33 && FREETZ_KERNEL_VERSION_2
i686_gcc-5.5.0_uClibc-1.0.14-nptl-freetz-r19984-shared-glibc.tar.lzma FREETZ_TARGET_ARCH_X86                           && FREETZ_TARGET_GCC_5   && FREETZ_TARGET_UCLIBC_1_0_14 && FREETZ_KERNEL_VERSION_2
i686_gcc-5.5.0_uClibc-1.0.41-nptl-freetz-r19984-shared-glibc.tar.lzma FREETZ_TARGET_ARCH_X86                           && FREETZ_TARGET_GCC_5   && FREETZ_TARGET_UCLIBC_1_0_41 && FREETZ_KERNEL_VERSION_2
i686_gcc-5.5.0_uClibc-1.0.41-nptl_kernel-3.12-freetz-r19984-shared-glibc.tar.lzmaFREETZ_TARGET_ARCH_X86                           && FREETZ_TARGET_GCC_5   && FREETZ_TARGET_UCLIBC_1_0_41 && FREETZ_KERNEL_VERSION_3_12
i686_gcc-5.5.0_uClibc-1.0.41-nptl_kernel-4.9-freetz-r19984-shared-glibc.tar.lzma FREETZ_TARGET_ARCH_X86                           && FREETZ_TARGET_GCC_5   && FREETZ_TARGET_UCLIBC_1_0_41 && FREETZ_KERNEL_VERSION_4_9
i686_gcc-8.3.0_uClibc-1.0.41-nptl_kernel-4.9-freetz-r19984-shared-glibc.tar.lzma FREETZ_TARGET_ARCH_X86                           && FREETZ_TARGET_GCC_8   && FREETZ_TARGET_UCLIBC_1_0_41 && FREETZ_KERNEL_VERSION_4_9
mipsel_gcc-4.6.4_uClibc-0.9.28-freetz-r19984-shared-glibc.tar.lzma FREETZ_TARGET_ARCH_MIPS && FREETZ_TARGET_ARCH_LE && FREETZ_TARGET_GCC_4_6 && FREETZ_TARGET_UCLIBC_0_9_28 && FREETZ_KERNEL_VERSION_2
mipsel_gcc-4.6.4_uClibc-0.9.29-freetz-r19984-shared-glibc.tar.lzma FREETZ_TARGET_ARCH_MIPS && FREETZ_TARGET_ARCH_LE && FREETZ_TARGET_GCC_4_6 && FREETZ_TARGET_UCLIBC_0_9_29 && FREETZ_KERNEL_VERSION_2
mipsel_gcc-4.6.4_uClibc-0.9.32.1-freetz-r19984-shared-glibc.tar.lzma FREETZ_TARGET_ARCH_MIPS && FREETZ_TARGET_ARCH_LE && FREETZ_TARGET_GCC_4_6 && FREETZ_TARGET_UCLIBC_0_9_32 && FREETZ_KERNEL_VERSION_2
mipsel_gcc-4.6.4_uClibc-0.9.33.2-freetz-r19984-shared-glibc.tar.lzma FREETZ_TARGET_ARCH_MIPS && FREETZ_TARGET_ARCH_LE && FREETZ_TARGET_GCC_4_6 && FREETZ_TARGET_UCLIBC_0_9_33 && FREETZ_KERNEL_VERSION_2
mips_gcc-4.6.4_uClibc-0.9.32.1-freetz-r19984-shared-glibc.tar.lzma FREETZ_TARGET_ARCH_MIPS && FREETZ_TARGET_ARCH_BE && FREETZ_TARGET_GCC_4_6 && FREETZ_TARGET_UCLIBC_0_9_32 && FREETZ_KERNEL_VERSION_2
mips_gcc-4.6.4_uClibc-0.9.33.2-freetz-r19984-shared-glibc.tar.lzma FREETZ_TARGET_ARCH_MIPS && FREETZ_TARGET_ARCH_BE && FREETZ_TARGET_GCC_4_6 && FREETZ_TARGET_UCLIBC_0_9_33 && FREETZ_KERNEL_VERSION_2
mips_gcc-4.7.4_uClibc-0.9.33.2-nptl-freetz-r19984-shared-glibc.tar.lzma FREETZ_TARGET_ARCH_MIPS && FREETZ_TARGET_ARCH_BE && FREETZ_TARGET_GCC_4_7 && FREETZ_TARGET_UCLIBC_0_9_33 && FREETZ_KERNEL_VERSION_2
mips_gcc-4.8.5_uClibc-0.9.33.2-nptl-freetz-r19984-shared-glibc.tar.lzma FREETZ_TARGET_ARCH_MIPS && FREETZ_TARGET_ARCH_BE && FREETZ_TARGET_GCC_4_8 && FREETZ_TARGET_UCLIBC_0_9_33 && FREETZ_KERNEL_VERSION_2
mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10-freetz-r19984-shared-glibc.tar.lzma FREETZ_TARGET_ARCH_MIPS && FREETZ_TARGET_ARCH_BE && FREETZ_TARGET_GCC_4_8 && FREETZ_TARGET_UCLIBC_0_9_33 && FREETZ_KERNEL_VERSION_3_10
mips_gcc-5.5.0_uClibc-0.9.33.2-nptl_kernel-3.10-freetz-r19984-shared-glibc.tar.lzma FREETZ_TARGET_ARCH_MIPS && FREETZ_TARGET_ARCH_BE && FREETZ_TARGET_GCC_5   && FREETZ_TARGET_UCLIBC_0_9_33 && FREETZ_KERNEL_VERSION_3_10
mips_gcc-5.5.0_uClibc-1.0.14-nptl_kernel-3.10-freetz-r19984-shared-glibc.tar.lzma FREETZ_TARGET_ARCH_MIPS && FREETZ_TARGET_ARCH_BE && FREETZ_TARGET_GCC_5   && FREETZ_TARGET_UCLIBC_1_0_14 && FREETZ_KERNEL_VERSION_3_10
mips_gcc-5.5.0_uClibc-1.0.14-nptl_kernel-4.4-freetz-r19984-shared-glibc.tar.lzma FREETZ_TARGET_ARCH_MIPS && FREETZ_TARGET_ARCH_BE && FREETZ_TARGET_GCC_5   && FREETZ_TARGET_UCLIBC_1_0_14 && FREETZ_KERNEL_VERSION_4_4
mips_gcc-5.5.0_uClibc-1.0.41-nptl_kernel-3.10-freetz-r19984-shared-glibc.tar.lzma FREETZ_TARGET_ARCH_MIPS && FREETZ_TARGET_ARCH_BE && FREETZ_TARGET_GCC_5   && FREETZ_TARGET_UCLIBC_1_0_41 && FREETZ_KERNEL_VERSION_3_10
mips_gcc-5.5.0_uClibc-1.0.41-nptl_kernel-4.4-freetz-r19984-shared-glibc.tar.lzma FREETZ_TARGET_ARCH_MIPS && FREETZ_TARGET_ARCH_BE && FREETZ_TARGET_GCC_5   && FREETZ_TARGET_UCLIBC_1_0_41 && FREETZ_KERNEL_VERSION_4_4
mips_gcc-8.3.0_uClibc-1.0.41-nptl_kernel-4.9-freetz-r19984-shared-glibc.tar.lzma FREETZ_TARGET_ARCH_MIPS && FREETZ_TARGET_ARCH_BE && FREETZ_TARGET_GCC_8   && FREETZ_TARGET_UCLIBC_1_0_41 && FREETZ_KERNEL_VERSION_4_9
```

fda77, 2022-07-07:
> Wenn du da noch Vereinfachungen hast oder Verkürzungen

Geht in 1 Zeile: f31204136494f83e70788e18db62680f46b66dfb