Freetz-NG/freetz-ng #581: Die 6591 07.39-97972 hat ein neuen oder anders gesagt ein anderen

freetz-ng-mod, 2022-07-06:
FRITZ.Box_6591_Cable-07.39-97972-LabBETA.image
    ... used by AVM ...... 4.9.279
    ... used by Freetz ... 4.9.250
ERROR: Freetz and AVM kernel versions do not match. Wrong kernel version configuration in Freetz?

fda77, 2022-07-06:
Wo die kleinen Sourcepakete herkommen weisst du ja..
Da nur die Docsis Dinger von avm x86/atom haben wird es noch nichts nutzbares geben.
Bevor dur fragst: Ich hab extra nochmal die komplette GPL gelesen, da gibt es KEINEN  Abschnitt "Für eine Labor von AVM braucht kein Sourcecode herausgerückt zu werden"!

freetz-ng-mod, 2022-07-06:
Ich könnte auch auf Klo gehen, ;-) man könnte vieles

fda77, 2022-07-06:
Damit du DAVOR auch das Hösschen runterziehst hab ich deinen Betrag gelikt

fda77, 2022-07-06:
Du könntest aber heraussuchen was von 6591 + 6660 die letzten revisionen mit der alten Kernelversion waren

fda77, 2022-07-06:
> die letzten revisionen mit der alten Kernelversion waren

freetz-ng-mod, 2022-07-06:
FRITZ.Box_6591_Cable-07.39-97061-LabBETA.image

fda77, 2022-07-06:
Dies c85bad2e5e387f19c238631b9429e518d4bb002e ?

fda77, 2022-07-06:
Schau am besten selbst durch: https://github.com/Freetz-NG/freetz-ng/commits/master/docs/JUIS.md

fda77, 2022-07-07:
Also, schreib avm ne mail, dann -> https://github.com/Freetz-NG/freetz-ng/discussions/134

closed 2022-07-07 by fda77