Freetz-NG/freetz-ng #582: Fehler bei bauen eines Images von der 7330 mit BU 2.24

freetz-ng-mod, 2022-07-06:
So, ich habe freetz neu geladen. Habe die Box 7330 ausgesucht und habe 
das an
```
FREETZ_BUILD_TOOLCHAIN
FREETZ_KERNEL_BINUTILS_2_24
```
das aus
```
FREETZ_HOSTTOOLS_DOWNLOAD
FREETZ_TOOLCHAIN_CCACHE
```
```
make -j20
Freetz-NG 20018-6e4062d7e master 2022-07-06
---> kernel ... 
downloading ... tools/freetz_download dl kernel_2.6.32.61-7330_06.50-r19566.patch.xz @MIRROR/ 6b908ee1569014e0cb21a1617f94666fb1f8eedef2369f79ae2f663824fe6b3a 

--2022-07-06 21:30:05--  https://freetz-ng.dropfiles.net/kernel_2.6.32.61-7330_06.50-r19566.patch.xz
Auflösen des Hostnamens freetz-ng.dropfiles.net (freetz-ng.dropfiles.net) … fehlgeschlagen: Der Name oder der Dienst ist nicht bekannt.
wget: Host-Adresse ‘freetz-ng.dropfiles.net’ kann nicht aufgelöst werden
Download failed - "https://freetz-ng.dropfiles.net/kernel_2.6.32.61-7330_06.50-r19566.patch.xz"  ->  error code 4

--2022-07-06 21:30:05--  https://freetz.wirsind.info/kernel_2.6.32.61-7330_06.50-r19566.patch.xz
Auflösen des Hostnamens freetz.wirsind.info (freetz.wirsind.info) … 188.165.115.52
Verbindungsaufbau zu freetz.wirsind.info (freetz.wirsind.info)|188.165.115.52|:443 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 404 Not Found
2022-07-06 21:30:05 FEHLER 404: Not Found.

Download failed - "https://freetz.wirsind.info/kernel_2.6.32.61-7330_06.50-r19566.patch.xz"  ->  error code 8

--2022-07-06 21:30:05--  https://freetz.magenbrot.net/kernel_2.6.32.61-7330_06.50-r19566.patch.xz
Auflösen des Hostnamens freetz.magenbrot.net (freetz.magenbrot.net) … 31.172.113.113
Verbindungsaufbau zu freetz.magenbrot.net (freetz.magenbrot.net)|31.172.113.113|:443 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 404 Not Found
2022-07-06 21:30:05 FEHLER 404: Not Found.

Download failed - "https://freetz.magenbrot.net/kernel_2.6.32.61-7330_06.50-r19566.patch.xz"  ->  error code 8

--2022-07-06 21:30:05--  https://raw.githubusercontent.com/Freetz-NG/dl-mirror/master/kernel_2.6.32.61-7330_06.50-r19566.patch.xz
Auflösen des Hostnamens raw.githubusercontent.com (raw.githubusercontent.com) … 2606:50c0:8001::154, 2606:50c0:8003::154, 2606:50c0:8002::154, ...
Verbindungsaufbau zu raw.githubusercontent.com (raw.githubusercontent.com)|2606:50c0:8001::154|:443 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 404 Not Found
2022-07-06 21:30:06 FEHLER 404: Not Found.

Download failed - "https://raw.githubusercontent.com/Freetz-NG/dl-mirror/master/kernel_2.6.32.61-7330_06.50-r19566.patch.xz"  ->  error code 8
make[1]: *** [make/linux/kernel.mk:54: dl/kernel_2.6.32.61-7330_06.50-r19566.patch.xz] Fehler 1
make: *** [Makefile:46: envira] Beendet
```

freetz-ng-mod, 2022-07-06:
Dir soll das nur sagen, dass du dich darum kümmern musst das. Ich habe das schon verstanden, was das bedeutet.

freetz-ng-mod, 2022-07-07:
Du bist öfters mal sehr herablassend. Und das ist eigentlich nicht gut. Da du auch sehr oft komische dinge, hier schreibst. Aber auch egal. Du machst das schon

freetz-ng-mod, 2022-07-07:
So wie man in Wald ruf, so kommt es auch zurück. Denke immer dran.

fda77, 2022-07-07:
Nix ist durch. Du benimmst dich ständig daneben, und heulst dann auch noch von wegen "herablassend" herum wenn du die passende Antwort bekommst.