Freetz-NG/freetz-ng #583: DNSmasq im Menü von FreetzNG

webfalter, 2022-07-07:
https://www.ip-phone-forum.de/threads/freetz-ng-7-39-auf-7590.311697/#post-2474451

Gibt es schon eine Lösung bei aktueller Labor der 7590 DNSmasq im Menü und das mit FreetzNG wieder geht ?
Habe es auch schon mit unterschiedlichen Einstellungen von DNS over TLS (DoT) versucht.

webfalter, 2022-07-07:
> Ich kann mit der "Fehlerbschreibung" dort nichts anfangen. Allgemein:
> 
>     * wenn dnsmasq gar nicht im webif ist, hat man es normal nicht mit eingebaut, bzw das falsche image von seinem imagedealer erwischt
 Mein Imagedealer wärst Du im meinem Fall

[image]

fda77, 2022-07-07:
Mit Imagedealer meinte ich Foren in denen es fertige Images gibt, bei denen man nicht genau weiss welche Pakete drin sind.
Als nächstes wären Logs von freetz interessant, zb /var/log/freetz.log oder im webif ob freetz das startet und ob da Fehlermeldungen kommen. Und wie du dnsmasq eingestellt hast

freetz-ng-mod, 2022-07-08:
Nur so zur Infomotion. Es wird mit gebaut, und es ist auf der Box. Nur wird es nicht benutzt. Das ist zwar von der 7530AX aber auch mit einer inhaus Version
[image]
[image]

pfichtner, 2022-07-08:
Ist dnsmasq eventuell "externalized"? 
Ich habe das mal bei einem Service gehabt, die Verbindug von "externalized" und "vor multid starten" führt dazu, dass der Service gar nicht gestartet wurde, da das initscript vermutlich nicht vor multid aufgerufen wurde und beim späteren Aufruf zurecht aussteigt. 
Vielleicht ist das auch hier der Fall und falls das ein generells Problem ist (vermute ich), dass die Kombination "vor multid starten" und "externalized" nicht funktioniert (was logisch und nachvollziehbr ist) könnte/sollte man überlegen, welche Möglichkeiten es gibt Anwender vor dieser dysfunktionalen Kombination zu schützen.

fda77, 2022-07-08:
Wahrscheinlich hat dnsmasq deshalb keine external Unterstützung :)
@freetz-ng-mod Wie, bei dir fehlt dnsmasq auch im Webif? Das hat eigentlich nichts mit der labor zu tun, das das Freetz-webif immer gleich ist. Ansonsten gilt das gleiche wie in https://github.com/Freetz-NG/freetz-ng/issues/583#issuecomment-1178364606

pfichtner, 2022-07-09:
"Deshalb"? "Wahrscheinlich" wegen was?

BlackStar999, 2022-07-09:
Was ich dazu sagen kann ist das ich DNSMASQ bis Version FritzOS 7.21 problemlos nutzen konnte

BlackStar999, 2022-07-10:
habe das selbe Problem bei folgenden Boxen:
FB7590
FB7530
FB7530 AX

pfichtner, 2022-07-11:
> Wahrscheinlich hat dnsmasq deshalb keine external Unterstützung

Also gehst Du davon aus, dass sich der Maintainer Gedanken gemacht hat, was passiert, wenn man "externializen" und "vor multid starten" gleichzeitig aktiviert, sich dieser Gefahr entsprechend bewusst war und aus dem Grund "externializen" für das Paket generell nicht anbietet. Dann hat der Maintainer aber arg weit gedacht und vielleicht zu vorsichtig, denn wenn dnsmasq "externalized" wäre würde es ja problemlos funktionieren, solange man "vor multid starten" nicht aktiviert. Halte ich für gewagt die Vermutung, dass sich der Maintainer deswegen gegen die Option des "externalizens" entschieden hat, aber wenn Du sagst, dass es wahrscheinlich deshalb war, dann glaube ich das natürlich. 

Wenn dem so ist: Maintainer anderer Pakete haben sich diese Gedanken offensichtlich nicht gemacht, darauf wollte ich hinweisen und die Diskussion anregen, ob man generell 
- die Option zum "externalizen" aus allen Paketen entfernt die potentiell vor multid gestartet werden können (fände ich schade/unklug)
- es einfach weiterhin Pech des Nutzers bleibt, wenn er das konfiguriert
- man es in die Doku aufnimmt, wenn nicht bereits geschehen (trotzdem bleibt die Gefahr groß dass Nutzer in dieses Problem laufen)
- man beim Herstellen einer deartigen Konfiguration dem Nutzer eines auf die Finger haut (wäre m.E. die beste Option)

fda77, 2022-07-14:
Ohne Logs ist nicht zu machen
@pfichtner: Frag doch den @Maintainer selbst

pfichtner, 2022-07-14:
Ich weiß, dass es bei unbound anders ist, dass unbound nicht startet, wenn man es externalized und "vor multid starten" aktiviert. Da brauche ich nix nachgucken. 
D.h. entweder man lebt damit, dass sich bei verschiedenen Diensten die Nutzer selbst in den Fuß schießen können oder man überlegt, wie es anders sein sollte. **Dann** kann man schauen, welche Dienste das betrifft und ob das (einer) dann zentral macht/machen kann oder ob es der jeweilige Maintainer tun sollte.

fda77, 2022-07-14:
Ist so. Ich versteh den Zusammenhang mit den Log trotzdem nicht

pfichtner, 2022-07-14:
Welches Log?

webfalter, 2022-07-15:
07.29 rev92051
```
/var/log/freetz.log 
1970-01-01 01:00:14 [   MODLOAD     ] Loading env ... done.
1970-01-01 01:00:14 [   MODLOAD     ] Loading flash ... done.
1970-01-01 01:00:14 [   MODLOAD     ] Loading users ... done.
1970-01-01 01:00:15 [   MODLOAD     ] Loading hosts ... done.
1970-01-01 01:00:16 [   MODLOAD     ] Loading config ... done.
1970-01-01 01:00:16 [   MODLOAD     ] Loading modules ... done.
1970-01-01 01:00:40 [  AVM-MULTID   ] Starting Dnsmasq ... done.
1970-01-01 01:00:45 [  AVM-MULTID   ] Starting AVM-multid ... done.
1970-01-01 01:00:47 [   AVM-DSLD    ] Starting AVM-dsld ... done.
1970-01-01 01:00:49 [    RCMOD      ] Freetz version ng-19890-0fb44ae61
1970-01-01 01:00:49 [    RCMOD      ] Crond is disabled.
1970-01-01 01:00:49 [    RCMOD      ] Starting WebCFG ... done.
1970-01-01 01:00:50 [    RCMOD      ] Syslogd is disabled.
1970-01-01 01:00:51 [    RCMOD      ] Starting Inetd ... done.
1970-01-01 01:00:52 [    RCMOD      ] Setting up SSH authorized-keys for root ... done.
1970-01-01 01:00:52 [    RCMOD      ] Setting up CA-bundle ... done.
1970-01-01 01:00:57 [    RCMOD      ] Starting callmonitor ... done.
1970-01-01 01:00:58 [    RCMOD      ] Setting up Addhole ... done.
1970-01-01 01:00:58 [    RCMOD      ] Bftpd is disabled.
1970-01-01 01:00:58 [    RCMOD      ] Setting up onlinechanged scripting ... done.
1970-01-01 01:00:59 [    RCMOD      ] external detected on nand.
1970-01-01 01:00:59 [   EXTERNAL    ] Starting external (/var/media/ftp/uStor01/external):                  
1970-01-01 01:00:59 [   EXTERNAL    ] Waiting for mod-startup:
1970-01-01 01:01:01 [    RCMOD      ] Startup finished.
1970-01-01 01:01:01 [   EXTERNAL    ] done.
1970-01-01 01:01:02 [   EXTERNAL    ] Dropbear is disabled.
1970-01-01 01:01:03 [   UDEVMOUNT   ] /dev/sda1 - mounting started
1970-01-01 01:01:03 [   EXTERNAL    ] privoxy is disabled.
1970-01-01 01:01:04 [   EXTERNAL    ] Starting Samba-smbd ... done.
1970-01-01 01:01:04 [   EXTERNAL    ] Tor Onion Router is disabled.
1970-01-01 01:01:04 [   EXTERNAL    ] Starting external finished.
1970-01-01 01:01:05 [   UDEVMOUNT   ] /dev/sda1 - mounted as /var/media/ftp/FritzBox (ext2)                 
1970-01-01 01:02:06 [ ONLINECHANGED ] [6922-online] approved
1970-01-01 01:02:06 [ ONLINECHANGED ] [6922-online] executing /etc/onlinechanged/00-get_ip                  
1970-01-01 01:02:06 [ ONLINECHANGED ] [6922-online] executing /tmp/flash/onlinechanged/onlinechanged-cgi    
1970-01-01 01:02:06 [ ONLINECHANGED ] [6922-online] executing /etc/onlinechanged/webdav_net                 
1970-01-01 01:02:06 [ ONLINECHANGED ] [6922-online]  * webdav_net online status_onlinev4                    
1970-01-01 01:02:06 [ ONLINECHANGED ] [6922-online] finished

```
07.39 rev97388
```
1970-01-01 01:00:15 [   MODLOAD     ] Loading env ... done.
1970-01-01 01:00:15 [   MODLOAD     ] Loading flash ... done.
1970-01-01 01:00:15 [   MODLOAD     ] Loading users ... done.
1970-01-01 01:00:16 [   MODLOAD     ] Loading hosts ... done.
1970-01-01 01:00:17 [   MODLOAD     ] Loading config ... done.
1970-01-01 01:00:17 [   MODLOAD     ] Loading modules ... done.
1970-01-01 01:00:18 [    RCMOD      ] Freetz version ng-19940-3b2588167
1970-01-01 01:00:18 [    RCMOD      ] Crond is disabled.
1970-01-01 01:00:19 [    RCMOD      ] Starting WebCFG ... done.
1970-01-01 01:00:19 [    RCMOD      ] Syslogd is disabled.
1970-01-01 01:00:20 [    RCMOD      ] Starting Inetd ... done.
1970-01-01 01:00:20 [    RCMOD      ] Setting up SSH authorized-keys for root ... done.
1970-01-01 01:00:21 [    RCMOD      ] Setting up CA-bundle ... done.
1970-01-01 01:00:30 [    RCMOD      ] Starting callmonitor ... done.
1970-01-01 01:00:31 [    RCMOD      ] Setting up Addhole ... done.
1970-01-01 01:00:31 [    RCMOD      ] Bftpd is disabled.
1970-01-01 01:00:31 [    RCMOD      ] Setting up onlinechanged scripting ... done.
1970-01-01 01:00:31 [    RCMOD      ] Updating lfs cache ... asynchronous.
1970-01-01 01:00:31 [    RCMOD      ] external detected on nand.
1970-01-01 01:00:31 [   EXTERNAL    ] Starting external (/var/media/ftp/uStor01/external):
1970-01-01 01:00:31 [   EXTERNAL    ] Waiting for mod-startup:
1970-01-01 01:00:34 [    RCMOD      ] Startup finished.
1970-01-01 01:00:35 [   EXTERNAL    ] done.
1970-01-01 01:00:37 [   EXTERNAL    ] Dropbear is disabled.
1970-01-01 01:00:37 [   EXTERNAL    ] privoxy is disabled.
1970-01-01 01:00:39 [   EXTERNAL    ] Starting Samba-smbd ... done.
1970-01-01 01:00:40 [   EXTERNAL    ] Tor Onion Router is disabled.
1970-01-01 01:00:40 [   EXTERNAL    ] Starting external finished.
1970-01-01 01:00:56 [   UDEVMOUNT   ] /dev/sda1 - mounting started
1970-01-01 01:00:58 [   UDEVMOUNT   ] /dev/sda1 - mounted as /var/media/ftp/FritzBox (ext2)
1970-01-01 01:01:06 [   AVM-DSLD    ] Stopping AVM-dsld ... done.
2022-07-15 17:47:35 [ ONLINECHANGED ] [8466-online] approved
2022-07-15 17:47:35 [ ONLINECHANGED ] [8466-online] executing /etc/onlinechanged/00-get_ip
2022-07-15 17:47:35 [ ONLINECHANGED ] [8466-online] executing /tmp/flash/onlinechanged/onlinechanged-cgi
2022-07-15 17:47:35 [ ONLINECHANGED ] [8466-online] executing /etc/onlinechanged/start_smbd
2022-07-15 17:47:37 [ ONLINECHANGED ] [8466-online] executing /etc/onlinechanged/webdav_net
2022-07-15 17:47:37 [ ONLINECHANGED ] [8466-online]  * webdav_net online status_onlinev4
2022-07-15 17:47:38 [ ONLINECHANGED ] [8466-online] finished
```

fda77, 2022-07-15:
Danke!!!
Offenbar funktionieren die wrapper-scripte (auf für dsld) in freetz nicht mehr: https://github.com/Freetz-NG/freetz-ng/blob/master/patches/scripts/106-wrapper_multid.sh
Ich vermute avm startet den multid nicht mehr aus der rc.net heraus
Dadurch wird das dnsmasq script überhaupt nicht ausgeführt, und das package registiert sich auch nicht im webif

PaulchenPlump, 2022-08-12:
Habe heute eine Inhouse-Version auf einer 7490 mit einem aktuellen Freetz-NG getestet: leider war kein Eintrag für Dnsmasq vorhanden. Kann das Problem bitte behoben werden?!

fda77, 2022-08-12:
Ist wirklich ein beschissener Service hier

webfalter, 2022-08-12:
Lol auch eine Art Probleme zu lösen  !!! Vielleicht liegt es am Image Dailer oder an deinem Bug Level worksforme

fda77, 2022-08-14:
Die Ursache steht doch oben. Nicht gelesen oder nicht verstanden?
Issues die niemanden interessieren werden halt automatisch geschlossen. Sonst endet es wie freetz.org mit 300 offenen tickets

PaulchenPlump, 2022-08-14:
Hallo @fda77, Hallo @webfalter,
ich kann den rauen Ton in keiner Weise nachvollziehen. Ich habe den Fall wieder eröffnet, da eine potenzielle Ursache genannt wurde, aber bisher keine Lösung damit einhergeht (oder habe ich es wirklich nicht verstanden?). Da ich mit dem Projekt wenig vertraut bin, kann ich es leider nicht einschätzen, geschweige denn eine Lösung beisteuern und bitte um Unterstützung.
Jetzt hoffe ich, dass wir zum konstruktiven Teil des Tickets zurückkehren können und das Problem aus der Welt schaffen können, statt uns hier "anzupflaumen".

fda77, 2022-08-15:
Es ist in der Freetz "community" normal dass herumgepflaumt wird. Das Schema hier ist wie üblich: jeder will, niemand macht.
Daher werden Issues die niemanden interessieren nach 2 Wochen Inaktivität geschlossen. Wie man oben sieht ist es für die meisten schon zuviel eine Logdatei hier anzuhängen. Ein Haufen ich-will Issues helfen niemandem.
Du könntest zb zu deinem lokalen Code-Dealer gehen und ihm sagen dass er dein Problem zu behebe hat. Danach macht dieser einen PR und alle können über was anderes herumpflaumen

closed 2022-09-01 by github-actions[bot]