Freetz-NG/freetz-ng #587: Probleme mit services file

fesc2000, 2022-07-10:
Im Zuge dieses pull-requests https://github.com/Freetz-NG/freetz-ng/pull/586 sehe ich mit dem /etc/services file von freetz einige Probleme:

- portmapper alias fehlt (siehe pull-reuquest), und auch viele andere (einfach mal ein diff machen), z.B.
chargen         19/tcp          ttytst source

- Einige fehlen komplett, z.B.
fsp             21/udp          fspd

- Einige service-kommentare sind nicht als kommentar markiert, z.B.
ssdp            1900/udp        Simple Service Discovery Protocol
pcp             5351/udp        Port Control Protocol
mdns            5353/tcp        Multicast Domain Name Service
mdns            5353/udp        Multicast Domain Name Service
llmnr           5355/udp        Link Local Multicast Name Resolution
...
und so weiter. D.h. die Kommentare werden also als alias interpretiert, mit werweis was für Effekten

- Default nfs4 port 2049 ist auskommentiert und durch 2047 ersetzt.

Wie man am pull-request sieht ist es nicht ohne Probleme einfach das komplette services-file zu ersetzen. Wäre es nicht besser die benötigten services dynamisch hinzuzufügen anstatt einfach alles zu ändern? Wie man sieht wird so manches intern benötigt.

Vielleicht als Anregung, Ich hatte mal ein skript geschrieben um services-files zu mergen:
[merge-services.gz](https://github.com/Freetz-NG/freetz-ng/files/9079413/merge-services.gz)

fda77, 2022-07-10:
Danke für den Quickfix!

>  einfach das komplette services-file zu ersetzen

Dieser Ansatz ist falsch. AVM nutzt die services Datei erst seit FOS 7.1x, davor gab es sie nur wenn man sie in Freetz auswählte, zB für NFS. Ein Workaound für das flash-Problem wäre auch gewesen die Option im menuconfig einfach nicht auzuwählen.

AVM nutzt aktuell 3 verschieden Dateien:
```
f15064ab952d3ebfc99df3077697658c  FRITZ.Box_7581-07.16/etc/services
4995ba7456dc1713b3d8804b0ff670b4  FRITZ.Box_7590-07.12/etc/services
9ebdc6f90a6070a1cdac5ecde97fbd33  FRITZ.Box_7530_AX-07.39-97402-LabBETA/etc/services
```

Also für 7581+7582, 7.39 inhaus+labor und manche (oder alle ???) ab FOS 7.1x

Man könnte die Dateien irgendwie mergen, evtl auch mit https://salsa.debian.org/md/netbase/-/blob/master/etc/services Das müsste man in Zukunft aber auch öfter mal machen...
Oder man bietet die hinzufüg-Option nur für Geräte ohne die Datei an (alte bekommen sie wie sie war, ging ja..), und appended Freetz-spezifische Dinge nur wenn es sie gibt

Btw, für die Puma7 hatte ich nfsd mal ausgemacht, weil avm schon einen Laufen hatte und es Problem gab

fda77, 2022-07-10:
Falls du Freetz auf einer Puma7 hast, könntest du b14d90966c4f86f44722ec598c993aa5fff9929f mal testen, davor aber d1c8224b5b0f70810ada57c848693ee0d9c7401c rückgängig machen

fesc2000, 2022-07-10:
Nein habe ich leider nicht, an meiner "produktivbox" kann ich im Augenblick nicht herumfummeln.

fda77, 2022-07-11:
Schade, es ist schwer Leute zu finden die imstande sind das zu testen und eine mögliche Fehlermeldung hier zu posten

Das sollte das services-Problem beheben

* 6826d8084 rename FREETZ_AVM_HAS_SERVICES
* d9b8f8d41 use FREETZ_AVM_HAS_ETCSERVICES
* f2fb7fd94 move nfs to add-patch
* 6e54b567a add custom etcservices
* 3113489ca convert etcservices to patch
* 3801e5858 props: add etc/services

closed 2022-07-11 by fda77