Freetz-NG/freetz-ng #604: bump libpng 1.2.59 to 1.6.37

freetz-ng-mod, 2022-08-12:
```
--- make/libs/libpng/libpng.mk_	2022-08-11 17:31:33.559017000 +0200
+++ make/libs/libpng/libpng.mk	2022-08-12 07:26:05.533833994 +0200
@@ -1,12 +1,12 @@
-$(call PKG_INIT_LIB, 1.2.59)
-$(PKG)_LIB_VERSION:=0.59.0
+$(call PKG_INIT_LIB, 1.6.37)
+$(PKG)_LIB_VERSION:=16.37.0
 $(PKG)_SOURCE:=$(pkg)-$($(PKG)_VERSION).tar.xz
-$(PKG)_HASH:=b4635f15b8adccc8ad0934eea485ef59cc4cae24d0f0300a9a941e51974ffcc7
+$(PKG)_HASH:=505e70834d35383537b6491e7ae8641f1a4bed1876dbfe361201fc80868d88ca
 $(PKG)_SITE:=@SF/libpng
 
-$(PKG)_BINARY:=$($(PKG)_DIR)/.libs/libpng12.so.$($(PKG)_LIB_VERSION)
-$(PKG)_STAGING_BINARY:=$(TARGET_TOOLCHAIN_STAGING_DIR)/usr/lib/libpng12.so.$($(PKG)_LIB_VERSION)
-$(PKG)_TARGET_BINARY:=$($(PKG)_TARGET_DIR)/libpng12.so.$($(PKG)_LIB_VERSION)
+$(PKG)_BINARY:=$($(PKG)_DIR)/.libs/libpng16.so.$($(PKG)_LIB_VERSION)
+$(PKG)_STAGING_BINARY:=$(TARGET_TOOLCHAIN_STAGING_DIR)/usr/lib/libpng16.so.$($(PKG)_LIB_VERSION)
+$(PKG)_TARGET_BINARY:=$($(PKG)_TARGET_DIR)/libpng16.so.$($(PKG)_LIB_VERSION)
 
 $(PKG)_DEPENDS_ON += zlib
 
@@ -31,9 +31,9 @@
 		DESTDIR="$(TARGET_TOOLCHAIN_STAGING_DIR)" \
 		install
 	$(PKG_FIX_LIBTOOL_LA) \
-		$(TARGET_TOOLCHAIN_STAGING_DIR)/usr/lib/libpng12.la \
-		$(TARGET_TOOLCHAIN_STAGING_DIR)/usr/lib/pkgconfig/libpng12.pc \
-		$(TARGET_TOOLCHAIN_STAGING_DIR)/usr/bin/libpng12-config
+		$(TARGET_TOOLCHAIN_STAGING_DIR)/usr/lib/libpng16.la \
+		$(TARGET_TOOLCHAIN_STAGING_DIR)/usr/lib/pkgconfig/libpng16.pc \
+		$(TARGET_TOOLCHAIN_STAGING_DIR)/usr/bin/libpng16-config
 
 $($(PKG)_TARGET_BINARY): $($(PKG)_STAGING_BINARY)
 	$(INSTALL_LIBRARY_STRIP)
@@ -48,7 +48,7 @@
 		$(TARGET_TOOLCHAIN_STAGING_DIR)/usr/lib/pkgconfig/libpng*.pc \
 		$(TARGET_TOOLCHAIN_STAGING_DIR)/usr/bin/libpng*-config \
 		$(TARGET_TOOLCHAIN_STAGING_DIR)/usr/include/png*.h \
-		$(TARGET_TOOLCHAIN_STAGING_DIR)/usr/include/libpng12 \
+		$(TARGET_TOOLCHAIN_STAGING_DIR)/usr/include/libpng16 \
 		$(TARGET_TOOLCHAIN_STAGING_DIR)/usr/share/man/man3/libpng*.3 \
 		$(TARGET_TOOLCHAIN_STAGING_DIR)/usr/share/man/man5/png.5
 
--- make/libs/libpng/Config.in	2022-08-11 17:31:33.559017000 +0200
+++ make/libs/libpng/Config.in	2022-08-12 07:26:42.833464754 +0200
@@ -1,4 +1,4 @@
-config FREETZ_LIB_libpng12
+config FREETZ_LIB_libpng16
 	bool "PNG library (libpng.so)"
 	select FREETZ_LIB_libm if FREETZ_TARGET_UCLIBC_HAS_multiple_libs
 	select FREETZ_LIB_libz
--- make/libs/libpng/external.files	2022-08-11 17:31:33.559017000 +0200
+++ make/libs/libpng/external.files	2022-08-12 07:26:21.585674663 +0200
@@ -1 +1 @@
-[ "$EXTERNAL_FREETZ_LIB_libpng12" == "y" ] && EXTERNAL_FILES+=" ${FREETZ_LIBRARY_DIR}/libpng12.so.0.59.0"
+[ "$EXTERNAL_FREETZ_LIB_libpng16" == "y" ] && EXTERNAL_FILES+=" ${FREETZ_LIBRARY_DIR}/libpng16.so.16.37.0"
--- make/libs/libpng/external.in	2022-08-11 17:31:33.559017000 +0200
+++ make/libs/libpng/external.in	2022-08-12 07:27:37.500929503 +0200
@@ -1,8 +1,8 @@
-config EXTERNAL_FREETZ_LIB_libpng12
-	depends on EXTERNAL_ENABLED && FREETZ_LIB_libpng12
-	bool "libpng12"
+config EXTERNAL_FREETZ_LIB_libpng16
+	depends on EXTERNAL_ENABLED && FREETZ_LIB_libpng16
+	bool "libpng16"
 	default n
 	help
 		externals the following file(s):
-		 ${FREETZ_LIBRARY_DIR}/libpng12.so.0.59.0
+		 ${FREETZ_LIBRARY_DIR}/libpng16.so.16.37.0
 
--- make/libs/libpng/Makefile.in	2022-08-11 17:31:33.559017000 +0200
+++ make/libs/libpng/Makefile.in	2022-08-12 07:28:45.428273299 +0200
@@ -1,3 +1,3 @@
-ifeq ($(strip $(FREETZ_LIB_libpng12)),y)
+ifeq ($(strip $(FREETZ_LIB_libpng16)),y)
 LIBS+=libpng
 endif
--- make/rrdtool/Config.in.libs	2022-08-11 17:31:33.651015000 +0200
+++ make/rrdtool/Config.in.libs	2022-08-12 07:29:57.087590412 +0200
@@ -3,7 +3,7 @@
 	select FREETZ_LIB_libart_lgpl_2
 	select FREETZ_LIB_libfreetype
 	select FREETZ_LIB_libm        if FREETZ_TARGET_UCLIBC_HAS_multiple_libs
-	select FREETZ_LIB_libpng12
+	select FREETZ_LIB_libpng16
 	select FREETZ_LIB_libz
 	default n
 	help
@@ -15,7 +15,7 @@
 	select FREETZ_LIB_libart_lgpl_2
 	select FREETZ_LIB_libfreetype
 	select FREETZ_LIB_libm        if FREETZ_TARGET_UCLIBC_HAS_multiple_libs
-	select FREETZ_LIB_libpng12
+	select FREETZ_LIB_libpng16
 	select FREETZ_LIB_libz
 	default n
 	help 
--- make/imagemagick/Config.in	2022-08-11 17:31:33.547017000 +0200
+++ make/imagemagick/Config.in	2022-08-12 07:51:19.244758952 +0200
@@ -39,7 +39,7 @@
 config FREETZ_PACKAGE_IMAGEMAGICK_png
 	bool "Support png"
 	default y
-	select FREETZ_LIB_libpng12
+	select FREETZ_LIB_libpng16
 
 config FREETZ_PACKAGE_IMAGEMAGICK_xml
 	bool "Support xml" 
--- make/netpbm/Config.in	2022-08-11 17:31:33.631015000 +0200
+++ make/netpbm/Config.in	2022-08-12 07:53:14.675042981 +0200
@@ -30,7 +30,7 @@
 config FREETZ_PACKAGE_NETPBM_PNGTOPNM
 	bool "Include pngtopnm"
 	depends on FREETZ_PACKAGE_NETPBM
-	select FREETZ_LIB_libpng12
+	select FREETZ_LIB_libpng16
 	select FREETZ_LIB_libz
 	default y
 	help
@@ -47,7 +47,7 @@
 config FREETZ_PACKAGE_NETPBM_PNMTOPNG
 	bool "Include pnmtopng"
 	depends on FREETZ_PACKAGE_NETPBM
-	select FREETZ_LIB_libpng12
+	select FREETZ_LIB_libpng16
 	select FREETZ_LIB_libz
 	default y
 	help	
```

fda77, 2022-08-12:
Die Programme die die png nutzen sind danach auch noch compilierbar? Auch für verschieden Gerätegenerationen wie 3020/7390/7590/6590?

Du hast übersehen:
```
/make/libs/leptonica/Config.in:        select FREETZ_LIB_libpng12
```

und https://freetz-ng.github.io/freetz-ng/SUPPORT.html#package-bump

freetz-ng-mod, 2022-08-12:
OHHH Sorry, dass ich das vergessen habe.
Ich weiß, dass es auf den 7490 7530AX 7590AX 1a lauft

harryboo, 2022-08-12:
Schonmal probiert NETPBM zu kompilieren?

Der Rest funktioniert...

closed 2022-08-28 by github-actions[bot]