Freetz-NG/freetz-ng #614: Fehler: Datei 'crt' des Pakets 'lighttpd' ist unbekannt.

hannsen, 2022-09-17:
Wenn man in der [lighttpd Konfiguration](http://fritz.box:81/cgi-bin/conf/lighttpd) auf "_Damit lighttpd mit SSL-Unterstützung gestartet werden kann, müssen Zertifikat & Schlüssel [hier](http://fritz.box:81/cgi-bin/file/lighttpd/crt) eingetragen sein._" klickt kommt die Meldung "Fehler: Datei 'crt' des Pakets 'lighttpd' ist unbekannt." und man kann somit kein SSL konfigurieren.

 
[image]
[image]

[config.txt](https://github.com/Freetz-NG/freetz-ng/files/9591918/config.txt)

hannsen, 2022-09-18:
```
server.modules += ( "mod_openssl" )
$SERVER["socket"] == ":443" {
  ssl.engine = "enable" 
  ssl.pemfile = "/var/www/htdocs/etc/lighttpd.pem" 
}
```

Ich habs jetzt auch so versucht, aber dann meckert es, dass es kein mod_openssl.so gibt.

Wie konfiguriere ich denn SSL für lighttpd?

hannsen, 2022-09-18:
Hab jetzt diese Zeilen in `/make/lighttpd/Config.in` hinzugefügt und nochmal ein Image erstellt, damit scheint es zu gehen:
```
   	config FREETZ_PACKAGE_LIGHTTPD_MOD_OPENSSL
           	bool "mod_openssl"
          	default y
           	help
                   	This option adds mod_openssl.so to the image.
```

fda77, 2022-09-19:
Sollte nun funktionieren, keines der 3 möglichen ssl-Module wurde wegen falschem Symbolnamen mit ins Image kopiert.

closed 2022-09-19 by fda77