Freetz-NG/freetz-ng #627: compiler error "libtinfow.so.6: error adding symbols: DSO missing from command line"

ls-2020, 2022-10-05:
i tried to update freetz-ng from an older version but compiling failed on my system (gcc-11.3.0/ncurses-6.3) with this error:
"libtinfow.so.6: error adding symbols: DSO missing from command line"

The following patch fixed the error - please update - thanks!

```
--- mconf-cfg.sh.org	2022-10-02 23:09:07.000000000 +0200
+++ mconf-cfg.sh	2022-10-04 18:20:49.484400779 +0200
@@ -23,7 +23,7 @@
 # find ncurses by pkg-config.)
 if [ -f /usr/include/ncursesw/ncurses.h ]; then
 	echo cflags=\"-D_GNU_SOURCE -I/usr/include/ncursesw\"
-	echo libs=\"-lncursesw\"
+	echo libs=\"-lncursesw -ltinfow\"
 	exit 0
 fi
```

ls-2020, 2022-10-05:
this patch fixes "make menuconfig" and should be applied to the file
/freetz-ng/source/host-tools/kconfig-v6.0/scripts/kconfig/mconf-cfg.sh

fda77, 2022-10-05:
You are changing a line which exists since 2018: https://github.com/torvalds/linux/commit/1c5af5cf9308fff327f52c7efd2dfa732d370871 How old was the "older" version of Freetz?
Seems the order of libs is wrong somewhere: https://stackoverflow.com/questions/19901934/libpthread-so-0-error-adding-symbols-dso-missing-from-command-line
Maybe a Freetz-patche caused it? https://github.com/Freetz-NG/freetz-ng/tree/master/tools/make/kconfig-host/patches

ls-2020, 2022-10-05:
yes, i think a freetz-patch must have caused this bug. i did not update freetz frequently, but the last image i managed to build without this bug was 7490_07.29.all_freetz-ng-19667M-82a75cde8_20220606-183819.image
i hope this helps to track down the freetz-patch that causes the compiler error.
i found somewhere (i think it was on github) a discussion and explanation about this DSO error, but unfortunately i cannot find it now

ls-2020, 2022-10-06:
i removed all kconfig-host/patches except 300-* and 310-* from a fresh clone, run "make menuconfig" and got the same error, so i guess it is something else that has been changed.
i try to update freetz once a month and the last image i managed to build without this error was "freetz-ng-19667M"
i hope this helps to find what causes the compiler error. if not i can only say that my patch to mconf-cfg.sh fixes this error.

closed 2022-10-30 by github-actions[bot]

fda77, 2022-11-05:
Seems only your system (Debian?!?) needs this patch, as nobody else complins.
Btw, on my  device the 'tinfow' lib you suggested not installed / you patch wont work for me
```
$ pkgconf --libs ncursesw
-lncursesw -ltinfo

$ pkgconf --libs ncurses
-lncurses -ltinfo

$ pkgconf --libs tinfo
-ltinfo

$ pkgconf --libs tinfow
Package tinfow was not found in the pkg-config search path.
Perhaps you should add the directory containing `tinfow.pc'
to the PKG_CONFIG_PATH environment variable
Package 'tinfow', required by 'virtual:world', not found

```

fda77, 2022-11-05:
https://patchwork.kernel.org/project/linux-kbuild/patch/20150425160546.GA4002@freedom/

ls-2020, 2023-01-02:
@fda77 
i still have to use my patch to fix this issue :(
i have read [this](https://github.com/Freetz-NG/freetz-ng/issues/671#issuecomment-1348787294), but i don't understand if this is a freetz or gentoo bug and what i have todo to avoid using the patch!?

gentoo:
```
$ pkgconf --libs ncursesw
-lncursesw -ltinfow 
$ pkgconf --libs ncurses
-lncurses -ltinfo 
$ pkgconf --libs tinfo
-ltinfo 
$ pkgconf --libs tinfow
-ltinfow 
$ pkgconf --libs tinfo
-ltinfo 
```

fda77, 2023-01-04:
I dont know why on very few devices the libs are not like on most oder systems

ls-2020, 2023-01-08:
@fda77 
as you closed this issue without a working fix please check out my new patch. i think this issue is caused by bad coding style in mconf-cfg.sh.  read the comment - they simply hardcoded values for libs because some distributions cannot find ncurses by pkg-config.  I changed the logic: use the hardcoded value only if pkgconf fails and returns an empty string. This works fine for me and should work for you and anybody else. Just check it out - i really hopy you add this patch to finally fix this annoying issue - thanks!

```
--- mconf-cfg.sh.org	2023-01-02 14:47:47.550346280 +0100
+++ mconf-cfg.sh	2023-01-08 09:39:40.867519496 +0100
@@ -23,8 +23,13 @@
 # find ncurses by pkg-config.)
 if [ -f /usr/include/ncursesw/ncurses.h ]; then
 	echo cflags=\"-D_GNU_SOURCE -I/usr/include/ncursesw\"
-#	echo libs=\"-lncursesw\"
-echo libs=\"-lncursesw -ltinfow\"
+LIBS=$(pkgconf --libs ncursesw | xargs)
+if [ -z "$LIBS" ]; then
+	echo libs=\"-lncursesw\"
+else
+	echo libs=\"$LIBS\"
+fi
+
 	exit 0
 fi
```

fda77, 2023-01-08:
Thx, i've changed syntax like in the head of source/host-tools/kconfig-v6.1/scripts/kconfig/mconf-cfg.sh
```
        if ${HOSTPKG_CONFIG} --exists $PKG; then
                echo cflags=\"$(${HOSTPKG_CONFIG} --cflags $PKG)\"
                echo libs=\"$(${HOSTPKG_CONFIG} --libs $PKG)\"
                exit 0
        fi
```
and addes it also no ncurses-without-wide.

Btw, this kconfig is from kernel 6.1. Can you download and run menuconfig/kconfig with it? If not you could report it upstream

fda77, 2023-01-13:
Thx. Maybe the kernel has set `HOSTPKG_CONFIG` (https://github.com/Freetz-NG/freetz-ng/issues/627#issuecomment-1374932571) and so the static `libs=` below are not used, but where in Freetz

> Check the default paths in case pkg-config is not installed.

```
$ pkgconf --exists ncurses ; echo $?
0

$ pkgconf --exists xxxx ; echo $?
1
```

So just setting `HOSTPKG_CONFIG` and remove the patch could be a better fix - and a fix for the other too:
```
grep HOSTPKG_CONFIG source/host-tools/kconfig-v6.1/ -r -l
source/host-tools/kconfig-v6.1/scripts/kconfig/qconf-cfg.sh
source/host-tools/kconfig-v6.1/scripts/kconfig/nconf-cfg.sh
source/host-tools/kconfig-v6.1/scripts/kconfig/gconf-cfg.sh
source/host-tools/kconfig-v6.1/scripts/kconfig/mconf-cfg.sh

```

fda77, 2023-01-13:
@ls-2020 Can you please test if it still works for you after PULL +  `make kconfig-host-recompile` ?

ls-2020, 2023-01-15:
> PULL + `make kconfig-host-recompile`
I tried this and got a strange error. This command tried to execute a non existent command 'Gentoo' and returned an error and 'make clean' returned the same error. Don't know what was wrong here, so i removed /freetz-ng and used a fresh clone of freetz-ng. Then i copied my .config , executed 'make oldconfig' and then 'make menuconfig'. This is how i always build freetz and it worked. So, yes this fix works for me (freetz-ng-20924M)

fda77, 2023-01-15:
Great :)
Only "pull" may be to less, if you have local changes, commits etc. This cleans everything local from clone: https://github.com/Freetz-NG/freetz-ng#delete-local-changes