Freetz-NG/freetz-ng #631: Error compiling minisatip: socketworks.c:1299:8: error: redefinition of 'struct mmsghdr'

fre73, 2022-10-22:
I am trying to compile freetz-ng for 6490 cable with minisatip using ready2go Virtualbox image.
Compiling aborts at the following:

> socketworks.c:1299:8: error: redefinition of 'struct mmsghdr'
>  struct mmsghdr
>         ^
> In file included from /home/freetz/freetz-ng/toolchain/build/i686_gcc-5.5.0_uClibc-1.0.42-nptl/i686-linux-uclibc/include/netinet/in.h:24:0,
>                  from /home/freetz/freetz-ng/toolchain/build/i686_gcc-5.5.0_uClibc-1.0.42-nptl/i686-linux-uclibc/include/arpa/inet.h:22,
>                  from socketworks.c:31:
> /home/freetz/freetz-ng/toolchain/build/i686_gcc-5.5.0_u
> [config.txt](https://github.com/Freetz-NG/freetz-ng/files/9843485/config.txt)
> Clibc-1.0.42-nptl/i686-linux-uclibc/include/sys/socket.h:102:8: note: originally defined here
>  struct mmsghdr
>         ^
> socketworks.c:1305:5: error: conflicting types for 'sendmmsg'
>  int sendmmsg(int rsock, struct mmsghdr *msg, int len, int t)
>      ^
> In file included from /home/freetz/freetz-ng/toolchain/build/i686_gcc-5.5.0_uClibc-1.0.42-nptl/i686-linux-uclibc/include/netinet/in.h:24:0,
>                  from /home/freetz/freetz-ng/toolchain/build/i686_gcc-5.5.0_uClibc-1.0.42-nptl/i686-linux-uclibc/include/arpa/inet.h:22,
>                  from socketworks.c:31:
> /home/freetz/freetz-ng/toolchain/build/i686_gcc-5.5.0_uClibc-1.0.42-nptl/i686-linux-uclibc/include/sys/socket.h:193:16: note: previous declaration of 'sendmmsg' was here
>  extern ssize_t sendmmsg (int __fd, struct mmsghdr *__vmessages,
>                 ^
> make[3]: *** [Make
> [config.txt](https://github.com/Freetz-NG/freetz-ng/files/9843484/config.txt)
> file:156: ../build/socketworks.o] Error 1
> make[3]: *** Waiting for unfinished jobs....
> make[2]: *** [Makefile:11: minisatip] Error 2
> make[1]: *** [make/pkgs/minisatip/minisatip.mk:47: source/target-i686_gcc-5.5.0_uClibc-1.0.42-nptl/minisatip-1.0.4/minisatip] Terminated
> make: *** [Makefile:46: envira] Terminated

CaptainJackDaniels, 2022-10-28:
make/pkgs/minisatip/patches/ and delete 100-fix-missing-sendmmsg.patch Then it should work. Since the patch only makes crap

fre73, 2022-10-29:
@CaptainJackDaniels Thank you. 
I now flashed Freetz with minisatip succesfull.
It is also listed at packages in Mod-Info but I can't find an option in Webinterface, and couldn't find process in process manager.

Can someone help :)

fre73, 2022-10-31:
If I try to run minisatip from command line I get:
"Segmentation fault"
Same if I add to rc.custom..

Any recommendations to resolve my issue?

fda77, 2022-11-06:
@fre73: There is no webif...
You could use "strace" to see during which command it crashes
Is the "100-fix-missing-sendmmsg.patch" file still needed for devices with older kernel/etc ?
--
There are 100s new release versions of minisatip within last months: https://github.com/catalinii/minisatip/releases

fre73, 2022-11-06:
Hello,
could you please tell me more, how to compile/use new minisatip version in freetz with minimal afford? :)
Here is strace:

`root@fritz:/var/tmp# strace minisatip -f -l http
execve("/usr/bin/minisatip", ["minisatip", "-f", "-l", "http"], 0xbff792ac /* 304 vars */) = 0
readlinkat(AT_FDCWD, "/proc/self/exe", "/usr/bin/minisatip", 4096) = 18
mmap2(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb7851000
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=0x3c} ---
+++ killed by SIGSEGV (core dumped) +++
Segmentation fault
`
And core dump as attachment.
[core.tar.gz](https://github.com/Freetz-NG/freetz-ng/files/9946285/core.tar.gz)

fda77, 2022-11-06:
"strace -ff" shows also forks/childs

fda77, 2022-11-06:
>  new minisatip version in freetz with minimal afford?

Let someone else do it ;-)

fre73, 2022-11-12:
@fda77 
How to compile it by myself?
Do I need x86 for my FritzBox 6490 Cable, or advice to buy another FritzBox Cable modell which may work better?
Should be price range for used till ~70€.

fda77, 2022-11-14:
When you compile the whole image, the program is automatic compiled. The archtecture is the correct for the device you selected.
I dont sell avm devices

closed 2022-11-15 by fda77