Freetz-NG/freetz-ng #635: [external] external file is not created

harryboo, 2022-11-03:
version: latest build with clean checkout
host: ubuntu 20 lts

have loaded my config (last build on 4.10.2022) but the external file (usb.ids) will not created.

[b.config.txt](https://github.com/Freetz-NG/freetz-ng/files/9932890/b.config.txt)

aus-sua, 2022-11-05:
Ich hänge mich hier einmal mit an,
ich denke "mein" Problem ist auch sein Problem.
---> Es werden keine Externals-Files erstellt.

Ich denke, bei meiner 7390 sieht man genauer, wo das Problem liegt.

Hier, ein BC wo es (auch) noch funktioniert 
[image]

und hier eine BS mit den fehlenden externals 
[image]

Das Problem ist wohl mit den Änderung der Pfade (z.B. make: fix some links....) zwischen:
feb2cd728d9f7 (das entspricht dem letzte Tag [ng22100], damit funktioniertes noch)
und z.B.
20d577d8b53 (das ist gleich nach den Pfadänderungen, damit tritt das Problem auf)
entstanden.
Da greift wohl ein script noch auf die älteren Pfadangaben zu und findet damit wohl
die separierbaren externals nicht (mehr)?

Hier noch das cmake-File (als zip) dazu
[cmake.log_18.10._20d577d8b5_v.06.87-old__.zip](https://github.com/Freetz-NG/freetz-ng/files/9944352/cmake.log_18.10._20d577d8b5_v.06.87-old__.zip)

fda77, 2022-11-05:
Ja, das Problem kam vom Verschieben in make/ . Bitte testen, ich geh davon dass das alles war
Es ist durchaus möglich dass noch die eine oder andere Stelle gefunden wird die übersehen wurde

aus-sua, 2022-11-06:
Bei mir passt es damit da jetzt wieder, echt Knorke  ;-)

Vielen Dank für den schnellen Fix!
:+1:

harryboo, 2022-11-06:
Funktioniert wieder, Danke...

fda77, 2022-11-06:
Danke für die Rückemeldungen!

closed 2022-11-06 by fda77