Freetz-NG/freetz-ng #643: Package Minisatip: Segmentation fault

fre73, 2022-11-15:
Hello,

after compiling freetz-ng for my FritzBox 6490 Cable with main branch I keep getting "Segmentation fault" Error trying to run minisatip.
In my old Issue #631 about compiling error, cause of patch, I also uploaded my configurations and core dump.
Compilation is now fixed, but this error persists.

Any Advice?
Maybe Bump minisatip version?

Tnanks @fda77 for your hard work :)

fda77, 2022-11-15:
Yes, check with strace when it crashes, but the longer version:
`strace -fff minisatip -f -l http`
My guess: something with avm's kernel-config,
Hard work would be a bump to newest version ;-)

fre73, 2022-11-15:
Thanks, I didn't know that it will be so difficult to run non avm Sat2IP Server on Fritzbox.
I think I will stick with another solution with Raspberry Pi and DVB-C USB Stick..

If u want I can deliver the strace, but I will now stick with my new solution :)

fda77, 2022-11-16:
Everyting is possible, avm uses sometime custim (dvb) hardware with kernelpatches or whatever...
But to fix a problem you first need to know what the problem is

closed 2022-12-11 by github-actions[bot]