Freetz-NG/freetz-ng #649: Aktuelles Problem mit Erstellung von FIT-Image fuer 5590 und andere Boxen

maverick3005, 2022-11-26:
Aktuell lässt sich mit dem Trunk wg. geänderter Parameter von "mkimage" offenbar kein Image mehr mit "default-settings"
mehr erstellen:

```
STEP 3: PACK/SIGN
  checking for left over version-control-system files
  integrate freetz info file into image
packing var.tar
  checking signature key files
    adding public signature key file
creating filesystem image (SquashFS4-xz)
  SquashFS block size: 64 kB (65536 bytes)
copying kernel image
  kernel image size: 3.2 MB
copying filesystem image
  filesystem image size: 29.3 MB
packing fitimg file
sh: 1: dtc: not found
/usr/src/freetz/freetz-ng/tools/fit/mkimage: Can't open build/modified/fit-image/image.itb.tmp: No such file or directory
Error: Bad parameters for FIT image type
Usage: /usr/src/freetz/freetz-ng/tools/fit/mkimage [-T type] -l image
          -l ==> list image header information
          -T ==> parse image file as 'type'
       /usr/src/freetz/freetz-ng/tools/fit/mkimage [-x] -A arch -O os -T type -C comp -a addr -e ep -n name -d data_file[:data_file...] image
          -A ==> set architecture to 'arch'
          -O ==> set operating system to 'os'
          -T ==> set image type to 'type'
          -C ==> set compression type 'comp'
          -a ==> set load address to 'addr' (hex)
          -e ==> set entry point to 'ep' (hex)
          -n ==> set image name to 'name'
          -d ==> use image data from 'datafile'
          -x ==> set XIP (execute in place)
       /usr/src/freetz/freetz-ng/tools/fit/mkimage [-D dtc_options] [-f fit-image.its|-f auto|-F] [-b <dtb> [-b <dtb>]] [-E] [-B size] [-i <ramdisk.cpio.gz>] fit-image
           <dtb> file is used with -f auto, it may occur multiple times.
          -D => set all options for device tree compiler
          -f => input filename for FIT source
          -i => input filename for ramdisk file
          -E => place data outside of the FIT structure
          -B => align size in hex for FIT structure and header
Signing / verified boot options: [-k keydir] [-K dtb] [ -c <comment>] [-p addr] [-r] [-N engine]
          -k => set directory containing private keys
          -K => write public keys to this .dtb file
          -G => use this signing key (in lieu of -k)
          -c => add comment in signature node
          -F => re-sign existing FIT image
          -p => place external data at a static position
          -r => mark keys used as 'required' in dtb
          -N => openssl engine to use for signing
       /usr/src/freetz/freetz-ng/tools/fit/mkimage -V ==> print version information and exit
Use '-T list' to see a list of available image types
ERROR: packing failed
FINISHED

```

Die Toolchain und host-tools werden lokal erstellt, da kein AVX2-Support vorhanden ist.
Als System rennt aktuell ein ubuntu 18.04 und alle dependancies scheinen auch erfüllt zu sein.

Bitte einmal einen Dry-run manuell mit frischem checkout durchlaufen lassen um das nachzuvollziehen,
da es im letzten Schritt dann dieses Pack-Problem gibt, wohl aufgrund geänderter command-line Switches ?

Frage: gibt es systembedingt eine neue fehlende Dependancie die noch nicht in der Liste der Dependancies erfasst wurde ?
(der "DTC not found error" ist etwas verwirrend. Stimmt dort ein Path nicht ? - oder fehlt etwas aus dem Ubuntu-Repo hier noch ?

maverick3005, 2022-11-26:
Also es fehlt eine Dependencie: "device-tree-compiler" aus dem Ubuntu-Repo - das fehlt bei den "requirements" und in den
entsprechenden DOCs als Hinweis und im dep-check.

Ferner ist mir aufgefallen das im Gegensatz zu anderen Boxen im Moment als Busybox-Default bei der 5590 die 1.35 ausgewählt ist, anstelle der 1.34 Version.

fda77, 2022-11-28:
Hi, man. Freetz hat nen eigenen dtc

closed 2022-11-28 by fda77

maverick3005, 2022-12-01:
> Hi, man. Freetz hat nen eigenen dtc

Es gibt offenbar einen Unterschied zwischen den prebuild host-tools und den self-build tools....

Bitte ueberprüfe einmal die Pfade zwischen Deinem Prebuild-Paket und den Scripts zur Selbsterstellung der host-tools.

Mein Logfile wies keinen "device-tree-compiler" innerhalb von freetz auf, bzw. es hat das binary im freetz-Path nicht gefunden,
daher half der "fix", das aus dem ubuntu-repo nachzuinstallen. Evtl. irgend wo eine path-variable vergessen bei den host-tools ?

fda77, 2022-12-01:
Daran lags wohl. Davon was hinzuschreiben fandest du überflüssig?