Freetz-NG/freetz-ng #653: Interessante Pakete Iptables-CGI und AVM-firewall werden in der Auswahl vermisst

karstengit, 2022-12-02:
Ich habe die aktuelle Version von Freetz-NG in Debian mit git heruntergeladen.
Nun versuche ich ein Image für eine 7430 zu bauen.

Leider sind die sehr interessanten Pakete Iptables-CGI (unter I) und AVM-firewall (unter Web-Interfaces) nicht in der Auswahl bei packages enthalten.
Dies ändert sich auch nicht wenn eine ältere Firmware-Version ausgewählt wird.

Woran liegt das oder sind diese Pakete irgendwo anders "versteckt"?

karstengit, 2022-12-02:
Ebenso fehlt NHIPT Iptables CGI, so daß keine interessante Option für eine Firewall zur Verfügung steht.

MasterRoCcO, 2022-12-02:
Benutze das mal `Level of user competence (Developer)  --->   ` und schaue an, es ist unter Unstable sehen

karstengit, 2022-12-02:
Das ändert leider ebenfalls nichts.

karstengit, 2022-12-02:
Gibt es übrigens eine Empfehlung für die Pakete/Abhängigkeiten in Debian GNU/Linux 11 (bullseye) ?

aus-sua, 2022-12-02:
:-|

Ich denke da einmal, ein einfaches:
RTFM
wirst Du nicht verstehen, deshalb:
Alle Deine Fragen werden da auch für Dich beantwortet:
https://freetz-ng.github.io/freetz-ng/

fda77, 2022-12-02:
Debian ist mir nur bekannt dass es mit Freetz verschiedene Probleme verursacht, keine Ahnung was genau ... nimm lieber Ubuntu oder Fedora.
Es gibt übrigens hier bei Github "Diskussions"

closed 2022-12-02 by fda77

karstengit, 2022-12-03:
Es wurde nämlich der Anleitung gefolgt und mit
git clone https://github.com/Freetz-NG/freetz-ng ~/freetz-ng
eine lokale Kopie heruntergeladen.

In der Anleitung stehen keine weiteren Schritte beschrieben was danach zu tun ist, damit alle Menüpunkte vollständig vorhanden sind.
Damit dürfte Freetz-NG wohl unbrauchbar sein.

Wenn darüber hinaus Freetz-NG nur mit bestimmten Distributionen kompilierbar ist, dann sollte die aktuelle gepflegte Distribution die funktioniert in der Anleitung ebenfalls beannt werden.

Schade, dann bleibt es wohl bei dem Stand vom alten Freetz.

karstengit, 2022-12-03:
Selbst in der Diskussion wurden bereits User mit exakt dem gleichen Problem unwirsch abgewiesen!
https://github.com/Freetz-NG/freetz-ng/discussions/624

fda77, 2022-12-04:
Spam nicht herum