Freetz-NG/freetz-ng #654: Paket AVM-firewall ist und bleibt verschollen

karstengit, 2022-12-04:
Da der Bug https://github.com/Freetz-NG/freetz-ng/issues/653 unwirsch und jäh ohne ausreichende Begründung abgewürgt worden ist, erfolgt hier die Dokumentation des zweiten erfolglosen Anlaufs zur Problemlösung.

Da Debian 11 das OS ist unter dem mit dem Desktop gearbeitet wird, wurde zuerst versucht Freetz-NG darunter zu kompilieren.
Dafür wurden die folgenden Pakete installiert:
```
bison flex subversion-tools paxtest
libncurses-dev zfs-dkms libcap-dev libacl1-dev libsqlite3-dev libzstd-dev
```
sowie manuell mit dpkg -i
http://ftp.de.debian.org/debian/pool/main/p/prelink/execstack_0.0.20131005-1+b10_amd64.deb

Nachdem der Pfad zu sbin manuell gesetzt worden ist
`export PATH="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin"`
startete der Kompiliervorgang, aber brach dann mit extrem unverständlichen Fehlermeldungen ab.

So wurde nun der zweite Versuch mit **RTFM** gestartet.
Es wurde ein vorhandenes kubuntu-22.04-desktop-amd64.iso als virtuelle Maschine in Virtualbox aufgesetzt.
Nach ca. 2h und 15GB Belegung auf der Platte war dann ein Image vorhanden, auf dem die empfohlenen Pakete installiert sind:
`sudo apt-get install pv cpio rsync kmod execstack imagemagick inkscape graphicsmagick subversion git bc unar wget sudo gcc g++ binutils autoconf automake autopoint libtool-bin make bzip2 libncurses5-dev libreadline-dev zlib1g-dev flex bison patch texinfo tofrodos gettext pkg-config ecj fastjar          perl libstring-crc32-perl ruby         gawk python2 python-is-python3 libusb-dev unzip intltool libacl1-dev libcap-dev libc6-dev-i386 lib32ncurses5-dev gcc-multilib bsdmainutils lib32stdc++6 libglib2.0-dev ccache cmake lib32z1-dev libsqlite3-dev sqlite3 libzstd-dev netcat curl uuid-dev libssl-dev libgnutls28-dev`

**Obwohl der Level auf Developer steht bleibt die AVM-firewall verschwunden:**
[image]

Diese scheint jedoch in den Quellen vorhanden zu sein wenn man ein grep danach ausführt:
```
config/.cache.in:               via AVM web interface. Nowadays the avm-firewall package is used
config/.cache.in:# INCLUDE_BEGIN make/pkgs/avm-firewall/Config.in
config/.cache.in:       bool "AVM-firewall"
config/.cache.in:               Configuration front-end for AVM-firewall.
config/.cache.in:# INCLUDE_END make/pkgs/avm-firewall/Config.in
```

Dafür ist immerhin Iptables unter Unstable auffindbar und auswählbar.

Der erste Test einer Kompilierung läuft derzeit noch.

Was muß nun getan werden um in den vollen Genuß aller Möglichkeiten der Firewall zu kommen?

fda77, 2022-12-04:
Nutz die "Diskussions" für deine Fragen.
Was du zu tun hast hast du selbst gepostet

closed 2022-12-04 by fda77

karstengit, 2022-12-04:
Es handelt sich hier definitiv um einen Bug.

Durch gezieltes abschätzen kann man in die **.config** manuell folgenden Eintrag machen, **nachdem menuconfig beendet worden ist**:

```
#
# Web interfaces
#

#
# Addhole (needs dnsmasq)
#
FREETZ_PACKAGE_AUTHORIZED_KEYS=y
FREETZ_PACKAGE_AVM_FIREWALL=y
```

Mit dem Eintrag **FREETZ_PACKAGE_AVM_FIREWALL=y** wird die AVM-Firewall hinein kompiliert.

```
installing packages
  syslogd-cgi-0.2.3 .................    3.21 Kb 
  inetd-0.2 .........................    3.17 Kb 
  authorized-keys-0.1 ...............    2.64 Kb 
  avm-firewall-2.0.4_rc5 ............   16.06 Kb 
  haserl-0.9.36 .....................   10.32 Kb 

```

berndy2001, 2022-12-04:
Und funktioniert das Paket auch korrekt?

karstengit, 2022-12-04:
Keine Ahnung - wahrscheinlich nicht, wenn AVM dafür gesorgt hat.

Aus den gleichen Gründen werden wahrscheinlich keine älteren Releases mehr bereitgestellt.
Gibt es irgendwo im Netz noch eine weitere Quelle von Backups alter Releases?

Auf der Box 7430 ist derzeit ein Firmware 06.83 installiert.
Leider kann man diese nicht sichern.

karstengit, 2022-12-05:
Es wurde zu Recht bemängelt das die Fehler in Debian 11 nicht vollständig dokumentiert worden sind.
Ein erneuter Aufruf von make hat die (extrem unverständliche) Fehlermeldung nicht wiederholt, sondern der Build-Vorgang startet und bricht erst später ab mit

```
---> tools/pseudo-host ... 
building ... cc -fno-strict-aliasing  -pipe -std=gnu99 -Wall -W -Wextra -Wno-deprecated-declarations -fPIC -D_LARGEFILE64_SOURCE -D_ATFILE_SOURCE -Wno-cast-function-type -Wno-nonnull-compare -fcommon -m32 -DPSEUDO_PREFIX='"/srv/test/freetz-ng/tools/build"' -DPSEUDO_SUFFIX='""' -DPSEUDO_BINDIR='"bin"' -DPSEUDO_LIBDIR='"lib"' -DPSEUDO_LOCALSTATEDIR='"var/pseudo"' -DPSEUDO_VERSION='"1.9.0"' -DUSE_MEMORY_DB  -DPSEUDO_PASSWD_FALLBACK='""'           -O2 -g  -D_GNU_SOURCE -c -o pseudo_wrappers.o pseudo_wrappers.c
cc -c   -pipe -std=gnu99 -Wall -W -Wextra -Wno-deprecated-declarations -fPIC -D_LARGEFILE64_SOURCE -D_ATFILE_SOURCE -Wno-cast-function-type -Wno-nonnull-compare -fcommon -m32 -DPSEUDO_PREFIX='"/srv/test/freetz-ng/tools/build"' -DPSEUDO_SUFFIX='""' -DPSEUDO_BINDIR='"bin"' -DPSEUDO_LIBDIR='"lib"' -DPSEUDO_LOCALSTATEDIR='"var/pseudo"' -DPSEUDO_VERSION='"1.9.0"' -DUSE_MEMORY_DB  -DPSEUDO_PASSWD_FALLBACK='""'           -O2 -g  pseudo_client.c
cc -c   -pipe -std=gnu99 -Wall -W -Wextra -Wno-deprecated-declarations -fPIC -D_LARGEFILE64_SOURCE -D_ATFILE_SOURCE -Wno-cast-function-type -Wno-nonnull-compare -fcommon -m32 -DPSEUDO_PREFIX='"/srv/test/freetz-ng/tools/build"' -DPSEUDO_SUFFIX='""' -DPSEUDO_BINDIR='"bin"' -DPSEUDO_LIBDIR='"lib"' -DPSEUDO_LOCALSTATEDIR='"var/pseudo"' -DPSEUDO_VERSION='"1.9.0"' -DUSE_MEMORY_DB  -DPSEUDO_PASSWD_FALLBACK='""'           -O2 -g  pseudo_ipc.c
cc  -pipe -std=gnu99 -Wall -W -Wextra -Wno-deprecated-declarations -fPIC -D_LARGEFILE64_SOURCE -D_ATFILE_SOURCE -Wno-cast-function-type -Wno-nonnull-compare -fcommon -m32 -DPSEUDO_PREFIX='"/srv/test/freetz-ng/tools/build"' -DPSEUDO_SUFFIX='""' -DPSEUDO_BINDIR='"bin"' -DPSEUDO_LIBDIR='"lib"' -DPSEUDO_LOCALSTATEDIR='"var/pseudo"' -DPSEUDO_VERSION='"1.9.0"' -DUSE_MEMORY_DB  -DPSEUDO_PASSWD_FALLBACK='""'           -O2 -g  -c -o pseudo_tables.o pseudo_tables.c
cc -c   -pipe -std=gnu99 -Wall -W -Wextra -Wno-deprecated-declarations -fPIC -D_LARGEFILE64_SOURCE -D_ATFILE_SOURCE -Wno-cast-function-type -Wno-nonnull-compare -fcommon -m32 -DPSEUDO_PREFIX='"/srv/test/freetz-ng/tools/build"' -DPSEUDO_SUFFIX='""' -DPSEUDO_BINDIR='"bin"' -DPSEUDO_LIBDIR='"lib"' -DPSEUDO_LOCALSTATEDIR='"var/pseudo"' -DPSEUDO_VERSION='"1.9.0"' -DUSE_MEMORY_DB  -DPSEUDO_PASSWD_FALLBACK='""'           -O2 -g  pseudo_util.c
In file included from pseudo_ports.h:13,
                 from pseudo.h:160,
                 from pseudo_client.c:34:
ports/linux/nostatx/portdefs.h:8:8: error: redefinition of ‘struct statx_timestamp’
    8 | struct statx_timestamp
      |        ^~~~~~~~~~~~~~~
In file included from /usr/include/bits/statx.h:31,
                 from /usr/include/sys/stat.h:446,
                 from pseudo_client.c:25:
/usr/include/linux/stat.h:56:8: note: originally defined here
   56 | struct statx_timestamp {
      |        ^~~~~~~~~~~~~~~
In file included from pseudo_ports.h:13,
                 from pseudo.h:160,
                 from pseudo_client.c:34:
ports/linux/nostatx/portdefs.h:15:8: error: redefinition of ‘struct statx’
   15 | struct statx
      |        ^~~~~
In file included from /usr/include/bits/statx.h:31,
                 from /usr/include/sys/stat.h:446,
                 from pseudo_client.c:25:
/usr/include/linux/stat.h:99:8: note: originally defined here
   99 | struct statx {
      |        ^~~~~
In file included from pseudo_ports.h:13,
                 from pseudo.h:160,
                 from pseudo_wrappers.c:30:
ports/linux/nostatx/portdefs.h:8:8: error: redefinition of ‘struct statx_timestamp’
    8 | struct statx_timestamp
      |        ^~~~~~~~~~~~~~~
In file included from /usr/include/bits/statx.h:31,
                 from /usr/include/sys/stat.h:446,
                 from pseudo_wrappers.c:24:
/usr/include/linux/stat.h:56:8: note: originally defined here
   56 | struct statx_timestamp {
      |        ^~~~~~~~~~~~~~~
In file included from pseudo_ports.h:13,
                 from pseudo.h:160,
                 from pseudo_wrappers.c:30:
ports/linux/nostatx/portdefs.h:15:8: error: redefinition of ‘struct statx’
   15 | struct statx
      |        ^~~~~
In file included from /usr/include/bits/statx.h:31,
                 from /usr/include/sys/stat.h:446,
                 from pseudo_wrappers.c:24:
/usr/include/linux/stat.h:99:8: note: originally defined here
   99 | struct statx {
      |        ^~~~~
In file included from pseudo_ports.h:13,
                 from pseudo.h:160,
                 from pseudo_util.c:28:
ports/linux/nostatx/portdefs.h:8:8: error: redefinition of ‘struct statx_timestamp’
    8 | struct statx_timestamp
      |        ^~~~~~~~~~~~~~~
In file included from /usr/include/bits/statx.h:31,
                 from /usr/include/sys/stat.h:446,
                 from pseudo_util.c:18:
/usr/include/linux/stat.h:56:8: note: originally defined here
   56 | struct statx_timestamp {
      |        ^~~~~~~~~~~~~~~
[... 5 lines cut ...]
      |        ^~~~~
In file included from /usr/include/bits/statx.h:31,
                 from /usr/include/sys/stat.h:446,
                 from pseudo_util.c:18:
/usr/include/linux/stat.h:99:8: note: originally defined here
   99 | struct statx {
      |        ^~~~~
[... cut ...]
In file included from /usr/include/bits/statx.h:39,
                 from /usr/include/sys/stat.h:446,
                 from pseudo_wrappers.c:24:
/usr/include/bits/statx-generic.h:56:5: note: previous declaration of ‘statx’ was here
   56 | int statx (int __dirfd, const char *__restrict __path, int __flags,
      |     ^~~~~
make[2]: *** [Makefile:158: pseudo_wrappers.o] Fehler 1
make[1]: *** [make/host-tools/pseudo-host/pseudo-host.mk:57: /srv/test/freetz-ng/tools/build/lib/libpseudo.so] Beendet
make: *** [Makefile:46: envira] Beendet

```

karstengit, 2022-12-09:
**Der Build-Prozess funktioniert nun in Debian 11.**

Dafür mußten die folgenden Pakete noch nachinstalliert werden:

`aptitude install uuid-dev libssl-dev libgnutls28-dev libglib2.0-dev-bin`

und der kryptische Tip "Nimm einfach fakeroot statt pseudo" von @fda77 hat geholfen.
Dafür muß in der **menuconfig** unter **Toolchain options -> Root emulator** auf **fakeroot** gestellt werden.

[image]

[image]

karstengit, 2022-12-09:
Allerdings ist nun wieder der Fehler mit den fehlenden Kernelmodulen da:

```
WARNING: File of kernel module not found: ppp_async.ko
WARNING: File of kernel module not found: ppp_deflate.ko
WARNING: File of kernel module not found: ppp_generic.ko
WARNING: File of kernel module not found: ppp_mppe_mppc.ko
WARNING: File of kernel module not found: sha1_generic.ko
WARNING: File of kernel module not found: slhc.ko
ERROR: Enable missing with 'make kernel-menuconfig' as (M)odule (and push a pull request). Use '/' to search in menuconfig.
make[1]: *** [Makefile:388: firmware-nocompile] Fehler 1
make: *** [Makefile:46: envira] Beendet

```

Die entsprechenden Kernelmodule sind jedoch alle ausgewählt!?

[image]

[image]

fda77, 2022-12-10:
Schreib dir alles auf was man für Debian braucht, dann kannst du es für später den PREREQUISITES hinzufügen. Für alle weiteren Fragen zu Debian bist dann aber du zuständig, ich hab damit nix zu tun!

> und der kryptische Tip "Nimm einfach fakeroot statt pseudo" von @fda77 hat geholfen.

Tud mir leid dich überfordert zu haben. Wie man im menuconfig sucht hast du ja schon selbst häufig gepostet

Ich wollte mit mal anschauen was das da mit den kernelmodulen ist, du hast es aber leider nicht geschafft zu schreiben um welchen Kernel es überhaupt geht

fda77, 2022-12-10:
@karstengit: Du hast ein Issue UND eine Diskussiuon mit gleichem Titel erstellt!?!
https://github.com/Freetz-NG/freetz-ng/discussions/657