Freetz-NG/freetz-ng #655: Paket PPP läßt sich nicht einbinden

karstengit, 2022-12-04:
Wird das Paket PPP eingebunden endet die Kompilierung mit:

```
WARNING: File of kernel module not found: ppp_async.ko
WARNING: File of kernel module not found: ppp_deflate.ko
WARNING: File of kernel module not found: ppp_generic.ko
WARNING: File of kernel module not found: ppp_mppe_mppc.ko
WARNING: File of kernel module not found: sha1_generic.ko
WARNING: File of kernel module not found: slhc.ko
ERROR: Enable missing with 'make kernel-menuconfig' as (M)odule (and push a pull request). Use '/' to search in menuconfig.
```
ohne das ein Image erstellt wird.

Die entsprechenden Kernelmodule sind jedoch in der Konfiguration bereits ausgewählt:

[image]

Was muß getan werden?

karstengit, 2022-12-04:
Hier ist die dazugehörige .config
[config.zip](https://github.com/Freetz-NG/freetz-ng/files/10148594/config.zip)

fda77, 2022-12-04:
Dein gepostetes Lesen -> machen

closed 2022-12-04 by fda77

karstengit, 2022-12-04:
Keine Ahnung - ein sha1_generic.ko läßt sich schwer erfinden.
Das Modul PPP ist mandantory und somit läßt sich ein Image für diese Box offensichtlich nicht kompilieren!

karstengit, 2022-12-04:
Der beschriebene Fehler mit den fehlenden Kernel-Modulen tritt ganz klar auf, wenn PPP unter den Weboberflächen aktiviert wird.

Wenn das Mal kein Bug ist ...

fda77, 2022-12-04:
Ganz klar kannst du nicht lesen