Freetz-NG/freetz-ng #659: Build von openssh für 3370 schlägt fehl

pfichtner, 2022-12-05:
Der Build von openssh für 3370 (uClibc 0.9.33.2) schlägt fehl, während es für die ältere 3270 (uClibc 0.9.32) funktioniert. 

```
---> kernel ...
building ... (cd openbsd-compat && make)
make[3]: Nothing to be done for 'all'.
/workspace/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/mips-linux-uclibc/bin/mips-linux-uclibc-gcc -march=34kc -mtune=34kc -msoft-float -Os -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 --std=gnu99 -pipe -Wall -Wpointer-arith -Wuninitialized -Wsign-compare -Wformat-security -Wsizeof-pointer-memaccess -Wno-pointer-sign -Wno-unused-result -fno-strict-aliasing -D_FORTIFY_SOURCE=2 -fno-builtin-memset   -ffunction-sections -fdata-sections -I. -I.  -D_XOPEN_SOURCE=600 -D_BSD_SOURCE -D_DEFAULT_SOURCE -DSSHDIR=\"/etc\" -D_PATH_SSH_PROGRAM=\"/usr/bin/ssh\" -D_PATH_SSH_ASKPASS_DEFAULT=\"/usr/lib/ssh-askpass\" -D_PATH_SFTP_SERVER=\"/usr/lib/sftp-server\" -D_PATH_SSH_KEY_SIGN=\"/usr/lib/ssh-keysign\" -D_PATH_SSH_PKCS11_HELPER=\"/usr/lib/ssh-pkcs11-helper\" -D_PATH_SSH_SK_HELPER=\"/usr/lib/ssh-sk-helper\" -D_PATH_SSH_PIDDIR=\"/var/run\" -D_PATH_PRIVSEP_CHROOT_DIR=\"/var/empty\" -DHAVE_CONFIG_H -c sandbox-seccomp-filter.c -o sandbox-seccomp-filter.o
/workspace/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/mips-linux-uclibc/bin/mips-linux-uclibc-gcc -march=34kc -mtune=34kc -msoft-float -Os -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 --std=gnu99 -pipe -Wall -Wpointer-arith -Wuninitialized -Wsign-compare -Wformat-security -Wsizeof-pointer-memaccess -Wno-pointer-sign -Wno-unused-result -fno-strict-aliasing -D_FORTIFY_SOURCE=2 -fno-builtin-memset   -ffunction-sections -fdata-sections -I. -I.  -D_XOPEN_SOURCE=600 -D_BSD_SOURCE -D_DEFAULT_SOURCE -DSSHDIR=\"/etc\" -D_PATH_SSH_PROGRAM=\"/usr/bin/ssh\" -D_PATH_SSH_ASKPASS_DEFAULT=\"/usr/lib/ssh-askpass\" -D_PATH_SFTP_SERVER=\"/usr/lib/sftp-server\" -D_PATH_SSH_KEY_SIGN=\"/usr/lib/ssh-keysign\" -D_PATH_SSH_PKCS11_HELPER=\"/usr/lib/ssh-pkcs11-helper\" -D_PATH_SSH_SK_HELPER=\"/usr/lib/ssh-sk-helper\" -D_PATH_SSH_PIDDIR=\"/var/run\" -D_PATH_PRIVSEP_CHROOT_DIR=\"/var/empty\" -DHAVE_CONFIG_H -c ssh-add.c -o ssh-add.o
sandbox-seccomp-filter.c: In function 'ssh_sandbox_violation':
sandbox-seccomp-filter.c:377:20: error: 'siginfo_t' has no member named 'si_arch'
      __func__, info->si_arch, info->si_syscall, info->si_call_addr);
                    ^
sandbox-seccomp-filter.c:377:35: error: 'siginfo_t' has no member named 'si_syscall'
      __func__, info->si_arch, info->si_syscall, info->si_call_addr);
                                   ^
sandbox-seccomp-filter.c:377:53: error: 'siginfo_t' has no member named 'si_call_addr'
      __func__, info->si_arch, info->si_syscall, info->si_call_addr);
                                                     ^
make[2]: *** [Makefile:195: sandbox-seccomp-filter.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [make/pkgs/openssh/openssh.mk:62: source/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/openssh-9.1p1/ssh] Terminated
make: *** [Makefile:46: envira] Terminated
```

Nach auskommentieren der Zeile [`#define SANDBOX_SECCOMP_FILTER_DEBUG 1`](https://github.com/openssh/openssh-portable/blob/73dcca12115aa12ed0d123b914d473c384e52651/sandbox-seccomp-filter.c#L26) in läuft der compile durch. 
Das korrespondiert mit der Änderung https://github.com/openssh/openssh-portable/commit/2580916e48721802220c61ce9e0df1297c00bc07?diff=unified

Natürlich könnte ich einen Patch erstellen, der den genannten Commit reverted, das würde ich allerdings ungern generell tun wollen sondern nur in den Fällen wollen, bei welchen es zu Problemen kommt (was, so wie es aussieht, nicht mit zu alten uClibcs zusammenhängt). 

Jemand eine Idee, wann der revert notwendig ist oder wie zu verfahren ist?

pfichtner, 2022-12-07:
Durch die Kommentare im Code (*Do not* use this in production) war ich mir nicht sicher, ob es sich eventuell um ein securityrelevantes Problem handeln könnte und hab es daher nicht als Bug sondern potentielles Securityproblem gemeldet. 
Rückmeldung: Stellt, solange es in der sanbox zu keiner Verletzung kommt, kein Securityproblem dar. 

Die Änderung war allerdings unbeabsichtigt und wird wieder zurückgenommen. Da mein Problem mit der (einer der) folgenden openssh Releases dann gefixt sein wird wäre mein Vorschlag, den Issue ohne Änderung zu schließen. Den Patch/Fix zu erstelle der mit der nächsten openssh Version wieder hinfällig ist rentiert sich kaum, zumindest nicht, solange a.) ich der einzige/einer von wenigen Betroffenen bin und b.)  das nächste openssh release in halbwegs absehbarerer Zeit erscheint (und in Freetz-NG übernommen wird). Bei anderen Meinungen erstelle ich den Patch.

fda77, 2022-12-07:
Ein Patch der diese 1 Zeile löscht ist ja in Sekunden gemacht, aber mir egal, mach wie du willst. Das Sicherheist problem soll wohl bei den Signalen sein, dass da was belaurscht werden kann. Ich glaub man könnte dieses Issue auch rückstandslos löschen.
Ich find die Patches von openssh -> openssh-portable eh etwas seltsam, vielleicht sollte man das normale Version verwenden?
Durch AVM's Linux-Verständnis (aus Windows-User Sicht...) ist die ganze Fritzbox aber eine Sicherheits-Zumutung, da grundsätzlich so gut wie alles als root läuft -.- Bei einem normalen Linux gibts viele Sicherheitslimien, Benutzerverwaltung, Selinux, Dateisystemberechtigungen usw. Wenn eines von AVM's selbstgebackener closed-source Daemons ne Lücke hat, hat man gleich kompletten Zugriff auf alles!

closed 2022-12-07 by pfichtner

pfichtner, 2022-12-07:
Der Grund den Patch nicht zu erstellen war nicht der, dass es zu viel Aufwand ist sondern aus dem Grund, dass man jetzt nicht vergessen sollte, diesen wieder zu entfernen, sobald man auf eine Version updated, welche https://github.com/openssh/openssh-portable/commit/bc234605fa3eb10f56bf0d74c8ecb0d91ada9d05 beinhaltet. 
Ich hoffe dabei, dass patch sich nur beschwert dass der Patch schon angewandt ist und eben nicht den Build abbricht. 
Ich gehe davon aus, dass es nichts gibt, wo man eintragen könnte, dass man den Patch löschen sollte, sobald man openssh updated oder es z.B. eine action gibt die darauf hinweist, oder? 

Bezüglich des Meldens hatte ich jetzt auch nicht die FritzBox im Kopf. Wäre das tatsächlich ein Securityproblem gewesen und ich hätte das auf einem öffentlichen Bugtracker eingestellt, wäre das für alle nutzenden Systeme/Distros/Appliances/... problematisch und Systeme eventuell dokumentiert angreifbar, die ich gar nicht kenne. Deswegen habe ich da mal Vorsicht walten lassen. 

Natürlich ist es alles andere als clever, sämtliche (viele) daemons schon alleine unter dem gleichen User laufen zu lassenm das muss noch nicht mal root sein, aber als root ist natürlich die Krönung von allem.

fda77, 2022-12-07:
Wenn man einen Patch anwenden will der nicht passt wird man das schon merken :) In Freetz sind viele upstream patches, meist als 000-*.upstream.patch: `find . -name *upstream*`, dann weiss man beim nächsten bump dass der weg kann wenn der nicht will. Ìch hab einfach dies unverändert genommen (ng erkennt das patchlevel) https://github.com/openssh/openssh-portable/commit/bc234605fa3eb10f56bf0d74c8ecb0d91ada9d05.patch @59724918d

pfichtner, 2022-12-07:
Jetzt sind zwei Patches drin die das gleiche machen

pfichtner, 2022-12-07:
> Habs gerad bemerkt Mist :)

Werfe meines gerne weg (ich schreibe das, bevor jetzt parallel jeder das eigene wegwirft)

fda77, 2022-12-07:
So jetzt passt es wieder. Es hat schon Vorteile wenn man im Commit ein Issue mit refs angibt oder wenn man es vergisst hier den Commit zu verlinken. Ich ging davon aus dass du das nicht gemacht hättest. "make openssh-autofix" hat bei mit keinen Fehler gemeldet, weil ich PULL vor einem COMMIT automatisch machen lasse

fda77, 2022-12-07:
Jo, es bleibt nicht unbemerkt dass der Patch weg muss. Bei neuen Versionen müssen auch häufig kleine Änderungen manuell vorgenommen werden

fda77, 2022-12-07:
Stimmt, hab ich nicht gesehen

commit c540618d (2022-12-07): fixes #659
commit 88e9de09 (2022-12-08): Revert "fixes #659"

This reverts commit c540618dbf5fbd37199c140243fe1df11e587f7a.