Freetz-NG/freetz-ng #662: Haserl mit lua: Error passing the lua library to the lua vm, bad header in precompiled chunk

berndy2001, 2022-12-08:
Habe meine 7590 heute auf FOS 7.50 aktualisiert, läuft eigentlich alles wie es soll, nur lua-scripts (über den inetd) machen Probleme:
```
root@fritz:/var/media/ftp/www/cgi-bin# /usr/bin/haserl --shell=lua test.cgi
Error passing the lua library to the lua vm: luascript.lua: bad header in precompiled chunk
```
sample-script:
```
#!/usr/bin/haserl --shell=lua
Content-type: text/html

<html>
<body>
<table border=1><tr>
<% 
	t = {'Red', 'Blue', 'Yellow', 'Cyan'}
	for k,v in ipairs(t) do
		io.write('<td bgcolor="'..v..'">'..v..'</td>')
	end
%>
</tr></table>
</body>
</html>
```
Jemand eine Idee dazu? Google Suche habe ich bemüht, wurde aber nicht schlau daraus.
Habe jetzt testweise wieder zu FritzOS 07.29/Freetz ng-19044-bc87a2031 (31.03.2022 19:34:30) geswitcht, damit läufts.

[config.txt](https://github.com/Freetz-NG/freetz-ng/files/10188785/config.txt)

fda77, 2022-12-09:
Wie, mit der FOS 7.50 geht nicht was mit 7.25 noch ging? Kann ich mir fast bei diesem Problem eigentlich nicht vorstellen. Wie ist es denn wenn du mit aktueller freetz-revision ein 7.25 baust? Es könnte auch sein dass ne lib aktualisiert wurde die lua nutzt, bei ncurses und readline ist ein bump letzter nicht so lange her
Btw lua 5.4 wäre aktuell, nicht wie in freetz 5.1 von 2006

berndy2001, 2022-12-09:
Gute Idee. Mit 07.29 klappt es auch nicht. Hab jetzt (weil du es erwähnt hast) testweise auch noch lua mit readline dazugenommen, brachte aber keine Änderung. Ich war auch der Meinung dass für den lua-Support von Haserl das lua-Paket nicht notwendig ist.

Config war die compressed, anbei die ganze.
[config.txt](https://github.com/Freetz-NG/freetz-ng/files/10196774/config.txt)

fda77, 2022-12-09:
Das readline ist optional, wenn du es eh nicht an hattest hat dessen bump nicht damti zu tun, prima.
Die Fos version auch nicht.
Ansonsten wurde in dem Bereich nicht viel verändert, zumindest fällt mir nichts dazu ein. VIelleicht noch das Verschieben innerhalb des make/ Verzeichnisses. Oder war deine alte Version noch mit einer älteren uclibc (ist schon länger her)

berndy2001, 2022-12-10:
Nachdem ich keine Idee habe um das zu debuggen, wollte ich ältere commits probieren, aber das scheitert daran, dass das File kernel_4.9.198-7590_07.25-r19566.patch.xz nicht mehr verfügbar ist.

berndy2001, 2022-12-10:
Ja, ist mir klar 😉
Lese gerade zu git bisect, wenn ich das richtig verstehe soll es das ermitteln der Mitte übernehmen. Hab das bei den bisherigen Tests eher geschätzt. Aber man lernt dazu...

fda77, 2022-12-10:
Das geht halt recht flott. Von deinen 200 Revisionen kann man auf 12 durch 5 Versuche eingrenzen

fda77, 2022-12-10:
Ungetestet, vielleicht hilft es dir: 0e4e5580f

closed 2022-12-10 by fda77

berndy2001, 2022-12-10:
die lua-binary 5.4 funktioniert.

Haserl mit lua geht jetzt die 5.1.5, die 5.4.4 nicht. Es kommt die vorhin genannte Fehlermeldung.

Ich habe mit 280358b getestet.

fda77, 2022-12-11:
Dass Lua ne shared lib erstllt und diese von haserl verwendet wird scheint freetz-spezifisch zu sein.
https://github.com/Freetz-NG/freetz-ng/blob/master/make/pkgs/haserl/patches/100-lualib_inc.patch  muss evtl je nach Version verschieden sein, ab dem "This file was automatically generated from haserl_lualib.lua. DO NOT EDIT"

fda77, 2022-12-11:
Diese lualib.inc kann nicht beim crosscompilieren erstellt werden und muss auf dem Zielsystem ausgeführt werden. Vielleicht klappt es jetzt! Daher geh ich davon aus dass das nur auf mips läuft, und noch nie (in freetz) auf atom und arm

berndy2001, 2022-12-14:
Hab jetzt nochmals mit dem aktuellen commit getestet, jetzt gibts nur ```Exception.```

fda77, 2022-12-14:
Hab keine Gerät mit dem ich das Testen könnte. Ich geh davon aus dass der Patch nicht angewendet wurde