Freetz-NG/freetz-ng #669: Failing to make menucconfig, getting Permission denied

dzelionis, 2022-12-11:
Hey, failing to build it, so here is a screenshot ofo the error i am getting and also my few additional step to debug it:

[image]

fda77, 2022-12-11:
Seems you have a permission problem with the local filesystem. Try to start over with a new checkout

dzelionis, 2022-12-11:
i have tried that already...did not work
for me its looks like miss gcc in the front 
[image]

fda77, 2022-12-13:
Sry, i have really no idea!

dzelionis, 2022-12-13:
i have checked your link but it's not the same,  i fail way earlier..

[image]

dzelionis, 2022-12-13:
also i have tried docker build environment got same issue

dzelionis, 2022-12-13:
```
domas.zelionis@LinuxBigMate:~/freetz-ng$ umask 0022
domas.zelionis@LinuxBigMate:~/freetz-ng$  git checkout master ; git fetch --all --prune ; git reset --hard origin/HEAD ; git clean -fd
Already on 'master'
Your branch is up to date with 'origin/master'.
Fetching origin

X11 forwarding request failed on channel 0
HEAD is now at 0e927ae97 juis: automatic update
domas.zelionis@LinuxBigMate:~/freetz-ng$ 
domas.zelionis@LinuxBigMate:~/freetz-ng$ make clean
Freetz-NG 20642-0e927ae97 master 2022-12-13
make -j -C /home/domas.zelionis/freetz-ng/source/host-tools/avm-rle-0 clean
make[2]: *** /home/domas.zelionis/freetz-ng/source/host-tools/avm-rle-0: No such file or directory.  Stop.
make[1]: [make/host-tools/avm-rle-host/avm-rle-host.mk:20: avm-rle-host-clean] Error 2 (ignored)
make -j -C /home/domas.zelionis/freetz-ng/source/host-tools/busybox-1.34.1 clean
make[2]: *** /home/domas.zelionis/freetz-ng/source/host-tools/busybox-1.34.1: No such file or directory.  Stop.
make[1]: [make/host-tools/busybox-host/busybox-host.mk:42: busybox-host-clean] Error 2 (ignored)
make -j -C /home/domas.zelionis/freetz-ng/source/host-tools/dtc-1.6.1 clean
make[2]: *** /home/domas.zelionis/freetz-ng/source/host-tools/dtc-1.6.1: No such file or directory.  Stop.
make[1]: [make/host-tools/dtc-host/dtc-host.mk:28: dtc-host-clean] Error 2 (ignored)
make -j -C /home/domas.zelionis/freetz-ng/source/host-tools/e2fsprogs-1.42.13 clean
make[2]: *** /home/domas.zelionis/freetz-ng/source/host-tools/e2fsprogs-1.42.13: No such file or directory.  Stop.
make[1]: [make/host-tools/e2fsprogs-host/e2fsprogs-host.mk:70: e2fsprogs-host-clean] Error 2 (ignored)
rm -f /home/domas.zelionis/freetz-ng/source/host-tools/e2fsprogs-1.42.13/.compiled
make -j -C /home/domas.zelionis/freetz-ng/source/host-tools/fakeroot-1.30.1/build/arch clean
make[2]: *** /home/domas.zelionis/freetz-ng/source/host-tools/fakeroot-1.30.1/build/arch: No such file or directory.  Stop.
make[1]: [make/host-tools/fakeroot-host/fakeroot-host.mk:79: fakeroot-host-clean] Error 2 (ignored)
make -j -C /home/domas.zelionis/freetz-ng/source/host-tools/fakeroot-1.30.1/build/biarch clean
make[2]: *** /home/domas.zelionis/freetz-ng/source/host-tools/fakeroot-1.30.1/build/biarch: No such file or directory.  Stop.
make[1]: [make/host-tools/fakeroot-host/fakeroot-host.mk:80: fakeroot-host-clean] Error 2 (ignored)
make -j -C /home/domas.zelionis/freetz-ng/source/host-tools/find-squashfs-0 clean
make[2]: *** /home/domas.zelionis/freetz-ng/source/host-tools/find-squashfs-0: No such file or directory.  Stop.
make[1]: [make/host-tools/find-squashfs-host/find-squashfs-host.mk:17: find-squashfs-host-clean] Error 2 (ignored)
make -j -C /home/domas.zelionis/freetz-ng/source/host-tools/genext2fs-4c1cc9468980448ca3e86db1cbe9600a4a084f5e clean
make[2]: *** /home/domas.zelionis/freetz-ng/source/host-tools/genext2fs-4c1cc9468980448ca3e86db1cbe9600a4a084f5e: No such file or directory.  Stop.
make[1]: [make/host-tools/genext2fs-host/genext2fs-host.mk:35: genext2fs-host-clean] Error 2 (ignored)
rm -f /home/domas.zelionis/freetz-ng/source/host-tools/genext2fs-4c1cc9468980448ca3e86db1cbe9600a4a084f5e/.configured /home/domas.zelionis/freetz-ng/source/host-tools/genext2fs-4c1cc9468980448ca3e86db1cbe9600a4a084f5e/.tests-passed
make -j -C /home/domas.zelionis/freetz-ng/source/host-tools/gmp-6.1.2 clean
make[2]: *** /home/domas.zelionis/freetz-ng/source/host-tools/gmp-6.1.2: No such file or directory.  Stop.
make[1]: [make/host-tools/gmp-host/gmp-host.mk:30: gmp-host-clean] Error 2 (ignored)
rm -f \
	/home/domas.zelionis/freetz-ng/source/host-tools/kconfig-v6.1/scripts/basic/.*.cmd \
	/home/domas.zelionis/freetz-ng/source/host-tools/kconfig-v6.1/scripts/kconfig/.*.cmd \
	/home/domas.zelionis/freetz-ng/source/host-tools/kconfig-v6.1/scripts/kconfig/lxdialog/.*.cmd \
	/home/domas.zelionis/freetz-ng/source/host-tools/kconfig-v6.1/scripts/kconfig/*.o \
	/home/domas.zelionis/freetz-ng/source/host-tools/kconfig-v6.1/scripts/kconfig/lxdialog/*.o \
	/home/domas.zelionis/freetz-ng/source/host-tools/kconfig-v6.1/scripts/kconfig/zconf.*.c \
	/home/domas.zelionis/freetz-ng/source/host-tools/kconfig-v6.1/scripts/basic/fixdep \
	/home/domas.zelionis/freetz-ng/source/host-tools/kconfig-v6.1/scripts/kconfig/conf \
	/home/domas.zelionis/freetz-ng/source/host-tools/kconfig-v6.1/scripts/kconfig/gconf.glade \
	/home/domas.zelionis/freetz-ng/source/host-tools/kconfig-v6.1/scripts/kconfig/qconf \
	/home/domas.zelionis/freetz-ng/source/host-tools/kconfig-v6.1/scripts/kconfig/gconf \
	/home/domas.zelionis/freetz-ng/source/host-tools/kconfig-v6.1/scripts/kconfig/nconf \
	/home/domas.zelionis/freetz-ng/source/host-tools/kconfig-v6.1/scripts/kconfig/mconf
make -j -f Makefile.freetz -C /home/domas.zelionis/freetz-ng/source/host-tools/libdtc-1.6.1/libfdt clean
make[2]: *** /home/domas.zelionis/freetz-ng/source/host-tools/libdtc-1.6.1/libfdt: No such file or directory.  Stop.
make[1]: [make/host-tools/libdtc-host/libdtc-host.mk:30: libdtc-host-clean] Error 2 (ignored)
make -j -C /home/domas.zelionis/freetz-ng/source/host-tools/lzma1-465/CPP/7zip/Compress/LZMA_Alone clean
make[2]: *** /home/domas.zelionis/freetz-ng/source/host-tools/lzma1-465/CPP/7zip/Compress/LZMA_Alone: No such file or directory.  Stop.
make[1]: [make/host-tools/lzma1-host/lzma1-host.mk:42: lzma1-host-clean] Error 2 (ignored)
make -j -C /home/domas.zelionis/freetz-ng/source/host-tools/lzma1-465/CPP/7zip/Compress/LZMA_Lib clean
[... cut ...] & echo; touch source/.echo_item_end; touch source/.echo_item_1st; fi; fi; echo -ne "\e[48;5;56mbuilding\e[49m ... "; cmd -C /home/domas.zelionis/freetz-ng/source/host-tools/kconfig-v6.1 menuconfig
building ... make[2]: make: Permission denied
make[2]: *** [Makefile:71: scripts_basic] Error 127
make[1]: *** [make/host-tools/kconfig-host/kconfig-host.mk:29: /home/domas.zelionis/freetz-ng/source/host-tools/kconfig-v6.1/scripts/kconfig/mconf] Terminated
make: *** [Makefile:46: envira] Terminated
```

dzelionis, 2022-12-13:
ok look like i found what is the issue 

looks like MAKE var is not defined, after adding 
```
MAKE = "make"
```
to ~/freetz-ng/source/host-tools/kconfig-v6.1/Makefile

```
domas.zelionis@LinuxBigMate:~/freetz-ng/source/host-tools/kconfig-v6.1$ make menuconfig 
make[1]: Entering directory '/home/domas.zelionis/freetz-ng/source/host-tools/kconfig-v6.1'
  UPD     scripts/kconfig/mconf-cfg
  HOSTCC  scripts/kconfig/mconf.o
  HOSTCC  scripts/kconfig/lxdialog/checklist.o
  HOSTCC  scripts/kconfig/lxdialog/inputbox.o
  HOSTCC  scripts/kconfig/lxdialog/menubox.o
  HOSTCC  scripts/kconfig/lxdialog/textbox.o
  HOSTCC  scripts/kconfig/lxdialog/util.o
  HOSTCC  scripts/kconfig/lxdialog/yesno.o
  HOSTCC  scripts/kconfig/confdata.o
  HOSTCC  scripts/kconfig/expr.o
  LEX     scripts/kconfig/lexer.lex.c
  YACC    scripts/kconfig/parser.tab.[ch]
  HOSTCC  scripts/kconfig/lexer.lex.o
  HOSTCC  scripts/kconfig/menu.o
  HOSTCC  scripts/kconfig/parser.tab.o
  HOSTCC  scripts/kconfig/preprocess.o
  HOSTCC  scripts/kconfig/symbol.o
  HOSTCC  scripts/kconfig/util.o
  HOSTLD  scripts/kconfig/mconf
make[1]: Leaving directory '/home/domas.zelionis/freetz-ng/source/host-tools/kconfig-v6.1'
```

dzelionis, 2022-12-13:
[image]

fda77, 2022-12-13:
You could add MAKE to this file: https://github.com/Freetz-NG/freetz-ng/blob/master/make/host-tools/kconfig-host/patches/300-main_makefile.patch
But it is defined here: https://github.com/Freetz-NG/freetz-ng/blob/master/Makefile#L98
This works for me, including number of cpus. So there should be something very odd with you system

dzelionis, 2022-12-13:
it failed while building busybox same issue, same fix.....
[image]

dzelionis, 2022-12-13:
regarding odd with my system, i have tried docker build environment and was getting the same error...so not sure....

dzelionis, 2022-12-13:
in mane Makefile:

[image]

fda77, 2022-12-14:
Your system seems somehow to deletc MAKE variable. Other variables (SED, TAR, MAKE1) are not used so its unknown if they are also be deleted

pfichtner, 2022-12-14:
> ```
> domas.zelionis@LinuxBigMate:~/freetz-ng$ umask 0022
> domas.zelionis@LinuxBigMate:~/freetz-ng$  git checkout master ; git fetch --all --prune ; git reset --hard origin/HEAD ; git clean -fd
> Already on 'master'
> Your branch is up to date with 'origin/master'.
> Fetching origin
> [...]
> ```

This wasn't really a fresh checkout. For a fresh checkout please do a `git clone` (don't forget the  `umask 0022` upfront). I guess it didn't work on your machine as well as not inside the docker environment because there **is** a **permission problem**. Did you use the existing directory inside the docker container via bind? 
I guess that if you'll do a **fresh clone** as described above everything will work as expected, if this won't work a **fresh clone** as described **inside the container** should do. In both cases: Don't forget to do the `umask 0022` upfront the `git clone`.

dzelionis, 2022-12-14:
> This wasn't really a fresh checkout. For a fresh checkout please do a `git clone` (don't forget the `umask 0022` upfront). I guess it didn't work on your machine as well as not inside the docker [...]

Thanks, i will give it another try by following your instructions coming days, will update you after.