Freetz-NG/freetz-ng #671: failing make menuconfig: "scripts/kconfig/lxdialog/checklist.o: undefined reference to symbol 'keypad'"

jpsollie, 2022-12-13:
I must have an unsupported version of ncurses (I think):
```
make menuconfig
Freetz-NG UNKNOWN
..................................................................................................
---> tools/kconfig-host ... 
preparing ... mkdir -p /home/janpieter/freetz-ng-master/source/host-tools/kconfig-v6.1; tar -C /home/janpieter/freetz-ng-master/source/host-tools -xf dl/kconfig-v6.1.tar.xz 
set -e; shopt -s nullglob; for i in /home/janpieter/freetz-ng-master/make/host-tools/kconfig-host/patches/*.patch*; do case $i in *.patch|*.patch.gz|*.patch.bzip2|*.patch.bz2|*.patch.bz|*.patch.xz|*.patch.lz|*.patch.lzma|*.patch.Z|*.patch.diff) ;; *) continue ;; esac; tools/freetz_patch /home/janpieter/freetz-ng-master/source/host-tools/kconfig-v6.1 $i; done; 
    applying patch file /home/janpieter/freetz-ng-master/make/host-tools/kconfig-host/patches/300-main_makefile.patch
    patching file Makefile
    ----------------------------------------------------------------------
    applying patch file /home/janpieter/freetz-ng-master/make/host-tools/kconfig-host/patches/310-fix_make.patch
    patching file scripts/kconfig/Makefile
    ----------------------------------------------------------------------
    applying patch file /home/janpieter/freetz-ng-master/make/host-tools/kconfig-host/patches/320-unchangable_style.patch
    patching file scripts/kconfig/mconf.c
    ----------------------------------------------------------------------
    applying patch file /home/janpieter/freetz-ng-master/make/host-tools/kconfig-host/patches/330-comment_style.patch
    patching file scripts/kconfig/mconf.c
    ----------------------------------------------------------------------
    applying patch file /home/janpieter/freetz-ng-master/make/host-tools/kconfig-host/patches/340-fail_on_unmet_direct_dependency.patch
    patching file scripts/kconfig/symbol.c
    ----------------------------------------------------------------------
    applying patch file /home/janpieter/freetz-ng-master/make/host-tools/kconfig-host/patches/350-max_item_count_for_choice_dialog.patch
    patching file scripts/kconfig/mconf.c
    ----------------------------------------------------------------------
    applying patch file /home/janpieter/freetz-ng-master/make/host-tools/kconfig-host/patches/360-save-hidden-prompts-to-file.patch
    patching file scripts/kconfig/menu.c
    ----------------------------------------------------------------------
    applying patch file /home/janpieter/freetz-ng-master/make/host-tools/kconfig-host/patches/370-multiline_comments_hack.patch
    patching file scripts/kconfig/mconf.c
    patching file scripts/kconfig/menu.c
    ----------------------------------------------------------------------
    applying patch file /home/janpieter/freetz-ng-master/make/host-tools/kconfig-host/patches/380-project_name.patch
    patching file scripts/kconfig/mconf.c
    ----------------------------------------------------------------------
    applying patch file /home/janpieter/freetz-ng-master/make/host-tools/kconfig-host/patches/390-menu_help.patch
    patching file scripts/kconfig/mconf.c
    ----------------------------------------------------------------------
    applying patch file /home/janpieter/freetz-ng-master/make/host-tools/kconfig-host/patches/391-no_symbol_prefix.patch
    patching file scripts/kconfig/lkc.h
    ----------------------------------------------------------------------
    applying patch file /home/janpieter/freetz-ng-master/make/host-tools/kconfig-host/patches/392-use_explicit_flex_call.patch
    patching file scripts/kconfig/Makefile
    ----------------------------------------------------------------------
    applying patch file /home/janpieter/freetz-ng-master/make/host-tools/kconfig-host/patches/393-no_POSIX_warning_from_bison.patch
    patching file scripts/kconfig/Makefile
    ----------------------------------------------------------------------
    applying patch file /home/janpieter/freetz-ng-master/make/host-tools/kconfig-host/patches/400-do_not_create_autoconf_stuff.patch
    patching file scripts/kconfig/confdata.c
    ----------------------------------------------------------------------
cmd() { make -j --quiet "$@"  || { [ "" = "0" ] && echo && cat .build.log 2>/dev/null; kill $$ 2>/dev/null || kill $$$$ 2>/dev/null; printf "\n\\033[33m%s\\033[m\n" "ERROR: ";  exit 1; } };                   mkdir -p source; [ -n "" ] && step="/" || step=""; [ -n "" ] && step="$step/"; case "" in BIN)      echo -n "package/"         >source/.echo_item_tmp ;; LIB)       echo -n "library/"         >source/.echo_item_tmp ;; HTL)       echo -n "tools/"           >source/.echo_item_tmp ;; KTC)   echo -n "toolchain/kernel$step" >source/.echo_item_tmp ;; TTC)  echo -n "toolchain/target$step" >source/.echo_item_tmp ;; KRN)  echo -n "kernel"                 >source/.echo_item_tmp ;; esac; if ! diff -q source/.echo_item_tmp source/.echo_item_new >/dev/null 2>&1 || [ ! -e source/.echo_item_1st ]; then   if [ -e source/.echo_item_end -a -e source/.echo_item_new -a -e source/.echo_item_1st ]; then echo -e "\e[48;5;26mdone\e[49m."; rm -f source/.echo_item_end source/.echo_item_1st; fi; [ -s source/.echo_item_tmp ] && cat source/.echo_item_tmp > source/.echo_item_new 2>/dev/null; [ -s source/.echo_item_new ] || cat source/.echo_item_old > source/.echo_item_new 2>/dev/null; if [ -s "source/.echo_item_new" ]; then echo -ne "\e[48;5;90m---> "; cat source/.echo_item_new 2>/dev/null | tee source/.echo_item_old; echo -ne "\e[49m ... "; [ "" != "0" ] && echo; touch source/.echo_item_end; touch source/.echo_item_1st; fi; fi; echo -ne "\e[48;5;56mbuilding\e[49m ... "; cmd -C /home/janpieter/freetz-ng-master/source/host-tools/kconfig-v6.1 menuconfig
building ...   HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/confdata.o
  HOSTCC  scripts/kconfig/expr.o
  LEX     scripts/kconfig/lexer.lex.c
  YACC    scripts/kconfig/parser.tab.[ch]
  HOSTCC  scripts/kconfig/menu.o
  HOSTCC  scripts/kconfig/preprocess.o
  HOSTCC  scripts/kconfig/symbol.o
  HOSTCC  scripts/kconfig/util.o
  UPD     scripts/kconfig/mconf-cfg
  HOSTCC  scripts/kconfig/mconf.o
  HOSTCC  scripts/kconfig/lxdialog/checklist.o
  HOSTCC  scripts/kconfig/lxdialog/inputbox.o
  HOSTCC  scripts/kconfig/lxdialog/menubox.o
  HOSTCC  scripts/kconfig/lxdialog/textbox.o
  HOSTCC  scripts/kconfig/lxdialog/util.o
  HOSTCC  scripts/kconfig/lxdialog/yesno.o
  HOSTCC  scripts/kconfig/lexer.lex.o
  HOSTCC  scripts/kconfig/parser.tab.o
  HOSTLD  scripts/kconfig/mconf
/usr/lib/gcc/x86_64-pc-linux-gnu/12/../../../../x86_64-pc-linux-gnu/bin/ld: scripts/kconfig/lxdialog/checklist.o: undefined reference to symbol 'keypad'
/usr/lib/gcc/x86_64-pc-linux-gnu/12/../../../../x86_64-pc-linux-gnu/bin/ld: /lib64/libtinfow.so.6: error adding symbols: DSO missing from command line
collect2: fout: ld gaf exit-status 1 terug
make[3]: *** [/home/janpieter/freetz-ng-master/source/host-tools/kconfig-v6.1/scripts/Makefile.host:120: scripts/kconfig/mconf] Error 1
make[2]: *** [Makefile:78: menuconfig] Error 2
make[1]: *** [make/host-tools/kconfig-host/kconfig-host.mk:29: /home/janpieter/freetz-ng-master/source/host-tools/kconfig-v6.1/scripts/kconfig/mconf] Beëindigd
make: *** [Makefile:46: envira] Beëindigd
```

Any hints on unsupported ncurses versions?
I'm using version 6.3, I tried installing 6.2 compatibility libraries, for ABI 5, but no result

fda77, 2022-12-13:
The older version was okay? Yesterday was a version bump, you could revert to check: https://github.com/Freetz-NG/freetz-ng/commit/7b4f0548fd407e01a866faf06aab45100dde6b42 (`make kconfig-host-recompile` to force recompile)
I'm using Fedora which also uses ncurses 6.3, that should not be the problem
There is another report about kconfig, but this is about 6.0: https://github.com/Freetz-NG/freetz-ng/issues/669

fda77, 2022-12-13:
https://github.com/openwrt/openwrt/issues/7308

fda77, 2022-12-13:
> make menuconfig
> Freetz-NG UNKNOWN

And why does the version detection not work for you? Is this not a reposity checkout?

jpsollie, 2022-12-13:
The one I'm pasting was download plain zip and extract, the git version is almost the same:
```
make menuconfig
Freetz-NG 18023-cb2778636 ng22100 2022-12-12
cmd() { make -j16  "$@"  || { [ "2" = "0" ] && echo && cat .build.log 2>/dev/null; kill $$ 2>/dev/null || kill $$$$ 2>/dev/null; printf "\n\\033[33m%s\\033[m\n" "ERROR: Build failed.";  exit 1; } };                  mkdir -p source; [ -n "" ] && step="/" || step=""; [ -n "" ] && step="$step/"; case "" in BIN)      echo -n "package/"         >source/.echo_item_tmp ;; LIB)       echo -n "library/"         >source/.echo_item_tmp ;; HTL)       echo -n "tools/"           >source/.echo_item_tmp ;; KTC)   echo -n "toolchain/kernel$step" >source/.echo_item_tmp ;; TTC)  echo -n "toolchain/target$step" >source/.echo_item_tmp ;; KRN)  echo -n "kernel"                 >source/.echo_item_tmp ;; esac; if ! diff -q source/.echo_item_tmp source/.echo_item_new >/dev/null 2>&1 || [ ! -e source/.echo_item_1st ]; then   if [ -e source/.echo_item_end -a -e source/.echo_item_new -a -e source/.echo_item_1st ]; then echo -e "\e[48;5;26mdone\e[49m."; rm -f source/.echo_item_end source/.echo_item_1st; fi; [ -s source/.echo_item_tmp ] && cat source/.echo_item_tmp > source/.echo_item_new 2>/dev/null; [ -s source/.echo_item_new ] || cat source/.echo_item_old > source/.echo_item_new 2>/dev/null; if [ -s "source/.echo_item_new" ]; then echo -ne "\e[48;5;90m---> "; cat source/.echo_item_new 2>/dev/null | tee source/.echo_item_old; echo -ne "\e[49m ... "; [ "2" != "0" ] && echo; touch source/.echo_item_end; touch source/.echo_item_1st; fi; fi; echo -ne "\e[48;5;56mbuilding\e[49m ... "; cmd -C /home/janpieter/modem/freetz-ng/source/host-tools/kconfig-v6.1 menuconfig
done.
---> tools/kconfig-host ... 
building ...   HOSTLD  scripts/kconfig/mconf
/usr/lib/gcc/x86_64-pc-linux-gnu/12/../../../../x86_64-pc-linux-gnu/bin/ld: scripts/kconfig/lxdialog/checklist.o: undefined reference to symbol 'keypad'
/usr/lib/gcc/x86_64-pc-linux-gnu/12/../../../../x86_64-pc-linux-gnu/bin/ld: /lib64/libtinfow.so.6: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[3]: *** [/home/janpieter/modem/freetz-ng/source/host-tools/kconfig-v6.1/scripts/Makefile.host:120: scripts/kconfig/mconf] Error 1
make[2]: *** [Makefile:78: menuconfig] Error 2
make[1]: *** [make/host-tools/kconfig-host/kconfig-host.mk:29: /home/janpieter/modem/freetz-ng/source/host-tools/kconfig-v6.1/scripts/kconfig/mconf] Beëindigd
make: *** [Makefile:46: envira] Beëindigd
```
I did this to see if any git issues were causing the problem, but apparently it wasn't true

jpsollie, 2022-12-13:
> * [kconfig: compiler error "libtinfow.so.6: error adding symbols: DSO missing from command line" #627](https://github.com/Freetz-NG/freetz-ng/issues/627)
> 
>     * https://patchwork.kernel.org/project/linux-kbuild/patch/20150425160546.GA4002@freedom/
> 
> 
> Are you a Debian user?

no, gentoo

jpsollie, 2022-12-13:
> The older version was okay? Yesterday was a version bump, you could revert to check: [7b4f054](https://github.com/Freetz-NG/freetz-ng/commit/7b4f0548fd407e01a866faf06aab45100dde6b42) (`make kconfig-host-recompile` to force recompile) I'm using Fedora which also uses ncurses 6.3, that should not be the problem There is another report about kconfig, but this is about 6.0: #669

I rebased and dropped that commit, but it didn't help:

```Freetz-NG 18022-92a1afab7 ng21090 2022-12-12
preparing ... mkdir -p /home/janpieter/modem/freetz-ng/source/host-tools/kconfig-v6.0; tar -C /home/janpieter/modem/freetz-ng/source/host-tools -xf dl/kconfig-v6.0.tar.xz 
set -e; shopt -s nullglob; for i in /home/janpieter/modem/freetz-ng/make/host-tools/kconfig-host/patches/*.patch*; do case $i in *.patch|*.patch.gz|*.patch.bzip2|*.patch.bz2|*.patch.bz|*.patch.xz|*.patch.lz|*.patch.lzma|*.patch.Z|*.patch.diff) ;; *) continue ;; esac; tools/freetz_patch /home/janpieter/modem/freetz-ng/source/host-tools/kconfig-v6.0 $i; done; 
    applying patch file /home/janpieter/modem/freetz-ng/make/host-tools/kconfig-host/patches/300-main_makefile.patch
    patching file Makefile
    ----------------------------------------------------------------------
    applying patch file /home/janpieter/modem/freetz-ng/make/host-tools/kconfig-host/patches/310-fix_make.patch
    patching file scripts/kconfig/Makefile
    ----------------------------------------------------------------------
    applying patch file /home/janpieter/modem/freetz-ng/make/host-tools/kconfig-host/patches/320-unchangable_style.patch
    patching file scripts/kconfig/mconf.c
    ----------------------------------------------------------------------
    applying patch file /home/janpieter/modem/freetz-ng/make/host-tools/kconfig-host/patches/330-comment_style.patch
    patching file scripts/kconfig/mconf.c
    ----------------------------------------------------------------------
    applying patch file /home/janpieter/modem/freetz-ng/make/host-tools/kconfig-host/patches/340-fail_on_unmet_direct_dependency.patch
    patching file scripts/kconfig/symbol.c
    ----------------------------------------------------------------------
    applying patch file /home/janpieter/modem/freetz-ng/make/host-tools/kconfig-host/patches/350-max_item_count_for_choice_dialog.patch
    patching file scripts/kconfig/mconf.c
    ----------------------------------------------------------------------
    applying patch file /home/janpieter/modem/freetz-ng/make/host-tools/kconfig-host/patches/360-save-hidden-prompts-to-file.patch
    patching file scripts/kconfig/menu.c
    ----------------------------------------------------------------------
    applying patch file /home/janpieter/modem/freetz-ng/make/host-tools/kconfig-host/patches/370-multiline_comments_hack.patch
    patching file scripts/kconfig/mconf.c
    patching file scripts/kconfig/menu.c
    ----------------------------------------------------------------------
    applying patch file /home/janpieter/modem/freetz-ng/make/host-tools/kconfig-host/patches/380-project_name.patch
    patching file scripts/kconfig/mconf.c
    ----------------------------------------------------------------------
    applying patch file /home/janpieter/modem/freetz-ng/make/host-tools/kconfig-host/patches/390-menu_help.patch
    patching file scripts/kconfig/mconf.c
    ----------------------------------------------------------------------
    applying patch file /home/janpieter/modem/freetz-ng/make/host-tools/kconfig-host/patches/391-no_symbol_prefix.patch
    patching file scripts/kconfig/lkc.h
    ----------------------------------------------------------------------
    applying patch file /home/janpieter/modem/freetz-ng/make/host-tools/kconfig-host/patches/392-use_explicit_flex_call.patch
    patching file scripts/kconfig/Makefile
    ----------------------------------------------------------------------
    applying patch file /home/janpieter/modem/freetz-ng/make/host-tools/kconfig-host/patches/393-no_POSIX_warning_from_bison.patch
    patching file scripts/kconfig/Makefile
    ----------------------------------------------------------------------
    applying patch file /home/janpieter/modem/freetz-ng/make/host-tools/kconfig-host/patches/400-do_not_create_autoconf_stuff.patch
    patching file scripts/kconfig/confdata.c
    ----------------------------------------------------------------------
cmd() { make -j16  "$@"  || { [ "2" = "0" ] && echo && cat .build.log 2>/dev/null; kill $$ 2>/dev/null || kill $$$$ 2>/dev/null; printf "\n\\033[33m%s\\033[m\n" "ERROR: Build failed.";  exit 1; } };                  mkdir -p source; [ -n "" ] && step="/" || step=""; [ -n "" ] && step="$step/"; case "" in BIN)      echo -n "package/"         >source/.echo_item_tmp ;; LIB)       echo -n "library/"         >source/.echo_item_tmp ;; HTL)       echo -n "tools/"           >source/.echo_item_tmp ;; KTC)   echo -n "toolchain/kernel$step" >source/.echo_item_tmp ;; TTC)  echo -n "toolchain/target$step" >source/.echo_item_tmp ;; KRN)  echo -n "kernel"                 >source/.echo_item_tmp ;; esac; if ! diff -q source/.echo_item_tmp source/.echo_item_new >/dev/null 2>&1 || [ ! -e source/.echo_item_1st ]; then   if [ -e source/.echo_item_end -a -e source/.echo_item_new -a -e source/.echo_item_1st ]; then echo -e "\e[48;5;26mdone\e[49m."; rm -f source/.echo_item_end source/.echo_item_1st; fi; [ -s source/.echo_item_tmp ] && cat source/.echo_item_tmp > source/.echo_item_new 2>/ [... cut ...]
/usr/lib/gcc/x86_64-pc-linux-gnu/12/../../../../x86_64-pc-linux-gnu/bin/ld: /lib64/libtinfow.so.6: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[3]: *** [/home/janpieter/modem/freetz-ng/source/host-tools/kconfig-v6.0/scripts/Makefile.host:104: scripts/kconfig/mconf] Error 1
make[2]: *** [Makefile:78: menuconfig] Error 2
make[1]: *** [make/host-tools/kconfig-host/kconfig-host.mk:28: /home/janpieter/modem/freetz-ng/source/host-tools/kconfig-v6.0/scripts/kconfig/mconf] Beëindigd
make: *** [Makefile:46: envira] Beëindigd
```

fda77, 2022-12-13:
https://freetz-ng.github.io/freetz-ng/NEWS.html#2022-10-16
> Die automatisch angebotene und nicht deaktivierbare .zip-Datei darf auf keinen Fall verwendet werden, da
> darin keine Dateiberechtigungen enthalten sind! In der .tar.gz fehlt mindestens die Versionsinformation.

This problem seems to exist for a long time, also before v6.0. There is something "special" with the ncurses libs.
You could compatre the output of "pkgconf ":  https://github.com/Freetz-NG/freetz-ng/issues/627#issuecomment-1304662814

jpsollie, 2022-12-13:
yes, that worked.  Thanks!
topic closed

closed 2022-12-13 by jpsollie

fda77, 2022-12-13:
Okay