Freetz-NG/freetz-ng #681: uudecode wird benötigt aber sharutils fehlt in docs/PREREQUISITES.md

pfichtner, 2022-12-19:
[uudecode](https://github.com/Freetz-NG/freetz-ng/blob/0606e890d03528c07fc417d83b0d79058fcfb167/.prerequisites#L46) wird von [`asterisk-gui` benötigt](https://github.com/Freetz-NG/freetz-ng/blob/0606e890d03528c07fc417d83b0d79058fcfb167/make/pkgs/asterisk-gui/asterisk-gui.mk#L6), ist aber zumindest bei ubuntu nach Installation aus [docs/PREREQUISITES.md](https://github.com/Freetz-NG/freetz-ng/blob/master/docs/PREREQUISITES.md) nicht vefügbar. 

Mögliche Lösung: Paket `sharutils` aufnehmen (Ubuntu und Fedora)
Alternative: Da `asterisk-gui` ziemlich abgehangen ist und vermutlich selten benutzt wird,  ignorieren, dann sollte `uudecode` aber konsequenterweise aus .prerequisites entfernt werden und mögliche build Versuche entsprechend fehlschlagen (auch nicht schön)

fda77, 2022-12-20:
Ich bin für aufnehmen, wenn ich diese 100 Pakete installiere erwarte ich dass ich danach gar nichts mehr nachinstallieren muss

closed 2022-12-20 by pfichtner

fda77, 2022-12-20:
Du hast bis Ubuntu 9 aus dem Jahr 2009 geprüft ob das Paket damals existierte und auch nicht umbenannt wurde? Normalerweise lässt man den alten Kram wie er ist und aktualisiert nur die Neueste, evtl noch die Vorletzte. Bei Fedora 37 stimmt es zumindest

pfichtner, 2022-12-20:
Ich habe genau das gemacht, was im commit Kommentar steht (deshalb schreib ich das auch rein): "Tested on Fedora 20/35/36 and Ubuntu 14.04/20.04/22.04". Das bedeutet im Umkehrschluss logischwerweise, dass ich es unter Ubuntu 9 **nicht** getetstet habe. 
Wie viele Leute wird es geben, die auf Basis einer >12 Jahre alten Distribution ihr Build-Environment aufsetzen wollen? Und falls ja, werden sie hier aufschlagen und Du darfst sie auslachen. Wenn Du dem zuvorkommen willst, nimm die sharutils für Ubuntu <14 raus oder sag Bescheid, dann tue ich dies.

pfichtner, 2022-12-20:
Nachtrag, Ubuntu 10 noch getestet, das kennt das package `sharutils` ebenfalls, die Installation scheitert, da die Paketlisten nicht mehr verfügbar und damit nicht mehr aktualisierbar sind.

pfichtner, 2022-12-20:
> Normalerweise lässt man den alten Kram wie er ist und aktualisiert nur die Neueste, evtl noch die Vorletzte. 

Ich dachte immer, die aufgelisteten Distributionen funktionieren? Das klingt jetzt aber eher so wie "die aktuelle funktioniert, vielleicht mit Glück auch noch die vorletzte, die vorvorletzte aber höchstwahrscheinlich nicht mehr"
Da stellt sich dann ja schon die Frage, ob der Bau mit der vorvorvorvorvorvorvorvorletzten beschriebenen Ubuntu Version 9.04 gelingen wird.

fda77, 2022-12-20:
Diese alberne Diskussionen nerven mich. Ich hab dazu weder Zeit noch Lust. Gab es git schon in u9? Mit allen Features die genutzt werden?

> dachte immer, die aufgelisteten Distributionen funktionieren 

gleich 2 Fehler, ein einem Satz, ja BEIDE funktionieren

pfichtner, 2022-12-20:
> > dachte immer, die aufgelisteten Distributionen funktionieren
> 
> gleich 2 Fehler, ein einem Satz, ja BEIDE funktionieren

Was für zwei Fehler? Mit DistributionEN meinte ich die aufgelisteten DistrubutionsVERSIONEN. 

> Diese alberne Diskussionen nerven mich. Ich hab dazu weder Zeit noch Lust. Gab es git schon in u9? Mit allen Features die genutzt werden?

Ich weiß nicht, was an der Diskussion albern ist, ich habe nur hinterfragt, ob mit all den aufgelisteten DistrubutionsVERSIONEN dann ein Build möglich ist, wenn mittlerweile benötigte Pakete dort nicht nachgezogen werden. Das scheint aber offensichtlich egal (bzw. wohl eine alberne Diskussion). Und nein: U9 ist verifiziert, da mir das zu aufwändig ist, das zu testen. Ich habe es daher einfach jetzt so wie von Dir vorgeschlagen umgesetzt.

pfichtner, 2022-12-20:
> Gab es git schon in u9? Mit allen Features die genutzt werden?

Weiß ich nicht, ich habe kein u9 zu Hand. Wenn ein Build mit U9 nicht (mehr) möglich ist, würde ich es aus dem PREREQUISITES rausnehmen. Ja, die Diskussion bringt nicht viel, was daran albern ist, verstehe ich allerdings nicht. Welchen Sinn macht es, eine DistrubutionsVERSION zu beschreiben, mit der ein Build nicht möglich ist? 

"Getestete Distributionen" bedeutet: "Hat irgendwann mal damit funktioniert" bedeutet aber eber nicht, dass ein aktuelles Freetz-NG damit noch baubar ist. Ist jetzt klar, geht aber aus dem Dokument nicht hervor. 
Kann ich, wenn gewünscht, ergänzen. 
Sieht so aus, als ob die Diskussion eben alles andere als albern ist/war. Zumindest mich hat noch kein Gespräch bislang dümmer gemacht als vorher.

pfichtner, 2022-12-21:
> > > dachte immer, die aufgelisteten Distributionen funktionieren
> > 
> > gleich 2 Fehler, ein einem Satz, ja BEIDE funktionieren
> 
> Was für zwei Fehler? Mit DistributionEN meinte ich die aufgelisteten DistrubutionsVERSIONEN.

Hab's, **diese** zwei Fehler meintest Du bezüglich "funktionieren"? 

```
[vagrant@fedora freetz-ng]$ make
Freetz-NG 20657-08a9aa19f master 2022-12-21
ERROR: The program 'python' was not found in path
ERROR: The program 'rpcgen' was not found in path
Makefile:180: *** Some build prerequisites are missing! See '.prerequisites' for why. Please install the missing packages before trying again. See https://freetz-ng.github.io/freetz-ng/PREREQUISITES for installation hints.  Stop.
make: *** [Makefile:46: envira] Terminated
```

Aber Du sagst ja, 
> ja BEIDE funktionieren

Was macht Dich so sicher, dass sie funktionieren, wie wird/wurde das sichergestellt und was habe ich falsch gemacht, dass es bei mir dann nicht funktioniert? 

```
[vagrant@fedora freetz-ng]$ history 

# aus https://github.com/Freetz-NG/freetz-ng/blob/master/docs/PREREQUISITES.md gecopy-pasted
sudo dnf -y groupinstall 'Development Tools' 'Development Libraries'
sudo dnf -y install pv cpio rsync kmod execstack sqlite.i686 sqlite-devel cmake zlib-devel.i686 libstdc++-devel.i686 openssl xz bc unar inkscape ImageMagick subversion ccache gcc gcc-c++ binutils autoconf automake libtool make bzip2 libglade2-devel qt5-qtbase-devel ncurses-devel ncurses-term zlib-devel flex bison patch texinfo gettext pkgconfig ecj perl perl-String-CRC32 wget glib2-devel git util-linux libacl-devel libattr-devel libcap-devel ncurses-devel.i686 glibc-devel.i686 libgcc.i686 libuuid-devel openssl-devel gnutls-devel libzstd-devel.x86_64 libstdc++-devel.x86_64 netcat curl sharutils

# Nur zur Sicherheit
sudo reboot

umask 0022
git clone https://github.com/Freetz-NG/freetz-ng
cd freetz-ng/
make menuconfig
make
```

```
[vagrant@fedora freetz-ng]$ hostnamectl status
   Static hostname: n/a                             
Transient hostname: fedora
         Icon name: computer-vm
           Chassis: vm 🖴
        Machine ID: 6d77623b6167497786915951951e51e3
           Boot ID: b0256a466b8f48c2bd6c878d2f8ad0b5
    Virtualization: oracle
  Operating System: Fedora Linux 36 (Thirty Six)    
       CPE OS Name: cpe:/o:fedoraproject:fedora:36
            Kernel: Linux 6.0.12-200.fc36.x86_64
      Architecture: x86-64
   Hardware Vendor: innotek GmbH
    Hardware Model: VirtualBox
```

fda77, 2022-12-21:
Ist das das gleich wie im anderen?

pfichtner, 2022-12-21:
Ja, deswegen auch die Vor- _"Originally posted by @pfichtner in https://github.com/Freetz-NG/freetz-ng/issues/681#issuecomment-1362057314"_ und Rückwärtsverlinkung _"@[pfichtner](https://github.com/pfichtner) pfichtner mentioned this issue Dec 21, 2022
python (u.a.) nach Anleitung aus docs/PREREQUISITES.md auf Fedora 36 nicht verfügbar #683"_

pfichtner, 2022-12-22:
> Was macht Dich so sicher, dass sie funktionieren, wie wird/wurde das sichergestellt und was habe ich falsch gemacht, dass es bei mir dann nicht funktioniert?

Hab's mir jetzt selbst erarbeitet, wie man automatisch verifiziert, ob das System nach Abarbeitung der Anleitung auch alles beinhaltet was Freetz-NG benötigt (unabhängig von der .config und damit dann eben "nur" eventuell benötigt). 

Ergebnis des Laufs: 
```
- ubuntu:20.04
ERROR: The program 'xml2-config' was not found in path

- ubuntu:22.04
ERROR: The program 'xml2-config' was not found in path

- fedora36
ERROR: The program 'fastjar' was not found in path
ERROR: The program 'javac' was not found in path
ERROR: The program 'python' was not found in path
ERROR: The program 'rpcgen' was not found in path
```

Für Ubuntu 20/22 fehlt `libxml2-dev` 
Für fedora36 python und rpcgen fehlt wie in #683 beschrieben `rpcgen` und `python-unversioned-command`

Ich weiß ja noch nicht, was Dir auf die Nüsse geht. Dass sich jemand die prerequisites anschaut, ausprobiert, Probleme meldet, Probleme meldet aber nicht korrigiert, Probleme korrigiert aber nicht so wie Du es machen würdest, ..., solange ich das nicht weiß, mach ich mal Punkt 1: Probleme melden/transparent machen/kommentieren. 

PS: Mein erster Ansatz war ein naives ausführen von `make allyesconfig`, aber das Schlug fehl (anderes und für mich irrelevantes Thema)
```
Freetz-NG 20659M-a9e9194a2 master 2022-12-21
ERROR: unmet direct dependencies detected for FREETZ_PACKAGE_SUNDTEK
  Depends on [n]: FREETZ_KERNEL_VERSION_2_6_19_MIN [=y] && !FREETZ_TARGET_ARCH_X86 [=n] && !FREETZ_SEPARATE_AVM_UCLIBC [=y]
  Selected by [y]:
  - FREETZ_PACKAGE_RRDSTATS_SEGMENTLOAD_SUNDTEK [=y] && FREETZ_PACKAGE_RRDSTATS [=y] && FREETZ_PACKAGE_RRDSTATS_SEGMENTLOAD [=y] && FREETZ_KERNEL_VERSION_2_6_19_MIN [=y] && !FREETZ_TARGET_ARCH_X86 [=n]
make[1]: *** [Makefile:543: allyesconfig] Error 1
make: *** [Makefile:46: envira] Terminated
```

pfichtner, 2022-12-22:
> - fedora36
> ERROR: The program 'fastjar' was not found in path
> ERROR: The program 'javac' was not found in path
> ERROR: The program 'python' was not found in path
> ERROR: The program 'rpcgen' was not found in path
> ```

> Für Ubuntu 20/22 fehlt `libxml2-dev` Für fedora36 python und rpcgen fehlt wie in #683 beschrieben `rpcgen` und `python-unversioned-command`

Nachtrag zu `fastjar` und `javac`. Habe mal reingeschaut: `fastjar` ist unter ubuntu da, da es explizit als benötigt aufgelistet ist. 

ecj hat unter beiden Distros eine Abhängigkeit zu `java-runtime-headless` und zieht somit java mit. Unter Ubuntu scheint das allerdings ein JDK inkl. `javac` zu sein, auf Fedora eine JRE ohne `javac` (dependsOn "`java-headless` or `java-17-headless`"). 

Für die vier Punkte müssten also folgende Pakete noch hinzugefügt werden: 
- `python`: `python-unversioned-command`
- `rpcgen`: `rpcgen`
- `javac`: `java-devel` (installiert unter fedora openjdk-17 gegenüber openjdk-11 unter ubuntu, sollte aber kein Problem sein, falls doch: `java-11-openjdk-devel`)
- `fastjar`: Unbekannt, `dnf search` als auch `command-not-found fastjar` schweigt sich aus

pfichtner, 2022-12-22:
- fedora:36
bash-5.2$ ecj
/usr/bin/ecj: line 8: /usr/share/java-utils/java-functions: No such file or directory
/usr/bin/ecj: line 27: set_jvm: command not found
/usr/bin/ecj: line 28: set_classpath: command not found
/usr/bin/ecj: line 29: set_flags: command not found
/usr/bin/ecj: line 30: set_options: command not found
/usr/bin/ecj: line 33: run: command not found

- ubuntu:22.04
builduser@c08bfb8817b5:~$ ecj
/usr/bin/ecj: 4: .: cannot open /usr/lib/java-wrappers/java-wrappers.sh: No such file

Wer oder was auch immer von ecj/javac abhängt: Es dürfte nicht funktionieren

classpath (als Nutzer in [.prerequisites](https://github.com/Freetz-NG/freetz-ng/blob/master/.prerequisites) gelistet) tut das auch mit genau dieser Fehlermeldung: 
```
checking if javac is a version of gcj... /usr/bin/javac: 4: .: cannot open /usr/lib/java-wrappers/java-wrappers.sh: No such file
checking if javac works... configure: error: The Java compiler javac failed (see config.log, check the CLASSPATH?)
```

Nach Installation von `openjdk-11-jdk-headless`
```
configure:35343: checking if javac works
configure:35368: javac  -source 1.5 -target 1.5 Object.java
warning: [options] bootstrap class path not set in conjunction with -source 5
error: Source option 5 is no longer supported. Use 6 or later.
error: Target option 1.5 is no longer supported. Use 1.6 or later.
configure:35371: $? = 2
configure:35375: error: The Java compiler javac failed (see config.log, check the CLASSPATH?)
```

"openjdk-11" ist also "zu neu" (bedeutet damit logischerweise, dass es mit openjdk-17 erst recht nicht funktionieren wird). 

Nach Installation von `openjdk-8-jdk-headless` baut `classpath` erfolgreich. 
Durch die Installation von `openjdk-8-jdk-headless` kann der (nicht funktionierende) ecj Kompiler (Paket ecj) entfernt werden.

pfichtner, 2022-12-22:
> Nach Installation von `openjdk-8-jdk-headless` baut `classpath` erfolgreich. Durch die Installation von `openjdk-8-jdk-headless` kann der (nicht funktionierende) ecj Kompiler (Paket ecj) entfernt werden.

Nachtrag: 
- Ubuntu: `openjdk-8-jdk-headless`
- Fedora: `java-1.8.0-openjdk-devel`

pfichtner, 2022-12-23:
ecj fuktioniert unter Ubuntu nach der Installation von `java-wrappers`, unter Fedora nach der Installation von `javapackages-tools`. Allerdings muss bei Fedora, wie gesagt, `ecj` noch als `javac` gelinkt werden, damit `javac` im PATH auch verfügbar ist. 
ecj funktioniert damit unter beiden Distros, der Bau des `classpath` Pakets funktioniert dann unter Ubuntu, bei Fedora scheitert es dann am genannten fehlenden Programm/Paket `fastjar`.

pfichtner, 2022-12-27:
Ich habe halt mal geschaut, welches program/binary zum Bau für was da ist und dann versucht die entsprechenden Pakete zu bauen. Das es nicht geht liegt aber nicht daran, dass die Pakete zu alt sind, sondern dass der `ecj` auf den neuen Systemen nach der Installation von `ecj` alleine nicht funktioniert. 
Fedora 37 ist derzeit noch nicht in den PREREQUISITES beschrieben, vertestet habe ich es aber: Ist analog zu Fedora 36 (inklusive dem gleichen Problem bezüglich `fastjar`)

fda77, 2022-12-30:
Ich update Fedora bei mir nun nur noch, da kann sich ein Package über die Zeit mal ändern

pfichtner, 2022-12-30:
Deswegen habe ich das ja vollautomatisiert, sobald sich die https://github.com/Freetz-NG/freetz-ng/blob/master/docs/PREREQUISITES.md ändern. 

Dazu wollte ich ja von Dir wissen, wie Du verifiziert, ob die Beschreibungen auch zum gewünschten Ergebnis führen, habe es aber ja mittlerweile (jetzt dann eben eventuell redundant) selbst implementiert.

fda77, 2022-12-30:
Ich wüsste nicht das ich was implementiert hätte

pfichtner, 2022-12-30:
> Ich wüsste nicht das ich was implementiert hätte

Wusste nicht ob es was gibt, deswegen hatte ich gefragt, jetzt weiß ich's

fda77, 2022-12-30:
Äh, keine Ahnung um was es geht! Könnte auch sein dass ich die 5 Threads durcheinander geworfen hab.
Fedroa37 hat jedenfals fastjar: `fastjar.aarch64 : Fast Java Archive (JAR) Tool`

pfichtner, 2022-12-30:
Ich hatte gefragt, wie sichergestellt wird, dass die Angaben in der https://github.com/Freetz-NG/freetz-ng/blob/master/docs/PREREQUISITES.md stimmen, also dass nach Abarbeitung des Befehle auch alle notwendigen Binaries, Libs etc. auf dem System wirklich vorhanden sind. 
Nachdem keine Antwort kam, habe ich das mittlerweile selbst implementiert. Du hast jetzt noch nachgeliefert, dass es bislang nichts gab, was das verifiziert hat.

pfichtner, 2022-12-30:
> Fedroa37 hat jedenfals fastjar: `fastjar.aarch64 : Fast Java Archive (JAR) Tool`

Fedora selbst wohl nicht, aber in [third party repos](https://fedora.pkgs.org/35/rpm-sphere-aarch64/fastjar-0.98-5.1.aarch64.rpm.html) wohl, oder wo hast Du es gefunden?

fda77, 2022-12-31:
Keine Ahnung! War einfach da. Evtl auch installiert als es das noch gab

commit 437272d9 (2022-12-20): PREREQUISITES: sharutils added, fixes #681

Tested on Fedora 20/35/36 and Ubuntu 14.04/20.04/22.04
commit ee6610fe (2022-12-20): Revert "PREREQUISITES: sharutils added, fixes #681"

This reverts commit 437272d9794e3d349df7070c460260ff01714ea2.