Freetz-NG/freetz-ng #683: python (u.a.) nach Anleitung aus docs/PREREQUISITES.md auf Fedora 36 nicht verfügbar

pfichtner, 2022-12-21:
> > > dachte immer, die aufgelisteten Distributionen funktionieren
> > 
> > gleich 2 Fehler, ein einem Satz, ja BEIDE funktionieren
> 
> Was für zwei Fehler? Mit DistributionEN meinte ich die aufgelisteten DistrubutionsVERSIONEN.

Hab's, **diese** zwei Fehler meintest Du bezüglich "funktionieren"? 

```
[vagrant@fedora freetz-ng]$ make
Freetz-NG 20657-08a9aa19f master 2022-12-21
ERROR: The program 'python' was not found in path
ERROR: The program 'rpcgen' was not found in path
Makefile:180: *** Some build prerequisites are missing! See '.prerequisites' for why. Please install the missing packages before trying again. See https://freetz-ng.github.io/freetz-ng/PREREQUISITES for installation hints.  Stop.
make: *** [Makefile:46: envira] Terminated
```

Aber Du sagst ja, 
> ja BEIDE funktionieren

Was macht Dich so sicher, dass sie funktionieren, wie wird/wurde das sichergestellt und was habe ich falsch gemacht, dass es bei mir dann nicht funktioniert? 

```
[vagrant@fedora freetz-ng]$ history 

# aus https://github.com/Freetz-NG/freetz-ng/blob/master/docs/PREREQUISITES.md gecopy-pasted
sudo dnf -y groupinstall 'Development Tools' 'Development Libraries'
sudo dnf -y install pv cpio rsync kmod execstack sqlite.i686 sqlite-devel cmake zlib-devel.i686 libstdc++-devel.i686 openssl xz bc unar inkscape ImageMagick subversion ccache gcc gcc-c++ binutils autoconf automake libtool make bzip2 libglade2-devel qt5-qtbase-devel ncurses-devel ncurses-term zlib-devel flex bison patch texinfo gettext pkgconfig ecj perl perl-String-CRC32 wget glib2-devel git util-linux libacl-devel libattr-devel libcap-devel ncurses-devel.i686 glibc-devel.i686 libgcc.i686 libuuid-devel openssl-devel gnutls-devel libzstd-devel.x86_64 libstdc++-devel.x86_64 netcat curl sharutils

# Nur zur Sicherheit
sudo reboot

umask 0022
git clone https://github.com/Freetz-NG/freetz-ng
cd freetz-ng/
make menuconfig
make
```

```
[vagrant@fedora freetz-ng]$ hostnamectl status
   Static hostname: n/a                             
Transient hostname: fedora
         Icon name: computer-vm
           Chassis: vm 🖴
        Machine ID: 6d77623b6167497786915951951e51e3
           Boot ID: b0256a466b8f48c2bd6c878d2f8ad0b5
    Virtualization: oracle
  Operating System: Fedora Linux 36 (Thirty Six)    
       CPE OS Name: cpe:/o:fedoraproject:fedora:36
            Kernel: Linux 6.0.12-200.fc36.x86_64
      Architecture: x86-64
   Hardware Vendor: innotek GmbH
    Hardware Model: VirtualBox
```

_Originally posted by @pfichtner in https://github.com/Freetz-NG/freetz-ng/issues/681#issuecomment-1362057314_

fda77, 2022-12-21:
> Aber Du sagst ja,

Auch zum zitieren: Geh mir nicht auf die Nüsse. Noch Fragen?

pfichtner, 2022-12-21:
Ja: Wieso auf die Nüsse gehen und wieso so agressiv? Ich habe nen Issue aufgemacht, da bei mir der build unter Fedora 36 (nicht meine Distro, aber ich wollte Fedora mal eine Chance geben) nicht funktioniert. Der Grund: Entweder bei mir ist irgendwas unerklärbares anders (deshalb habe ich alle vermeintlich notwendigen Informationen zur Verfügung gestellt) oder es fehlen doch Pakete in der PREREQUISITES.md. 

Bei mir startet der Build nach Installation von `rpcgen` und `python-unversioned-command`
`sudo dnf -y install rpcgen python-unversioned-command`

pfichtner, 2022-12-21:
> Auch zum zitieren: Geh mir nicht auf die Nüsse. Noch Fragen?

Ja, ein paar. 

Was ist denn Deine Erwartung an die Community?  Was geht Dir denn auf die Nüsse? 
- Dass Probleme gemeldet werden? 
- Dass Änderungen durch "andere" gemacht werden? 
- Dass Änderungen durch "andere" gemacht werden die nicht in Deinem Sinne sind? 
- Dass Sachen diskutiert werden? 
- Dass Sachen hinterfragt werden? 

Der Umgangston hier wurde ja kürzlich schon mal kritisiert. Vielleicht liegt es ja daran, dass Du andere Erwartungen und/oder Vorstellungen an kolloborative Zusammenarbeit (oder vielleicht sogar gar kein Interesse daran) hast. Wenn Dir also alle Leute nur auf die Nüsse gehen, dann fehlt offensichtlich eine Beschreibung eines erwartenden Verhaltens hier.

fda77, 2022-12-27:
Ich hab keine Lust auf elend lange überflüssige Diskussionen, wie du sie gerne führst. Es ist nunmal so dass nach ändern der Sprache dies sie nicht auf die aktuelle Shell auswirkt. Das exit rauswerfen ist destruktiv, bessere Lösungen mag es geben. Das ist jedenfalls eine einfache

pfichtner, 2022-12-27:
Darum geht es hier doch überhaupt nicht, das ist ja auch der Grund, warum ich diesen Issue aus der Diskussion erzeugt habe. 
In diesem Issue geht es darum, dass unter Fedora36  `python` und `rpcgen` fehlen (nach der Abarbeitung der PREREQUISITES.md). `python` ist kein optionales Programm, d.h. dass ohne dies keine Firmware-Images gebaut werden können. 

Deshalb war die Reaktion auf das Melden dieses Problems für mich auch nicht nachvollziehbar. 

Das betrifft sowohl Fedora 36, als auch die noch nicht beschriebene Version 37.

fda77, 2022-12-27:
Gut möglich, Fedora hat auf python3 umgestellt, und das läuft noch nicht 100%, zb war bis vor kurzem nur ein trac dort verfügbar das python2 voraussetzte. Daher hatte ich das wahrscheinlich vorinstalliert

pfichtner, 2022-12-27:
Ja, alles in #681 beschrieben, bezüglich python2/3 siehe https://github.com/Freetz-NG/freetz-ng/issues/681#issuecomment-1362640102

pfichtner, 2023-01-08:
Wie geht es mit dem Issue weiter? Zur Zeit kann nach Befolgen der Fedora 36 Installationsanleitung nichts gemacht werden, da `python` nicht gefunden wird. 
Während `python` und `rpcgen` noch trivial korrigierbar ([siehe Kommentar](https://github.com/Freetz-NG/freetz-ng/issues/683#issuecomment-1362084703)) sind, gilt dies nicht für `fastjar`. Das ist zwar (noch) nicht Inhalt dieses Issues, ich würde aber gerne alle gefundenen/offenen Punkte gleich miterledigen. Für `fastjar` gäbe es ja die Möglichkeit es als host-tools bereitzustellen (davon bin ich wieder abgerückt). Die Entscheidung wäre allerdings (vermutlich) eine Diskussion.

fda77, 2023-01-08:
- fastjar war nur für jam/classpath? Wenn niemandem 10 Jahre lang auffiel dass da was fehlt würde ich das einfach lassen. Wahrscheinlich gibt es ohne Update eh einige Sicherheitsproblem. Oder was hast du damit vor?
 - python. wofür braucht man das nochmal? Bei Fedora gibts jedenfalls nur noch v3 (was mit anderen Dinge kleinere Probleme machte). Ist da alles kompatibel?
 - rpcgen nfs und autofs nutze ich, das war das evtl schon da

pfichtner, 2023-01-08:
> * fastjar war nur für jam/classpath? Wenn niemandem 10 Jahre lang auffiel dass da was fehlt würde ich das einfach lassen. Wahrscheinlich gibt es ohne Update eh einige Sicherheitsproblem. Oder was hast du damit vor?

Ja, wobei es ja unter Ubuntu da ist (und bei jedem der es irgendwann mal auf seinem heute noch laufendem System installiert hat). Ich habe selbst nichts damit vor, ich habe nur meinen Ideen umgesetzt, dabei kam dann eben auch die automatische Validierung der Abschnitte der unterschiedlichen Distributionsversionen der [PREREQUISITES.md](https://github.com/Freetz-NG/freetz-ng/blob/master/docs/PREREQUISITES.md) raus, wobei sich eben herausgestellt hatte, dass die hier beschriebenen Pakete bei Fedora in der Anleitung fehlen. 
Mögliche Lösungen [habe ich hier mal zsuammengeschrieben](https://github.com/pfichtner/freetz-ng/discussions/2#discussioncomment-4625199). 

>     * python. wofür braucht man das nochmal? Bei Fedora gibts jedenfalls nur noch v3 (was mit anderen Dinge kleinere Probleme machte). Ist da alles kompatibel?

`python` als Kommando wurde mittlerweile verbannt, man muss als Interpreter entweder `python2` oder `python3` verwenden. Damit `python` wieder als Kommando verfügbar ist gibt es die beschriebenen Pakete (setzen unter ubuntu und fedora einen symlink auf python3) . 
Kompatibel: Für Ubuntu steht ja in den [PREREQUISITES.md](https://github.com/Freetz-NG/freetz-ng/blob/master/docs/PREREQUISITES.md)  auch `python-is-python3`, wenn es dort kompatibel ist, ist es es für Fedora vermutlich auch und umgekehrt. 
Möchte man wirklich abwärtskompatibel sein/bleiben, müsste der symlink vermutlich auf python2 und nicht python3 zeigen, aber das sollte eben auch auf beiden Distributionen (bzw. der entsprechenden Installationsanleitung) identisch für beide sein. 
Nicht kompatibel ist der Zustand so wie er jetzt ist: `python` wird als Befehl nicht gefunden. Damit schetert schon die Prüfung der [.prerequisites](https://github.com/Freetz-NG/freetz-ng/blob/master/.prerequisites) da 
`program     python`
dort ohne Einschränkung gefordert wird. Was es tatsächlich erfordert weiß ich nicht. Somit ist dann eben kein make Aufruf möglich, da dort ja via `tools/prerequisites` festgestellt wird, dass `python` nicht auf dem System installiert ist (bzw. es kein binary `python` gibt).  

>     * rpcgen nfs und autofs nutze ich, das war das evtl schon da

Ja, fehlt aber eben auch in den [PREREQUISITES.md](https://github.com/Freetz-NG/freetz-ng/blob/master/docs/PREREQUISITES.md)

fda77, 2023-01-09:
Wie gesagt, bei Fedora gibts an Python v3 oder nix.
Wenn ich noch wüsste wofür man das braucht... ob das evtl von hippys tool war? Oder er hatte perl verwendet. Jedenfalls gibts auch ein python27-host

pfichtner, 2023-01-09:
> Wie gesagt, bei Fedora gibts an Python v3 oder nix. Wenn ich noch wüsste wofür man das braucht... ob das evtl von hippys tool war? Oder er hatte perl verwendet. Jedenfalls gibts auch ein python27-host in freetz

Wie [in](https://github.com/Freetz-NG/freetz-ng/issues/683#issuecomment-1362084703) vorgeschlagen, sollte dafür `python-unversioned-command` noch installiert werden, das setzt den genannten symlink. So macht es Ubuntu mit deim in den [PREREQUISITES.md](https://github.com/Freetz-NG/freetz-ng/blob/master/docs/PREREQUISITES.md) aufgeführten  `python-is-python3` auch. Unter Ubuntu baut damit bei mir auch Freetz-NG Images für sechs verschiedene FritzBox-Modelle. Ob Python da aber aufgerufen wird weiß ich natürlich nicht, d.h. das ist kein Nachweis. 
Ich forsche die Woche mal, warum und wofür Freetz-NG python braucht.

fda77, 2023-01-09:
Mindestens das braucht jedenfalls python2: https://github.com/Freetz-NG/freetz-ng/blob/master/make/host-tools/mklibs-host/patches/016-use_python2.patch
Wahrscheinlich sollte mal jemand das package aktualisieren...
mklibs ist jedenfalls optional und braucht man bei Geräten mit mehr als 8mb flash nicht mehr so oft

pfichtner, 2023-01-09:
Das braucht (nach dem Patchen) sogar explizit `python2` als command. Das ist ja auch vorhanden, nur eben in den aktuellen Distros kein symlink mehr von `python` auf `pythonX`. 
Dann wäre (Stand jetzt und ohne Aktualisierung des genannten Pakets) eigentlich richtiger, wenn in den [.prerequisites](https://github.com/Freetz-NG/freetz-ng/blob/master/.prerequisites) statt `python` auf `python2` aufgeführt wäre. 
Das ist der Grund, warum die Distros ([zumindest ist das die Begründung bei ubuntu](https://askubuntu.com/questions/1296790/python-is-python3-package-in-ubuntu-20-04-what-is-it-and-what-does-it-actually)) den Link nicht mehr setzen: "Skripte" sollen explizit angeben, welchen Interpreter/Version sie benötigen, also eben z.B.  `python2` oder `python3` und nicht allgemein `python`

fda77, 2023-01-09:
Wenn du es aktualisierst braucht es python3 ... und dann hast du wieder die Experten mit uraltem Krempel  am Hals die nur 2 haben ... https://freetz-ng.github.io/freetz-ng/NEWS.html#2022-08-24
Davon abgesehen braucht das in freetz immer p2, mit p3 geht nämlich schlicht nicht

pfichtner, 2023-01-09:
> Wenn du es aktualisierst braucht es python3 ... und dann hast du wieder die Experten mit uraltem Krempel am Hals die nur 2 haben ... 

Wenn Du was updatest, was statt python2 dann python3 braucht, wie willst Du aus der Nummer rauskommen? 
Ich glaube Du verstehst mich falsch: Ich will nix updaten! Fakt ist derzeit: 
- Mit der aktuellen PREREQUISITES.md  für Ubuntu zeigt "python" auf "python3" (`python-is-python3`), d.h. Skripte python als Interreter aufrufen würden landen in nem python3. Macht nichts, solange Skripte `python2` als Interpreter nutzen (in Deinem Beispiel der Fall). Das Paket ist eigentlich unnötig, ist vermutlich nur dafür da, damit `tools/prerequisites` nicht abbricht, denn da wird aber auf `python` geprüft, welches gar nicht benötigt wird. 
- Mit der aktuellen PREREQUISITES.md  für Fedora gibt es kein "python"), d.h. Skripte python als Interreter aufrufen würden funktioneren nicht. Macht nichts, solange Skripte `python2` als Interpreter nutzen (in Deinem Beispiel der Fall).  `tools/prerequisites` bricht ab, da auf dem System kein "python" vorhanden ist (und auch nicht gebraucht wird). 

> Davon abgesehen braucht das in freetz immer p2, mit p3 geht nämlich schlicht nicht

Genau das ist ja der Grund, warum der Interpreter python nicht mehr da ist: Das Skript muss "sagen", ob es ein python2 oder python3 Skript ist. Und wenn es aktualisert wird und python3 braucht, dann muss man in die PREREQUISITES.md neben python2 auch python3 aufnehmen/installiert haben. 

Das unter Ubuntu genutzte Paket sorgt nur dafür, dass `tools/prerequisites` nicht abbricht, weil damit dann ein (falsches) `python` (nämlich `python3`) vorhanden ist. 
Die von mir vorgeschlagene Lösung für Fedora ist also falsch: Die Installation von `python-unversioned-command` würde zum gleichen falschen Zustand führen. Deswegen weiterhin mein Vorschlag: Wenn die Skripte python2 brauchen (und das auch als Interpreter brauchen), dann sollte in der [.prerequisites](https://github.com/Freetz-NG/freetz-ng/blob/master/.prerequisites) auch `python2` (und nicht `python`) stehen. 
Sollte es zum Zustand kommen, in welcher Freetz-NG Skripte python3 brauchen, ist das entsprechend zu ergänzen oder zu ersetzen (je nachdem).

fda77, 2023-01-09:
Fedora hat KEIN python2, damit ist dein Ziel mit den prerequisites nicht erreichbar!
Daher musst du mklibs aktualisieren, ganz einfach. Bei Ubuntu fliegt das auch irgend wann mal raus.
Dass mklibs nur für uralten Kram genutzt wird, den heut niemand mehr nutzt - geschenkt :D

pfichtner, 2023-01-09:
> Fedora hat KEIN python2

Ok, das war/ist mir neu, dann ist mir jetzt klar, was/warum Du das geschrieben hattest. 
Und ja, wenn die Skripte mit python3 nicht laufen ist dann logischerweise die Konsequenz, dass man aus der Nummer nur rauskommt alles updazudaten was python2 braucht. 

Dann stelle ich mir allerdings die Frage, was `python-is-python3` in den Ubuntu Paketen zu suchen hat? :confused:

fda77, 2023-01-09:
Fedora ist nicht Ubuntu
Ubuntu ist nicht Fedora
Fedora ist nicht Ubuntu
Ubuntu ist nicht Fedora
Fedora ist nicht Ubuntu
Ubuntu ist nicht Fedora
Fedora ist nicht Ubuntu
Ubuntu ist nicht Fedora
Fedora ist nicht Ubuntu
Ubuntu ist nicht Fedora
Fedora ist nicht Ubuntu
Ubuntu ist nicht Fedora
Fedora ist nicht Ubuntu
Ubuntu ist nicht Fedora
Fedora ist nicht Ubuntu
Ubuntu ist nicht Fedora
Fedora ist nicht Ubuntu
Ubuntu ist nicht Fedora
Fedora ist nicht Ubuntu
Ubuntu ist nicht Fedora
Fedora ist nicht Ubuntu
Ubuntu ist nicht Fedora
Fedora ist nicht Ubuntu
Ubuntu ist nicht Fedora
Fedora ist nicht Ubuntu
Ubuntu ist nicht Fedora
Fedora ist nicht Ubuntu
Ubuntu ist nicht Fedora
Fedora ist nicht Ubuntu
Ubuntu ist nicht Fedora
Fedora ist nicht Ubuntu
Ubuntu ist nicht Fedora
Fedora ist nicht Ubuntu
Ubuntu ist nicht Fedora
Fedora ist nicht Ubuntu
Ubuntu ist nicht Fedora
Fedora ist nicht Ubuntu
Ubuntu ist nicht Fedora
Fedora ist nicht Ubuntu
Ubuntu ist nicht Fedora
Fedora ist nicht Ubuntu
Ubuntu ist nicht Fedora
Fedora ist nicht Ubuntu
Ubuntu ist nicht Fedora
Fedora ist nicht Ubuntu
Ubuntu ist nicht Fedora
Fedora ist nicht Ubuntu
Ubuntu ist nicht Fedora
Fedora ist nicht Ubuntu
Ubuntu ist nicht Fedora
Fedora ist nicht Ubuntu
Ubuntu ist nicht Fedora
Fedora ist nicht Ubuntu
Ubuntu ist nicht Fedora
Fedora ist nicht Ubuntu
Ubuntu ist nicht Fedora
v

pfichtner, 2023-01-09:
Hat auch niemand behauptet. Ich habe ja nur festgestellt, dass mit einem aktuellen Fedora die Pakete die python erfordern nicht baubar sind und versuche dafür Lösungen zu finden. 

Erklärt auch nicht, wofür `python-is-python3` unter Ubuntu installiert wird. 

git blame ist "schwierig", wenn es sich um eine Zeile handelt :smirk: 
```

❯ git diff c14608bd26e15bd8c69dfa1cc34d34ed0c02266d..5d8d1b278ce04ef889ce8df06ab2a0df78335d95 | grep python3
+sudo apt-get -y install pv cpio rsync kmod execstack imagemagick inkscape graphicsmagick subversion git bc unar wget sudo gcc g++ binutils autoconf automake autopoint libtool-bin make bzip2 libncurses5-dev libreadline-dev zlib1g-dev flex bison patch texinfo tofrodos gettext pkg-config ecj fastjar          perl libstring-crc32-perl ruby         gawk python2 python-is-python3 libusb-dev unzip intltool libacl1-dev libcap-dev libc6-dev-i386 lib32ncurses5-dev gcc-multilib bsdmainutils lib32stdc++6 libglib2.0-dev ccache cmake lib32z1-dev libsqlite3-dev sqlite3 netcat curl libzstd-dev
```

NEIN! Kein "blame"! Nur zur zeitlichen Einsortierung (Kommentar lässt keinen Aufschluss drüber warum)

fda77, 2023-01-09:
davon ist auszugehen

Wie kann man das eigentlich übersehen??

> Bei Fedora gibts jedenfalls nur noch v3
 
> Wie gesagt, bei Fedora gibts an Python v3 oder nix.

> Fedora hat KEIN python2

pfichtner, 2023-01-09:
> Wie kann man das eigentlich übersehen??

Hab's doch schon beim dritten Mal gesehen! :smirk:

pfichtner, 2023-01-09:
Für mich ergibt sich folgendes Bild: 
- Freetz-NG braucht python2 (nicht python, nicht python3 als Kommando)
- `tools/prerequisites` prüft auf das Vorhandensein von `python`
"Problem" unter Fedora (ohne größere Eingriffe) nicht lösbar
- Unter Ubuntu wird python3 installiert, wird aber von Freetz-NG gar icht benötigt. Sorgt aber dafür, dass python als Kommando exisitiert (das aber niemand braucht, da das Freetz-NG `python2` und nicht `python` braucht)

Das führt bei mir immer noch zu folgendem Lösungsvorschlag: 
 `tools/prerequisites` bzw. [.prerequisites](https://github.com/Freetz-NG/freetz-ng/blob/master/.prerequisites) sollte `python2` statt `python` aufführen
- wird unter Fedora/Distros ohe python2 zu keinen Veränderungen führen, ob python nicht da ist und/oder python2 nicht ist egal. Durch die Prüfung durchzukommen weil `python` auf `python3` zeigt bringt ja nix, wenn das Skript dann versucht `python2` aufzurufen was nicht da ist
- man kommt unter Ubuntu nur durch die Prüfung durch, wenn auch `python2` da ist (und eben nicht wenn `python` und `python3` da sind die aufeinander zeigen aber gar kein python2 Interpreter installiert ist, der dann vom Skript versucht wird aufzurufen. 

Nichts desto trotz sollte die Abhängigkeit zu python2 aufgelöst werden, ich gucke die Woche mal.

fda77, 2023-01-09:
Also ungefähr so wie vor 4 Stunden gemach?!?

fda77, 2023-01-09:
> Unter Ubuntu wird python3 installiert, wird aber von Freetz-NG gar icht benötigt.

ja?

pfichtner, 2023-01-09:
ja

fda77, 2023-01-09:
"ja vermutung" ist für mich "keine ahnung". 
Schau doch einfach mal nach

pfichtner, 2023-01-09:
> "ja vermutung" ist für mich "keine ahnung". 

Ja, für mich auch :smile: 

> Schau doch einfach mal nach

```
❯ grep -lR python3 2>/dev/null
.github/docker/firmware/Dockerfile
.github/workflows/make_freetz.yml
docs/PREREQUISITES.md
```

Wenn Du das mit "nachsehen" meintest würde ich sagen, sieht gut aus.

pfichtner, 2023-01-09:
python generell: 

```
❯ grep -lR python 2>/dev/null | grep -v '.git' | grep -v '.patch$' | grep -v '^docs/' | grep -v '^.github'
tools/genin
make/host-tools/python-host/python-host.mk
make/host-tools/Makefile.in
make/host-tools/scons-host/scons-host.mk
make/libs/libdnet/libdnet.mk
make/libs/iksemel/iksemel.mk
make/libs/libxml2/libxml2.mk
make/libs/external.in
make/libs/Config.in
make/pkgs/pyload/Config.in
make/pkgs/python-yenc/python-yenc.mk
make/pkgs/python-yenc/Config.in
make/pkgs/owfs/owfs.mk
make/pkgs/bind/bind.mk
make/pkgs/python-pyopenssl/python-pyopenssl.mk
make/pkgs/python-pyopenssl/Config.in
make/pkgs/python-bjoern/python-bjoern.mk
make/pkgs/python-bjoern/Config.in
make/pkgs/python-pycurl/python-pycurl.mk
make/pkgs/python-pycurl/Config.in
make/pkgs/obexftp/obexftp.mk
make/pkgs/gdb/gdb.mk
make/pkgs/cifsmount/cifsmount.mk
make/pkgs/README.md
make/pkgs/python-cheetah/python-cheetah.mk
make/pkgs/python-cheetah/Config.in
make/pkgs/avahi/avahi.mk
make/pkgs/xsltproc/xsltproc.mk
make/pkgs/cryptsetup/cryptsetup.mk
make/pkgs/asterisk/generate-menuconfig.py
make/pkgs/python-imaging-library/python-imaging-library.mk
make/pkgs/python-imaging-library/Config.in
make/pkgs/python-pyrrd/python-pyrrd.mk
make/pkgs/hplip/hplip-config-update.py
make/pkgs/unbound/unbound.mk
make/pkgs/python/python-module-macros.mk.in
make/pkgs/python/python.mk
make/pkgs/python/files/root/usr/bin/python2.7
make/pkgs/python/python-filelists.mk.in
make/pkgs/python/external.files
make/pkgs/python/external.in.libs
make/pkgs/python/external.in
make/pkgs/python/Config.in.libs
make/pkgs/python/Config.in
make/pkgs/collectd/collectd.mk
make/pkgs/python-mechanize/python-mechanize.mk
make/pkgs/python-mechanize/Config.in
make/pkgs/python-pyserial/python-pyserial.mk
make/pkgs/rrdtool/rrdtool.mk
make/pkgs/util-linux/util-linux.mk
make/pkgs/python-pycrypto/python-pycrypto.mk
make/pkgs/python-pycrypto/Config.in
.prerequisites
Makefile
fwmod
```

(fwmod Treffer ist nur Kommentar)

fda77, 2023-01-09:
Muss man halt mal alles durchsuchen. Und avm's kernel, die haben da auch gerne irgend welche zusätze (zb execstack) drin

pfichtner, 2023-01-09:
Habe mir mal das mklibs Thema angeschaut, was wäre denn da die vermeintlich bessere Wahl? 
- auf die aktuelle Version zu gehen? Die hätte dann python3 als Abängigkeit oder
- eine alternative Implementierung zu suchen/zu erstellen die ganz ohne python auskommt

Selbst wenn mklibs das einzige Paket in Freetz-NG ist/wäre welches python benötigt und dieses Paket kaum gebraucht wird, so tut die Abhängigkeit zu python nicht so sehr weh, dass man sich den Schuh mit einer python-losen-mklibs Version antut, oder?

fda77, 2023-01-10:
Gefällt dir nicht was ich oben dazu schrieb und hättest nun gern ne andere Antwort, oder warum fragst du nochmal?

pfichtner, 2023-01-10:
> Gefällt dir nicht was ich oben dazu schrieb und hättest nun gern ne andere Antwort, oder warum fragst du nochmal?

Wenn Du den Satz " Wahrscheinlich sollte mal jemand das package aktualisieren..." meinst hatte ich den nicht als alternativlos interpretiert, das nur eine Aktualisierung die Lösung ist. 

Ich habe jetzt mal patches um von mklibs_0.1.34 auf mklibs_0.1.40, mklibs_0.1.43, mklibs_0.1.44 und mklibs_0.1.45 zu kommen, so dass zumindest mal die tarballs gezogen, entpackt und gepatcht werden können, mehr nicht. Ist nicht viel, aber auch nicht nichts. 

Um weiter zukommen fehlen mir Makefile- und Freetz-NG Build-Kenntnisse. 
Aktuell stehe ich hier: 

```
builduser@c83a353c3cd8:~$ make mklibs-host-recompile
[...]
   applying patch file /workspace/make/host-tools/mklibs-host/patches/040-deprecated_throw.freetz.patch
    patching file src/mklibs-readelf/elf.cpp
    patching file src/mklibs-readelf/elf_data.hpp
    patching file src/mklibs-readelf/elf.hpp
    ----------------------------------------------------------------------
mkdir -p /workspace/tools/build/bin/; cp /workspace/source/host-tools/mklibs-0.1.45/src/mklibs /workspace/tools/build/bin/mklibs;
configuring ... (conf_cmd() { ./configure  "$@"  || { [ "2" = "0" ] && echo && cat .build.log 2>/dev/null; kill $ 2>/dev/null || kill $$ 2>/dev/null; printf "\n\\033[33m%s\\033[m\n" "ERROR: ";  exit 1; } }; cd /workspace/source/host-tools/mklibs-0.1.45; { rm -f config.{cache,status}; } ;   PATH="/workspace/tools/path:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin" CC="gcc" CXX="g++" CFLAGS="-Wall -O2" CXXFLAGS="-Wall -O2" LDFLAGS=""  CONFIG_SITE="" conf_cmd   --prefix=/   )
environment: line 1: ./configure: No such file or directory

ERROR: 
make[1]: *** [make/host-tools/mklibs-host/mklibs-host.mk:18: /workspace/source/host-tools/mklibs-0.1.45/.configured] Terminated
make: *** [Makefile:46: envira] Terminated
```

Nach mklibs_0.1.34 gibt es kein `configure` mehr, alles was ich jetzt machen würde wäre trial and error und deshalb nicht sinnvoll. 

Von daher habe ich [den aktuellen Arbeitsstand von mklibs-0.1.45 mal hier abgelegt](https://github.com/pfichtner/freetz-ng/tree/mklibs-1.45) in der Hoffnung/mit dem Wunsch, dass sich jeder der dazu beitragen kann einbringt um das Update hinzukriegen (ob es am Ende fruchtet kann ich natürlich nicht zusichern, selbst wenn es am Ende baut bedeuetet das ja nicht, dass es auch nutzbar ist). 

Den Kommentar zur Diskussion machen?

pfichtner, 2023-01-10:
Da war viel Handarbeit nötig, sieht aber nicht gut aus der Patch, da werde ich nochmal nacharbeiten (müssen). 
https://github.com/pfichtner/freetz-ng/blob/1c8ad25ef71429044de74bbb909fbfed5c4bf836/make/host-tools/mklibs-host/patches/040-deprecated_throw.freetz.patch

Ich habe auch blindlings versucht einfach alle Patches anwendbar zu kriegen, ohne zu wissen und/oder einschätzen zu können ob sie überhaupt nicht notwendig sind.

fda77, 2023-01-10:
Direkt unter 
> Fedora hat KEIN python2, damit ist dein Ziel mit den prerequisites nicht erreichbar!

steht
> Daher musst du mklibs aktualisieren, ganz einfach. Bei Ubuntu fliegt das auch irgend wann mal raus.

 ;-)

Ich glaub da hatte nur jemand sowas wie https://docs.python.org/3/library/2to3.html drüberlaufen lassen

pfichtner, 2023-01-10:
> steht
> 
> > Daher musst du mklibs aktualisieren, ganz einfach. Bei Ubuntu fliegt das auch irgend wann mal raus.

> ;-)

Ja, meine Frage war (habe ich bei meiner Suche gefunden) ob statt dem Original-mklibs auch ein anderes Paket was identisches macht, eine denkbare Alternative wäre um gegebenenfalls ganz von python auszukommen (was wahrscheinlich eh nicht oder nicht dauerhaft erreichbar ist). Von daher vermutlich den Aufwand nicht wert und vermutlich auch nicht sinnvoll. Wollte es trotzdem zur Diskussion stellen. 

> Ich glaub da hatte nur jemand sowas wie https://docs.python.org/3/library/2to3.html drüberlaufen lassen

Möglich, habe es mir nicht angeschaut. Wäre natürlich auch noch ne Option: Die bestehende mklibs-1.0.34 lediglich die python2 Skripte von 2 auf 3 patchen. Aber ich vermute es wäre generell mal gut das doch recht alte Paket mal zu aktualisieren.

fda77, 2023-01-10:
Irgend wie kapierst du nie was ich meine...
HIER sind doch patches für NEUE 45 verlinkt: https://github.com/Freetz-NG/freetz-ng/blob/master/make/host-tools/mklibs-host/mklibs-host.mk#L8
Und wenn man da drauchfklickert kommt man irgend wann zu https://github.com/openwrt/openwrt/blob/master/tools/mklibs/patches/100-apply-2to3.patch
Aber selbst damit ist das wohl noch nicht 100% python3 kompatibel und man darf an dem mist noch herumfixen. Was hast du dir daran eigentlich angeschaut?

fda77, 2023-01-10:
Falls du es oben überlesen haben solltest: Ich bin der Meinung dass man diesen Uralten Mist den seit 10 Jahren niemand vermisst hat einfach so lassen sollte wie er ist! Wenn man Langeweile hat gibts genügend aktuelle Dinge zu tun

pfichtner, 2023-01-10:
> Irgend wie kapierst du nie was ich meine... 

Sorry :disappointed:

> HIER sind doch patches für NEUE 45 verlinkt: https://github.com/Freetz-NG/freetz-ng/blob/master/make/host-tools/mklibs-host/mklibs-host.mk#L8 Und wenn man da drauchfklickert kommt man irgend wann zu [...]

https://github.com/Freetz-NG/freetz-ng/tree/master/make/host-tools/mklibs-host/patches beinhaltet ja neben den openwrt-Patches noch weitere Patches von Freetz, gerade der eine Freetz-Patch war für die letzte mklib Version massiv anzupassen (damit er anwendbar ist). 

Du hast Recht, es ist sehr fraglich daran rumzudoktoren, wenn es eh kaum gebraucht wird. 
Ich bin da ja nur gelandet, da ich Fedora-basierte Images gebaut habe, bei denen man `make` bedingt durch die Prüfung auf `python` abbricht. Das sollte meiner Meinung nach korrigiert werden, da ansonsten niemand mit einem neu aufgesetzten Fedora-System anhand der [PREREQUISITES.md](https://github.com/Freetz-NG/freetz-ng/blob/master/docs/PREREQUISITES.md) Images bauen kann. Meine verfolgten Lösungsansätze wie z.B. die Aktualisierung von `mklibs` sind da wohl wirklich der falsche/kein sinnvoller Weg. 

Was ist Dein Vorschlag was geändert werden sollte, damit auf Distributionen ohne `python` `make` aufrufbar/durchführbar ist? [Zeile 36 in den .prerequisites](https://github.com/Freetz-NG/freetz-ng/blob/a7696a688d276947d775ce78b86ae13c65526003/.prerequisites#L36) verhindert dies ja aktuell. Dann doch `python-unversioned-command` in die [PREREQUISITES.md](https://github.com/Freetz-NG/freetz-ng/blob/master/docs/PREREQUISITES.md) aufnehmen?

fda77, 2023-01-10:
Ich weiss nicht auf welcher Seite das Problem ist.

Wie (oben schon) gesagt, man müsste zuerst mal herausfinden was alles in freetz python braucht. Wenn es nur mklibs wäre könnte man es ja rausnehmen. Wenn niemand guggt, weiss es niemand

fda77, 2023-01-10:
Teste mal ob das läuft: [mklibs.diff.txt](https://github.com/Freetz-NG/freetz-ng/files/10387535/mklibs.diff.txt)
Falls nicht, ob es im gleichen Fall noch vorher mit p2 ging

pfichtner, 2023-01-10:
> Wie (oben schon) gesagt, man müsste zuerst mal herausfinden was alles in freetz python braucht. Auch schon oben: auf die avm-kernel achten. Wenn es nur mklibs wäre könnte man es ja rausnehmen. Solange niemand guggt, weiss es niemand

Wenn man `python-unversioned-command` aufnehmen würde und das identisch zu ubuntus `python-is-python3` (wovon ich ausgehe ud was trivial zu validieren ist) ist welches lediglich einen symlink python anlegt der auf python3 zeigt, sollte damit ja eigentlich damit nicht etwas nicht mehr gehen, was vorher ging. 
Wäre, wenn ich mich nirgends täusche, der kleinstmögliche Zugriff mit keinem Risiko bzw. geringstem Risiko und würde schon mal das Problem lösen, dass  `make` wieder ausgeführt werden kann.

pfichtner, 2023-01-10:
> Teste mal ob das läuft: [mklibs.diff.txt](https://github.com/Freetz-NG/freetz-ng/files/10387535/mklibs.diff.txt) Falls nicht, ob es im gleichen Fall noch vorher mit p2 ging

Danke, schaue ich mir morgen mal an.

fda77, 2023-01-10:
Wenn ein script für p2 ist läuft es nur damit, das gilt für p3 genau so 
Und da ich kein p2 mehr hab mach ich mir darum einfach keine Gedanken

pfichtner, 2023-01-10:
Ich weiß, und so ist es ja auch richtig, das hilft aber ja alles nichts beim angesprochenen Problem. Ja, ich habe mich auch verrannt, bitte nochmal gedanklich zum Ursprungsproblem zurück: Jemand installiert alle beschrieben Abhängigkeiten auf Fedora und hat damit keine Chance Freetz-NG zu nutzen. 
Für mich ist jede Lösung "ok", von es gibt eine technische Lösung, in den PREREQUISITES.md steht ein Hinweis das es nicht geht bis hin zu: Ok, haben User die sich ein frisches Fedora aufsetzen Pech gehabt, geht halt doch nicht.

pfichtner, 2023-01-11:
> Und da ich kein p2 mehr hab mach ich mir darum einfach keine Gedanken

Wie kann dann auf Deinem System  `tools/prerequisites` erfolgreich durchlaufen werden, wenn es dort kein `python` Kommando gibt? :confused:

fda77, 2023-01-11:
commented [2 weeks ago](https://github.com/Freetz-NG/freetz-ng/issues/683#issuecomment-1366036071)
> Gut möglich, Fedora hat auf python3 umgestellt, und das läuft noch nicht 100%, zb war bis vor kurzem nur ein trac dort verfügbar das python2 voraussetzte. Daher hatte ich das wahrscheinlich vorinstalliert

fda77, 2023-01-11:
Statt hier wochelang immer das gleiche zu fragen, such doch einfach mal heraus wo in freetz überall python genutzt wird.
Wie lang sollen wir das noch durchkauen?

pfichtner, 2023-01-11:
Deswege war ich ja verwirrt, zuerst schriebst Du, DU hast python2 noch weil Du es mal installiert hattest, dann widerrum, dass Du froh bist kein python2 mehr zu haben. 

> Statt hier wochelang immer das gleiche zu fragen, such doch einfach mal heraus wo in freetz überall python genutzt wird. Wie lang sollen wir das noch durchkauen? 

Deine Hoffnung ist vermutlich, dass mklibs mit dem Patch kein pthon2 mehr braucht  und man den Eintrag "python" gänzlich aus ".prerequisites" entfernen kann. 
Ich werde die Woche mir den Patch anschauen und mal versuchen meine sechs Image-Typen ohne Vorhandensein von python/python2 zu bauen.

fda77, 2023-01-11:
Hä??? Keine Ahnung

pfichtner, 2023-01-11:
Dann reden wir (vielleicht seit Tagen) aneinander vorbei, ich versuche es nochmal punktuell aufzuführen (wenn es eh alles klar ist: Ignorieren, wenn ich micht irgendwo täusche, bitte richtig stellen)

- make liest `Makefile`
- Makefile ruft `tools/prerequisites` auf
- `tools/prerequisites` liest `.prerequisites`
- `.prerequisites` beinhaltet "program     python" (kein "depends", also mandatory='y')
- tools/prerequisites ruft `[ -x "$(command -v python)" ]` auf, schlägt fehlt wenn kein `python` im `PATH`

Ich fang jetzt an Deinen Patch zu testen, an der Konstellation/dem Problem/diesem Issues ändern wird er aber nichts: 
Solange es den genannten Eintrag in der `.prerequisites` und kein `python` im `PATH` gibt ist kein `make` möglich (unabhängig davon was man bauen will/was in der `.config` steht: Immer).

fda77, 2023-01-11:
such doch einfach mal heraus wo in freetz überall python genutzt wird.

pfichtner, 2023-01-11:
In der Hoffnung beim filtern nicht zu viel gefiltert zu haben: 
```
make/pkgs/asterisk/generate-menuconfig.py:#!/usr/bin/env python2
make/pkgs/hplip/hplip-config-update.py:#!/usr/bin/env python
```

Dann gibt es noch `python-host`, davon hängt `scons-host` ab. 

Generell natürlich gefährlich, würde `mklibs-host` nicht "/usr/bin/env python" zu "/usr/bin/env python2" per Patch ändern, würde es beim Durchsuchen des Repos durch die Lappen gehen. 
Also noch die aktuellen host-tools runtergeladen, neben (logischerweise) zahlreicher Treffer in `tools/build/usr/lib/python2.7` wird an folgenden Stellen noch python genutzt/referenziert: 
```
tools/build/bin/mklibs:#! /usr/bin/env python2
tools/build/usr/bin/pydoc:#!/usr/bin/python2.7
tools/build/usr/bin/python2.7-config:#!/usr/bin/python2.7
tools/build/usr/bin/smtpd.py:#!/usr/bin/python2.7
tools/build/usr/bin/2to3:#!/usr/bin/python2.7
tools/build/usr/bin/idle:#!/usr/bin/python2.7
tools/build/usr/bin/python2-config:#!/usr/bin/python2.7
tools/build/usr/bin/python-config:#!/usr/bin/python2.7
```

fda77, 2023-01-11:
Aha... die Scripte des python2-host brauchen ein p2. Das kann man (in freetz) einfach in den Suchpfad einbinden durch links in tools/path/. Die "#!/usr/bin/python2.7" müsste man aber noch durch "env" aufrufe ersetzen

mklibs könnte das p2-host dann auch nutzen.. falls der bump-patch nicht geht

Von asterisk+hplip sind das nur developer-script zum einmaligen generieren. Die Versionen sind wahrscheinlich so alt das nach einem bump die .py eh nicht mehr passen würden. - kann man eigentlich ignorieren

Die make/pkgs/python* Dinger brauchen das python-host auch

fda77, 2023-01-12:
So, es gibt jetzt python 2+3 und fastjar als hosttools. Sonst gibt das nur Probleme mit neuen/alten Linux Versionen die nur die eine oder andere version haben. Es gibt auch in vielen Packages und kernel python scripte, da hab ich nicht lust alle zu testen und zu schauen ob die überhaupt gentzt werden.
Ein Nachteil ist dass die dl-hostst nun von 15 auf 35 mb angewachsen sind

Das mklibs/strip libs scheint aber vor als auch nach dem bump nicht ganz sauber zu laufen, evtl kam das von den "sepearate uclib" änderungen. Müsste mal jemand mit viel Zeit fixen, hat aber bislang niemand vermisst
Ist eh komisch, die neueste version von debian sagt dass es python3 kompatibel ist, aber ohne den 2to3 patch von openwrt geht das nicht. Und dann gibt es noch 3 weitete Inkompatibilitäten die ich selbst behenem musste

javac passt wohl noch nicht so ganz, auf meinem Fedora zumindest. Gibts da was kleines was man als Hosttool aufnehmen könnte?

pfichtner, 2023-01-12:
Mega! :clap: 
Ich habe es gestern nur mit viel Gewalt hinbekommend, dass mklibs überhaupt aufgerufen/genutzt wird und habe den Abgleich zwischen der Ergebnis des Aufrufs der alten und der neuen mklibs Version noch nicht fertig (soll ich das jetzt überhaupt noch weiter verfolgen?). 

Bezüglich javac, damit funktioniert dann auch ecj unter fedora und ist als javac ausführbar
```
dnf -y install javapackages-tools
update-alternatives --install /usr/bin/javac javac /usr/bin/ecj 20
```

[Siehe](https://github.com/pfichtner/pfichtner-freetz/blob/main/provisioning/fedora:36/51-fix-freetz-ng-prerequisites.sh), da ist dann auch noch `rpcgen` (und `python-unversioned-command` was als Quick-Fix diente um durch die Prüfung durchzukommen, diente und jetzt dann rausfliegt). 

Bei Ubuntu ist das Pendant zu fedoras `javapackages-tools` entsprechend `java-wrappers` und müsste auch noch in die [PREREQUISITES.md](https://github.com/Freetz-NG/freetz-ng/blob/master/docs/PREREQUISITES.md) hinzugefügt werden (getestet unter [u20](https://github.com/pfichtner/pfichtner-freetz/blob/main/provisioning/ubuntu:20.04/51-fix-freetz-ng-prerequisites.sh)/[u22](https://github.com/pfichtner/pfichtner-freetz/blob/main/provisioning/ubuntu:22.04/51-fix-freetz-ng-prerequisites.sh))

fda77, 2023-01-12:
Du hast doch so viele Docker container... Da würde mich interessieren ob dies p3 mklibs fehlschlägt wo es früher noch ging. Oder ob es jemals ging. Bzw wann es kaputt ging, falls es kaputt ging... Ich hoffe du versteht wie ich das meine
Die Option heisst FREETZ_STRIP_LIBRARIES und ist nur sichbar bei nicht-separater uclib, also fos <7.2 (oder max 7390, die alten bauen eh schneller)

javapackages-tools hat ja nur 37 k. Schafft man sich mit dem "update-alternatives" nicht neue Problem? Blöd ist jedenfalls dass .prerequisites das nicht erkennt

Was ist mit rpcgen? Es fehlt einfach?

pfichtner, 2023-01-12:
> Du hast doch so viele Docker container... Da würde mich interessieren ob dies p3 mklibs fehlschlägt wo es früher noch ging mit p2. Oder ob es jemals ging. Bzw wann es kaputt ging, falls es kaputt ging... Ich hoffe du versteht wie ich das meine 

Ich glaube ja :smirk: Ich habe aber hier (noch) kein fehlschlagendes mklibs, zumindest sehe ich nicht, dass etwas nicht geht. Hast Du ne .config? 

> Die Option heisst FREETZ_STRIP_LIBRARIES und ist nur sichbar bei nicht-separater uclib, also fos <7.2 (oder max 7390, die alten bauen eh schneller)

Ich hatte extra ne 3030 gebaut, die Option hatte/hab ich gesetzt, trotzdem führt `make` nicht zum Aufruf von `mklibs`, ein Aufruf direkter Aufruf von `fwmod -a` dagegen schon

> javapackages-tools hat ja nur 37 k. Schafft man sich mit dem "update-alternatives" nicht neue Problem? 
`update-alternatives` macht ja nichts anderes als nen Link (plus ein paar interne Verwaltungsinformationen) zu setzen. Man kann natürlich auch knallhart selbst nen symlink `javac` setzen der auf `ecj` zeigt. Via `update-alternatives` ist der Link aber sauber eingetragen/verwaltet. Ich habe gerade keine Idee, welche Probleme das schaffen könnte. 

> Blöd ist jedenfalls dass .prerequisites das nicht erkennt

Das war nicht erkannt wird? 
Man könnte alternativ in .prerequisites und
```
make/pkgs/classpath/classpath.mk:$(PKG)_BUILD_PREREQ += fastjar javac
make/pkgs/classpath/classpath.mk:$(PKG)_BUILD_PREREQ_HINT := Hint: on Debian-like systems fastjar is provided by the package of the same name, whereas javac (Java compiler) is provided by many packages. You could use either of them interchangeably.
make/pkgs/classpath/classpath.mk:$(PKG)_CONFIGURE_ENV += JAVAC=javac
```
javac durch ecj ersetzen. Dann bräuchte man kein `javac` (ungetestet). 

> Was ist mit rpcgen? Es fehlt einfach?

Exakt

fda77, 2023-01-12:
ecj würde auch nichts bringen, den Befehl gibt es bei mir, aber ohne update-alternatives geht classpath nicht

Die _BUILD_PREREQ waren übrigens das alte System (vor ng) als in der .prereq... noch keine Conditionen möglich waren, also alles immer da sein musste

Ich hatte einfach eine 7170 ausgewählt und "Run mklibs on the built root filesystem - EXPERIMENTAL " aktiviert. 
```
shrinking shared libs
    Converting symlink '/lib/libcapi20.so' from '/./lib/libcapi20.so.3.0.4' to 'libcapi20.so.3.0.4'
    Converting symlink '/lib/libfwsign.so.0' from '/./lib/libfwsign.so' to 'libfwsign.so'
    Converting symlink '/lib/libcapi20.so.3' from '/./lib/libcapi20.so.3.0.4' to 'libcapi20.so.3.0.4'
    Converting symlink '/lib/libled2.so.2.2' from '/./lib/libled2.so.2.2.0' to 'libled2.so.2.2.0'
    Converting symlink '/lib/libled2.so.2' from '/./lib/libled2.so.2.2.0' to 'libled2.so.2.2.0'
calling mklibs-readelf --print-needed /home/cam/freetz-ng/build/modified/filesystem/lib/libc.so.0
...
abort()ing pseudo client by server request. See https://wiki.yoctoproject.org/wiki/Pseudo_Abort for more details on this.
...
Check logfile /.fakeroot-cache/pseudo.log
...
Command failed with status 250
```

Mit Fakeroot gehts!!!1
evtl ist das wieder dieses komisch Problem was ich öfter mal hab, und weshalb man übrhaupt zwischen pseudo und fakeroot auswählen kann. An mklibs+p3 liegts dann mal nicht

pfichtner, 2023-01-12:
> ecj würde auch nichts bringen, den Befehl gibt es bei mir, aber ohne update-alternatives geht classpath nicht

Mein Vorschlag war ja, das auch in den Treffern von classpath zu ändern. Hattest Du das probiert? Das meinte ich mit "ungetestet". 

> Ich hatte einfach eine 7170 ausgewählt und "Run mklibs on the built root filesystem - EXPERIMENTAL " aktiviert.

Schaue ich mir heute Abend mal an. 

> ...
> abort()ing pseudo client by server request. See https://wiki.yoctoproject.org/wiki/Pseudo_Abort for more details on this.
> ...
> Check logfile /.fakeroot-cache/pseudo.log
> ...
> Command failed with status 250
> ```
> 
> Mit Fakeroot gehts!!!1 EDIT: Das Problem hatte ich aber auch schon mit der p2 version evtl ist das wieder dieses komisch Problem was ich öfter mal hab, und weshalb man übrhaupt zwischen pseudo und fakeroot auswählen kann. An mklibs+p3 liegts dann mal nicht

Ok, dann hat sich das o.g. erledigt? 

Und ich habe genau [das umgekehrte Problem entdeckt](https://github.com/pfichtner/pfichtner-freetz/issues/13): Ein Fall wo pseudo geht, aber fakeroot nur unter u20, nicht aber unter u22.

[15 more replies]

closed 2023-01-15 by pfichtner