Freetz-NG/freetz-ng #690: SSLH fails to start in recent builds. SSLH has a new config file structure, which is not taken into consideration by the Freetz GUI, causing bad syntax in the config file..

Spamm00r, 2023-01-02:
SSLH has been working on my last build that I made in January 2022 for my 7590 Boxes. 
In mid 2022, the SSLH Version in Freetz was updated. SSLH has had some changes in the config file, that were not taken into consideration by the Freetz GUI to adjust it to the new config file structure, thats why SSLH fails with bad syntax error.

You can start SSLH manually, by taking the config file from the new package and making the necessary config changes in the default basic config file. However making any changes through the freetz GUI does fail.

See the following discussion for details:

https://www.ip-phone-forum.de/threads/sslh-startet-nicht-mit-aktueller-freetz-ng-keine-fehlermeldung-in-syslog.314410/#post-2501027

Please fix.

Thanks!
 
- §1 Write in german or english
 - §2 Attach your .config after renaming to config.txt
 - §3 Describe steps to reproduce the behavior
 - §4 "Error 1" is not an error description
 - §5 Add logs, in most cases this are more than 10 lines
 - §6 Embed images by dragging them into this text box
 - §7 Github does not allow to attach all file types, append .txt
```
 - §8 Format code and logs by wrapping it with ``` like this line
```

pfichtner, 2023-01-04:
Github makes it so easy for you to make the changes yourself. You don't need anything except your browser. Just edit the file https://github.com/Freetz-NG/freetz-ng/blob/master/make/pkgs/sslh/files/root/etc/default.sslh/default-config.sh online, github will automatically create a fork of the project for you, since you won't have permission to actually change the file in the repository. After saving the file, open a pull request (PR) to commit the changes to the original repository which can be done in the browser as well. So nothing more needed than a browser an some clicks.

pfichtner, 2023-01-04:
PS: Of course you should change all files needed in your fork like 
https://github.com/Freetz-NG/freetz-ng/blob/9d366bef12afcd358e049ad4efba394245ea6019/make/pkgs/sslh/files/root/etc/default.sslh/sslh_conf#L6

fda77, 2023-01-04:
https://www.ip-phone-forum.de/threads/sslh-startet-nicht-mit-aktueller-freetz-ng-keine-fehlermeldung-in-syslog.314410/post-2501027

links to https://github.com/yrutschle/sslh/commit/64c3e0ed1e2a30170e2fa4d9979b332b69090765 which was changed  Sep 4, 2022

version 1.22c used in freetz was released Sep 5, 2021

> typical ippf bullshit

closed 2023-01-04 by fda77

noherbert, 2023-01-18:
Hi guys,
sorry for bugging you but i recently switched from freetz to freetz-ng and got the same issue. When i checked about the problem i found this closed ticket here but i can confirm that this _also affects version "v1.22c" which is used by default in freetz-ng_:
> root@fritz:/var/flash# sslh -F /var/mod/etc/sslh.conf -f
> Unknown settings:
> /verbose:[false]

Here you can see that the verbose-option now takes a number instead of a boolean:

> root@fritz:/var/flash# sslh --version
> sslh-fork v1.22c
> root@fritz:/var/flash# sslh --help
> sslhcfg: invalid option "--help"
>  [-Vfin] [-F <file>] [-v <n>] [--transparent] [-t <n>] [-u <str>] [-P <file>] [-C <path>] [--syslog-facility=<str>] [--on-timeout=<str>] [--prefix=<str>] [-p <host:port>]... [--ssh=<host:port>]... [--tls=<host:port>]... [--openvpn=<host:port>]... [--tinc=<host:port>]... [--xmpp=<host:port>]... [--http=<host:port>]... [--adb=<host:port>]... [--socks5=<host:port>]... [--syslog=<host:port>]... [--anyprot=<host:port>]...
>   -F, --config=<file>      	Specify configuration file
>   -v, --verbose=<n>        	
>   -V, --version            	Print version information and exit
>   -f, --foreground         	Run in foreground instead of as a daemon
>   -i, --inetd              	Run in inetd mode: use stdin/stdout instead of network listen
>   -n, --numeric            	Print IP addresses and ports as numbers
>   --transparent            	Set up as a transparent proxy
>   -t, --timeout=<n>        	Set up timeout before connecting to default target
>   -u, --user=<str>         	Username to change to after set-up
>   -P, --pidfile=<file>     	Path to file to store PID of current instance
>   -C, --chroot=<path>      	Root to change to after set-up
>   --syslog-facility=<str>  	Facility to syslog to
>   --on-timeout=<str>       	Target to connect to when timing out
>   --prefix=<str>           	Reserved for testing
>   -p, --listen=<host:port> 	Listen on host:port
>   --ssh=<host:port>        	Set up ssh target
>   --tls=<host:port>        	Set up TLS/SSL target
>   --openvpn=<host:port>    	Set up OpenVPN target
>   --tinc=<host:port>       	Set up tinc target
>   --xmpp=<host:port>       	Set up XMPP target
>   --http=<host:port>       	Set up HTTP (plain) target
>   --adb=<host:port>        	Set up ADB (Android Debug) target
>   --socks5=<host:port>     	Set up socks5 target
>   --syslog=<host:port>     	Set up syslog target
>   --anyprot=<host:port>    	Set up default target

  
After trying with the following changes on my FritzBox 7490 it was working so i would propose the following PR:
[https://github.com/Freetz-NG/freetz-ng/pull/704](https://github.com/Freetz-NG/freetz-ng/pull/704)

Thank you for checking the changes and also thank you for all your efforts you do on this project in general!

fda77, 2023-01-18:
Thx! Great when it works now: Maybe
> https://github.com/yrutschle/sslh/commit/64c3e0ed1e2a30170e2fa4d9979b332b69090765 which was changed Sep 4, 2022
was also merged into other version, i had not verified it

Spamm00r, 2024-02-06:
This still doesn't work. I circumvented this problem starting sslh manually over a year now..
Seeing that a newer version of sslh was implemented, I updated to the latest version, hoping this problem would be fixed, and build an image with it today and starting SSLH still fails, with the same problem I described in my OP.

How come this wasn't fixed for over a year now? Everyone who is using SSLH should be stumbling over this problem. Am I the only one using SSLH?

fda77, 2024-02-06:
Yes!
Instead of hoping+waiting you could check changes (none) here: https://github.com/Freetz-NG/freetz-ng/commits/master/make/pkgs/sslh