Freetz-NG/freetz-ng #691: Avahi fails to start in freetz.

Spamm00r, 2023-01-02:
I have been using freetz for years, but never build Avahi. So I can not say whether it ever worked with Freetz.

Now I have need for Avahi and I build it into my freetz build for my 7590 box, which goes through without errors.

However after flashing, avahi fails to start. No errors in Syslog about this.

- §1 Write in german or english
 - §2 Attach your .config after renaming to config.txt
 - §3 Describe steps to reproduce the behavior
 - §4 "Error 1" is not an error description
 - §5 Add logs, in most cases this are more than 10 lines
 - §6 Embed images by dragging them into this text box
 - §7 Github does not allow to attach all file types, append .txt
```
 - §8 Format code and logs by wrapping it with ``` like this line
```

pfichtner, 2023-01-04:
If there is no error in syslog then how do you know it won't start? What makes you sure it will not run?

pfichtner, 2023-01-04:
PS: Of course you should change all files needed in your fork like https://github.com/Freetz-NG/freetz-ng/blob/9d366bef12afcd358e049ad4efba394245ea6019/make/pkgs/sslh/files/root/etc/default.sslh/sslh_conf#L6 as well

Spamm00r, 2023-01-04:
Because the service does not start automatically and also stops immediately if I start it manually.

pfichtner, 2023-01-04:
And nothing (no error message) is printed out? 
Then login onto the box (or use Rudi-Shell) and run `avahi` manually and check the output. You **may** have to add `-r /path/to/the/avahi.config`.

closed 2023-01-22 by github-actions[bot]