Freetz-NG/freetz-ng #692: Installation bei 7490 nicht möglich

Hoerli1337, 2023-01-08:
Hi!
Ich habe versucht eine 7490 mit freetz-ng zu versorgen.
Dazu habe ich eine VM mit Debian 11 aufgesetzt und alle notwendigen Abhängigkeiten installiert.
Das Erstellen des Images war soweit auch erfolgreich.
Beim Aufspielen über das Webinterface, lässt mich die Box das nicht tun.
[image]
- FritzBox 7490 (DE-Version)
- Firmware 113.07.29
- Im Mesh aufgenommen
- Habe die Box vor ca. 6 Monaten mal zurückgesetzt, seitdem läuft sie durchgehend
- Konfiguration über menuconfig weitestgehend erst mal auf Standard belassen
Hier ist die Config:
[config.txt](https://github.com/Freetz-NG/freetz-ng/files/10368782/config.txt)

Muss ich noch etwas bestimmtes Einstellen, damit Freetz-NG auf der Box läuft oder hab ich gerade nur extremes Pech?
Muss ich die Firmware mit dem Flash-Tool installieren?

pfichtner, 2023-01-08:
Über AVM's Weboberfläche können nur von AVM freigegebene/durch AVM signierte Images eingespielt werden. Du musst die Firmware über den Bootloader einspielen (flashen).

fda77, 2023-01-08:
Da es kein Issue ist...