Freetz-NG/freetz-ng #697: Build von tools/patchelf-host schlägt fehl

sebastianertz, 2023-01-14:
config:
```
FREETZ_USER_LEVEL_EXPERT=y
FREETZ_TYPE_7530=y
FREETZ_REMOVE_LANGUAGE=y
FREETZ_REMOVE_BRANDING=y
FREETZ_REMOVE_BRANDING_1und1=y
FREETZ_PACKAGE_DNSMASQ=y
FREETZ_PACKAGE_DNSMASQ_ADDHOLE=y
FREETZ_PACKAGE_VNSTAT_CGI=y
# FREETZ_HOSTTOOLS_DOWNLOAD is not set
FREETZ_SIZEINFO_UNCOMPRESSED=y
```

Fehlermeldung:
```
---> tools/patchelf-host ... 
building ... Making all in src
g++ -std=c++17 -DPACKAGE_NAME=\"patchelf\" -DPACKAGE_TARNAME=\"patchelf\" -DPACKAGE_VERSION=\"0.17.2\" -DPACKAGE_STRING=\"patchelf\ 0.17.2\" -DPACKAGE_BUGREPORT=\"\" -DPACKAGE_URL=\"\" -DPACKAGE=\"patchelf\" -DVERSION=\"0.17.2\" -DHAVE_CXX17=1 -I.    -Wall -std=c++11 -D_FILE_OFFSET_BITS=64     -Wall -O2 -MT patchelf.o -MD -MP -MF .deps/patchelf.Tpo -c -o patchelf.o patchelf.cc
patchelf.cc: In member function ‘void ElfFile<Elf_Ehdr, Elf_Phdr, Elf_Shdr, Elf_Addr, Elf_Off, Elf_Dyn, Elf_Sym, Elf_Verneed, Elf_Versym>::normalizeNoteSegments()’:
patchelf.cc:966:17: error: use of ‘auto’ in lambda parameter declaration only available with ‘-std=c++14’ or ‘-std=gnu++14’
  966 |             [&](auto & shdr) { return rdi(shdr.sh_offset) >= start_off && rdi(shdr.sh_offset) < end_off; });
      |                 ^~~~
patchelf.cc: In lambda function:
patchelf.cc:966:48: error: request for member ‘sh_offset’ in ‘shdr’, which is of non-class type ‘int’
  966 |             [&](auto & shdr) { return rdi(shdr.sh_offset) >= start_off && rdi(shdr.sh_offset) < end_off; });
      |                                                ^~~~~~~~~
patchelf.cc:966:84: error: request for member ‘sh_offset’ in ‘shdr’, which is of non-class type ‘int’
  966 |             [&](auto & shdr) { return rdi(shdr.sh_offset) >= start_off && rdi(shdr.sh_offset) < end_off; });
      |                                                                                    ^~~~~~~~~
In file included from /usr/include/c++/10/bits/stl_algobase.h:71,
                 from /usr/include/c++/10/algorithm:61,
                 from patchelf.cc:19:
/usr/include/c++/10/bits/predefined_ops.h: In instantiation of ‘bool __gnu_cxx::__ops::_Iter_pred<_Predicate>::operator()(_Iterator) [with _Iterator = __gnu_cxx::__normal_iterator<Elf32_Shdr*, std::vector<Elf32_Shdr, std::allocator<Elf32_Shdr> > >; _Predicate = ElfFile<Elf_Ehdr, Elf_Phdr, Elf_Shdr, Elf_Addr, Elf_Off, Elf_Dyn, Elf_Sym, Elf_Verneed, Elf_Versym>::normalizeNoteSegments<Elf32_Ehdr, Elf32_Phdr, Elf32_Shdr, unsigned int, unsigned int, Elf32_Dyn, Elf32_Sym, Elf32_Verneed, short unsigned int>::<lambda(int&)>]’:
/usr/include/c++/10/bits/stl_algobase.h:1929:14:   required from ‘_RandomAccessIterator std::__find_if(_RandomAccessIterator, _RandomAccessIterator, _Predicate, std::random_access_iterator_tag) [with _RandomAccessIterator = __gnu_cxx::__normal_iterator<Elf32_Shdr*, std::vector<Elf32_Shdr, std::allocator<Elf32_Shdr> > >; _Predicate = __gnu_cxx::__ops::_Iter_pred<ElfFile<Elf_Ehdr, Elf_Phdr, Elf_Shdr, Elf_Addr, Elf_Off, Elf_Dyn, Elf_Sym, Elf_Verneed, Elf_Versym>::normalizeNoteSegments<Elf32_Ehdr, Elf32_Phdr, Elf32_Shdr, unsigned int, unsigned int, Elf32_Dyn, Elf32_Sym, Elf32_Verneed, short unsigned int>::<lambda(int&)> >]’
/usr/include/c++/10/bits/stl_algobase.h:1974:23:   required from ‘_Iterator std::__find_if(_Iterator, _Iterator, _Predicate) [with _Iterator = __gnu_cxx::__normal_iterator<Elf32_Shdr*, std::vector<Elf32_Shdr, std::allocator<Elf32_Shdr> > >; _Predicate = __gnu_cxx::__ops::_Iter_pred<ElfFile<Elf_Ehdr, Elf_Phdr, Elf_Shdr, Elf_Addr, Elf_Off, Elf_Dyn, Elf_Sym, Elf_Verneed, Elf_Versym>::normalizeNoteSegments<Elf32_Ehdr, Elf32_Phdr, Elf32_Shdr, unsigned int, unsigned int, Elf32_Dyn, Elf32_Sym, Elf32_Verneed, short unsigned int>::<lambda(int&)> >]’
/usr/include/c++/10/bits/stl_algo.h:3934:28:   required from ‘_IIter std::find_if(_IIter, _IIter, _Predicate) [with _IIter = __gnu_cxx::__normal_iterator<Elf32_Shdr*, std::vector<Elf32_Shdr, std::allocator<Elf32_Shdr> > >; _Predicate = ElfFile<Elf_Ehdr, Elf_Phdr, Elf_Shdr, Elf_Addr, Elf_Off, Elf_Dyn, Elf_Sym, Elf_Verneed, Elf_Versym>::normalizeNoteSegments<Elf32_Ehdr, Elf32_Phdr, Elf32_Shdr, unsigned int, unsigned int, Elf32_Dyn, Elf32_Sym, Elf32_Verneed, short unsigned int>::<lambda(int&)>]’
/usr/include/c++/10/bits/stl_algo.h:471:47:   required from ‘bool std::none_of(_IIter, _IIter, _Predicate) [with _IIter = __gnu_cxx::__normal_iterator<Elf32_Shdr*, std::vector<Elf32_Shdr, std::allocator<Elf32_Shdr> > >; _Predicate = ElfFile<Elf_Ehdr, Elf_Phdr, Elf_Shdr, Elf_Addr, Elf_Off, Elf_Dyn, Elf_Sym, Elf_Verneed, Elf_Versym>::normalizeNoteSegments<Elf32_Ehdr, Elf32_Phdr, Elf32_Shdr, unsigned int, unsigned int, Elf32_Dyn, Elf32_Sym, Elf32_Verneed, short unsigned int>::<lambda(int&)>]’
patchelf.cc:965:34:   required from ‘void ElfFile<Elf_Ehdr, Elf_Phdr, Elf_Shdr, Elf_Addr, Elf_Off, Elf_Dyn, Elf_Sym, Elf_Verneed, Elf_Versym>::normalizeNoteSegments() [with Elf_Ehdr = Elf32_Ehdr; Elf_Phdr = Elf32_Phdr; Elf_Shdr = Elf32_Shdr; Elf_Addr = unsigned int; Elf_Off = unsigned int; Elf_Dyn = Elf32_Dyn; Elf_Sym = Elf32_Sym; Elf_Verneed = Elf32_Verneed; Elf_Versym = short unsigned int]’
patchelf.cc:804:5:   required from ‘void ElfFile<Elf_Ehdr, Elf_Phdr, Elf_Shdr, Elf_Addr, Elf_Off, Elf_Dyn, Elf_Sym, Elf_Verneed, Elf_Versym>::rewriteSectionsLibrary() [with Elf_Ehdr = Elf32_Ehdr; Elf_Phdr = Elf32_Phdr; Elf_Shdr = Elf32_Shdr; Elf_Addr = unsigned int; Elf_Off = unsigned int; Elf_Dyn = Elf32_Dyn; Elf_Sym = Elf32_Sym; Elf_Verneed = Elf32_Verneed; Elf_Versym = short unsigned int]’
patchelf.cc:1023:9:   required from ‘void ElfFile<Elf_Ehdr, Elf_Phdr, Elf_Shdr, Elf_Addr, Elf_Off, Elf_Dyn, Elf_Sym, Elf_Verneed, Elf_Versym>::rewriteSections() [with Elf_Ehdr = Elf32_Ehdr; Elf_Phdr = Elf32_Phdr; Elf_Shdr = Elf32_Shdr; Elf_Addr = unsigned int; Elf_Off = unsigned int; Elf_Dyn = Elf32_Dyn; Elf_Sym = Elf32_Sym; Elf_Verneed = Elf32_Verneed; Elf_Versym = short unsigned int]’
patchelf.cc:1328:11:   required from ‘void ElfFile<Elf_Ehdr, Elf_Phdr, Elf_Shdr, Elf_Addr, Elf_Off, Elf_Dyn, Elf_Sym, Elf_Verneed, Elf_Versym>::modifySoname(ElfFile<Elf_Ehdr, Elf_Phdr, Elf_Shdr, Elf_Addr, Elf_Off, Elf_Dyn, Elf_Sym, Elf_Verneed, Elf_Versym>::sonameMode, const string&) [with Elf_Ehdr = Elf32_Ehdr; Elf_Phdr = Elf32_Phdr; Elf_Shdr = Elf32_Shdr; Elf_Addr = unsigned int; Elf_Off = unsigned int; Elf_Dyn = Elf32_Dyn; Elf_Sym = Elf32_Sym; Elf_Verneed = Elf32_Verneed; Elf_Versym = short unsigned int; std::string = std::__cxx11::basic_string<char>]’
patchelf.cc:1928:29:   required from ‘void patchElf2(ElfFile&&, const FileContents&, const string&) [with ElfFile = ElfFile<Elf32_Ehdr, Elf32_Phdr, Elf32_Shdr, unsigned int, unsigned int, Elf32_Dyn, Elf32_Sym, Elf32_Verneed, short unsigned int>; FileContents = std::shared_ptr<std::vector<unsigned char> >; std::string = std::__cxx11::basic_string<char>]’
patchelf.cc:1980:185:   required from here
/usr/include/c++/10/bits/predefined_ops.h:316:23: error: no match for call to ‘(ElfFile<Elf_Ehdr, Elf_Phdr, Elf_Shdr, Elf_Addr, Elf_Off, Elf_Dyn, Elf_Sym, Elf_Verneed, Elf_Versym>::normalizeNoteSegments<Elf32_Ehdr, Elf32_Phdr, Elf32_Shdr, unsigned int, unsigned int, Elf32_Dyn, Elf32_Sym, Elf32_Verneed, short unsigned int>::<lambda(int&)>) (Elf32_Shdr&)’
  316 |  { return bool(_M_pred(*__it)); }
      |                ~~~~~~~^~~~~~~
patchelf.cc:966:13: note: candidate: ‘ElfFile<Elf_Ehdr, Elf_Phdr, Elf_Shdr, Elf_Addr, Elf_Off, Elf_Dyn, Elf_Sym, Elf_Verneed, Elf_Versym>::normalizeNoteSegments<Elf32_Ehdr, Elf32_Phdr, Elf32_Shdr, unsigned int, unsigned int, Elf32_Dyn, Elf32_Sym, Elf32_Verneed, short unsigned int>::<lambda(int&)>’
  966 |             [&](auto & shdr) { return rdi(shdr.sh_offset) >= start_off && rdi(shdr.sh_offset) < end_off; });
      |             ^
patchelf.cc:966:13: note:   no known conversion for argument 1 from ‘Elf32_Shdr’ to ‘int&’
In file included from /usr/include/c++/10/bits/stl_algobase.h:71,
                 from /usr/include/c++/10/algorithm:61,
                 from patchelf.cc:19:
/usr/include/c++/10/bits/predefined_ops.h: In instantiation of ‘bool __gnu_cxx::__ops::_Iter_pred<_Predicate>::operator()(_Iterator) [with _Iterator = __gnu_cxx::__normal_iterator<Elf64_Shdr*, std::vector<Elf64_Shdr, std::allocator<Elf64_Shdr> > >; _Predicate = ElfFile<Elf_Ehdr, Elf_Phdr, Elf_Shdr, Elf_Addr, Elf_Off, Elf_Dyn, Elf_Sym, Elf_Verneed, Elf_Versym>::normalizeNoteSegments<Elf64_Ehdr, Elf64_Phdr, Elf64_Shdr, long unsigned int, long unsigned int, Elf64_Dyn, Elf64_Sym, Elf64_Verneed, short unsigned int>::<lambda(int&)>]’:
[... cut ...]
  966 |             [&](auto & shdr) { return rdi(shdr.sh_offset) >= start_off && rdi(shdr.sh_offset) < end_off; });
      |             ^
patchelf.cc:966:13: note:   no known conversion for argument 1 from ‘Elf64_Shdr’ to ‘int&’
make[3]: *** [Makefile:376: patchelf.o] Error 1
make[2]: *** [Makefile:447: all-recursive] Error 1
make[1]: *** [make/host-tools/patchelf-host/patchelf-host.mk:18: /home/se/freetz-ng/freetz-ng/source/host-tools/patchelf-0.17.2/src/patchelf] Terminated
make: *** [Makefile:46: envira] Terminated
```

fda77, 2023-01-14:
Komisch, ging vorhin noch

fda77, 2023-01-14:
@er13: Passt das so noch? Alternativ könnte man auch `std=c++11` lassen und https://github.com/NixOS/patchelf/commit/15f02b0f7a2721062a89d545d9f3696499f0d8de reverten, braucht man nicht

closed 2023-01-14 by fda77