Freetz-NG/freetz-ng #706: Fehler bei rrdtool in FW. 07.39/07.50

leo-022, 2023-01-19:
Firmware: 259.07.39 rev102523 (Fritzbox 7590 AX)
Freetz: ng-20983M-f2333d0b2

In der Beta-Firmware bekomme ich leider ununterbrochen rrd-Fehlermeldungen:

```
Jan 19 16:55:52 fritz user.err avmcounterd[2413]: update of rrd for 'cpuusage' FAILED: ERROR: unknown option '-s'
Jan 19 16:55:52 fritz user.err avmcounterd[2413]: update of rrd for 'dev_guest' FAILED: ERROR: unknown option '-s'
Jan 19 16:55:52 fritz user.err avmcounterd[2413]: update of rrd for 'dev_lan' FAILED: ERROR: unknown option '-s'
Jan 19 16:55:52 fritz user.err avmcounterd[2413]: update of rrd for 'dsliface_internet' FAILED: ERROR: unknown option '-s'
Jan 19 16:55:52 fritz user.err avmcounterd[2413]: update of rrd for 'dsliface_mstv' FAILED: ERROR: unknown option '-s'
Jan 19 16:55:52 fritz user.err avmcounterd[2413]: update of rrd for 'ethgstats' FAILED: ERROR: unknown option '-s'
Jan 19 16:55:52 fritz user.err avmcounterd[2413]: update of rrd for 'local' FAILED: ERROR: unknown option '-s'
Jan 19 16:55:52 fritz user.err avmcounterd[2413]: update of rrd for 'memory' FAILED: ERROR: unknown option '-s'
Jan 19 16:55:52 fritz user.err avmcounterd[2413]: update of rrd for 'neigh10307' FAILED: ERROR: unknown option '-s'
Jan 19 16:55:52 fritz user.err avmcounterd[2413]: update of rrd for 'neigh10613' FAILED: ERROR: unknown option '-s'
Jan 19 16:55:52 fritz user.err avmcounterd[2413]: update of rrd for 'neigh10908' FAILED: ERROR: unknown option '-s'
Jan 19 16:55:52 fritz user.err avmcounterd[2413]: update of rrd for 'neigh11252' FAILED: ERROR: unknown option '-s'
Jan 19 16:55:52 fritz user.err avmcounterd[2413]: update of rrd for 'neigh115238' FAILED: ERROR: unknown option '-s'
Jan 19 16:55:52 fritz user.err avmcounterd[2413]: update of rrd for 'neigh11853' FAILED: ERROR: unknown option '-s'
Jan 19 16:55:52 fritz user.err avmcounterd[2413]: update of rrd for 'neigh11934' FAILED: ERROR: unknown option '-s'
Jan 19 16:55:52 fritz user.err avmcounterd[2413]: update of rrd for 'neigh126476' FAILED: ERROR: unknown option '-s'
Jan 19 16:55:52 fritz user.err avmcounterd[2413]: update of rrd for 'neigh15111' FAILED: ERROR: unknown option '-s'
Jan 19 16:55:52 fritz user.err avmcounterd[2413]: update of rrd for 'neigh17614' FAILED: ERROR: unknown option '-s'
Jan 19 16:55:52 fritz user.err avmcounterd[2413]: update of rrd for 'neigh199472' FAILED: ERROR: unknown option '-s'
Jan 19 16:55:52 fritz user.err avmcounterd[2413]: update of rrd for 'neigh20103' FAILED: ERROR: unknown option '-s'
Jan 19 16:55:52 fritz user.err avmcounterd[2413]: update of rrd for 'neigh29230' FAILED: ERROR: unknown option '-s'
Jan 19 16:55:52 fritz user.err avmcounterd[2413]: update of rrd for 'neigh3106' FAILED: ERROR: unknown option '-s'
Jan 19 16:55:52 fritz user.err avmcounterd[2413]: update of rrd for 'neigh3109' FAILED: ERROR: unknown option '-s'
Jan 19 16:55:52 fritz user.err avmcounterd[2413]: update of rrd for 'neigh3118' FAILED: ERROR: unknown option '-s'
Jan 19 16:55:52 fritz user.err avmcounterd[2413]: update of rrd for 'neigh36381' FAILED: ERROR: unknown option '-s'
Jan 19 16:55:52 fritz user.err avmcounterd[2413]: update of rrd for 'neigh7299' FAILED: ERROR: unknown option '-s'
Jan 19 16:55:52 fritz user.err avmcounterd[2413]: update of rrd for 'neigh75602' FAILED: ERROR: unknown option '-s'
Jan 19 16:55:52 fritz user.err avmcounterd[2413]: update of rrd for 'neigh8746' FAILED: ERROR: unknown option '-s'
Jan 19 16:55:52 fritz user.err avmcounterd[2413]: update of rrd for 'neigh8923' FAILED: ERROR: unknown option '-s'
Jan 19 16:55:52 fritz user.err avmcounterd[2413]: update of rrd for 'neigh9210' FAILED: ERROR: unknown option '-s'
Jan 19 16:55:52 fritz user.err avmcounterd[2413]: update of rrd for 'neigh9926' FAILED: ERROR: unknown option '-s'
Jan 19 16:55:52 fritz user.err avmcounterd[2413]: update of rrd for 'neigh9979' FAILED: ERROR: unknown option '-s'
Jan 19 16:55:52 fritz user.err avmcounterd[2413]: update of rrd for 'onlinemonitor0' FAILED: ERROR: unknown option '-s'
Jan 19 16:55:52 fritz user.err avmcounterd[2413]: update of rrd for 'prio' FAILED: ERROR: unknown option '-s'
Jan 19 16:55:52 fritz user.err avmcounterd[2413]: update of rrd for 'temps' FAILED: ERROR: unknown option '-s'
```

Ursache hierfür ist, dass AVM nun selbst rrdtool in der Firmware benutzt, und zwar die Version 1.7.2. Das in Freetz enthaltene rrdtool ist älter (1.2.30).

Da im erstellten Freetz-Image die Original-AVM-Version nicht mehr vorhanden ist und deshalb die von Freetz gebaute auch für die AVM-Aufrufe verwendet wird, kommt es zu ```unknown option '-s'```.

Den Parameter ```-s``` in der Funktion ```update```  gibt es in der alten Version nicht.

In Freetz enthaltene Version:
```
RRDtool 1.2.30  Copyright 1997-2008 by Tobias Oetiker <tobi@oetiker.ch>
               Compiled Oct 30 2022 10:51:16

Usage: rrdtool [options] command command_options

* update - update an RRD

	rrdtool update filename
		--template|-t ds-name:ds-name:...
		time|N:value[:value...]

		at-time@value[:value...]

		[ time:value[:value...] ..]

RRDtool is distributed under the Terms of the GNU General
Public License Version 2. (www.gnu.org/copyleft/gpl.html)

For more information read the RRD manpages
```

In original-AVM-Firmware 07.39 enthaltene Version:

```
RRDtool 1.7.2  Copyright by Tobias Oetiker <tobi@oetiker.ch>
               Compiled Oct 25 2022 13:26:07

Usage: rrdtool [options] command command_options

* update - update an RRD

	rrdtool update filename
		[--template|-t ds-name:ds-name:...]
		[--skip-past-updates|-s]
		[--daemon|-d <address>]
		time|N:value[:value...]

		at-time@value[:value...]

		[ time:value[:value...] ..]

RRDtool is distributed under the Terms of the GNU General
Public License Version 2. (www.gnu.org/copyleft/gpl.html)

For more information read the RRD manpages
```

In der Release-Versiion 07.50 für die 7590 zeigt sich dieselbe Situation.

fda77, 2023-01-19:
Oh, blöd. Mir ist auch schon aufgefallen dass AVM nach `/tmp/rrd/` diverse .rrd Dateien ablegt, und ich zufällig auch dieses Verzeichnis genutzt hab. Und das obwohl `FREETZ_REMOVE_AVMCOUNTERD` ausgewählt war.
Am besten würde man rrdtool in Freetz updaten, aber das braucht ganz komische libs (cairo, pango), bei denen es irgend welche Problem gab, hab vergessen welche
Als Workaround könnte man die Zeile `select FREETZ_PACKAGE_RRDTOOL` in https://github.com/Freetz-NG/freetz-ng/blob/master/make/pkgs/rrdstats/Config.in#L3 löschen und rrdtool deaktivieren, dann müste das neuere von AVM genutzt werden

leo-022, 2023-01-20:
Im Prinzip würde das funktionieren. Die Werte werden in die entsprechenden Dateien geschrieben, allerdings kann man keine Diagramme anzeigen.

Leider hat AVM rrdtool etwas verstümmelt:
```
ERROR: the instance of rrdtool has been compiled without graphics
```

fda77, 2023-01-20:
Gut scheidet die AVM-Binary somit aus. Bliebe noch die von Freetz umzubenennen und den Namen auch in rrdstats zu ändern...
Ich hab mir (lib)pango nochmal angeschaut, die verwendet das meson buildsystem (https://mesonbuild.com/) das man nur dafür noch in feetz einbauen müsste. Das kann ich nicht "mal kurz"

leo-022, 2023-01-21:
Das mit dem Umbenennen funktioniert. Ich verwende nun das alte Freetz-rrdtool ausschließlich zum Generieren der Grafiken, für das Datensammeln wird das AVM-rrdtool verwendet. Workaround dazu ist folgender:

In https://github.com/Freetz-NG/freetz-ng/blob/master/make/pkgs/rrdtool/rrdtool.mk in der Zeile `$(PKG)_TARGET_BINARY:=$($(PKG)_DEST_DIR)/usr/bin/rrdtool` für `rrdtool` einen neuen Namen vergeben, z. B. `rrdtoolneuername `.

In https://github.com/Freetz-NG/freetz-ng/blob/master/make/pkgs/rrdstats/files/root/usr/lib/cgi-bin/rrdstats/main.cgi
alle Vorkommen von `rrdtool graph` durch `rrdtoolneuername graph` ersetzen.

fda77, 2023-01-24:
Die version 1.8 sollte mit den Parametern von AVM kompatibel sein, sie ist aber 2,8mb grösser als die alte 1.2. Ich kann mir nicht vorstellen dass die Graphen so viele besser aussehen werden dass sich das lohnt. Falls die Binary überhaupt läuft. Zum testen am besten vorher ein Backup der rrds erstellen...

leo-022, 2023-01-24:
Ich bekomme leider die Fehlermeldung `rm: cannot remove 'builddir/': Is a directory`:

```
~/freetz-ng-test$ make
Freetz-NG 21044M-cecac2361 master 2023-01-24
JUIS: Using cached value ... http://download.avm.de/labor/MOVE21/7590AX/FRITZ.Box_7590_AX-07.39-102523-LabBETA.image ... done.
---> library/glib2 ...
configuring ... cat include/meson.cross/mips-linux-uclibc > source/target-mips_gcc-8.3.0_uClibc-1.0.42-nptl_kernel-4.9/glib2-2.75.2/meson.freetz
(meson_cmd() { meson setup "$@"  || { [ "2" = "0" ] && echo && cat .build.log 2>/dev/null; kill $ 2>/dev/null || kill $$ 2>/dev/null; printf "\n\\033[33m%s\\033[m\n" "ERROR: Build failed.";  exit 1; } }; cd source/target-mips_gcc-8.3.0_uClibc-1.0.42-nptl_kernel-4.9/glib2-2.75.2; {  rm -f config.{cache,status} builddir/; } ;    PATH="/home/leo/freetz-ng-test/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.42-nptl_kernel-4.9/mips-linux-uclibc/bin:/home/leo/freetz-ng-test/toolchain/build/mips_gcc-8.3.0/mips-unknown-linux-gnu/bin:/home/leo/freetz-ng-test/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.42-nptl_kernel-4.9/mips-linux-uclibc/bin-ccache:/home/leo/freetz-ng-test/toolchain/build/mips_gcc-8.3.0/mips-unknown-linux-gnu/bin-ccache:/home/leo/freetz-ng-test/tools/path:/home/leo/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/usr/lib/wsl/lib:/mnt/c/Program Files/WindowsApps/CanonicalGroupLimited.Ubuntu20.04onWindows_2004.2022.8.0_x64__79rhkp1fndgsc:/mnt/c/Program Files/Common Files/Oracle/Java/javapath:/mnt/c/Program Files (x86)/Common Files/Oracle/Java/javapath:/mnt/c/WINDOWS/system32:/mnt/c/WINDOWS:/mnt/c/WINDOWS/System32/Wbem:/mnt/c/WINDOWS/System32/WindowsPowerShell/v1.0/:/mnt/c/WINDOWS/System32/OpenSSH/:/mnt/c/Program Files/PuTTY/:/mnt/c/Users/leo/AppData/Local/Microsoft/WindowsApps:/mnt/c/texlive/2022/bin/win32:/mnt/c/Program Files (x86)/LilyPond/usr/bin:/snap/bin:/usr/sbin" FREETZ_LIBRARY_DIR="/usr/lib/freetz" FREETZ_KERNEL_VERSION_MAJOR="4.9" FREETZ_TARGET_ARCH="mips" FREETZ_TARGET_ARCH_ENDIANNESS_DEPENDENT="mips" CC="/home/leo/freetz-ng-test/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.42-nptl_kernel-4.9/mips-linux-uclibc/bin/mips-linux-uclibc-gcc" CXX="/home/leo/freetz-ng-test/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.42-nptl_kernel-4.9/mips-linux-uclibc/bin/mips-linux-uclibc-g++-wrapper" CFLAGS="-march=34kc -mtune=34kc -msoft-float -Ofast -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Wl,-I/usr/lib/freetz/ld-uClibc.so.1" CXXFLAGS="-march=34kc -mtune=34kc -msoft-float -Ofast -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Wl,-I/usr/lib/freetz/ld-uClibc.so.1" LDFLAGS="" PKG_CONFIG_PATH="/home/leo/freetz-ng-test/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.42-nptl_kernel-4.9/mips-linux-uclibc/bin/../lib/pkgconfig" PKG_CONFIG_LIBDIR="/home/leo/freetz-ng-test/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.42-nptl_kernel-4.9/mips-linux-uclibc/bin/../lib/pkgconfig"  meson_cmd --cross-file meson.freetz builddir . -D selinux=disabled -D xattr=false -D libmount=disabled -D man=false -D dtrace=false -D systemtap=false -D sysprof=disabled -D gtk_doc=false -D bsymbolic_functions=false -D tests=false -D installed_tests=false -D nls=disabled -D oss_fuzz=disabled -D glib_debug=disabled -D glib_assert=false -D glib_checks=false -D libelf=disabled -D multiarch=false   )
rm: cannot remove 'builddir/': Is a directory

ERROR: Malformed value in cross file variable ar.

ERROR: Build failed.
make[1]: *** [make/libs/glib2/glib2.mk:103: source/target-mips_gcc-8.3.0_uClibc-1.0.42-nptl_kernel-4.9/glib2-2.75.2/.configured] Terminated
make: *** [Makefile:46: envira] Terminated
```

fda77, 2023-01-24:
Du hattest aber vorher schon mal angefangen zu bauen und abgebrochen?
"make glib2-recompile"  sollte helfen (und "make help" )
Das "Malformed value in cross file variable ar." find ich aber auch komisch

leo-022, 2023-01-24:
Das "Malformed value in cross file variable ar."  bekomme ich auch bei `make glib2-recompile`.

fda77, 2023-01-24:
Meson wird geprüft/installiert und das `'builddir/': Is a directory` sollte behoben sein: c9ebeb9fe +680fec29b
Schau doch mal was in der cross file drin ist, die wird generiert:
`source/target-mips_gcc-8.3.0_uClibc-1.0.42-nptl_kernel-4.9/glib2-2.75.2/meson.freetz`
Ich hatte das nur mir 7490/7590 mips getestet, ich glaub du hast ARM

leo-022, 2023-01-24:
Ich habe hier gerade die 7590 AX. `Malformed value in cross file variable ar` kommt nach wie vor.
In `meson.freetz` steht folgendes:
```

[constants]
arch              = 'mips-linux-uclibc'

[target_machine]
system            = 'linux'
cpu_family        = 'mips'
cpu               = 'mips'
endian            = 'big'

[built-in options]
prefix            = '/usr'
c_args            = ['-march=34kc', '-mtune=34kc', '-msoft-float', '-Ofast', '-pipe', '-Wa,--trap', '-D_LARGEFILE_SOURCE', '-D_LARGEFILE64_SOURCE', '-D_FILE_OFFSET_BITS=64', '-Wl,-I/usr/lib/freetz/ld-uClibc.so.1',]
c_link_args       = []
cpp_args          = ['-march=34kc', '-mtune=34kc', '-msoft-float', '-Ofast', '-pipe', '-Wa,--trap', '-D_LARGEFILE_SOURCE', '-D_LARGEFILE64_SOURCE', '-D_FILE_OFFSET_BITS=64', '-Wl,-I/usr/lib/freetz/ld-uClibc.so.1',]
cpp_link_args     = []

[properties]
needs_exe_wrapper = true
pkg_config_libdir = '/home/leo/freetz-ng-test/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.42-nptl_kernel-4.9/mips-linux-uclibc/bin/../lib/pkgconfig'

[binaries]
ar                = arch + '-ar'
as                = arch + '-as'
c                 = arch + '-gcc'
cc                = arch + '-gcc'
cpp               = arch + '-g++-wrapper'
cxx               = arch + '-g++-wrapper'
ld                = arch + '-ld'
ldconfig          = arch + '-ldconfig'
nm                = arch + '-nm'
ranlib            = arch + '-ranlib'
objcopy           = arch + '-objcopy'
readelf           = arch + '-readelf'
strip             = arch + '-strip --remove-section={.comment,.note,.pdr}'
pkgconfig         = 'pkg-config'
python            = 'python3'
cmake             = 'cmake'

```

CaptainJackDaniels, 2023-01-24:
Für die 6430 6490 und 6590 bekommt man auch den Fehler 
**_ninja: build stopped: subcommand failed._**
make[1]: *** [make/libs/glib2/glib2.mk:107: source/target-i686_gcc-5.5.0_uClibc-1.0.42-nptl/glib2-2.75.2/builddir/glib/libglib-2.0.so.0.7502.0] Beendet
make: *** [Makefile:46: envira] Beendet
`
`

leo-022, 2023-01-24:
Nach dem Test mit dem Editieren der `600-macros.mk` bekam ich die aufschlussreiche Fehlermeldung
`Meson version is 0.53.2 but project requires >= 0.60.0`

Ich hatte Meson unter Ubuntu mit `apt-get install meson` installiert und dachte, da müsste ja die neueste Version genommen werden. Man muss aber mit Pythons `pip install meson` installieren, dann erhält man anstatt der Version 0.53.2 die Version 1.0.0.

fda77, 2023-01-24:
Was hast du denn für ein Linux das eine zu alter Version installiert?
Bei Fedora 37
```
meson -V
0.63.3
```
Am besten wäre es das dann wohl meson als Hosttool aufzunehmen. Alles run um python ist irgendwie Mist...

leo-022, 2023-01-24:
Das Image wurde jetzt erfolgreich gebaut.

Es läuft, nur etwas mit der Schrift stimmt nicht:
[image]

fda77, 2023-01-25:
Keine Ahnung... aber immerhin werden überhaupt Grafiken erstellt.
Sehen die "besser" aus? Im "tmp" Verzeichnis von rrdstats könnten noch alte Grafiken liegen mit denen man vergleichen könnte.
Am besten rufst du einen "rrdtool graph" Befehl mal im Terminal auf, vielleicht zeigt der was an. Evtl gibts da auch ein debug oder verbose. Es könnte sein dass bei der der 10 Libs noch irgend Optionen aktiviert werden müssen. Oder dass einfach ein paar Daten (Fonts?) vergessen wurden ins Image zu kopierenn

> Was hast du denn für ein Linux das eine zu alter Version installiert?

CaptainJackDaniels, 2023-01-25:
Für die 6430, 6490 und 6590 bekommt man immer noch den Fehler auch nach dem heutigen Update

CaptainJackDaniels, 2023-01-25:
Was soll das denn bitte. Darf man hier keine Fehler mehr posten?

CaptainJackDaniels, 2023-01-25:
Wie soll man bitte mit den rrdtool 1.8 ohne glib2 baue? 
Und ein Fehler bild wurde richtig gepostet.

leo-022, 2023-01-25:
Wenn ich `rrdtool graph` direkt aufrufe, bekomme ich die Fehlermeldung `Fontconfig error: Cannot load default config file: No such file: (null)`.

rrdtool verwendet fontconfig, siehe rrdbuild.txt. Da die Fritzbox keine großartige Schriftenverwaltung hat, genügt es hier, wenn sich unter `/etc/fonts` eine entsprechende `fonts.conf` befindet:
```
<?xml version='1.0'?>
<!DOCTYPE fontconfig SYSTEM 'fonts.dtd'>
<fontconfig>
  <dir>/usr/share/rrdtool/fonts</dir>
</fontconfig>
```
Man könnte auch ein anderes Fontverzeichnis wählen.

Die Schrift wird jetzt dargestellt, nur keine Sonderzeichen. In der `/usr/lib/cgi-bin/rrdstats/main.cgi` müssen folgende Einträge angepasst werden:
```
NBSP="$(echo -e '\240')"
AUML="$(echo -e '\344')"
GRAD="$(echo -ne '\260')"
```

PS. Das Linux mit der zu alten Meson-Version war Ubuntu 20.04, Ubuntu 22.04.1 LTS installiert die Meson-Version 0.61.2

fda77, 2023-01-25:
Prima dass du das gefunden hast.
Leider bringen die Unicode mit dem alten rrdtool 1.2 komische Zeichen und man muss das unterscheiden.
Freetz hat mittlerweile ein eigenes meson 1.0 dabei, so ist die Version vom System egal

Welches Package von Freetz soll die "fonts.conf " denn erstellen? Dieses: https://github.com/Freetz-NG/freetz-ng/tree/master/make/libs/fontconfig ? Da könnte man ein files/root/etc/fonts/ Unterverzeichnis dazumachen.

Wobei es seltsam ist dass rrdtool eine Font fest mit dabei hat: https://github.com/Freetz-NG/freetz-ng/tree/master/make/pkgs/rrdtool/files/root/usr/share/rrdtool/fonts/
Stattdessen könnte man ein Fontpackage aufnehmen, zb https://dejavu-fonts.github.io/  mit
```
./ttf/DejaVuSans.ttf
./ttf/DejaVuSansMono.ttf
./ttf/DejaVuSans-Bold.ttf
./ttf/DejaVuSansMono-BoldOblique.ttf
./ttf/DejaVuSansCondensed-Bold.ttf
./ttf/DejaVuSansMono-Oblique.ttf
./ttf/DejaVuSansCondensed-Oblique.ttf
./ttf/DejaVuSansCondensed-BoldOblique.ttf
./ttf/DejaVuSans-ExtraLight.ttf
./ttf/DejaVuSansMono-Bold.ttf
./ttf/DejaVuSans-Oblique.ttf
./ttf/DejaVuSans-BoldOblique.ttf
./ttf/DejaVuSansCondensed.ttf
```

leo-022, 2023-01-26:
Scheinbar hat freetz-rrdtool eine Font-Datei dabei, weil es in der Box keine Schriftverwaltung mit eigenen Schriftarten gibt.
Da es sich ja im Moment nur um eine Schrift handelt, muss eigentlich kein Package die `fonts.conf` erstellen. Man kann das statisch machen. Ich hatte einfach nur diesen Inhalt nach `/etc/fonts/fonts.conf` geschrieben, somit wird auf die Schrift verwiesen, die rrdtool dabei hat:
```
<?xml version='1.0'?>
<!DOCTYPE fontconfig SYSTEM 'fonts.dtd'>
<fontconfig>
  <dir>/usr/share/rrdtool/fonts</dir>
  <cachedir>/var/cache/fontconfig</cachedir>
  <cachedir prefix="xdg">fontconfig</cachedir>
</fontconfig>
```

fda77, 2023-01-26:
Es gibt halt nur 1 Font um nicht viel Platz im Flash zu belegen
Die Font die Freetz dabei hat wurde wohl vor 17 Jahren umbenannt: https://linux.debian.bugs.dist.narkive.com/jkSLjdrH/bug-360293-dejavusansmono-roman-ttf-renamed-to-dejavusansmono-ttf Die 1.8 erkennt sie noch unter dem alten Namen?
Das schlechte an so einem Blob ist auch dass die Sache mit der Lizenz kompliziert ist. Ein eigenes Package ist einfacher
Sehen die 1.8 Grafiken überhaupt irgendwie "besser" aus??

fda77, 2023-01-26:
Es gibt jetzt das neue dejavu-fonts-ttf Package das von rrdstats 1.8 ausgewählt wird. In diesem ist 1 Font ausgewählt. Im menuconfig kann man noch einige andere auswählen, ka ob rrdstats die automatisch nutzen würde.
Die fonts.conf wird von fontconfig erstellt, leicht angepasst wie ich sie in einem normalen Linux gefunden hab. Dort ist der Pfad /usr/share/fonts drin, das Unterverzeichnis /usr/share/fonts/dejavu-sans-fonts" von dejavu-fonts-ttf müsste eigentlich gefunden werden

leo-022, 2023-01-26:
Einen Fehler gab es noch, im Webinterface wurde "Â°C" angezeigt. Ich habe das berichtigt, siehe 32b54d696f1f87fdacce51569c075d47ea8bf0f3. Gleichzeitig habe ich alle Darstellungen angepassr, so dass die Grafiken jetzt einheitlicher aussehen.

Die Darstellung der png hat sich m.E. nicht groß geändert. Jetzt funktioniert aber auch svg, und das sieht schon besser aus. Vielleicht kann man das irgendwann umschaltbar machen.
[image]

fda77, 2023-01-27:
svg funktioniert weil ich das in cairo angeschaltet gelassen hatte: https://github.com/Freetz-NG/freetz-ng/blob/master/make/libs/cairo/cairo.mk#L52 Das kann man zb super vergrössern. Per default wären auch noch pdf, ps, xml aktiviert, bei denen wusste ich aber nicht was das bringen soll.
Vielleicht gibt es da noch andere nützliche Optionen in den ganzen Libs? Manche könnten aber nochmal weitere libs brauchen.

Dh für die 1.8 könnte man immer svg verwenden, zb in dieses if: https://github.com/Freetz-NG/freetz-ng/blob/master/make/pkgs/rrdstats/files/root/usr/lib/cgi-bin/rrdstats/main.cgi#L56 eine Variable mit je nach Version svg oder png. Ich habs es aber immer noch nicht geschafft die 1.8 auf meine Fritzbox zu packen...

Bei Openwrt wird aktuell auch überlegt ob auf 1.8 von der alten 1.0 aktualisiert werden soll
 - https://forum.openwrt.org/t/139473
 - https://github.com/openwrt/luci/issues/6139

fda77, 2023-01-27:
Mit 368979043bd4f9900789c46c58ff38d2260cc1c9 kann man einfacher experimentieren. Bei svg scheinen die Grafiken eine beliebige Grösse zu haben, da die Font grösser ist.
Die rrdtool binary heisst nun "rrdtool-freetz", worauf ein link "rrdtool" verweist, solang es dies nicht als binary von avm gibt: 2a47ada61a9dd47929dcb0a57869f5933f1d7aa1 + d08be5530
So kann man auch das kleine rrdtool 1.2 mit ganz neuem fos verwenden

leo-022, 2023-01-27:
Mit der Größe experimentiere ich auch gerade. WIDTH und HEIGHT bewirken bei svg schon etwas, nur scheinen die dort zugrundegelegten Pixel größer zu sein. Bei Verkleinerung der Werte ist die Schrift zu breit und es erfolgen Zeilenumbrüche. Mal sehen, wie es mit einer kleineren Schriftgröße ist.

Wo kommt überhaupt das `_cgi_width` her, auf dem die Grafik-Größe aufbaut?

fda77, 2023-01-27:
Das ist die konfigurierbare Breite ders Webif: 
[image]

[image]

leo-022, 2023-01-27:
Mit `let WIDTH=_cgi_width-290` und `--font LEGEND:7` sieht es gut aus.

[image]

fda77, 2023-01-27:
Das was abgezogen wird ist das Menü, es hat eine feste Breite. Wenn man rrdstats auf einem eigenen Port (nicht über Status>RRDstats) aufruft ist das Menu nicht sichtbar und die Grafiken breiter
Diese Breite sind "Pixel" da gibt es eigentlich keine unterschiedliche Auffassung. Vielleicht sollte man den svg im html-Code sagen wie gross sie sein sollen?

leo-022, 2023-01-27:
> Vielleicht sollte man den svg im html-Code sagen wie gross sie sein sollen?

Das wäre vielleicht am besten.

fda77, 2023-01-27:
https://www.w3schools.com/tags/att_img_width.asp

fda77, 2023-01-27:
Ne, dieser feste Wert der Abgezogen wird ist gar nicht das Menü... könnte der rand um das Bild sein.
Wenn man im html-img tag width= _cgi_width - 50 angibt passt es ganz gut

leo-022, 2023-01-27:
Bei `_cgi_width` gar nichts abziehen und bei `RRDTOOL graph` eine feste Größe vorgeben, z. B. `--width 500 --height 250`.
Und im HTML nur `width=\"$WIDTH\"` ung HEIGHT weglassen, das berechnet sich automatisch.
So sollte es gut aussehen.

fda77, 2023-01-27:
565? Das sind doch die 710-145 wie es bisher war Für ein SVG bei einem Rand von 20 wären es 665.
Das Verhältnis 5:2 kann man konfigurieren. Ein SVG scheint sich das zu merken und macht die Höhe entsprechend

leo-022, 2023-01-27:
Ja, das ist wie bisher. Das scheint vom Aussehen her das beste zu sein, wird aber ohne das im HTML-Code zu skalieren zu breit. Wenn man die Werte weiter erhöht, wird die Schrift wieder zu klein, die wäre dann aber richtig, wenn man das svg im Browser z. B. mit Rechtsklick öffnet, da muss man nicht extra vergrößern.

leo-022, 2023-01-27:
Was ist denn das `--disable-rrdtool-tag`?
Man kann in der alten Version auch `--imgformat PNG` verwenden.

fda77, 2023-01-27:
Schau mal wie das bei dir passt .. oder auch nicht.
Ich denke das ganze Problem ist dass alle Bildformate eine Grösse (in Pixel) haben, ausser SVG (die kann man mit dem Texteditor erstellen). Und deshalb muss es in den IMG-tag, ein bisschen kleiner als der festgelegte "Hauptinhalt"

fda77, 2023-01-27:
"disable-rrdtool-tag" ist das Tobias Irgendwer oben rechts, in der 1.2 hat es das noch nicht gegeben
"imgformat" theoretisch ja, ich hab aber kein freetz mit der 1.2 um das zu testen. Das es bislang so ging hab ich es einfach gelassen

leo-022, 2023-01-27:
Wenn ich deine neue Version einfach so auf der laufenden Box nutze, zeigt es gar keine Grafiken mehr an. Muss erstmal schauen, woran das liegt.

fda77, 2023-01-27:
Mir kommen die Bilder auch "irgendwie" anders vor... Vielleicht liegts daran dass rrdtool nun 15 Jahre neuer ist. Ich glaube auch das Padding der bilder selbst ist kleiner und der Text näher am Rand, das sieht man bei dunklem Skin gut
1.8: [image]
1.2: [image]

Daran wird man sich gewöhnen...
Übrigens klappt es nicht dass /etc/fonts/fonts.conf auf die Box kopiert wird! Das sollte die lib make/libs/fontconfig/ machen, aber scheinbar gibt es in freetz dieses system mit "files" nicht für libs
Aber die fonts werden dennoch gefunden, da das Default-Verzeichnis /usr/share/fonts/ genutzt wird ^^

leo-022, 2023-01-27:
Hier noch einmal ein Vergleich alt/neu.
1.2 png:[image]
1.8 svg:[image]

Von den Probortionen her identisch. Bei svg ist die Qualität naturgemäß besser. Das "RRDTOOL / TOBI OETIKER" am rechten Rand war in der alten Version vorhanden, in meiner Erinnerung war das auch immer so. Als Tribut an den Entwickler von rrdtool könnte man das auch stehen lassen.

Ich denke, das Projekt Umstellung auf svg ist damit erfolgreich abgeschlossen.

fda77, 2023-01-27:
Die svg sehen schon besser aus, auf deinen Bildern sieht man das an der Beschriftung.
Das "rrdtool-tag" kann man halt erst seit 15 Jahren deaktivieren, mit der 1.2 gab es die Option nicht. Am besten macht man in Freetz auch ne Option, dann kann es jeder selbst einstellen. Bei anderen Projekten wie Cacti ist es auch aus.
Man könnte auch noch einen dark-mode der Graphen einbauen
Offen ist auch noch die fehlende fonts.cfg

fda77, 2023-01-28:
f2748b9ae3759597437b0e2c3051c0c4c1d82b26 & 2b3c779ba786e97c833abafcd20021affaa97316

closed 2023-01-28 by fda77

leo-022, 2023-01-28:
Noch 2 Fehler auf der Konfigurationsseite RRDstats:
* `--disable-rrdtool-tag` wirkt auch in Version 1.2, default wird dort ebenfalls die Info am rechten Rand angezeigt.
* "Zeige den Namen des Authors von rrdtool auf allen Graphen" - ein "h" bei Autor zuviel.

fda77, 2023-01-29:
von https://github.com/Cacti/cacti/issues/2205

Although I've haven't found a hint within RRDtool's Changelog, graph option "--disable-rrdtool-tag" will be supported by RRDtool >= 1.5.

Such doch die Datei mit dem Schreibfehler unterhalb https://github.com/Freetz-NG/freetz-ng/tree/master/make/pkgs/rrdstats und dann oben rechts auf den Stift zum editieren

leo-022, 2023-01-29:
> Mit "wirkt" meinst du dass man damit dieses Tag entfernen kann? In 1.2?

Genau, ich habe es ausprobiert.

fda77, 2023-01-29:
Das ist das gleiche Prinzip mit dem Stift... Diesmal halt das "if" mit dem "ABANDON" in der rrdstats.cgi um "tobitag" wegmachen, damit es für alle sichtbar ist

fda77, 2023-01-29:
Sieht ganz nett aus mit
 showframe aus
 darkmode an
 tobitag aus
Nur die Defaultbreite von 1004 der zusätzlichen Webserver ist vielleicht ein bisschen viel

[image]

fda77, 2023-01-29:
Nur die Ausrichtung von SmartHome passt nicht mehr so ganz:
[image]

leo-022, 2023-01-29:
Das liegt an der Formatierung mit Tab, die sieht bei png und svg unterschiedlich aus. Vielleicht hängt das auch mit den unterschiedlichen Pixelgrößen zusammen. Da bin ich noch nicht ganz dahinter gestiegen, ich habe dazu bisher im Netz auch noch nichts gefunden. Mit einer proportionalen Schriftart bekomme ich gar kein System in die Abstände. Vielleicht sollte man das in der Monospace-Schrift generell mit Leerzeichen formatieren.

fda77, 2023-01-29:
Ich hab halt das Pech genau an der Tabgrenze zu sein. Daher am besten ne minimale Zeichenzahl und dann tabs drüber. Ich hab da schon ne Idee...

fda77, 2023-01-29:
Mit len15() hab ich die String mit Leerzeichen bis auf 15 aufgefüllt, und den Header min/avg/max/cur oben drüber. Der ist aber nicht 100% gut ausgerichtet, dafür müsste man eine andere Schriftart nehmen. Am alten rrdtool 1.2 will ich aber eigentlich nicht mehr ändern  c643269cc