Freetz-NG/freetz-ng #711: rc.rextd: syntax error in line 28 and line 42

PeterPawn, 2023-01-25:
https://github.com/Freetz-NG/freetz-ng/blob/c37854e759738c0b164cc966041604d35bd93ee8/make/pkgs/mod/files/root/etc/init.d/rc.rextd#L28

Wrong redirection attempt and after fixing this error, `shellcheck` finds further problems:

```
vidar:/tmp $ shellcheck -a rc.rextd

In rc.rextd line 13:
start() {
        ^-- SC1009 (info): The mentioned syntax error was in this brace group.

In rc.rextd line 36:
        if modlib_check_supervisor; then
        ^-- SC1046 (error): Couldn't find 'fi' for this 'if'.
        ^-- SC1073 (error): Couldn't parse this if expression. Fix to allow more checks.

In rc.rextd line 42:
        else
        ^-- SC1047 (error): Expected 'fi' matching previously mentioned 'if'.
            ^-- SC1072 (error): Unexpected keyword/token. Fix any mentioned problems and try again.

For more information:
  https://www.shellcheck.net/wiki/SC1046 -- Couldn't find 'fi' for this 'if'.
  https://www.shellcheck.net/wiki/SC1047 -- Expected 'fi' matching previously...
  https://www.shellcheck.net/wiki/SC1072 -- Unexpected keyword/token. Fix any...
vidar:/tmp $
```

fda77, 2023-01-25:
Ah, ein kleiner Typo. Wer hätte das nach den obskuren Berichten im Forum erahnt.
Hast du bemerkt dass multid sich anders verhält (nicht forkt) je nachdem von wem er gestartet wird? 

shellsheck ist ganz nett, kennt aber vieles nicht, zb
```
[ $var -eq 123 2>/dev/null ]
```
Wird nicht akzeptiert, das `2>/dev/null`  muss weg. Wenn `var` keine Zahl ist würde man dann aber einen Fehler sehen

Auch wird sourcen mit Parameter bemängelt:  `. /abc xxx` - das klappt aber mit der Busybox

closed 2023-01-25 by fda77

PeterPawn, 2023-01-26:
> Wird nicht akzeptiert, das 2>/dev/null muss weg.

Das ist ja auch ein Syntaxfehler. Diese Notation des `test`-Kommandos endet erst an der schließenden eckigen Klammer und eventuelle Redirections gehören DAHINTER. Oder man nimmt gleich die andere Form (also das Kommando `test`), da ist der richtige Platz für eine Redirection eindeutig. Und wenn man's dann noch "richtig sicher" machen will, berücksichtigt man auch noch die Hinweise zur Verwendung von `-a` und `-o` in solchen Konstrukten (z.B. hier: https://pubs.opengroup.org/onlinepubs/9699919799/utilities/test.html im Abschnitt "Application Usage"). Die "fetten" eckigen Klammern in der Spezifikation gehören zur Darstellung der Syntax des Kommandos - entscheidend sind die "äußeren" bei der Notation mit der Klammer.

Jedenfalls "versteht" das dann auch wieder `shellcheck` korrekt und auch die BusyBox-Shell führt das korrekt aus:

```
peh@vidar:~/freetz-ng> printf "#! /bin/sh\nvar="ABC"\ntest "\$var" -eq 123\n" | busybox sh
sh: ABC: bad number
peh@vidar:~/freetz-ng> printf "#! /bin/sh\nvar="ABC"\ntest "\$var" -eq 123 2>/dev/null\n" | busybox sh
peh@vidar:~/freetz-ng> printf "#! /bin/sh\nvar="ABC"\ntest "\$var" -eq 123 2>/dev/null\n" | shellcheck -a -
peh@vidar:~/freetz-ng> (keine Fehlermeldungen von `shellcheck`)
---> oder mit den eckigen Klammern formuliert:
peh@vidar:~/freetz-ng> printf "#! /bin/sh\nvar="ABC"\n[ "\$var" -eq 123 ] && echo yes || echo no\n" | busybox sh
sh: ABC: bad number
no
peh@vidar:~/freetz-ng> printf "#! /bin/sh\nvar="ABC"\n[ "\$var" -eq 123 ] 2>/dev/null && echo yes || echo no\n" | busybox sh
no
peh@vidar:~/freetz-ng> printf "#! /bin/sh\nvar="ABC"\n[ "\$var" -eq 123 2>/dev/null ] && echo yes || echo no\n" | shellcheck -a -

In - line 3:
[ $var -eq 123 2>/dev/null ] && echo yes || echo no
^-- SC1073 (error): Couldn't parse this test expression. Fix to allow more checks.
               ^-- SC1072 (error): Expected test to end here (don't wrap commands in []/[[]]). Fix any mentioned problems and try again.

For more information:
  https://www.shellcheck.net/wiki/SC1072 -- Expected test to end here (don't ...
  https://www.shellcheck.net/wiki/SC1073 -- Couldn't parse this test expressi...

peh@vidar:~/freetz-ng> printf "#! /bin/sh\nvar="ABC"\n[ "\$var" -eq 123 ] 2>/dev/null && echo yes || echo no\n" | shellcheck -a -
peh@vidar:~/freetz-ng> (auch hier keine Fehlermeldungen von `shellcheck` mehr, wenn die Redirection HINTER der schließenden Klammer erfolgt)
```

Laß Dich nicht von den Escape-Zeichen vor dem Variablennamen irritieren - die sind nur der Tatsache geschuldet, daß diese Eingaben direkt in einer Shell erfolgen und ohne das Escaping eine Auflösung versucht würde.

Das IST also immer noch ein Syntaxfehler und wenn eine Shell das kommentarlos schluckt, dann liegt das nur daran, daß sie dieses Konstrukt falsch zerlegt.

fda77, 2023-01-26:
Achso! Dann hab ich das Redirect aber konsequent schon immer und überall falsch IN die Klammer geschrieben.
Wobei beide Varianten auf allen Systemen die ich kurz getestet hab gleich funktionierten und sich gleich verhielten.

Das mit dem teilweise undefinierten `-a` und `-o` ist bekannt, für Freetz ist mir das aber zu aufwändig.

Witzig dass auf den Screenshots auf https://github.com/koalaman/shellcheck#readme auch ein vidar eingeloggt ist