Freetz-NG/freetz-ng #718: OpenVPN 2.6 - falscher Pfad bei Verwendung von BusyBox "IP"

feedzapper, 2023-01-28:
Bei setzen von ...
Use 'ip' instead of 'ifconfig' and 'route'  unter Package "OpenVPN"

OpenVPN 2.6 sucht unter /usr/sbin/ip

fda77, 2023-01-28:
Das ist erst seit der neuen 2.6 so? Und ging vorher?
Wie kommtst du überhaupt auf die Idee dass es nich geht? 
Ich seh
```
checking for ifconfig... (cached) /sbin/ifconfig
checking for route... (cached) /sbin/route
checking for ip... (cached) /sbin/ip
```

fda77, 2023-01-28:
durch diesen comit entfallen die "cached", sollte aber ansonsten gleich sein

feedzapper, 2023-01-28:
genau - seit OpenVPN 2.6.
Habe es extra heute Morgen "NEU" ausgecheckt.

`Jan  1 01:01:03 fritz daemon.notice openvpn[5774]: OpenVPN 2.6.0 mips-unknown-linux-gnu [SSL (OpenSSL)] [EPOLL] [MH/PKTINFO] [AEAD]
Jan  1 01:01:03 fritz daemon.notice openvpn[5774]: library versions: OpenSSL 1.1.1s  1 Nov 2022
Jan  1 01:01:03 fritz kern.notice FREETZ: [RCMOD] Starting OpenVPN ... done.`
........
........
`Jan 28 12:05:34 fritz daemon.notice openvpn[5828]: OPTIONS IMPORT: --sndbuf/--rcvbuf options modified
Jan 28 12:05:34 fritz daemon.notice openvpn[5828]: Socket Buffers: R=[237568->475136] S=[237568->475136]
Jan 28 12:05:34 fritz daemon.notice openvpn[5828]: OPTIONS IMPORT: --ifconfig/up options modified
Jan 28 12:05:34 fritz daemon.notice openvpn[5828]: OPTIONS IMPORT: route options modified
Jan 28 12:05:34 fritz daemon.notice openvpn[5828]: OPTIONS IMPORT: route-related options modified
Jan 28 12:05:34 fritz daemon.notice openvpn[5828]: OPTIONS IMPORT: --ip-win32 and/or --dhcp-option options modified
Jan 28 12:05:34 fritz daemon.notice openvpn[5828]: OPTIONS IMPORT: peer-id set
Jan 28 12:05:34 fritz daemon.notice openvpn[5828]: OPTIONS IMPORT: data channel crypto options modified
Jan 28 12:05:34 fritz daemon.notice openvpn[5828]: TUN/TAP device tun0 opened
Jan 28 12:05:34 fritz daemon.notice openvpn[5828]: TUN/TAP TX queue length set to 1000
Jan 28 12:05:34 fritz daemon.notice openvpn[5828]: /usr/sbin/ip link set dev tun0 up mtu 1420
Jan 28 12:05:34 fritz daemon.err openvpn[5828]: Linux ip link set failed: could not execute external program
Jan 28 12:05:34 fritz daemon.notice openvpn[5828]: Exiting due to fatal error`

fda77, 2023-01-28:
Waschreinlich ist du ne kaputt config von irgend wo her imortiert, der Pfad ist ja richtig gesetzt beim compilieren

closed 2023-01-28 by fda77

feedzapper, 2023-01-28:
Also daran liegt es definitiv NICHT .
Der Server macht nur einen PUSH "route 192.x.x.x xxxxx etc." 
Die BIN werden ja mit dem Path auf dem Client aufgerufen (hier die 7590 mit OpenVPN 2.6)
also mit Freetz-NG 20998 und der 2.5.8 hat er noch /sbin/ip benutzt
Ist aber nun auch egal, ich habe mir eine kopie von "IP" nach /usr/sbin kopiert 
WOW und die performance mit 2.6 ist deutlich besser - komme nun über 30 mbit mit iperf und das obwohl der Server noch mit 
2.5.8 läuft auf einem Netgear R7000 
Kann natülich sein, das openvpn und inetd noch performance gefressen haben ?! - was Du ja nun auch beseitigt hast ...
anyway ,-)
läuft :-)

fda77, 2023-01-28:
> was Du ja nun auch beseitigt hast ...

wie "ich" hab das "auch" beseitigt. Du warst wohl zu lahm im DEBen forum. Wird freetz nur von Idioten verwendet? Wie wärs wenn du mal openvpn anrufst und denen die Ohren vollnölst?