Freetz-NG/freetz-ng #727: iptables "static" Option broken bei 7590 und weiteren Modellen & iptables-Kernel Module Options fuer FOS 7.50

maverick3005, 2023-02-02:
Beim Versuch iptables in der "Default" config zu bauen ist die Vorgabe "static binary" broken.

Beim Bau eines Images für die 7590 ist ein erfolgreicher Image Bau nur noch als shared Module moeglich.

Das Image wurde fuer eine 7590 mit Version FOS 7.29 gebaut.

Beim Bau eines Images fuer FOS 7.50 ist es nach wie vor erforderlich, das "make kernelmenuconfig" extra noch genutzt werden muss,
damit die "fehlenden" Module zum verlinken auch noch vorhanden sind. (Ist wie gesagt bei 7.29 nicht erforderlich!)

Am Ende des Imagebaus erscheinen dann in "gelb" die Hinweise auf die fehlenden Kernelmodule mit dem Hinweis die Developer zu informieren - was ich hiermit mache.

Wäre eine Angleichung 7.29/7.50 bei iptables moeglich, das dieser extra Schritt beim bereits vorhandenen 7.50er Kernel Source so wie bei der 7.29 hier nicht mehr manuell noetig ist jedesmal ? - somit die entsprechenden noetigen Kernel-Module ueber das "make menuconfig" auch bei der Kernel-config wie bisher mit ausgewählt werden ?

Ein freundliches "einmal drüber schaun" ohne ein lappidares "invalid" oder "closed" wäre hier wirklich nett ;-.)

fda77, 2023-02-02:
Die letzte grössere Änderung an iptables ist über 2 Jahre her: https://github.com/freetz-ng/freetz-ng/commit/05a098bea

maverick3005, 2023-02-02:
Hier der Unterschied zwischen 7.29 + 7.50 kernels:

```
WARNING: File of kernel module not found: ip6_tables.ko
WARNING: File of kernel module not found: ip6table_filter.ko
WARNING: File of kernel module not found: ip_tables.ko
WARNING: File of kernel module not found: iptable_filter.ko
WARNING: File of kernel module not found: nf_log_common.ko
WARNING: File of kernel module not found: nf_log_ipv4.ko
WARNING: File of kernel module not found: nf_log_ipv6.ko
WARNING: File of kernel module not found: x_tables.ko
WARNING: File of kernel module not found: xt_LOG.ko
WARNING: File of kernel module not found: xt_iprange.ko
WARNING: File of kernel module not found: xt_multiport.ko
WARNING: File of kernel module not found: xt_tcpudp.ko
ERROR: Enable missing with 'make kernel-menuconfig' as (M)odule (and push a pull request). Use '/' to search in menuconfig.

```

Evtl. koenntest Du Dich doch einfach doch "erwärmen", die Unterschiede zu den kernel-Sourcen von 7.29 zu 7.50
aktuell wieder mit in Sync zu bringen das bei der aktuellen 7.50er die bald auf allen Geräten Standard ist, diese extra config Schritte nicht jedes mal immer noch extra erfolgen muessen ? (...mal lieb nachgefragt im Sinne vieler User....)

maverick3005, 2023-02-02:
Da das "static" iptables compilen nebenbei auch broken ist "out of the box" und "static" nach wie vor die Vorgabe ist, wäre
es evtl. auch sinnvoll den Default auch gleich mit auf "shared" Module zu setzen ?

fda77, 2023-02-02:
Nein, auf keinen Fall. Die Module aktiviert ihr schön selbst. Wie das geht hast du ja mit gepostet.
Es gibt ca 75 verschieden Kernel und jeder Otto braucht ein anderes Modul. Da hätt ich nichts anderes zu tun.
Wenn dir das zu stressig ist, bleib halt bei der alten Version. Oder such dir jemanden der es für dich macht

closed 2023-02-02 by fda77