Freetz-NG/freetz-ng #740: Fehlerhafte Images fuer 7490 / 7590 mit aktuellem Trunk

maverick3005, 2023-02-20:
Ich habe nach einiger Zeit mal 2 frische minimal netinstall-Images für ubuntu 20.04 latest + debian 11 latest mal neu aufgesetzt
und festgestellt das keine validen Images mehr zum flashen erstellt werden. Das Problem kann ich mit pushfirmware
und freetz-flashing nachstellen.

Zuerst einmal noch ein Winweis zu fehlenden Dependencies aus der PREREQUISITES-Docu zu debian11:
Bei einem installierten minimal System fehlen hier noch: "zip (unzip) python patchutils ftp und ncftp (für pushfirmware)",
für ältere Tags fehlt dann noch execstack (genommen vom debian 10 repo für Tag 22100/zum testen.

Ansonsten beide Enviroments entsprechen der aktuellen Doku der Requirements-Liste für debian11 + ubuntu 20.04

Auf beiden Systemen tritt direkt nach dem make immer folgendes Problem auf:

```
[15137.389723] conftest[1740129]: segfault at 0 ip 00007f0611cd0ba7 sp 00007ffdb8472308 error 4 in libc-2.31.so[7f0611c4e000+15a000]
[15137.389936] Code: 00 00 00 48 39 d1 0f 82 87 63 06 00 0f 1f 80 00 00 00 00 48 89 f8 48 83 fa 10 0f 82 fa 00 00 00 48 83 fa 20 0f 87 3f 01 00 00 <0f> 10 06 0f 10 4c 16 f0 0f 11 07 0f 11 4c 17 f0 c3 0f 1f 84 00 00

```
Wieso entsteht hier auf beiden Systemen direkt nach dem make ein crash im Hintergrund ?

Ich habe die Problematik einige Schritte rückwaerts verfolgt und die letzte funktioniernde Version bei mir ist jetzt
TAG "67657a58"

TAG: "09fb889" erzeugt bereits keine bootfähigen Images mehr. Eine serielle Console für weitere Infos müsste ich erst
noch anschliessen damit ich da mehr logfiles habe, bei der 7490 hängt der boot (wahrschlich Kernel-Halt), bei der 7590
entsteht ein bootloop.

Das Ergebnis ist reproduzierbar für die 7490 und die 7590, tritt jedoch nicht bei einer 5590 auf, hier kann der aktuelle Trunk
weiterhin verwendet werden und es werden valide Images erstellt. Offenbar tritt der Fehler bei FIT-Boxen nicht auf?

Meine Vermutung wäre, das evtl. eine Dependancy noch fehlt oder ein library/dev welches noch nicht dokumentiert ist ?

Was ich nicht aktiv habe ist "Freetz-Mount/Udev-Mount", was bislang auch nie ein Problem war.
Die Toolchain + Hosttools erstelle ich jeweils selber. Uboot wird nicht statisch mit openssl erstellt.

Meine Vermutung liegt im Bereich patchelf/bump patchelf-host (gcc5/gcc7)/install strip 2.31-Fixes
Irgendwie muss sich ich hier ein Problem eingeschlichen haben ?

Meine test-config habe ich hier mal von der 7490 angehängt:

[config.txt](https://github.com/Freetz-NG/freetz-ng/files/10778348/config.txt)

maverick3005, 2023-02-20:
Was mir auch noch aufgefallen ist:

im /tmp Ordner verbleiben nach jedem Compilat Unmengen an temp. Muell welcher nicht gecleaned wird.
Ferner habe ich die tests auch mit ccache und auch ohne durchgeführt.

fda77, 2023-02-20:
Für debian 11 diese Datei editieren: https://github.com/Freetz-NG/freetz-ng/blob/master/docs/PREREQUISITES/packages/Debian11 

> Wieso entsteht hier auf beiden Systemen direkt nach dem make ein crash im Hintergrund ?

5. Gebot nochmal durchlesen: https://github.com/Freetz-NG/freetz-ng/blob/master/.github/ISSUE_TEMPLATE/1-bug_report.md

> Das Ergebnis ist reproduzierbar für die 7490

Meine 7490 läuft mit 21235-@cc99bd755 seit 1 woche

> Meine Vermutung wäre, das evtl. eine Dependancy noch fehlt oder ein library/dev welches noch nicht dokumentiert ist ?

Anhand welche Meldung vermutest du das?

> Die Toolchain [..] erstelle ich jeweils selber. 

wie jeder andere auch: https://freetz-ng.github.io/freetz-ng/NEWS.html#2022-08-24

Das herumgejammere mit den Toolchains könntest du einstellen, es gibt seit nem halben Jahr (wegen Leuten wie dir) keine mehr. Scheint aber ein Reflex zu sein. Was muss ich noch machen damit sich das herumspricht?

> Uboot wird nicht statisch mit openssl erstellt.

Uboot ist für FIT ... oben hiess es 7490+7590

> Meine Vermutung liegt im Bereich patchelf/bump patchelf-host (gcc5/gcc7)/install strip 2.31-Fixes

Warum?

> im /tmp Ordner verbleiben nach jedem Compilat Unmengen an temp. Muell welcher nicht gecleaned wird.

Bei mir ist nix

> Ferner habe ich die tests auch mit ccache und auch ohne durchgeführt.

Gibt es irgend welche bekannten Probleme mit ccache? Ich habs immer an, auch für nicht-freetz

Ich würde auf etwas externes vermuten, zb imatisch...
Oder was mit pseudo/fakeroot: https://github.com/Freetz-NG/freetz-ng/discussions/731

maverick3005, 2023-02-20:
Zur dem:
"Meine Vermutung liegt im Bereich patchelf/bump patchelf-host (gcc5/gcc7)/install strip 2.31-Fixes"

TAG "67657a58" ist noch "heile", Ab dem TAG: "09fb889" werden "invalide" images für 7490+7590 gebaut.
Irgendwo in diesem Bereich (ich vermute die patchelf-Ursache könnte das Problem sein?)

Ich muss erst eine serielle Schnittstelle anbauen um genau zu sehen was nach einem Flash dann tatächlich
die Ursache ist.

Die Umschaltung auf fakeroot habe ich noch nicht durch, da ich für einen Fehlerreport erst mal weitestgehend
die commits einkreisen wollte, ab dem die Probleme dann auftreten.

Wieso bei ubuntu 20+debian11 dann jeweils immer jede Menge Muell im /tmp Verzeichnis stehen bleiben,
ferner die conftest-Crashmeldung ständig bei einem make entsteht kann ich auch nur mitteilen.

Daher war ja auch meine Frage, ob eine mir bislang unbekannte System-Dependancie auf einem "minimal"-Install-Image
+ Dependencies nach Liste evtl. noch fehlen, welche jedoch als "Fehler" oder "fehlende" Dependancy aktuell nicht
bemeckert wird beim compilen.

fda77, 2023-02-20:
Ich hab an meiner 7490 ne Console, aber die läuft ja :D
Du könntest dir mal Syslog über Netzwerk einrichten, das erspart mit Glück eine Console

Daher kann ich deinen patchelf-Hokuspukus nicht nachvollziehen. Wäre das auch bei anderen ein Problem, würde man das bestimmt schon gehört haben. Selbst als die (nicht so verbreitete) 7583 nicht sofort aktualisiert wurde gab es überraschenderweise direkt Beschwerden. Daher tippe ich auf irgend was in deinem Script. Führst du eigentlich "make olddefconfig" nach deiner sed-Orgie aus?
Solang es nur 1 Reporter 1es Problems gibt liegt es oft an diesem selbst

Schau halt welche Dateien in tmp sind - und dann wer diese erstellt

maverick3005, 2023-02-20:
Interessant ist jedenfalls der conftest dauercrash der im Anschluss ans compilat im dmesg mit obriger Fehlermeldung
jedes mal entsteht. Die Diskussion mit dem pseuso/fakeroot habe ich mitbekommen. Extra Tests stehen hier noch aus.

Ich habe bei meinen Tests jeweils vollständig (ohne eine bereits vorhandene .config) erstmalig angestart und dann
eine komplette initiale .config erstmal erstellt.

syslog ? Dazu büsste die Box erst einmal ueberhaupt booten. Ich bau eine serielle Console dran und schau
was der Kernel in der Bootphase dann meldet und wie weit das Ganze dann noch kommt. :-)

Ansonsten beschreibe ich Dir keinen "Kokuspokus", sondern lediglich 2 Tags, ab wann keine validen Boot-Images mehr produziert werden, damit Du einen Anhaltspunkt hast ab welchen commits die Fehler dann auftreten.

Und ja es kann noch noch einen Unterschied machen, ob Ich ich ein masives "Full-Developement-System" frisch aufsetze und dann die Dependancies für Freetz noch im Anchluss einpflege, oder mit einem absoluten Minimal-Image starte und dann lediglich die reinen Freetz-Dependancies adde. 

Die Frage richtete sich unter anderem auch noch an weitere Teilnehmer des Projekts,
um herauszufinden, welche zusätzlichen Deps ggfs. bei einer absoluten Minimal-Netinstall dann noch extra fehlen "könnten".

...ich bin ja immer noch dabei den exakten commit herauszufinden, ab der dann "nichts mehr geht" - bislang habe ich nur den Bereich erst einmal eingegrenzt bei dem die weiteren commits das Ergebnis "invalid" machen.

fda77, 2023-02-20:
> syslog ? Dazu büsste die Box erst einmal ueberhaupt booten. 

Genau. Hättest du es aber schon vorher gemacht käme vielleicht noch kurz was durch. Man weiss es nicht

>  conftest dauercrash

Ich hab keine Ahnung was das ist oder von wem das aufgerufen wird, hab ich noch nie gehört. Siehe 5tes Gebot ^^

> Die Frage richtete sich unter anderem auch noch an weitere Teilnehmer des Projekts,

Viel Glück, idr kommt da nichts

Für maximale Kompatibilität (und Langsamheit) könntest du das recht neue FREETZ_ANCIENT_SYSTEM aktivieren, dann hast du auch dein altes patchelf, nichts vorcompiliertes, kein ccache usw

maverick3005, 2023-02-20:
Diese Meldung:

```
[15137.389723] conftest[1740129]: segfault at 0 ip 00007f0611cd0ba7 sp 00007ffdb8472308 error 4 in libc-2.31.so[7f0611c4e000+15a000]
[15137.389936] Code: 00 00 00 48 39 d1 0f 82 87 63 06 00 0f 1f 80 00 00 00 00 48 89 f8 48 83 fa 10 0f 82 fa 00 00 00 48 83 fa 20 0f 87 3f 01 00 00 <0f> 10 06 0f 10 4c 16 f0 0f 11 07 0f 11 4c 17 f0 c3 0f 1f 84 00 00

```
erscheint direkt nach dem "make" (wenn unter anderm noch keine .config existiert und man frisch durchstartet nach einem auschecken auf debian11+ubuntu 20.04)

Ich habe unter Anderem noch ein älteres debian10-system, bei den diese Crash-Meldung nicht auftritt zum Vergleich auch noch einmal durchlaufen lassen.

Lass bitte mal das Ticket noch eine Weile offen. Ich liefere da kurzftistig noch logfinfos von einer seriellen Console nach.

"Ancient_System" auf aktuellem System kann ich auch nochmal durchspielen mit dem aktuellem Trunk.

fda77, 2023-02-20:
Wenn man nach conftest segfault  libc sucht findet man zb https://forums.gentoo.org/viewtopic-t-1142799-start-0.html - du sollst die glibc deines System neubauen. Berichte zu diesem Fehler gibt es auch schon 2010 mit libc-2.11 https://marc.info/?l=gentoo-user&m=127244141415669
Keine Ahnung warum das bei deinem Linux so ist, vielleicht sind einfach deshalb die Images kaputt? Was hast du denn mit deinem Linux angestellt? Vielleicht um irgend was anderes zu Compilieren System-Dateien ersetzt oder verschoben?

FREETZ_ANCIENT_SYSTEM  setzt einfach die ganzen Dinge die du oben beschrieben hast

maverick3005, 2023-02-20:
Ich hab nichts weiter gemacht ausser die offizielle debian-11.6.0 .netinstall image genommen, bzw. das ubuntu 20.04 "mini.iso"
und im Anschluss ein textbasiertes minimalsystem mit lediglich openssh-server erstellt. Im Anchluss dann die kompletten dokumentierten dependencies ausgecheckt.
Selbstredend auch apt update / upgrade. Die Systeme sind frisch aufgesetzt und auf dem aktuellen Stand - einzig den ganzen Grafik-Overhead eines typischen 6+ GB überladenen Systems habe ich nicht und arbeite ausschliesslich in der ssh-shell.

Eben um genau solche side-effekte zu vermeiden sind die beiden Test-Systeme minimal komplett firsch aufgesetzt.
Da ist weder etwas "ersetzt" noch selber gebildet oder anderweitig modifiziert.

Danke für die Links. Evtl. ist auch ein "security-update" beim "apt update/upgrade" auf letzten Stand jetzt das Problem...

TWinkeri, 2023-02-20:
Ich habe übrigens genau dassselbe Problem unter Debian 11 "bullseye": keine Images bootet.

maverick3005, 2023-02-20:
...sag ich doch, ist keine Einbilung gewesen bei 2 komplett frischen Systemen. Gehe mal zurück zum commit "67657a58",
dann rennen die Images für 7490, 7590 auch wieder. Auch mit einem alten debian10 werden hier dann wieder valide images erstellt. :-(

maverick3005, 2023-02-20:
Das hier ist uebrigens der Muell im /tmp Folder der nach dem compiling jedes mal stehen bleibt:
```

-rw------- 1 freetz freetz     0 Feb 20 13:02 cc0a9Dba.le
-rw------- 1 freetz freetz     0 Feb 20 13:08 cc0JYLMH.c
-rw------- 1 freetz freetz     0 Feb 20 13:02 cc1c5qDL.c
-rw------- 1 freetz freetz 21187 Feb 20 13:08 cc2Oquai.ld
-rw------- 1 freetz freetz 21187 Feb 20 13:02 cc31tzvo.ld
-rw------- 1 freetz freetz     0 Feb 20 13:02 cc6bLHkW.le
-rw------- 1 freetz freetz     0 Feb 20 13:08 cc8VkJvb.o
-rw------- 1 freetz freetz     0 Feb 20 13:02 ccBWkk8A.o
-rw------- 1 freetz freetz     0 Feb 20 13:02 ccC64HCT.c
-rw------- 1 freetz freetz 21187 Feb 20 13:02 ccCcCWXg.ld
-rw------- 1 freetz freetz     0 Feb 20 13:08 ccCK2ofZ.le
-rw------- 1 freetz freetz     0 Feb 20 13:08 ccD2Sz5A.o
-rw------- 1 freetz freetz     0 Feb 20 13:02 ccDPpPSb.le
-rw------- 1 freetz freetz 21187 Feb 20 13:02 ccg1SyC2.ld
-rw------- 1 freetz freetz     0 Feb 20 13:08 ccGkC23C.c
-rw------- 1 freetz freetz 21187 Feb 20 13:02 cchtnghf.ld
-rw------- 1 freetz freetz 21187 Feb 20 13:08 ccJhUGeF.ld
-rw------- 1 freetz freetz 21187 Feb 20 13:02 ccl0Qbgm.ld
-rw------- 1 freetz freetz     0 Feb 20 13:08 ccl8DEX8.le
-rw------- 1 freetz freetz     0 Feb 20 13:02 ccnczzz9.le
-rw------- 1 freetz freetz     0 Feb 20 13:08 ccnrMJun.o
-rw------- 1 freetz freetz     0 Feb 20 13:08 ccOp5F0T.c
-rw------- 1 freetz freetz     0 Feb 20 13:08 ccpNw8lS.le
-rw------- 1 freetz freetz     0 Feb 20 13:02 ccQJ8krU.c
-rw------- 1 freetz freetz     0 Feb 20 13:02 ccSffD7X.o
-rw------- 1 freetz freetz 21187 Feb 20 13:08 ccsl8qV7.ld
-rw------- 1 freetz freetz 21187 Feb 20 13:02 ccSMPwD5.ld
-rw------- 1 freetz freetz     0 Feb 20 13:02 ccsZiOWy.o
-rw------- 1 freetz freetz     0 Feb 20 13:02 ccT6ereW.c
-rw------- 1 freetz freetz     0 Feb 20 13:02 ccV36p50.o
-rw------- 1 freetz freetz     0 Feb 20 13:02 ccW1EeHU.le
-rw------- 1 freetz freetz     0 Feb 20 13:02 ccW7SjxW.c
-rw------- 1 freetz freetz     0 Feb 20 13:02 ccwtDiRz.o
-rw------- 1 freetz freetz     0 Feb 20 13:02 ccYqf5KN.c
-rw------- 1 freetz freetz     0 Feb 20 13:02 ccyXHu76.le
-rw------- 1 freetz freetz     0 Feb 20 13:02 cczwDbBB.o

```

Der inhalt einer dieser Dateien:

```
cat cc2Oquai.ld

Usage: /home/freetz/freetz-ng/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.42-nptl_kernel-4.9/mips-linux-uclibc/bin/../lib/gcc/mips-linux-uclibc/8.3.0/../../../../mips-linux-uclibc/bin/ld [options] file...
Options:
  -a KEYWORD                  Shared library control for HP/UX compatibility
  -A ARCH, --architecture ARCH
                              Set architecture
  -b TARGET, --format TARGET  Specify target for following input files
  -c FILE, --mri-script FILE  Read MRI format linker script
  -d, -dc, -dp                Force common symbols to be defined
  --dependency-file FILE      Write dependency file
  --force-group-allocation    Force group members out of groups
  -e ADDRESS, --entry ADDRESS Set start address
  -E, --export-dynamic        Export all dynamic symbols
  --no-export-dynamic         Undo the effect of --export-dynamic
  --enable-non-contiguous-regions
                              Enable support of non-contiguous memory regions
  --enable-non-contiguous-regions-warnings
                              Enable warnings when --enable-non-contiguous-regions may cause unexpected behaviour
  -EB                         Link big-endian objects
  -EL                         Link little-endian objects
  -f SHLIB, --auxiliary SHLIB Auxiliary filter for shared object symbol table
  -F SHLIB, --filter SHLIB    Filter for shared object symbol table
  -g                          Ignored
  -G SIZE, --gpsize SIZE      Small data size (if no size, same as --shared)
[... 275 lines cut ...]
                              an ISA mode switch
  --compact-branches          Generate compact branches/jumps for MIPS R6
  --no-compact-branches       Generate delay slot branches/jumps for MIPS R6
elf32ltsmipn32:
  --insn32                    Only generate 32-bit microMIPS instructions
  --no-insn32                 Generate all microMIPS instructions
  --ignore-branch-isa         Accept invalid branch relocations requiring
                              an ISA mode switch
  --no-ignore-branch-isa      Reject invalid branch relocations requiring
                              an ISA mode switch
  --compact-branches          Generate compact branches/jumps for MIPS R6
  --no-compact-branches       Generate delay slot branches/jumps for MIPS R6
elf64ltsmip:
  --insn32                    Only generate 32-bit microMIPS instructions
  --no-insn32                 Generate all microMIPS instructions
  --ignore-branch-isa         Accept invalid branch relocations requiring
                              an ISA mode switch
  --no-ignore-branch-isa      Reject invalid branch relocations requiring
                              an ISA mode switch
  --compact-branches          Generate compact branches/jumps for MIPS R6
  --no-compact-branches       Generate delay slot branches/jumps for MIPS R6

Report bugs to <https://sourceware.org/bugzilla/>

```

maverick3005, 2023-02-20:
@fda77 

Bist Du sicher das hier alles passt ?:

https://github.com/Freetz-NG/freetz-ng/commit/832d680eaa3b4f37e5c63bec65c449cd6e4c10ac

```
-$(PKG)_CONFIGURE_OPTIONS += --prefix=/usr
+$(PKG)_CONDITIONAL_PATCHES+=$(if $(FREETZ_TOOLS_PATCHELF_VERSION_ABANDON),abandon,current)
+$(PKG)_REBUILD_SUBOPTS += FREETZ_TOOLS_PATCHELF_VERSION_ABANDON

+$(PKG)_CONFIGURE_OPTIONS += --prefix=/    <---- ???
```

Nach dem commit ist das Prefix jetzt "/", ich sehe aber keinen weiteren Zusammenbau eines Zielpfades augrund zwei verschiedener patchelf versionen ?

maverick3005, 2023-02-20:
> Wenn man nicht precompiled-tools hat kann man die patchelf version auswählen, bootet das image wenn du die version nimmst die noch mit gcc4 compilierbar ist? ubuntu20 hat aber bestimmt was neueres
> 

Das habe ist selbstverständlich auch durch mit gleichem Ergebnis :-(
Der system-gcc ist version 10.2.1 auf der debian11 (also von den default Dependencies vom freetz)

Irgend einer der nachfolgenden commits sorgt jedenfalls für das Problem mit den invaliden images.
Ich taste mich dann nochmal commitweise bis an den eigentlichen commit heran, wo das Problem dann entsteht.

Dauert aber etwas länger step-by-step.

maverick3005, 2023-02-20:
Beim Durchlauf des compilings sehe ich an verschiedenen stellen beim "ld" auch eine errormeldung mehrfach, wärend das
compiling jedoch weiterhin durchrennt. Ist wohl eher kosmetisch. Sieht dann so auch als ob der aktuelle Linker mit einem
ungültigen älteren Command noch aufgerufen wird an verschiedenen Stellen historisch bedingt.

freetz-user, 2023-02-20:
Nur so zur Information alle MIPS Boxen, die mit patchelf 0.17.2 gebaut werden booten nicht. 
Wurde hier aber schon angesprochen 
https://www.ip-phone-forum.de/threads/aktuell-bootet-kein-7490-und-7590ax-image-mehr-selbst-getestet.314675/
hier gibt es auch schon ein script der euch hilft 
https://www.ip-phone-forum.de/threads/aktuell-bootet-kein-7490-und-7590ax-image-mehr-selbst-getestet.314675/post-2505600
oder auch hier gibt es was für freetz 
https://www.ip-phone-forum.de/threads/mir-sind-bis-jetzt-diese-fehler-bekannte-in-freetz-ng.314827/

maverick3005, 2023-02-20:
Ich habe mich jetzt zum Tag: "a844465f0" vorgearbeitet und "gcc7" setting passt tatächlich nicht. Lasse gerade das ganze nochmal fuer "gcc5" durchlaufen, ansonsten: der Beitrag weiter höher bestätigt das mit den mips boxen.

Das ist wohl auch der Grund wieso es auf der 5590 im aktuellen Trunk diesbzgl. niemals Probleme gab, und das jetzt mit der 7490/7590 und wohl weiteren MIPS doch mehreren Usern aufgefallen ist.

Rückmeldung kommt gleich....

freetz-user, 2023-02-20:
https://github.com/Freetz-NG/freetz-ng/commit/a979218b72b67fb247020af2238eb75099a8b46f
der geht noch 

https://github.com/Freetz-NG/freetz-ng/commit/832d680eaa3b4f37e5c63bec65c449cd6e4c10ac
ab hier geht nichts mehr außer man stellt es auf gcc5 also patchelf bei den Mips Geräten

--- config/ui/toolchain.in    2023-02-16 18:56:14.044248727 +0100
+++ config/ui/toolchain.in    2023-02-18 02:40:20.685358453 +0100
@@ -807,13 +807,13 @@
 
     config FREETZ_TOOLS_PATCHELF_VERSION_ABANDON
         bool "gcc 5"
-        depends on !FREETZ_HOSTTOOLS_DOWNLOAD
+        depends on !FREETZ_HOSTTOOLS_DOWNLOAD && FREETZ_TARGET_ARCH_MIPS
         help
             For usage with old Ubuntu 14
 
     config FREETZ_TOOLS_PATCHELF_VERSION_CURRENT
         bool "gcc 7"
-        depends on !FREETZ_ANCIENT_SYSTEM
+        depends on !FREETZ_ANCIENT_SYSTEM && !FREETZ_TARGET_ARCH_MIPS
 
 endchoice

freetz-user, 2023-02-20:
Aber das steht ja alles schon im ippf

fda77, 2023-02-20:
Ich wüsste nicht welchen Unterschied es hier mit mips etc gibt. Es sind wohl mehr verschieden binutils versionen die genutzt werden. Es ist ja nur die 1 der 4(?) die problem beim "strip" hat. Und das auch nur falls eine binary vorher mit der neuen patchelf bearbeitet wurde...

freetz-user, 2023-02-20:
Es ist auch so, wenn man einmal mitt 0.17.2 gebaut hat dann auf 0.14.5 zurückgeht, bootet die MIPS geriet auch nicht mehr. Frag nicht, warum ist halt so.

maverick3005, 2023-02-20:
Also ich habe, da ich frisch auschecke und keine vorhandene .config verwendet habe immer die "defaultwerte" von AVM und freetz beim testcompile verwendet. In ein paar Minuten weiss ich da mehr. Zwischen 7490/7590 werden auch andere AVM binutils als Vorgabe schon verwendet, daran wirds wohl nicht liegen.

fda77, 2023-02-20:
@freetz-user dort steht aber noch viel mehr quatsch

Der Link https://www.ip-phone-forum.de/threads/aktuell-bootet-kein-7490-und-7590ax-image-mehr-selbst-getestet.314675/post-2505600 beeinhaltet nur diesen 10 Satz:
> Mal an das mit patchelf also gcc 5 in der config gedacht. Wo es ja hier in den Thread geht

Man kann sich ja mal vertippen oder Gross und Klein vertauschen, aber das ergibt absolut keinen Sinn

freetz-user, 2023-02-20:
Ich weiß nur, dass mit den Sachen die Mips Geräte Booten

maverick3005, 2023-02-20:
So, Zwischenergebnis:

mit "gcc5" gesetzt geht der trunk bis a844465f0b

Die Kollegen hatten derweil wohl recht mit Ihrem Findings für MIPS.

Ich werde das jetzt nochmal mit dem tagesaktuellen trunk auf 3 verschiedenen Systemen durchtesten.

@fda77 
Könntest Du der Einfachheit halber, gerade weil "gcc7" eine Vorgabe ist, und nirgendwo im freetzbereich dokumentiert ist,
das Ganze einfach wieder reverten ? Auch auf debian11+ubuntu20.04 kommt ja das gleiche Ergebnis, und
das sind beides nun wirklich keine "Ancient-Systeme" ;-) - 0.14.5 tut was es soll

Gibt es wirklich einen guten Grund wieso wir ein "aktuelleres" Patchelf benötigen ? Irgendwelche PUMA-Kisten ggfs. ?

fda77, 2023-02-20:
Bei mir wurde gcc 12 installiert, daher ist für ein Linux das noch nicht mal gcc 7 hat recht betagt.
Du brauchst weder theoretisch noch praktisch was neueres ausser FOS 1.0. Eigentlich brauchst du sogar gar keine Internetverbindung.
Bei Problemen würde ich eher alte Geräte rauswerfen statt neue Software. Wie wäre es wenn jemand mal versucht das problem zu beheben?

maverick3005, 2023-02-20:
Du hast wahrscheinlich gerade ein ubuntu 22+ nehme ich mal an ?

Ubuntu 20.04 + debian 11.6 sind mit Ihren Updates nach wie vor tagesaktuell. Allerdings sind hier neuere GCC optional ebenso vorhanden - nur werden die nicht mit den PREREQUISITES automatisch sofort mit installiert. Das wäre dann ein User-Addon und ggfs. über "alternatives" nachzuhelfen.

Ändert aber nicht viel daran, das man ohne google/research/fremdforum-Suche mit den frischen Defaults auch erst darauf kommen muss, wieso bei einem recht aktuellen Linux eine einzelne versteckte Option für MIPS extra "downgegraded" werden muss ;-)

fda77, 2023-02-20:
>  Wie wäre es wenn jemand mal versucht das Problem zu beheben? Und nicht irgend ein drumherumgepfusche

maverick3005, 2023-02-20:
> > Wie wäre es wenn jemand mal versucht das Problem zu beheben? Und nicht irgend ein drumherumgepfusche

Das wäre sicherlich eine gute Idee....

Übrigens hat debian11.6 per default: "gcc version 10.2.1 20210110 (Debian 10.2.1-6)" 
ubuntu 20.04: "gcc version 9.4.0 (Ubuntu 9.4.0-1ubuntu1~20.04.1)" - gcc 10 ist hier ueber das repo verfügbar optional.
(mit "Testing-Repos gehts sowieso noch aktueller manuell)

Da sind wir doch von <= GCC 7 immer noch weit entfernt ;-)

Allerdings würde die PREREQUISITES-Dokumentation dann schon obsolut werden und man kann auch gleich sagen:

Ubuntu "22.04 only" und dann eine dazu passende, vollständige PreReq-Liste - dann hätten alle dieselbe Basis.

fda77, 2023-02-20:
Um kein gcc7 zu haben musst du Ubuntu 14 (von 2014) verwenden. Davon abgesehen supporte ich gar keine Ubuntu sondern verwende Fedora. Also installiert das bitte

maverick3005, 2023-02-20:
Ok, wir haben da aneinander vobei diskutiert:

Es ging nur darum das Du eine Option mit dem Namen "gcc-5" hast, welche aber auch auf recht aktuellen Systemen
nach wie vor gesetzt sein muss. (also gcc 8-10+)

Mit der Option "gcc-7" und das mal zu fixen, das es für gcc 7-10+ dann auch tatsächlich tut, was es soll, geb ich Dir schon recht.

fda77, 2023-02-20:
Einer von uns beiden schielt wohl etwas

[image]

Da du das angeblich nicht gelesen hast, woher weisst du überhaupt dass die option mit patchelf zu tun hat?

maverick3005, 2023-02-20:
Ich rede doch die ganze Zeit von dieser Option. Passt doch jetzt eh (nach dem man weiss wozu das gesetzt werden muss)

Die "Beschriftung" ist halt aktuell eher irreführend für die verwendeten "Host-Systeme" offenbar ;-)

so... Bestätigt für die 7490:

aktueller trunk compiliert mit Option "gcc-5" auf

debian10/gcc 8.3 - OK
debian11/gcc 10 - OK
ubuntu 22.04/gcc 9 - OK

fda77, 2023-02-20:
Oben hiess es doch dass die strip und patchelf Zeilen müssten wieder zurückgetauscht werden!?!?!

fda77, 2023-02-20:
Wenn es deiner Meinung nach ein Bug in der neuen Versionson von patchelf ist, mach dort ein Issue auf: https://github.com/NixOS/patchelf/

fda77, 2023-02-20:
https://github.com/NixOS/patchelf/issues/371
> If the order of the commands is inverted (first strip and then patchelf) the issue doesn't happen.

https://github.com/Freetz-NG/freetz-ng/commit/09fb889ea988a8c807a5fd467deae80d4c2e744a

maverick3005, 2023-02-20:
> Oben hiess es doch dass die strip und patchelf Zeilen müssten wieder zurückgetauscht werden!?!?!

Ich meinte eigentlich die Option gcc5/gcc7 komplett wieder entfernen und es auf "gcc-5" "compat" belassen, sonst nichts ändern. Aber passt schon - sofern man kein allerneustes Fedora oder Ubuntu 22.04+ oder debian-sid verwendet muss MIPS
derweil immer auf gcc-5 erstellt werden. Wird sich schon rumsprechen irgendwann ;-)

@fda77 
Danke trotzdem fürs Bug meldem, so tief bin ich in der Materie leider nicht drin.

fda77, 2023-02-20:
Du hast noch nicht mal auf die Links geklickt?

maverick3005, 2023-02-20:
Klar, war eine Erwähnung, jetzt muss man schaun ob da jemand auch "Deutsch kann" :p

fda77, 2023-02-20:
[image]

Wie man das Kontextmenü öffnet weisst du?

fda77, 2023-02-20:
[image]

maverick3005, 2023-02-20:
...habe ich ja nie gebraucht auf github, ich kanns ja.

Fazit jedoch: Problem bei MIPS im Grunde gefunden und nen Quickfix ist vorhanden.

fda77, 2023-02-20:
Im OP schreibst du 7490+7590. Bei welchen gibts das Problem denn nicht? Und welche Arch haben die?

maverick3005, 2023-02-20:
Bei den 7490+7590 gibt es das Problem. 5590 ging schon immer mit gcc-7 Setting.

Du hast ja jetzt einen Revert gemacht. Kann ich jetzt von gcc-5 wieder auf gcc-7/default umschalten für gcc 9/10 / mips ?

fda77, 2023-02-20:
ne
ich nenne besser die "gcc-7 Setting." um

closed 2023-02-20 by fda77

TWinkeri, 2023-02-20:
Läuft leider immer noch nicht:
```
mkdir -p packages/target-mips_gcc-5.5.0_uClibc-1.0.42-nptl_kernel-3.10/busybox/; cp source/target-mips_gcc-5.5.0_uClibc-1.0.42-nptl_kernel-3.10/busybox-1.36.0/busybox packages/target-mips_gcc-5.5.0_uClibc-1.0.42-nptl_kernel-3.10/busybox/busybox; [ "y" == "y" ] && patchelf --set-interpreter /usr/lib/freetz/ld-uClibc.so.1 packages/target-mips_gcc-5.5.0_uClibc-1.0.42-nptl_kernel-3.10/busybox/busybox || true; /home/USER/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.42-nptl_kernel-3.10/mips-linux-uclibc/bin/mips-linux-uclibc-strip --remove-section={.comment,.note,.pdr} packages/target-mips_gcc-5.5.0_uClibc-1.0.42-nptl_kernel-3.10/busybox/busybox;
/home/USER/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.42-nptl_kernel-3.10/mips-linux-uclibc/bin/mips-linux-uclibc-strip: packages/target-mips_gcc-5.5.0_uClibc-1.0.42-nptl_kernel-3.10/busybox/stCdEThd: error: PHDR segment not covered by LOAD segment
/home/USER/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.42-nptl_kernel-3.10/mips-linux-uclibc/bin/mips-linux-uclibc-strip: packages/target-mips_gcc-5.5.0_uClibc-1.0.42-nptl_kernel-3.10/busybox/stCdEThd[.interp]: file format not recognized
make[1]: *** [make/busybox/busybox.mk:61: packages/target-mips_gcc-5.5.0_uClibc-1.0.42-nptl_kernel-3.10/busybox/busybox] Fehler 1
make: *** [Makefile:46: envira] Beendet
```
Ich muss allerdings dazu sagen, dass ich dass unter Debian "bullseye-backports" versucht habe. Werde jetzt nochmals zurückkehren zu "stable" und erneut versuchen.

maverick3005, 2023-02-20:
Lass die Option bitte erst einmal auf "gcc-5" stehen für alle Boxen.

Ich hab das für gcc 8-10 schon durchgespielt. Auf meiner debian 11.6 rennt das problemlos im Moment.

fda77, 2023-02-20:
Das gcc WAR die minimale version die man zum compilieren von patchelf brauchte. Wenn das nicht erfüllt ist kommt keine binary raus und es bricht recht früh ab. Also sind keine gcc-orgien nötig. "PHDR segment not covered by LOAD segment" kommt von falscher Reihenfolge mit patchelf 0.17 -> issues dort lelsen

cawidtu, 2023-02-25:
Mit dem aktuellen Trunk f8e44f348a315f25c81d12c84b1919134b5c5eda und der Option "Patchelf version new" hing das Image auf einer 5491 mit Firmware 07.29 beim Booten. Nach Zurückstellen auf "Patchelf version old" lief es.

fda77, 2023-02-25:
Bis vor kurzem konnte man zwischen 0.14 und 0.17 auswählen. Jetzt nach dem downgrade (bis es upstream gefixt ist) zwischen 0.14 und 0.15.
Bitte sicherstellen dass 
 - bei dir `tools/patchelf` auch wirklich die version ist von der du berichtest, das wird nicht unbedingt immer neu gebaut
 - auch schauen dass in sources/, packages/ etc nicht noch mit dem "schlechten" 0.17 patchelf  bearbeitete dateien lagen (zb distclean oder neuer checkout). Bei mir war das ein problem

cawidtu, 2023-02-28:
Nach einem ``make toolchain-distclean`` + ``make clean`` lief es mit "patchelf version new", d. h. der Version 0.15. Danke.

fda77, 2023-02-28:
Prima :) Ich bin da anfangs auch reingefallen und dachte die 0.17 funktioniert...